raspberrypi/linux #1971: build failure: drivers/built-in.o: In function `hcd_init_fiq'

clandmeter, 2017-04-21:
Running make with [config](http://tpaste.us/JMjg) on;

```
gcc -v
Using built-in specs.
COLLECT_GCC=gcc
COLLECT_LTO_WRAPPER=/usr/libexec/gcc/armv6-alpine-linux-musleabihf/6.3.0/lto-wrapper
Target: armv6-alpine-linux-musleabihf
Configured with: /home/buildozer/aports/main/gcc/src/gcc-6.3.0/configure --prefix=/usr --mandir=/usr/share/man --infodir=/usr/share/info --build=armv6-alpine-linux-musleabihf --host=armv6-alpine-linux-musleabihf --target=armv6-alpine-linux-musleabihf --with-pkgversion='Alpine 6.3.0' --enable-checking=release --disable-fixed-point --disable-libstdcxx-pch --disable-multilib --disable-nls --disable-werror --disable-symvers --enable-__cxa_atexit --enable-default-pie --enable-cloog-backend --enable-languages=c,c++,objc,java,fortran,ada --with-arch=armv6zk --with-tune=arm1176jzf-s --with-fpu=vfp --with-float=hard --with-abi=aapcs-linux --disable-libquadmath --disable-libssp --disable-libmpx --disable-libmudflap --disable-libsanitizer --enable-shared --enable-threads --enable-tls --disable-libitm --with-system-zlib --with-linker-hash-style=gnu
Thread model: posix
gcc version 6.3.0 (Alpine 6.3.0)
```

Results in:

```
  LD [M]  drivers/usb/serial/usbserial.o
  LD      drivers/usb/built-in.o
  LD      drivers/built-in.o
  LD      vmlinux.o
  MODPOST vmlinux.o
  GEN     .version
  CHK     include/generated/compile.h
drivers/built-in.o: In function `hcd_init_fiq':
arm_pmu.c:(.text+0xd657c): undefined reference to `claim_fiq'
arm_pmu.c:(.text+0xd6638): undefined reference to `set_fiq_handler'
arm_pmu.c:(.text+0xd6690): undefined reference to `__set_fiq_regs'
arm_pmu.c:(.text+0xd6778): undefined reference to `enable_fiq'
make[2]: *** [/home/clandmeter/aports/main/linux-rpi/src/linux-4.9/Makefile:969: vmlinux] Error 1
make[1]: *** [Makefile:150: sub-make] Error 2
make: *** [Makefile:24: __sub-make] Error 2
```

pelwell, 2017-04-21:
Although the config file contains the required information, saying a bit more about your config settings would have saved some time in finding the problem.

Your build is failing because the dwc_otg USB driver relies on FIQ support (CONFIG_FIQ), but this config option isn't directly settable by the user. The standard RPi defconfigs (bcmrpi_defconfig for Pi0 and Pi1, bcm2709_defconfig for Pi2 and Pi3) enable CONFIG_ARCH_BCM, which selects CONFIG_FIQ, but your config doesn't. The dependency of dwc_otg on CONFIG_FIQ should arguably be made explicit in its Kconfig, but that will just make the driver unselectable.

Without FIQ support you will have to use the upstream dwc2 driver (CONFIG_USB_DWC2), but I think you'll struggle to get something working without enabling either ARCH_BCM or ARCH_BCM2835.

Our guide to building a kernel is [here](https://www.raspberrypi.org/documentation/linux/kernel/building.md), and there are a few notes on configuration [here](https://www.raspberrypi.org/documentation/linux/kernel/configuring.md).

popcornmix, 2017-04-26:
I've explicitly added the dependency on FIQ to dwc_otg (use without FIQ is not supported).
That should resolve your build issue (although dcw_otg won't be available without FIQ defined).
See: https://github.com/raspberrypi/linux/commit/8b68f431a6168c65f67f0c6b645e9c925f90a7c7

JamesH65, 2017-05-19:
Closing this issue as questions answered/issue resolved.

closed 2017-05-19 by JamesH65