raspberrypi/linux #1972: ARM64: RPI3 Mermory card failing to boot

raidensnake, 2017-04-22:
Whenever I try to boot kernels 4.9 to 4.11 using debian arn64 even on a clean install it fails to boot saying mmc: hardware interrupt error failed to initialise sdio. It fails to start. I even tried changing memory cards but to no avail. The only version that's bootable is 4.8.10, Us anyone able to find out why this is happening?

lategoodbye, 2017-04-27:
AFAIK support for ARM64 won't be provide here.

Btw you need to provide more information.
Which kernel do you use (mainline, foundation, exact version)?
Which devicetree do you use?
Can you provide a kernel config?
Does the problem occur with every sd card?

raidensnake, 2017-04-29:
1. I tried the master, stable, rpi-4.9.y rpi-4.10.y and rpi-4.11.y branches of this repo. 2. the kernel config is the bcmrpi3_defconfig default found in arch/arm64/configs folder for all of those. and 3. yes it happens on every single card and raspberry pi 3 I try. and 4. it also does the same for 32-bit as well on both rpi-4.10.y and rpi-4.11.y branches.

lategoodbye, 2017-04-29:
Can you please provide a kernellog from startup until the issue?

raidensnake, 2017-04-30:
Unless there's a way of outputting it to a file I can only take a picture.

pelwell, 2017-04-30:
`dmesg > file.txt`

raidensnake, 2017-04-30:
Any idea where I can place it? I can't boot the os.

raidensnake, 2017-04-30:
Being as I can't even boot to the os to run that command. I posted pictures of what it displays.

[image]
[image]
[image]

lategoodbye, 2017-05-01:
Could you please add the output before Exception stack timestamp 1.3xxx?

raidensnake, 2017-05-01:
I'm not able to as the HDMI only displays when the first screenshot is. It says no signal otherwise.

pelwell, 2017-05-01:
The usual way to get kernel tracing output is using a serial port connected to header pins 6 (GND), 8 (TXD0) and 10 (RXD0). To enable it you would need to add `enable_uart=1` to config.txt and `console=serial0,11520 loglevel=8` to cmdline.txt, removing `quiet` if it is present.

You can often get to see more logging output on screen by adding the following to config.txt:
```
framebuffer_width=3840
framebuffer_height=2160
```

It might also save time if you post the contents of your config.txt and cmdline.txt.

raidensnake, 2017-05-01:
1. I don't have the kit to do the serial part. And 2. The config.txt and codeine.txt files look like this.

Cmdline.txt:

consoleblank=0 console=ttyAMA0,115200 kgdboc=ttyAMA0,115200 console=tty1 root=/dev/mmcblk0p2

Config.txt:

arm_control=0x200
kernel=kernel8.img
hdmi_drive=2
dtparam=audio=on
dtparam=i2c_arm=on
dtparam=i2s=on
dtparam=spi=on

lategoodbye, 2017-05-04:
Did you tried without arm_control parameter?

raidensnake, 2017-05-04:
Yes and it wouldn't boot at all. Just a rainbow screen.

pelwell, 2017-05-05:
I just built an Pi3 ARMv8 kernel using bcmrpi3_defconfig, set device_tree=broadcom/bcm2710-rpi-3-b.dtb (`ARCH=arm64` puts the dtb files in the broadcom subdirectory) and it booted just fine.

My build script uses the `dtbs_install` make target, which creates and populates the broadcom subdirectory, so if you aren't careful you can end up using the old `ARCH=arm` dtbs instead. I found that with the new kernel and old dtb the mmc0/mmc1 assignments were backwards; DT aliases to manage the mapping were added at the end of March.

raidensnake, 2017-05-05:
Do you have an example? This is how I'm compiling it:

Here's the complete bash command list I use: (note: sdc is the memory card(

git clone --depth=1 -b rpi-4.11.y https://github.com/raspberrypi/linux
cd linux
mkdir mnt
cd mnt
mkdir fat32
mkdir ext4
cd ../
mount /dev/sdc1 mnt/fat32
mount /dev/sdc2 mnt/ext4
make ARCH=arm64 CROSS_COMPILE=aarch64-linux-gnu- bcmrpi3_defconfig
make -j8 ARCH=arm64 CROSS_COMPILE=aarch64-linux-gnu- Image modules dtbs
cp arch/arm64/boot/dts/broadcom/*.dtb mnt/fat32/
cp arch/arm64/boot/dts/overlays/*.dtb* mnt/fat32/overlays
cp arch/arm64/boot/Image mnt/fat32/kernel8.img
make ARCH=arm64 CROSS_COMPILE=aarch64-linux-gnu- INSTALL_MOD_PATH=mnt/ext4 modules_install
umount mnt/fat32
umount mnt/ext4

pelwell, 2017-05-06:
That looks alright - it doesn't use the dtbs_install rule, and explicitly copies out of the Broadcom directory into the fat32 root, so you should be using the correct dtb.

raidensnake, 2017-05-06:
The problem is when I try this it doesn't boot and gives me that error.

pelwell, 2017-05-06:
I understand that. If you did manage to get some more kernel logging output, from a serial port or a screenshot, then that might give us enough information to see what is causing your problem,. Until then, considering that nobody else has reported a problem, and since we don't support 64-bit mode, I can't help you any further.

raidensnake, 2017-05-07:
is there an option of outputting the log using another working pi? if so how can I go about doing that?

lategoodbye, 2017-05-07:
Yes, that would be a nice hack. I recommend a RPi 1 or 2 (now called dumper). Here my untested suggestion:

1. Enable 1 UART of the dumper and but disable its console ( remove console=ttyAMA0,115200 from cmdline.txt )
2. Make sure the debug UART of the affected RPI3 is enabled ( enable_uart=1 in config.txt )
3. Power off both Pis
4. connect GND lines of both pis with each other
5. cross the rx/tx lines between them ( rx of RPI3 to tx of other RPI and vice versa)
6. power on the dumper and control it via the other UART or HDMI
7. configure the trace UART which is connected to the RPI 3 ( sudo stty -F /dev/ttyAMA0 115200 )
8. start dumping into a file ( cat /dev/ttyAMA0 > dump.txt )
9. now start the affected RPI 3
10. wait a moment
11. stop the trace on the dumper with Strg+C

pelwell, 2017-05-07:
You can also follow the same steps using a Pi3 as the receiver if you put `dtoverlay=pi3-disable-bt` in config.txt.

raidensnake, 2017-05-07:
well now for some reason the pi has gone to an sd card error to a blank screen and doing absolutely nothing.

lategoodbye, 2017-05-07:
Maybe the filesystem is corrupt now?

JamesH65, 2017-05-19:
Inclined to close this as we don't specifically support ARM64 and conversation has dried up.. You could certainly continue the conversation on the forums, which is probably a  better place for it anyway. Put in a link back to this issue for context.

closed 2017-05-19 by JamesH65

raidensnake, 2017-06-08:
I tried doing the debug however it's still failing to boot.

raidensnake, 2017-06-08:
it's not outputting anything.