raspberrypi/linux #1974: Error : Rapberry Pi 3 : unrecognized command line option ‘-mgeneral-regs-only’

diparthshah, 2017-04-25:
hello,
 i am trying to compile kernel for rpi3 on 64 bit host system(ubuntu) and have configured toolchain as mentioned in the document.

i am facing this error : unrecognized command line option ‘-mgeneral-regs-only’ while compiling.

ps: this is the toolchain i am currently using : 
/tools/arm-bcm2708/gcc-linaro-arm-linux-gnueabihf-raspbian-x64/bin 

defconfig file : arch/arm64/configs/bcmrpi3_defconfig

any help related to this issue will be appreciated.

popcornmix, 2017-04-26:
Are you following this: https://www.raspberrypi.org/documentation/linux/kernel/building.md

popcornmix, 2017-04-27:
This is a newer gcc compiler and woul be a better choice:
https://github.com/raspberrypi/tools/tree/master/arm-bcm2708/arm-rpi-4.9.3-linux-gnueabihf

closed 2017-04-27 by diparthshah