raspberrypi/linux #1975: updates around 4/23/17 to jessie armhf broke postfix dns queries

rfournerat, 2017-04-25:
Had a RPI3 running the latest stable Jessie.  Did an apt-get update then apt-get upgrade on 4/23/17.  After the upgrade, postfix was unable to do DNS queries.  Specifically, it fails looking up MX hosts and A records.  However, nslookup and dig seem to work correctly.

I suspect (I do not know) the problem has something to do with one or more of these updates:

    libraspberrypi-bin:armhf 1.20170303-1 => 1.20170405-1
    libraspberrypi-dev:armhf 1.20170303-1 => 1.20170405-1
    libraspberrypi0:armhf 1.20170303-1 => 1.20170405-1
    raspberrypi-bootloader:armhf 1.20170303-1 => 1.20170405-1
    raspberrypi-kernel:armhf 1.20170303-1 => 1.20170405-1

Other than postfix, I haven't noticed any other problems.

Linux www 4.4.50-v7+ #970 SMP Mon Feb 20 19:18:29 GMT 2017 armv7l GNU/Linux

thanks,
Robert

popcornmix, 2017-04-26:
What was the previous kernel version?

rfournerat, 2017-04-26:
While there was no update to postfix, I did an "apt-get install --reinstall postfix", and that seemed to fix the issue.

closed 2017-04-26 by rfournerat