raspberrypi/linux #1977: VCHIQ functional test broken since 4.11rc1

lategoodbye, 2017-04-26:
Hi,

recently i found that vchiq_test -f doesn't work anymore with current mainline (4.11-rc6) and linux-next (20170404) on my Raspberry Pi Zero. The issue is always reproducible, but the error behavior isn't deterministic. Sometimes vchiq_test hangs and sometimes i get an error message from vchiq_test like this (but never errors from the kernel):

$ vchiq_test -f 10
Functional test - iters:10
======== iteration 1 ========
vchiq_test: 1502: expected callback reason VCHIQ_MESSAGE_AVAILABLE, got 1
vchiq_test: 1524: expected callback reason VCHIQ_BULK_TRANSMIT_DONE, got 5
vchiq_test: 863: func_error != 0

I've bisected this regression to this commit:

    00a19f3e25c0c40e0ec77f52d4841d23ad269169 is the first bad commit
    commit 00a19f3e25c0c40e0ec77f52d4841d23ad269169
    Author: Rabin Vincent <rabinv@axis.com>
    Date: Tue Nov 8 09:21:19 2016 +0100

    ARM: 8627/1: avoid cache flushing in flush_dcache_page()

    When the data cache is PIPT or VIPT non-aliasing, and cache operations
    are broadcast by the hardware, we can always postpone the flush in
    flush_dcache_page(). A similar change was done for ARM64 in commit
    b5b6c9e9149d ("arm64: Avoid cache flushing in flush_dcache_page()").

    Reviewed-by: Catalin Marinas <catalin.marinas@arm.com>
    Signed-off-by: Rabin Vincent <rabinv@axis.com>
    Signed-off-by: Russell King <rmk+kernel@arm.linux.org.uk>

It seems that staging/vc04_services/interface/vchiq_arm/vchiq_2835_arm relies on the behavior of flush_dcache_page before this patch get applied.

Here the related [thread](http://lists.infradead.org/pipermail/linux-rpi-kernel/2017-April/006149.html) on linux-rpi-kernel.

It would be nice to get a solution which is acceptable in mainline.

ED6E0F17, 2017-04-27:
00a19f3e depends on !cache_ops_need_broadcast() :
````
#if !defined(CONFIG_SMP) || __LINUX_ARM_ARCH__ >= 7
#define cache_ops_need_broadcast()      0
````

It seems sensible to revert the regression for armv6 only.

lategoodbye, 2017-04-27:
@ED6E0F17 Could you please explain why this is the right solution? Regarding to Russel's [explanation](http://lists.infradead.org/pipermail/linux-rpi-kernel/2017-April/006252.html) the issue must be in the driver. Playing with fundamental ARM parts could break other platforms.

Btw i already found a workaround by calling flush_cache_all() after get_user_pages().

ED6E0F17, 2017-04-27:
I forgot to ask if this is broken on Armv7, but I  was working with RMK`s comment "What I'm willing to do is to temporarily drop the offending patch".

The last thing that I considered is that the upstream patch might be broken, and reverting it is best avoided if you have another solution.

pelwell, 2017-04-27:
I'm looking through this now, and the failure mechanism isn't obvious. I haven't got it to fail on ARMv7, but that is probably because we define CONFIG_SMP on those platforms.

lategoodbye, 2017-04-27:
@ED6E0F17 I must confess that didn't test it on ARMv7 or ARMv8.

RMK's suggested to enable debugging for vchiq. After setting all debugfs entries to loglevel info the dmesg is flooded. So i'm skeptical that will helps.

pelwell, 2017-04-27:
I've been looking for non-standard behaviour by the driver. It's using the standard DMA map/unmap functions now, and the only questionable code I've seen is using memcpy to copy in the read data to partial cache lines. The destination is to a user page, and the context may be the application thread or it may be a kernel thread...

lategoodbye, 2017-04-27:
Can you point me to the line?
Does it mean it could be the problem?

pelwell, 2017-04-27:
I was thinking of [this line](https://github.com/raspberrypi/linux/blob/rpi-4.11.y/drivers/staging/vc04_services/interface/vchiq_arm/vchiq_2835_arm.c#L603) and [this one](https://github.com/raspberrypi/linux/blob/rpi-4.11.y/drivers/staging/vc04_services/interface/vchiq_arm/vchiq_2835_arm.c#L610), but I haven't identified the mechanism yet.

lategoodbye, 2017-04-28:
I tested ARMv7 today and got a NULL pointer dereference in free_pagelist. So i better fix this before.

lategoodbye, 2017-04-29:
@ED6E0F17 Thanks for extending vchiq_test, i will update ASAP.

lategoodbye, 2017-05-02:
Okay, here are the results for mainline tree + bcm2836 dtb (vchiq node patched) + multi_v7_defconfig on Raspberry Pi 2 :

df044ebfc5bfba6810e3940db458d7fb4572bc6f -> don't build on v7 because of implicite declaration of v7_dma_map_area
cf9caf1929882b66922aee698e99e6c8f357bee5 -> NULL pointer dereference in memcpy of free_pagelist

So i switched to raspberry pi tree + bcm2836 dtb (vchiq node patched) + multi_v7_defconfig on Raspberry Pi 2 :

525571c4df4066a61166b4b44cf5d757c2c53f5e -> vchiq_test hang, unable to terminate program

Here an strace until the hang (unfortunately not always the same position):

```
pi@raspberrypi:~$ strace -f vchiq_test -f 1
execve("/usr/bin/vchiq_test", ["vchiq_test", "-f", "1"], [/* 13 vars */]) = 0
brk(0)                                  = 0x3a000
uname({sys="Linux", node="raspberrypi", ...}) = 0
access("/etc/ld.so.nohwcap", F_OK)      = -1 ENOENT (No such file or directory)
mmap2(NULL, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0xb6f57000
access("/etc/ld.so.preload", R_OK)      = 0
open("/etc/ld.so.preload", O_RDONLY|O_CLOEXEC) = 3
fstat64(3, {st_mode=S_IFREG|0644, st_size=42, ...}) = 0
mmap2(NULL, 42, PROT_READ|PROT_WRITE, MAP_PRIVATE, 3, 0) = 0xb6f56000
close(3)                                = 0
open("/usr/lib/arm-linux-gnueabihf/libarmmem.so", O_RDONLY|O_CLOEXEC) = 3
read(3, "\177ELF\1\1\1\0\0\0\0\0\0\0\0\0\3\0(\0\1\0\0\0h\5\0\0004\0\0\0"..., 512) = 512
lseek(3, 17960, SEEK_SET)               = 17960
read(3, "\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0"..., 960) = 960
lseek(3, 17696, SEEK_SET)               = 17696
read(3, "A.\0\0\0aeabi\0\1$\0\0\0\0056\0\6\6\10\1\t\1\n\3\f\1\22\4\24"..., 47) = 47
fstat64(3, {st_mode=S_IFREG|0644, st_size=18920, ...}) = 0
mmap2(NULL, 83236, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0xb6f15000
mprotect(0xb6f1a000, 61440, PROT_NONE)  = 0
mmap2(0xb6f29000, 4096, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x4000) = 0xb6f29000
mprotect(0xbea85000, 4096, PROT_READ|PROT_WRITE|PROT_EXEC|PROT_GROWSDOWN) = 0
close(3)                                = 0
munmap(0xb6f56000, 42)                  = 0
open("/etc/ld.so.cache", O_RDONLY|O_CLOEXEC) = 3
fstat64(3, {st_mode=S_IFREG|0644, st_size=24350, ...}) = 0
mmap2(NULL, 24350, PROT_READ, MAP_PRIVATE, 3, 0) = 0xb6f51000
close(3)                                = 0
access("/etc/ld.so.nohwcap", F_OK)      = -1 ENOENT (No such file or directory)
open("/opt/vc/lib/libvchiq_arm.so", O_RDONLY|O_CLOEXEC) = 3
read(3, "\177ELF\1\1\1\0\0\0\0\0\0\0\0\0\3\0(\0\1\0\0\0(\26\0\0004\0\0\0"..., 512) = 512
lseek(3, 29248, SEEK_SET)               = 29248
read(3, "\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0"..., 1040) = 1040
lseek(3, 21226, SEEK_SET)               = 21226
read(3, "A;\0\0\0aeabi\0\0011\0\0\0\5ARM1176JZF-S\0\6\7"..., 60) = 60
fstat64(3, {st_mode=S_IFREG|0644, st_size=30288, ...}) = 0
mmap2(NULL, 4096, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0xb6f50000
mmap2(NULL, 88456, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0xb6eff000
mprotect(0xb6f05000, 61440, PROT_NONE)  = 0
mmap2(0xb6f14000, 4096, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x5000) = 0xb6f14000
close(3)                                = 0
access("/etc/ld.so.nohwcap", F_OK)      = -1 ENOENT (No such file or directory)
open("/opt/vc/lib/libvcos.so", O_RDONLY|O_CLOEXEC) = 3
read(3, "\177ELF\1\1\1\0\0\0\0\0\0\0\0\0\3\0(\0\1\0\0\0\234*\0\0004\0\0\0"..., 512) = 512
lseek(3, 171748, SEEK_SET)              = 171748
read(3, "\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0"..., 1360) = 1360
lseek(3, 37806, SEEK_SET)               = 37806
read(3, "A;\0\0\0aeabi\0\0011\0\0\0\5ARM1176JZF-S\0\6\7"..., 60) = 60
fstat64(3, {st_mode=S_IFREG|0644, st_size=173108, ...}) = 0
mmap2(NULL, 105624, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0xb6ee5000
mprotect(0xb6eee000, 65536, PROT_NONE)  = 0
mmap2(0xb6efe000, 4096, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x9000) = 0xb6efe000
close(3)                                = 0
access("/etc/ld.so.nohwcap", F_OK)      = -1 ENOENT (No such file or directory)
open("/lib/arm-linux-gnueabihf/libpthread.so.0", O_RDONLY|O_CLOEXEC) = 3
read(3, "\177ELF\1\1\1\0\0\0\0\0\0\0\0\0\3\0(\0\1\0\0\0lY\0\0004\0\0\0"..., 512) = 512
lseek(3, 120788, SEEK_SET)              = 120788
read(3, "\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0"..., 1520) = 1520
[... 104 lines cut ...]
[pid   512] <... munmap resumed> )      = 0
[pid   510] ioctl(3, 0xc01cc402 <unfinished ...>
[pid   512] mprotect(0xb6400000, 139264, PROT_READ|PROT_WRITE) = 0
[pid   510] <... ioctl resumed> , 0xbea85b24) = -1 EIO (Input/output error)
[pid   512] ioctl(3, 0xc014c407 <unfinished ...>
[pid   510] ioctl(3, 0xc01cc402, 0xbea85b24) = 0
[pid   510] ioctl(3, 0xc01cc402, 0xbea85b24) = -1 EIO (Input/output error)
[pid   510] ioctl(3, 0xc01cc402, 0xbea85b24) = 0
[pid   510] ioctl(3, 0xc01cc402, 0xbea85b24) = -1 EIO (Input/output error)
[pid   510] ioctl(3, 0xc403 <unfinished ...>
[pid   512] <... ioctl resumed> , 0xb6d50d6c) = 1
[pid   512] ioctl(3, 0xc411 <unfinished ...>
[pid   510] <... ioctl resumed> , 0xa002) = 0
[pid   512] <... ioctl resumed> , 0xa002) = 0
[pid   510] ioctl(3, 0xc01cc402 <unfinished ...>
[pid   512] ioctl(3, 0xc014c407 <unfinished ...>
[pid   510] <... ioctl resumed> , 0xbea85b24) = 0
[pid   510] ioctl(3, 0x400cc404, 0xbea85b7c) = 0
[pid   512] <... ioctl resumed> , 0xb6d50d6c) = 1
[pid   510] ioctl(3, 0xc014c405 <unfinished ...>
[pid   512] ioctl(3, 0xc014c407 <unfinished ...>
[pid   510] <... ioctl resumed> , 0xbea85b74) = 0
[pid   510] ioctl(3, 0xc014c405, 0xbea85b74) = 0
[pid   510] futex(0x2e4e4, FUTEX_WAIT_PRIVATE, 0, NULL
```

pelwell, 2017-05-02:
Returning to the initial bug report, I think the problem may lie in this commit: https://github.com/raspberrypi/linux/commit/cf9caf1929882b66922aee698e99e6c8f357bee5

The loop calling sg_set_page sets up a scatterlist comprising whole pages, which looks dangerous when you follow it up with DMA_FROM_DEVICE cache maintenance (invalidate).

Any comments, @Electron752?

ED6E0F17, 2017-05-02:
There is an issue with corrupt messages on Linux 4.5 and 4.9 when the "lazy cache flushing" patch is backported.

The issue with the test stalling seems to be new to Linux 4.11, and I sometimes see it running slowly on Arm v7, although I cannot make it fail on Pi2.

pelwell, 2017-05-03:
See https://github.com/raspberrypi/linux/pull/1987 for a fix for the corruption.

lategoodbye, 2017-05-03:
My tests with a Raspberry Pi 1 B (mainline tree + bcm2835_defconfig) were also successful. Because of the other issue i'm not able to test with a Raspberry Pi 2 (mainline tree + multi_v7_defconfig). This needs more investigation.

@pelwell Thanks. Do you plan to submit the patch upstream?

pelwell, 2017-05-03:
Do battle with the Linux kernel patch submission process? Try and stop me.

lategoodbye, 2017-05-03:
Sure, if it makes your job a little bit easier.

pelwell, 2017-05-04:
Sorry - that was a jokey way of saying yes, I will submit it upstream.

lategoodbye, 2017-05-04:
Fine, but please add a Fixes tag on submission.

pelwell, 2017-05-04:
Patch submitted, including #1990.

closed 2017-05-04 by lategoodbye