raspberrypi/linux #1979: USB devices stopped working

jkuek, 2017-04-28:
Had a new issue this morning. I think it was triggered when sierra_net was unloaded (modprobe -r sierra_net)

It seemed the system was still running afterwards (as there are cron messages in syslog) but it seems like USB wasn't working at all, as I couldn't SSH into the device over ethernet, and the HDMI output was blank, and pressing the keyboard didn't turn it on.

I had to hard power-cycle the unit to bring it back to full operation.

```
Apr 28 04:14:09 raspberrypi kernel: [558865.601947] WARN::dwc_otg_hcd_handle_hc_fsm:2619: Unexpected state received on hc=6 fsm=1 on transfer to device 5 ep 0x7
Apr 28 08:14:18 raspberrypi kernel: [573274.900590] sierra_net 1-1.4:1.7 wwan0: Session idle, 0x00
Apr 28 08:14:18 raspberrypi kernel: [573274.914656] WARN::dwc_otg_hcd_handle_hc_fsm:2619: Unexpected state received on hc=5 fsm=11 on transfer to device 5 ep 0x1
Apr 28 08:16:42 raspberrypi kernel: [573419.379630] ------------[ cut here ]------------
Apr 28 08:16:42 raspberrypi kernel: [573419.379708] WARNING: CPU: 0 PID: 0 at net/sched/sch_generic.c:306 dev_watchdog+0x268/0x28c()
Apr 28 08:16:42 raspberrypi kernel: [573419.379749] NETDEV WATCHDOG: eth0 (smsc95xx): transmit queue 0 timed out
Apr 28 08:16:42 raspberrypi kernel: [573419.379761] Modules linked in: sierra_net cfg80211 rfkill joydev snd_usb_audio snd_hwdep snd_usbmidi_lib snd_rawmidi snd_seq_device sierra usbserial snd_soc_bcm2835_i2s regmap_mmio snd_soc_core snd_pcm_dmaengine snd_bcm2835 snd_pcm snd_timer snd spi_bcm2835 bcm2835_gpiomem bcm2835_wdt evdev uio_pdrv_genirq uio ipv6 [last unloaded: sierra_net]
Apr 28 08:16:42 raspberrypi kernel: [573419.379906] CPU: 0 PID: 0 Comm: swapper Not tainted 4.4.50+ #970
Apr 28 08:16:42 raspberrypi kernel: [573419.379920] Hardware name: BCM2708
Apr 28 08:16:42 raspberrypi kernel: [573419.379989] [<c0016d18>] (unwind_backtrace) from [<c0013c3c>] (show_stack+0x20/0x24)
Apr 28 08:16:42 raspberrypi kernel: [573419.380052] [<c0013c3c>] (show_stack) from [<c02e49dc>] (dump_stack+0x20/0x28)
Apr 28 08:16:42 raspberrypi kernel: [573419.380093] [<c02e49dc>] (dump_stack) from [<c0021f2c>] (warn_slowpath_common+0x8c/0xc4)
Apr 28 08:16:42 raspberrypi kernel: [573419.380146] [<c0021f2c>] (warn_slowpath_common) from [<c0021fa4>] (warn_slowpath_fmt+0x40/0x48)
Apr 28 08:16:42 raspberrypi kernel: [573419.380183] [<c0021fa4>] (warn_slowpath_fmt) from [<c04bb860>] (dev_watchdog+0x268/0x28c)
Apr 28 08:16:42 raspberrypi kernel: [573419.380228] [<c04bb860>] (dev_watchdog) from [<c0064f5c>] (call_timer_fn+0x40/0x120)
Apr 28 08:16:42 raspberrypi kernel: [573419.380284] [<c0064f5c>] (call_timer_fn) from [<c006526c>] (run_timer_softirq+0x14c/0x280)
Apr 28 08:16:42 raspberrypi kernel: [573419.380318] [<c006526c>] (run_timer_softirq) from [<c00253c0>] (__do_softirq+0x124/0x314)
Apr 28 08:16:42 raspberrypi kernel: [573419.380353] [<c00253c0>] (__do_softirq) from [<c0025904>] (irq_exit+0xbc/0x11c)
Apr 28 08:16:42 raspberrypi kernel: [573419.380412] [<c0025904>] (irq_exit) from [<c005a608>] (__handle_domain_irq+0x60/0xb8)
Apr 28 08:16:42 raspberrypi kernel: [573419.380449] [<c005a608>] (__handle_domain_irq) from [<c0010748>] (handle_IRQ+0x2c/0x30)
Apr 28 08:16:42 raspberrypi kernel: [573419.380502] [<c0010748>] (handle_IRQ) from [<c000940c>] (bcm2835_handle_irq+0x3c/0x58)
Apr 28 08:16:42 raspberrypi kernel: [573419.380541] [<c000940c>] (bcm2835_handle_irq) from [<c05768e4>] (__irq_svc+0x44/0x5c)
Apr 28 08:16:42 raspberrypi kernel: [573419.380557] Exception stack(0xc0821f18 to 0xc0821f60)
Apr 28 08:16:42 raspberrypi kernel: [573419.380574] 1f00:                                                       00000000 00000000
Apr 28 08:16:42 raspberrypi kernel: [573419.380620] 1f20: 00000000 c0822f74 c088d5f0 c0820000 c08220ac 00000000 c0887d04 c0887268
Apr 28 08:16:42 raspberrypi kernel: [573419.380646] 1f40: c0887268 c0821f74 c0821f68 c0821f68 c0010804 c0010808 60000013 ffffffff
Apr 28 08:16:42 raspberrypi kernel: [573419.380681] [<c05768e4>] (__irq_svc) from [<c0010808>] (arch_cpu_idle+0x30/0x40)
Apr 28 08:16:42 raspberrypi kernel: [573419.380726] [<c0010808>] (arch_cpu_idle) from [<c004d7ec>] (default_idle_call+0x34/0x48)
Apr 28 08:16:42 raspberrypi kernel: [573419.380783] [<c004d7ec>] (default_idle_call) from [<c004d930>] (cpu_startup_entry+0x130/0x17c)
Apr 28 08:16:42 raspberrypi kernel: [573419.380813] [<c004d930>] (cpu_startup_entry) from [<c05722e0>] (rest_init+0x6c/0x84)
Apr 28 08:16:42 raspberrypi kernel: [573419.380855] [<c05722e0>] (rest_init) from [<c07c2ca0>] (start_kernel+0x358/0x3cc)
Apr 28 08:16:42 raspberrypi kernel: [573419.380894] ---[ end trace 2dc24c872aa9e8d2 ]---

```

P33M, 2017-05-04:
There have been a few bugfixes that are now available in rpi-update. Can you retest with an updated Pi?

jkuek, 2017-05-04:
Unfortunately, I've only seen this issue occur once, and I'm not sure how to reproduce it. I can certainly update my Pi and see what happens.

JamesH65, 2017-05-17:
@jkuek  Anything to report after the update? Would like to close this if possible, you can always reopen if the error reoccurs.

jkuek, 2017-05-17:
I have not seen this error again. Suggest it is closed for now.

closed 2017-05-17 by jkuek