raspberrypi/linux #1980: Module dvb_usb_dibusb_common cannot be loaded: 'Kernel tainted'

juandiazq, 2017-04-29:
I cannot load the modules dib3000mc and dvb_usb_a800 because the MODULE_LICENSE() macro are not declared. This patch would fix the problem:

[fix_dibusb_load_module.patch.zip](https://github.com/raspberrypi/linux/files/965771/fix_dibusb_load_module.patch.zip)

6by9, 2017-04-29:
It is already defined in files 1, 2, and 4 that you've modified.
https://github.com/raspberrypi/linux/blob/rpi-4.9.y/drivers/media/usb/dvb-usb/dibusb-common.c#L20
https://github.com/raspberrypi/linux/blob/rpi-4.9.y/drivers/media/usb/dvb-usb/dibusb-mb.c#L471
https://github.com/raspberrypi/linux/blob/rpi-4.9.y/drivers/media/usb/dvb-usb/dibusb-mc.c#L149
It is missing in drivers/media/usb/dvb-usb/dibusb-mc-common.c.

That is an upstream file, so the issue ought to be reported upstream - https://github.com/raspberrypi/linux/blob/rpi-4.9.y/REPORTING-BUGS. It looks like it got missed as part of a cleanup/fix last year - https://github.com/raspberrypi/linux/commit/e91455a1495ada59e90ae26bc3e68be3b7656570
Once accepted it will automatically filter down with the next kernel release, or we can cherrypick it back to earlier trees.

juandiazq, 2017-04-29:
Thank you, I reported it on kernel bugzilla: https://bugzilla.kernel.org/show_bug.cgi?id=195613

JamesH65, 2017-05-19:
Closing this issue as questions answered/issue resolved.

closed 2017-05-19 by JamesH65