raspberrypi/linux #1981: PDM Input Mode Operation

pyrog, 2017-04-30:
Hi,

How to use the **Pulse Digital Modulation** (PDM) mode to connect one or two _"PDM microphones"_ directly to the SoC pins ?

The [Broadcom BCM2835 data sheet](https://cdn-shop.adafruit.com/product-files/2885/BCM2835Datasheet.pdf) describe registers to use.

The [ALSA SoC I2S Audio Layer](https://github.com/raspberrypi/linux/blob/linux_stable/sound/soc/bcm/bcm2835-i2s.c) source code use the I2S audio layer in PCM mode only (but have the definition of all registers).

Thanks,

—
Yves

See also: [Introduction to MEMS Microphones](https://curiouser.cheshireeng.com/2014/10/21/introduction-to-mems-microphones/) and [Using a PDM Microphone](https://curiouser.cheshireeng.com/2014/11/04/using-a-pdm-microphone/).

P33M, 2017-05-02:
This repository is for kernel issues and PRs that implement additional features or fix bugs. As this is a feature that doesn't yet have support, this isn't the best place to have the discussion.

You're likely to get a more lively discussion going on the Interfacing section of the community forum - there are several knowledgeable people there that would probably be interested in exploring this feature.

https://www.raspberrypi.org/forums/viewforum.php?f=44

closed 2017-05-02 by P33M

pyrog, 2017-05-02:
Thank you for the link 😄