raspberrypi/linux #1984: V4L2: duplicate frame capture on STREAM_ON

AriHC, 2017-05-02:
When taking pictures with V4L2 on the RPi3, the driver seems to capture 2 frames when the stream is first turned on, resulting in approximately off-by-one frames to be returned when multiple buffers are pulled from the stream.

That is, as best I can tell the following is happening:

Enqueue buffer 1
Stream On
Camera captures frames a, b (a split second apart)
Dequeue - returns frame a data in buffer 1
... wait a few seconds
Enqueue buffer 2
Camera captures frame c
Dequeue - returns frame b data in buffer 2

I expect frame c to be returned there; frame b should not exist/be returned at all because it was never requested. If I enqueue and dequeue another buffer, it will be filled with the data c from the previous capture, etc.

Additionally, both YU12 and JPEG formats (the only ones I've tested with currently) have issues with frame timestamps due to this, but in different ways:
For YU12: The first frame returned has a nonsense timestamp, e.g. 4154505010.207436s. (so using my example above, it looks like "a" is actually the bad frame, not "b")
For JPEG: The frames are timestamped based on when they were theoretically taken - that is to say, frame a has the expected timestamp, but when the second dequeue returns frame b, the timestamp is what I would expect frame c to have (which is clearly incorrect because b was taken around the same time as a)

6by9, 2017-05-02:
Sounds like a duplicate of #1297.
There is an outstanding PR at #1359 which should deal with it, but I haven't had a chance to test. Any possibility you can try it out?

AriHC, 2017-05-02:
You're right, does sound like a dupe. If I have a chance I'll test, though for now my code is just working around this and it's a low priority. Worth noting the PR (at least according to its commit message) only purports to fix JPEG, I suppose we'll see if it works for YUV as well. Further comments will be left on the bug this is a duplicate of.

closed 2017-05-02 by AriHC