raspberrypi/linux #1985: Connection flood to port 445 on mounting cifs volume (4.9.24-v7+)

Esowteric, 2017-05-02:
Existing bug reports:

-- https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=856843
-- https://bugs.launchpad.net/ubuntu/+source/linux/+bug/1686099

I've got a flood on port 445 across my LAN. If I block that port, I get a flood instead via port 139.

There was 1MB/s upload/download between my Windows 10 machine and a Raspberry Pi 3 (4.9.24-v7+).

I blocked that and had the same flood between two Raspberry Pi 3s.

The same thing happened if I fired up my Lubuntu boxes (I can't remember if it was the 4.4.0-75-generic or the 4.8.0-29-generic, and I can't re-enable Samba right now on the network, as it's "live" and I don't want to disrupt the web service).

Regards,
Eric Twose.

Esowteric, 2017-05-02:
Wireshark screenshot of port 445 traffic between two RPis ...

[image]

6by9, 2017-05-02:
Fixed upstream with a stable tag, so it should come via upstream fairly soon.
https://github.com/torvalds/linux/commit/62a6cfddcc0a5313e7da3e8311ba16226fe0ac10

6by9, 2017-05-02:
Checking the upstream 4.9 tree, the fix should be in 4.9.25 which is already merged into this repo. 
AIUI It hasn't been bumped via apt-get as yet, but rpi-update should get it.
https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux-stable.git/commit/?id=0b7c970663411cbca20517f38544983ff4fd9650

popcornmix, 2017-05-02:
Yes, confirmed it is in latest ``rpi-update`` kernel.

Esowteric, 2017-05-02:
It's been reported with Ubuntu 16.04 LTS 4.4.0-75, as well. If it's the same issue, will the backports cover this, too?

6by9, 2017-05-02:
Not a question to ask here as this kernel only relates to Raspberry Pi.

It was applied to the 4.4 tree 5 days ago, so should be in 4.4.64 and later. When Ubuntu bump the kernel in their repos is totally up to them.
https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux-stable.git/log/?id=refs/tags/v4.4.65

Esowteric, 2017-05-02:
Brilliant. Thanks, guys.

JamesH65, 2017-05-19:
Closing this issue as questions answered/issue resolved.

closed 2017-05-19 by JamesH65