raspberrypi/linux #1988: kernel stack trace warning in sme.c

herbertp, 2017-05-03:
Always happens when I bring up wifi.

```
[   73.362633] ------------[ cut here ]------------
[   73.362761] WARNING: CPU: 3 PID: 76 at net/wireless/sme.c:728 __cfg80211_connect_result+0x2cc/0x3e4 [cfg80211]()
[   73.362771] Modules linked in: bnep hci_uart btbcm bluetooth joydev evdev uvcvideo brcmfmac hid_logitech_hidpp videobuf2_vmalloc videobuf2_memops brcmutil videobuf2_v4l2 videobuf2_core cfg80211 v4l2_common videodev media hid_logitech_dj rfkill snd_bcm2835 snd_pcm snd_timer snd spi_bcm2835 bcm2835_gpiomem i2c_bcm2708 bcm2835_wdt uio_pdrv_genirq uio i2c_dev fuse ipv6
[   73.362923] CPU: 3 PID: 76 Comm: kworker/u8:1 Not tainted 4.4.50-v7+ #970
[   73.362932] Hardware name: BCM2709
[   73.363017] Workqueue: cfg80211 cfg80211_event_work [cfg80211]
[   73.363061] [<800187c0>] (unwind_backtrace) from [<80014094>] (show_stack+0x20/0x24)
[   73.363087] [<80014094>] (show_stack) from [<80321ce4>] (dump_stack+0xd4/0x118)
[   73.363110] [<80321ce4>] (dump_stack) from [<8002541c>] (warn_slowpath_common+0x98/0xc8)
[   73.363134] [<8002541c>] (warn_slowpath_common) from [<80025508>] (warn_slowpath_null+0x2c/0x34)
[   73.363223] [<80025508>] (warn_slowpath_null) from [<7f1941f4>] (__cfg80211_connect_result+0x2cc/0x3e4 [cfg80211])
[   73.363398] [<7f1941f4>] (__cfg80211_connect_result [cfg80211]) from [<7f16c580>] (cfg80211_process_wdev_events+0x160/0x1a0 [cfg80211])
[   73.363556] [<7f16c580>] (cfg80211_process_wdev_events [cfg80211]) from [<7f16c5fc>] (cfg80211_process_rdev_events+0x3c/0x70 [cfg80211])
[   73.363713] [<7f16c5fc>] (cfg80211_process_rdev_events [cfg80211]) from [<7f16728c>] (cfg80211_event_work+0x24/0x2c [cfg80211])
[   73.363805] [<7f16728c>] (cfg80211_event_work [cfg80211]) from [<8003c930>] (process_one_work+0x154/0x458)
[   73.363829] [<8003c930>] (process_one_work) from [<8003cc88>] (worker_thread+0x54/0x500)
[   73.363849] [<8003cc88>] (worker_thread) from [<80042954>] (kthread+0xec/0x104)
[   73.363870] [<80042954>] (kthread) from [<8000fbe8>] (ret_from_fork+0x14/0x2c)
[   73.363881] ---[ end trace 7adda82b71f20594 ]---
```

```
[   55.796184] ------------[ cut here ]------------
[   55.796455] WARNING: CPU: 3 PID: 115 at net/wireless/sme.c:738 __cfg80211_connect_result+0x2d4/0x3e8 [cfg80211]
[   55.796463] Modules linked in: bnep hci_uart btbcm bluetooth binfmt_misc evdev joydev hid_logitech_hidpp brcmfmac spidev brcmutil hid_logitech_dj cfg80211 rfkill snd_bcm2835 snd_pcm snd_timer snd i2c_bcm2835 spi_bcm2835 bcm2835_gpiomem fixed uio_pdrv_genirq uio i2c_dev fuse ipv6
[   55.796580] CPU: 3 PID: 115 Comm: kworker/u8:1 Not tainted 4.9.24-v7+ #993
[   55.796585] Hardware name: BCM2835
[   55.796788] Workqueue: cfg80211 cfg80211_event_work [cfg80211]
[   55.796823] [<8010fb3c>] (unwind_backtrace) from [<8010c058>] (show_stack+0x20/0x24)
[   55.796839] [<8010c058>] (show_stack) from [<80455200>] (dump_stack+0xd4/0x118)
[   55.796856] [<80455200>] (dump_stack) from [<8011d2c4>] (__warn+0xf8/0x110)
[   55.796869] [<8011d2c4>] (__warn) from [<8011d3ac>] (warn_slowpath_null+0x30/0x38)
[   55.797067] [<8011d3ac>] (warn_slowpath_null) from [<7f232510>] (__cfg80211_connect_result+0x2d4/0x3e8 [cfg80211])
[   55.797453] [<7f232510>] (__cfg80211_connect_result [cfg80211]) from [<7f208b84>] (cfg80211_process_wdev_events+0x168/0x1a8 [cfg80211])
[   55.797826] [<7f208b84>] (cfg80211_process_wdev_events [cfg80211]) from [<7f208c00>] (cfg80211_process_rdev_events+0x3c/0x70 [cfg80211])
[   55.798196] [<7f208c00>] (cfg80211_process_rdev_events [cfg80211]) from [<7f203284>] (cfg80211_event_work+0x24/0x2c [cfg80211])
[   55.798396] [<7f203284>] (cfg80211_event_work [cfg80211]) from [<80136570>] (process_one_work+0x158/0x454)
[   55.798415] [<80136570>] (process_one_work) from [<801368d0>] (worker_thread+0x64/0x568)
[   55.798431] [<801368d0>] (worker_thread) from [<8013c7d8>] (kthread+0x10c/0x124)
[   55.798447] [<8013c7d8>] (kthread) from [<80108148>] (ret_from_fork+0x14/0x2c)
[   55.798455] ---[ end trace 6c952b165f7e4957 ]---
```

pelwell, 2017-05-03:
A few questions:
1) Which Pi model is this?
2) Which WiFi interface?
3) Which distribution?
4) When you say "bring up wifi", is it enabled from boot or are you performing some manual step to start it?

herbertp, 2017-05-03:
1. Raspberry Pi 3 Model B V1.2
2. Built in Wifi (wlan0)
3. Raspbian (8.0)
4. I manually configure it with essid and key (iwconfig)

pelwell, 2017-05-03:
It looks like a lookup of a BSS is failing. Run this before running iwconfig:
```
sudo sh -c "echo 1 > /sys/kernel/debug/tracing/events/cfg80211"
```
and this afterwards:
```
sudo cat /sys/kernel/debug/tracing/trace > ~pi/cfg80211_trace.txt
```
Then paste the results somewhere - a gist, pastebin etc., - or as a comment here if it isn't too large.

pelwell, 2017-05-05:
For comparison, I ran a similar test on a standard Raspbian release running dhdpcd. In order to enable tracing before the wlan interface comes up I disabled dhcpcd from starting automatically:
```
sudo systemctl disable dhcpcd
```
and rebooted. Then I ran:
```
sudo sh -c "echo 1 >/sys/kernel/debug/tracing/events/cfg80211/enable"
sudo systemctl start dhcpcd
# Wait for the wlan interface to come up
sudo sh -c "echo 0 >/sys/kernel/debug/tracing/events/cfg80211/enable"
sudo cat /sys/kernel/debug/tracing/trace > cfg80211_trace.txt
sudo systemctl enable dhcpcd
```
and this was the output:
```
# tracer: nop
#
# entries-in-buffer/entries-written: 17/17   #P:4
#
#                              _-----=> irqs-off
#                             / _----=> need-resched
#                            | / _---=> hardirq/softirq
#                            || / _--=> preempt-depth
#                            ||| /     delay
#           TASK-PID   CPU#  ||||    TIMESTAMP  FUNCTION
#              | |       |   ||||       |         |
  wpa_supplicant-453   [002] ....   134.238655: rdev_scan: phy0
  wpa_supplicant-453   [002] ....   134.243161: rdev_return_int: phy0, returned: 0
     kworker/1:1-33    [001] ....   135.070706: cfg80211_return_bss: 80:2a:a8:**:**:**, band: 0, freq: 2462
     kworker/1:1-33    [001] ....   135.070725: cfg80211_return_bss: b0:d5:cc:**:**:**, band: 0, freq: 2422
     kworker/1:1-33    [001] ....   135.070741: cfg80211_return_bss: 8e:15:44:**:**:**, band: 0, freq: 2437
     kworker/1:1-33    [001] ....   135.070749: cfg80211_return_bss: 44:d9:e7:**:**:**, band: 0, freq: 2437
     kworker/1:1-33    [001] ....   135.070761: cfg80211_return_bss: 88:15:44:**:**:**, band: 0, freq: 2437
     kworker/1:1-33    [001] ....   135.070770: cfg80211_return_bss: b8:27:eb:**:**:**, band: 0, freq: 2442
     kworker/1:1-33    [001] ....   135.070778: cfg80211_return_bss: 46:d9:e7:**:**:**, band: 0, freq: 2437
     kworker/1:1-33    [001] ....   135.070788: cfg80211_return_bss: 8e:15:44:**:**:**, band: 0, freq: 2437
     kworker/1:1-33    [001] ....   135.070796: cfg80211_return_bss: b8:27:eb:**:**:**, band: 0, freq: 2442
     kworker/1:1-33    [001] ....   135.070806: cfg80211_return_bss: a0:f3:c1:**:**:**, band: 0, freq: 2462
     kworker/1:1-33    [001] ....   135.070824: cfg80211_return_bss: 88:15:44:**:**:**, band: 0, freq: 2462
     kworker/1:1-33    [001] ....   135.070835: cfg80211_return_bss: 8e:15:44:**:**:**, band: 0, freq: 2462
     kworker/1:1-33    [001] ....   135.070845: cfg80211_return_bss: 8e:15:44:**:**:**, band: 0, freq: 2462
     kworker/1:1-33    [001] ....   135.070855: cfg80211_return_bss: e0:91:f5:**:**:**, band: 0, freq: 2462
     kworker/1:1-33    [001] ....   135.070864: cfg80211_scan_done: aborted: false, scan start (TSF): 0, tsf_bssid: 00:00:00:00:00:00
```

JamesH65, 2017-05-19:
@herbertp Is this still an issue? Did you try @pelwell suggestions? Anything to report?

Which kernel version are you using?

ED6E0F17, 2017-05-20:
I was trying to ignore this harmless warning, but I am still seeing it on Linux 4.12 with ZeroW
````
/ # mount -t debugfs none /sys/kernel/debug
/ # ifconfig wlan0 up
/ # echo 1 >/sys/kernel/debug/tracing/events/cfg80211/enable
/ # /system/bin/iw wlan0 connect -w honeypot
wlan0 (phy #0): connected to 6a:9c:a6:f2:f3:35
/ # dmesg|tail -n 30
[    2.212255] dwc2 20980000.usb: new device is high-speed
[    2.214945] Freeing unused kernel memory: 1024K
[    2.319083] mmc1: new high speed SDIO card at address 0001
[    2.372373] dwc2 20980000.usb: new device is high-speed
[    2.477505] dwc2 20980000.usb: new address 1
[    2.496259] g_cdc gadget: high-speed config #1: CDC Composite (ECM + ACM)
[    2.522359] brcmfmac: Firmware version = wl0: May 27 2016 00:13:38 version 7.45.41.26 (r640327) FWID 01-df77e4a7
[   14.020051] random: crng init done
[  108.794798] brcmfmac: power management disabled
[  108.797633] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[  147.455393] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[  147.455861] ------------[ cut here ]------------
[  147.455927] WARNING: CPU: 0 PID: 23 at /linux/net/wireless/sme.c:756 __cfg80211_connect_result+0x378/0x3e8
[  147.455945] Modules linked in:
[  147.455976] CPU: 0 PID: 23 Comm: kworker/u2:1 Not tainted 4.12.0-rc1-armv6+ #63
[  147.455989] Hardware name: BCM2835
[  147.456033] Workqueue: cfg80211 cfg80211_event_work
[  147.456106] [<c010f08c>] (unwind_backtrace) from [<c010c724>] (show_stack+0x20/0x24)
[  147.456155] [<c010c724>] (show_stack) from [<c04ce85c>] (dump_stack+0x20/0x28)
[  147.456194] [<c04ce85c>] (dump_stack) from [<c011caf8>] (__warn+0xe4/0x110)
[  147.456223] [<c011caf8>] (__warn) from [<c011cbf0>] (warn_slowpath_null+0x30/0x38)
[  147.456256] [<c011cbf0>] (warn_slowpath_null) from [<c0a59118>] (__cfg80211_connect_result+0x378/0x3e8)
[  147.456296] [<c0a59118>] (__cfg80211_connect_result) from [<c0a2c5e4>] (cfg80211_process_wdev_events+0x19c/0x228)
[  147.456445] [<c0a2c5e4>] (cfg80211_process_wdev_events) from [<c0a2c6ac>] (cfg80211_process_rdev_events+0x3c/0x74)
[  147.456491] [<c0a2c6ac>] (cfg80211_process_rdev_events) from [<c0a26518>] (cfg80211_event_work+0x24/0x2c)
[  147.456528] [<c0a26518>] (cfg80211_event_work) from [<c0136718>] (process_one_work+0x140/0x4c8)
[  147.456558] [<c0136718>] (process_one_work) from [<c0136adc>] (worker_thread+0x3c/0x634)
[  147.456592] [<c0136adc>] (worker_thread) from [<c013d5d8>] (kthread+0x120/0x15c)
[  147.456625] [<c013d5d8>] (kthread) from [<c0108288>] (ret_from_fork+0x14/0x2c)
[  147.456642] ---[ end trace 8a4aed0b1c487a85 ]---

/ # echo 0 >/sys/kernel/debug/tracing/events/cfg80211/enable
/ # cat /sys/kernel/debug/tracing/trace
# tracer: nop
#
# entries-in-buffer/entries-written: 4/4   #P:1
#
#                              _-----=> irqs-off
#                             / _----=> need-resched
#                            | / _---=> hardirq/softirq
#                            || / _--=> preempt-depth
#                            ||| /     delay
#           TASK-PID   CPU#  ||||    TIMESTAMP  FUNCTION
#              | |       |   ||||       |         |
              iw-327   [000] ....   146.876143: rdev_connect: phy0, netdev:wlan0(7), bssid: 00:00:00:00:00:00, ssid: honeypot, auth type: 8, privacy: false, wpa versions: 0, flags: 0, previous bssid: 00:00:00:00:00:00
              iw-327   [000] ....   146.918381: rdev_return_int: phy0, returned: 0
     kworker/0:2-128   [000] ....   147.455275: cfg80211_return_bss: 6a:9c:a6:f2:f3:35, band: 0, freq: 2442
    kworker/u2:1-23    [000] ....   147.455835: cfg80211_get_bss: phy0, band: 0, freq: 0, 6a:9c:a6:f2:f3:35, buf: 0x68, bss_type: 0, privacy: 2
/ # 
````
I have not tested other wifi devices to see if this is an upstream issue, or something specific to the RPI hardware.

ED6E0F17, 2017-05-21:
My laptop on Linux 4.10 (with an atheros wifi chip)connects to the same AP without the warning.

Adding brcmfmac tracing gives:
````
     kworker/0:1-13    [000] ....   132.871847: brcmf_hexdump: hexdump [addr=ce66a800, length=64]
     kworker/0:1-13    [000] ....   132.871856: brcmf_dbg: brcmf_inform_single_bss: bssid: 6a:9c:a6:f2:f3:35
     kworker/0:1-13    [000] ....   132.871866: brcmf_dbg: brcmf_inform_single_bss: Channel: 7(2442)
     kworker/0:1-13    [000] ....   132.871872: brcmf_dbg: brcmf_inform_single_bss: Capability: 401
     kworker/0:1-13    [000] ....   132.871878: brcmf_dbg: brcmf_inform_single_bss: Beacon interval: 100
     kworker/0:1-13    [000] ....   132.871883: brcmf_dbg: brcmf_inform_single_bss: Signal: -6000
     kworker/0:1-13    [000] ....   132.871910: cfg80211_return_bss: 6a:9c:a6:f2:f3:35, band: 0, freq: 2442
     kworker/0:1-13    [000] ....   132.871917: brcmf_dbg: brcmf_update_bss_info: Exit
     kworker/0:1-13    [000] .n..   132.871950: brcmf_dbg: brcmf_bss_connect_done: Report connect result - connection succeeded
    kworker/u2:0-5     [000] ....   132.872072: cfg80211_get_bss: phy0, band: 0, freq: 0, 6a:9c:a6:f2:f3:35, buf: 0x68, bss_type: 0, privacy: 2
     kworker/0:1-13    [000] ....   132.875718: brcmf_dbg: brcmf_bss_connect_done: Exit
     kworker/0:1-13    [000] ....   132.875743: brcmf_dbg: brcmf_net_setcarrier: Enter, bsscfgidx=0 carrier=1
     kworker/0:1-13    [000] ....   132.875756: brcmf_dbg: brcmf_txflowblock_if: enter: bsscfgidx=0 stop=0x0 reason=4 state=0
     kworker/0:1-13    [000] .n..   132.876444: brcmf_dbg: brcmf_fil_iovar_data_set: ifidx=0, name=nd_hostip_clear, len=0
    kworker/u2:1-23    [000] ....   132.876596: brcmf_hexdump: hexdump [addr=ce66ea80, length=12]

pelwell, 2017-05-23:
If I leave dhcpcd disabled. enable tracing and connect using `iw` I get:
```
  wpa_supplicant-447   [001] ....  1155.573909: rdev_connect: phy0, netdev:wlan0(3), bssid: 80:2a:a8:**:**:**, ssid: XXXXXXXX, auth type: 0, privacy: true, wpa versions: 2, flags: 0, previous bssid: 00:00:00:00:00:00
  wpa_supplicant-447   [001] ....  1155.579059: rdev_return_int: phy0, returned: 0
     kworker/3:2-123   [003] ....  1155.690906: cfg80211_return_bss: 80:2a:a8:**:**:**, band: 0, freq: 2462
    kworker/u8:1-117   [001] ....  1155.690997: cfg80211_get_bss: phy0, band: 0, freq: 0, 80:2a:a8:**:**:**, buf: 0x50, bss_type: 0, privacy: 2
    kworker/u8:1-117   [001] ....  1155.691007: cfg80211_return_bss: 80:2a:a8:**:**:**, band: 0, freq: 2462
```
Notice the rdev_connect message includes the bssid of the AP, but in your trace the corresponding ssid is all zeroes. I think this is what is causing the problem, but I can't explain the difference.

ED6E0F17, 2017-05-23:
I think that my bssid is showing as zeros because the AP  is "hidden", but I cannot say that  @herbertp has the same setup.

All I can tell so far is that "cfg80211_get_bss"  is returning  early with a NULL, before reaching "trace_cfg80211_return_bss(&res->pub);", which may be due to the hidden ssid.

pelwell, 2017-05-23:
Thanks for the hint - I'll experiment with hidden SSIDs.

herbertp, 2017-05-24:
On Tue, May 23, 2017 at 09:59:37AM -0700, ED6E0F17 wrote:
> I think that my bssid is showing as zeros because the AP is
> "hidden", but I cannot say that @herbertp has the same setup.

Sorry for not getting back earlier ...

Yes, the AP is "hidden" here to, so the ESSID doesn't 
show up in scans.

I will see if I can do some testing in the next few
days if it is still required.

Thanks,
Herbert

pelwell, 2017-05-24:
I've been experimenting with a NetGear AP, and even though I've disabled SSID broadcasting I still see the correct bssid value in the rdev_connect tracing. In a way that shouldn't be surprising because the WiFi stack must surely required a MAC address before it will consider connecting.

In my current testing I've disabled dhcpcd and edited /etc/network/interfaces to look like this:
```
pi@raspberrypi ~ $ cat /etc/network/interfaces
# interfaces(5) file used by ifup(8) and ifdown(8)

# Please note that this file is written to be used with dhcpcd
# For static IP, consult /etc/dhcpcd.conf and 'man dhcpcd.conf'

# Include files from /etc/network/interfaces.d:
source-directory /etc/network/interfaces.d

auto lo
iface lo inet loopback

iface eth0 inet manual

#allow-hotplug wlan0
#iface wlan0 inet manual
#    wpa-conf /etc/wpa_supplicant/wpa_supplicant.conf
#
#allow-hotplug wlan1
#iface wlan1 inet manual
#    wpa-conf /etc/wpa_supplicant/wpa_supplicant.conf

#auto wlan0
#allow-hotplug wlan0
iface wlan0 inet dhcp
   wpa-scan-ssid 1
   wpa-ap-scan 1
   wpa-key-mgmt WPA-PSK
   wpa-proto RSN WPA
   wpa-pairwise CCMP TKIP
   wpa-group CCMP TKIP
   wpa-ssid "PiTest"
   wpa-psk "supersecretpassword"
```
I can then boot without wlan0 coming up, but running `ifup wlan0` will connect to the AP. The test script is:
```
#!/bin/sh
sudo sh -c "echo 1 > /sys/kernel/debug/tracing/events/cfg80211/enable"
sudo ifup wlan0
sudo sh -c "echo 0 > /sys/kernel/debug/tracing/events/cfg80211/enable"
sudo cat /sys/kernel/debug/tracing/trace
```
Run it with `./wlantest > wlan_log.txt`.

If you don't mind, can you run my test or something equivalent  (@ED6E0F17 - perhaps just move the point where you enable tracing before the ifup) to capture more of the cfg80211 activity?

ED6E0F17, 2017-05-25:
I can use wpa_supplicant to connect to the AP without getting the WARN; I will do some more testing.

ED6E0F17, 2017-05-26:
I am not going to claim that this is a userspace issue, but wpa_supplicant seems to be doing something different than iw (and iwconfig from the initial report). There is a related issue that "iw wlan0 link" will fail if iw is used to connect to the hidden AP, but works perfectly well if wpa_supplicant makes the connection.

As mentioned above, the obvious difference is the zeroed bssid, but also the auth type:
````
  wpa_supplicant-700   [000] ....   473.889793: rdev_connect: phy0, netdev:wlan0(7), bssid: 6a:9c:a6:f2:f3:35, ssid: honeypot, auth type: 0, privacy: false, wpa versions: 0, flags: 0, previous bssid: 00:00:00:00:00:00
  wpa_supplicant-700   [000] ....   473.926417: rdev_return_int: phy0, returned: 0
     kworker/0:0-562   [000] ....   474.000242: cfg80211_return_bss: 6a:9c:a6:f2:f3:35, band: 0, freq: 2442
    kworker/u2:0-5     [000] ....   474.000597: cfg80211_get_bss: phy0, band: 0, freq: 0, 6a:9c:a6:f2:f3:35, buf: 0x68, bss_type: 0, privacy: 2
    kworker/u2:0-5     [000] ....   474.000613: cfg80211_return_bss: 6a:9c:a6:f2:f3:35, band: 0, freq: 2442
  wpa_supplicant-700   [000] ....   474.002766: rdev_change_station: phy0, netdev:wlan0(7), station mac: 6a:9c:a6:f2:f3:35, station flags mask: 2, station flags set: 2, station modify mask: 0, listen interval: -1, aid: 0, plink action: 0, plink state: 0, uapsd queues: 0, vlan:
  wpa_supplicant-700   [000] ....   474.003706: rdev_return_int: phy0, returned: 0

              iw-972   [000] ....   719.058571: rdev_connect: phy0, netdev:wlan0(7), bssid: 00:00:00:00:00:00, ssid: honeypot, auth type: 8, privacy: false, wpa versions: 0, flags: 0, previous bssid: 00:00:00:00:00:00
              iw-972   [000] ....   719.101307: rdev_return_int: phy0, returned: 0
     kworker/0:3-878   [000] ....   719.666659: cfg80211_return_bss: 6a:9c:a6:f2:f3:35, band: 0, freq: 2442
    kworker/u2:1-702   [000] ....   719.666979: cfg80211_get_bss: phy0, band: 0, freq: 0, 6a:9c:a6:f2:f3:35, buf: 0x68, bss_type: 0, privacy: 2

ED6E0F17, 2017-05-26:
Resolved with:
````
diff --git a/net/wireless/scan.c b/net/wireless/scan.c
--- a/net/wireless/scan.c
+++ b/net/wireless/scan.c
@@ -544,6 +544,8 @@ static bool is_bss(struct cfg80211_bss *a, const u8 *bssid,
                return false;
        if (ssidie[1] != ssid_len)
                return false;
+       if (!ssidie[2]) /* Hidden AP */
+               return true;
        return memcmp(ssidie + 2, ssid, ssid_len) == 0;
 }

````
giving:
````
/ # /system/bin/iw wlan0 link
Connected to 6a:9c:a6:f2:f3:35 (on wlan0)
        SSID: \x00\x00\x00\x00\x00\x00\x00\x00
        SSID: honeypot
        freq: 2442
        RX: 202 bytes (3 packets)
        TX: 798 bytes (9 packets)
        signal: -62 dBm
        tx bitrate: 65.0 MBit/s
````
The underlying issue may be related to https://github.com/raspberrypi/linux/issues/1742

pelwell, 2017-05-27:
Interesting. Do the kernel backtraces also cease with your patch?

JamesH65, 2017-09-13:
@pelwell @ED6E0F17 Any news on this? Does Stretch make any difference? Can it be closed?

pelwell, 2017-09-13:
I was waiting for a response to my question, but I would guess that answer would have been "Yes". The next step would be the creation of a proper patch with an explanation of the problem and why this is the correct solution, then it needs to go upstream.

JamesH65, 2017-12-04:
@ED6E0F17 We need a response really, to progress any further.

herbertp, 2017-12-05:
What is the simplest way to test?
I.e. what do I need to install to test the patch?

Thanks in advance,
Herbert

ED6E0F17, 2017-12-06:
I remember spending a lot of time looking at the code, and I didn`t get any closer to understanding it, but  I think this is a symptom of a bug somewhere else in the wireless stack. I noticed that the Broadcom code was getting a lot of refactoring, but I am not actively testing new kernels to see if this issue has been fixed upstream.

ED6E0F17, 2017-12-06:
The main reason that I lost interest in this issue is that it does not affect wpa_supplicant, which is required for wpa2 - so this is a harmless warning that only affects hidden APs with wep or no encryption.

JamesH65, 2018-05-11:
@pelwell ISTR a recent issue with hidden SSID's, was that only on the 3B+? Perhaps related?

ED6E0F17, 2018-05-12:
I get the same result on 3B+

````
[  124.033832] [<ffffff98e93651a8>] __cfg80211_connect_result+0x2a0/0x350
[  124.033846] [<ffffff98e93407b8>] cfg80211_process_wdev_events+0x130/0x180
[  124.033860] [<ffffff98e934083c>] cfg80211_process_rdev_events+0x34/0x70
[  124.033873] [<ffffff98e933b7bc>] cfg80211_event_work+0x1c/0x30
[  124.033887] [<ffffff98e8ab4adc>] process_one_work+0x1e4/0x360
[  124.033899] [<ffffff98e8ab4ca0>] worker_thread+0x48/0x470
[  124.033913] [<ffffff98e8aba954>] kthread+0x12c/0x130
[  124.033928] [<ffffff98e8a84bd8>] ret_from_fork+0x10/0x18
````
(Firmware version = wl0: Feb 27 2018 03:15:32 version 7.45.154 (r684107 CY) FWID 01-4fbe0b04)

rinllow, 2018-05-16:
[ 1986.561944] [<c0017f0c>] (unwind_backtrace) from [<c00135c8>] (show_stack+0x10/0x14)
[ 1986.569699] [<c00135c8>] (show_stack) from [<c06397ec>] (dump_stack+0x90/0xa4)
[ 1986.576934] [<c06397ec>] (dump_stack) from [<c0027adc>] (warn_slowpath_common+0x80/0xb0)
[ 1986.585034] [<c0027adc>] (warn_slowpath_common) from [<c0027bf0>] (warn_slowpath_null+0x1c/0x24)
[ 1986.593829] [<c0027bf0>] (warn_slowpath_null) from [<c0611088>] (cfg80211_process_wdev_events+0x188/0x190)
[ 1986.603494] [<c0611088>] (cfg80211_process_wdev_events) from [<c06110bc>] (cfg80211_process_rdev_events+0x2c/0x60)
[ 1986.613853] [<c06110bc>] (cfg80211_process_rdev_events) from [<c060ca70>] (cfg80211_event_work+0x14/0x1c)
[ 1986.623433] [<c060ca70>] (cfg80211_event_work) from [<c003bfe8>] (process_one_work+0x138/0x348)
[ 1986.632142] [<c003bfe8>] (process_one_work) from [<c003c248>] (worker_thread+0x50/0x4f8)
[ 1986.640244] [<c003c248>] (worker_thread) from [<c00412ac>] (kthread+0xf0/0x108)
[ 1986.647563] [<c00412ac>] (kthread) from [<c00101e8>] (ret_from_fork+0x14/0x2c)

ED6E0F17, 2018-05-19:
I believe that the brcmfmac driver is reading the SSID from the firmware in the function  brcmf_get_assoc_ies(), and that may be where it is getting the zeroed-out SSID that it is later trying to compare to the Hidden SSID.

````
[   62.990310] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[  105.347026] brcmfmac: brcmf_set_wpa_version setting wpa_auth to 0x0
[  105.348032] brcmfmac: brcmf_set_auth_type automatic, auth type (8)
[  105.348998] brcmfmac: brcmf_set_wsec_mode pval (0) gval (0)
[  105.349977] brcmfmac: brcmf_set_sharedkey key len (0)
[  105.350001] brcmfmac: brcmf_cfg80211_connect SSID "honeypot", len (8)
[  105.916385] brcmfmac: brcmf_is_linkup Processing set ssid
[  105.916405] brcmfmac: brcmf_notify_connect_status Linkup
[  105.919988] brcmfmac: brcmf_get_assoc_ies req len (118) resp len (147)
[  105.921224] brcmfmac: brcmf_inform_single_bss bssid: 84:9c:a6:f2:f3:34
[  105.921247] brcmfmac: brcmf_inform_single_bss Channel: 11(2462)
[  105.921258] brcmfmac: brcmf_inform_single_bss Capability: 401
[  105.921267] brcmfmac: brcmf_inform_single_bss Beacon interval: 100
[  105.921276] brcmfmac: brcmf_inform_single_bss Signal: -6800
[  105.921340] brcmfmac: brcmf_bss_connect_done Report connect result - connection succeeded
[  105.921425] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[  105.924831] ------------[ cut here ]------------
[  105.931083] WARNING: CPU: 0 PID: 23 at net/wireless/sme.c:752 __cfg80211_connect_result+0x384/0x3f4
[  105.937898] Modules linked in:
[  105.941498] CPU: 0 PID: 23 Comm: kworker/u2:1 Not tainted 4.16.9-armv6+ #7

chakradivi, 2018-07-03:
Hi All,

I'm doing the testing on Raspberry Pi Zero W, and still seeing the kernel tainting issue while turning wlan0 down and up "ifdown -a && rm -rf /var/run/wpa_supplicant/wlan0 && ifup -a" .

I'm using latest Linux kernel from Buildroot "Linux buildroot 4.14.39" and wifi firmware version " Firmware version = wl0: Oct 23 2017 03:55:53 version 7.45.98.38 (r674442 CY) FWID 01-e58d219f".

Testing process:
Everytime i turned wlan0 down and then up - udhcpc says : no lease, failing
After few iterations - kernel is getting tainted and scan errors 

[ 1326.465528] ------------[ cut here ]------------
[ 1326.473513] WARNING: CPU: 0 PID: 91 at net/wireless/sme.c:752 __cfg80211_connect_result+0x368/0x3d8 [cfg80211]
[ 1326.488505] Modules linked in: 8021q garp stp llc ipv6 brcmfmac brcmutil cfg80211 rfkill
[ 1326.501727] CPU: 0 PID: 91 Comm: kworker/u2:2 Tainted: G        W       4.14.39 #1
[ 1326.512116] Hardware name: BCM2835
[ 1326.519273] Workqueue: cfg80211 cfg80211_event_work [cfg80211]
[ 1326.528205] [<c0016298>] (unwind_backtrace) from [<c0013bdc>] (show_stack+0x20/0x24)
[ 1326.538998] [<c0013bdc>] (show_stack) from [<c0638ab4>] (dump_stack+0x20/0x28)
[ 1326.549293] [<c0638ab4>] (dump_stack) from [<c0021f4c>] (__warn+0xe4/0x10c)
[ 1326.559366] [<c0021f4c>] (__warn) from [<c0022040>] (warn_slowpath_null+0x30/0x38)
[ 1326.571078] [<c0022040>] (warn_slowpath_null) from [<bf0391f4>] (__cfg80211_connect_result+0x368/0x3d8 [cfg80211])
[ 1326.589613] [<bf0391f4>] (__cfg80211_connect_result [cfg80211]) from [<bf010c28>] (cfg80211_process_wdev_events+0x13c/0x1b4 [cfg80211])
[ 1326.610460] [<bf010c28>] (cfg80211_process_wdev_events [cfg80211]) from [<bf010cdc>] (cfg80211_process_rdev_events+0x3c/0x74 [cfg80211])
[ 1326.631813] [<bf010cdc>] (cfg80211_process_rdev_events [cfg80211]) from [<bf00b208>] (cfg80211_event_work+0x24/0x2c [cfg80211])
[ 1326.652031] [<bf00b208>] (cfg80211_event_work [cfg80211]) from [<c003940c>] (process_one_work+0x1e4/0x458)
[ 1326.669811] [<c003940c>] (process_one_work) from [<c00396c8>] (worker_thread+0x48/0x5a8)
[ 1326.686288] [<c00396c8>] (worker_thread) from [<c003f4ec>] (kthread+0x124/0x164)
[ 1326.698160] [<c003f4ec>] (kthread) from [<c000fbec>] (ret_from_fork+0x14/0x28)
[ 1326.709785] ---[ end trace 19a248e831285db0 ]---
 
[ 1326.265587] brcmfmac: brcmf_cfg80211_escan: Connecting: status (3)
[ 1326.274128] brcmfmac: brcmf_cfg80211_scan: scan error (-11)
[ 1326.282327] brcmfmac: brcmf_cfg80211_escan: Connecting: status (3)
[ 1326.290863] brcmfmac: brcmf_cfg80211_scan: scan error (-11)

Once failed - it is getting fixed only after reboot of the board.

Is this issue related rpi wifi firmware or the mainline kernel ?

ED6E0F17, 2018-07-06:
The firmware must be good enough, because at least one piece of code is working normally.

The driver is connecting to the AP but failing to report that properly because the reported SSID does not match the requested SSID. I am leaning towards a race condition in the driver or with the wifi chip.

There have been some small changes in the upstream code that I can test, and I do get a better understanding of the Linux wifi stack every time I look at this.

Evanok, 2018-10-19:
Any update on this warning ?

JamesH65, 2018-10-19:
Might be worth trying the latest bleeding edge 4.18 kernel to see if it helps.

pelwell, 2018-10-19:
...which you can download with:
```
$ sudo BRANCH=next rpi-update
```

JamesH65, 2019-07-30:
Lots of changes in the driver since this issue was last visited. Please try the latest kernel code, and report back any issues.

This issue will be closed within 30 days unless further interactions are posted. If you wish this issue to remain open, please add a comment. A closed issue may be reopened if requested.

kamarya, 2019-08-01:
I have the same issue with 4.19.58 and 4.19.60. I do not see @ED6E0F17 patch in there.
I have also a more complicated situation reported in #3114 that might be related to this issue.

satmandu, 2019-08-07:
I've been seeing a similar error on arm64 4.19.64-v8-g76b2727ef (and earlier 4.19 builds this past week) on a Raspberry Pi 4B. I'm just connecting to an access point, which has a somewhat low signal since this RPI4B is now in a metal case.
```
[ 3803.041177] WARNING: CPU: 3 PID: 23210 at net/wireless/sme.c:752 __cfg80211_connect_result+0x338/0x358 [cfg80211]
[ 3803.041184] Modules linked in: ipt_MASQUERADE nf_conntrack_netlink nfnetlink iptable_nat nf_nat_ipv4 xt_addrtype iptable_filter xt_conntrack nf_nat nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 br_netfilter bridge stp llc overlay binfmt_misc dm_mod bcm2835_v4l2(C) bcm2835_mmal_vchiq(C) v4l2_common videobuf2_vmalloc videobuf2_memops videobuf2_v4l2 videobuf2_common sg videodev media brcmfmac vc_sm_cma(C) spidev brcmutil sha256_generic cfg80211 rfkill vchiq(C) i2c_bcm2835 fixed spi_bcm2835 uio_pdrv_genirq uio sch_fq_codel ip_tables x_tables ipv6
[ 3803.041414] CPU: 3 PID: 23210 Comm: kworker/u8:1 Tainted: G        WC        4.19.64-v8-g76b2727ef #1
[ 3803.041421] Hardware name: Raspberry Pi 4 Model B Rev 1.1 (DT)
[ 3803.041498] Workqueue: cfg80211 cfg80211_event_work [cfg80211]
[ 3803.041514] pstate: 00000005 (nzcv daif -PAN -UAO)
[ 3803.041597] pc : __cfg80211_connect_result+0x338/0x358 [cfg80211]
[ 3803.041676] lr : __cfg80211_connect_result+0x338/0x358 [cfg80211]
[ 3803.041682] sp : ffffff800d99bcc0
[ 3803.041689] x29: ffffff800d99bcc0 x28: ffffff8008c16000 
[ 3803.041706] x27: ffffff8011c33cb8 x26: ffffffc0b9b25db8 
[ 3803.041722] x25: dead000000000100 x24: dead000000000200 
[ 3803.041738] x23: ffffff800d99bd08 x22: ffffff8008c18688 
[ 3803.041753] x21: ffffffc0bb3e6000 x20: ffffffc0bb3e7008 
[ 3803.041769] x19: ffffffc0a42f7a18 x18: 0000000000000000 
[ 3803.041784] x17: 0000000000000000 x16: 0000000000000000 
[ 3803.041790] x15: ffffff8008888408 x14: 21dd00000007430c 
[ 3803.041796] x13: 0000000000000000 x12: ffffff8008d11000 
[ 3803.041802] x11: ffffff8008c2c000 x10: ffffff8008d11f20 
[ 3803.041808] x9 : 0000000000000000 x8 : 0000000000000004 
[ 3803.041815] x7 : ffffff8008d11000 x6 : 00000040b73ad000 
[ 3803.041821] x5 : 0000000000000002 x4 : ffffffc0bffac9e0 
[ 3803.041827] x3 : ffffff8008bfe018 x2 : a67f504a94025600 
[ 3803.041833] x1 : 0000000000000000 x0 : 0000000000000024 
[ 3803.041839] Call trace:
[ 3803.041872]  __cfg80211_connect_result+0x338/0x358 [cfg80211]
[ 3803.041902]  cfg80211_process_wdev_events+0x120/0x188 [cfg80211]
[ 3803.041931]  cfg80211_process_rdev_events+0x40/0x98 [cfg80211]
[ 3803.041960]  cfg80211_event_work+0x24/0x38 [cfg80211]
[ 3803.041968]  process_one_work+0x1c4/0x420
[ 3803.041971]  worker_thread+0x50/0x408
[ 3803.041975]  kthread+0x108/0x138
[ 3803.041980]  ret_from_fork+0x10/0x1c
[ 3803.041983] ---[ end trace b35e6f498f375be0 ]---
```

Here is dmesg: https://paste.ubuntu.com/p/2b6YSz5XtX/
and debug trace: https://paste.ubuntu.com/p/2QTRjNx5vp/

DenyDarko, 2019-09-02:
Same for me on Pi3B+.
It started with those warnings after the update to Buster:

`Sep  2 16:44:15 raspberrypi kernel: [98886.816211] bond0: Releasing backup interface eth0
Sep  2 16:44:15 raspberrypi kernel: [98886.816226] bond0: making interface wlan0 the new active one
Sep  2 16:44:26 raspberrypi kernel: [98897.538659] bond0: Releasing backup interface wlan0
Sep  2 16:44:37 raspberrypi kernel: [98908.048919] brcmfmac: power management disabled
Sep  2 16:44:42 raspberrypi kernel: [98913.427638] ------------[ cut here ]------------
Sep  2 16:44:42 raspberrypi kernel: [98913.427775] WARNING: CPU: 2 PID: 25435 at net/wireless/sme.c:752 __cfg80211_connect_result+0x3c0/0x410 [cfg80211]
Sep  2 16:44:42 raspberrypi kernel: [98913.427778] Modules linked in: bonding nft_counter ipt_REJECT nf_reject_ipv4 xt_tcpudp xt_state xt_conntrack nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 nft_compat nf_tables nfnetlink evdev r820t rt$
Sep  2 16:44:42 raspberrypi kernel: [98913.427899] CPU: 2 PID: 25435 Comm: kworker/u8:0 Tainted: G        WC        4.19.66-v7+ #1253
Sep  2 16:44:42 raspberrypi kernel: [98913.427901] Hardware name: BCM2835
Sep  2 16:44:42 raspberrypi kernel: [98913.427996] Workqueue: cfg80211 cfg80211_event_work [cfg80211]
Sep  2 16:44:42 raspberrypi kernel: [98913.428015] [<80111f38>] (unwind_backtrace) from [<8010d4b0>] (show_stack+0x20/0x24)
Sep  2 16:44:42 raspberrypi kernel: [98913.428024] [<8010d4b0>] (show_stack) from [<808191e0>] (dump_stack+0xd4/0x118)
Sep  2 16:44:42 raspberrypi kernel: [98913.428033] [<808191e0>] (dump_stack) from [<801209c8>] (__warn+0x104/0x11c)
Sep  2 16:44:42 raspberrypi kernel: [98913.428040] [<801209c8>] (__warn) from [<80120b18>] (warn_slowpath_null+0x50/0x58)
Sep  2 16:44:42 raspberrypi kernel: [98913.428128] [<80120b18>] (warn_slowpath_null) from [<7f28b6f8>] (__cfg80211_connect_result+0x3c0/0x410 [cfg80211])
Sep  2 16:44:42 raspberrypi kernel: [98913.428295] [<7f28b6f8>] (__cfg80211_connect_result [cfg80211]) from [<7f25b368>] (cfg80211_process_wdev_events+0x104/0x160 [cfg80211])
Sep  2 16:44:42 raspberrypi kernel: [98913.428461] [<7f25b368>] (cfg80211_process_wdev_events [cfg80211]) from [<7f25b404>] (cfg80211_process_rdev_events+0x40/0x98 [cfg80211])
Sep  2 16:44:42 raspberrypi kernel: [98913.428625] [<7f25b404>] (cfg80211_process_rdev_events [cfg80211]) from [<7f2552d4>] (cfg80211_event_work+0x24/0x2c [cfg80211])
Sep  2 16:44:42 raspberrypi kernel: [98913.428712] [<7f2552d4>] (cfg80211_event_work [cfg80211]) from [<8013bf0c>] (process_one_work+0x170/0x458)
Sep  2 16:44:42 raspberrypi kernel: [98913.428720] [<8013bf0c>] (process_one_work) from [<8013c250>] (worker_thread+0x5c/0x5a4)
Sep  2 16:44:42 raspberrypi kernel: [98913.428728] [<8013c250>] (worker_thread) from [<8014253c>] (kthread+0x138/0x168)
Sep  2 16:44:42 raspberrypi kernel: [98913.428736] [<8014253c>] (kthread) from [<801010ac>] (ret_from_fork+0x14/0x28)
Sep  2 16:44:42 raspberrypi kernel: [98913.428740] Exception stack(0x8d73dfb0 to 0x8d73dff8)
Sep  2 16:44:42 raspberrypi kernel: [98913.428744] dfa0:                                     00000000 00000000 00000000 00000000
Sep  2 16:44:42 raspberrypi kernel: [98913.428749] dfc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
Sep  2 16:44:42 raspberrypi kernel: [98913.428753] dfe0: 00000000 00000000 00000000 00000000 00000013 00000000
Sep  2 16:44:42 raspberrypi kernel: [98913.428757] ---[ end trace e699c425d6179cb5 ]---
Sep  2 16:44:47 raspberrypi kernel: [98918.352822] bond0: Enslaving eth0 as a backup interface with an up link
Sep  2 16:44:47 raspberrypi kernel: [98918.904660] brcmfmac: power management disabled
Sep  2 16:44:47 raspberrypi kernel: [98918.905140] bond0: Enslaving wlan0 as a backup interface with a down link
Sep  2 16:44:47 raspberrypi kernel: [98918.951085] bond0: link status definitely up for interface eth0, 100 Mbps full duplex
Sep  2 16:44:47 raspberrypi kernel: [98918.951097] bond0: making interface eth0 the new active one
Sep  2 16:44:47 raspberrypi kernel: [98918.951186] bond0: first active interface up!
Sep  2 16:44:53 raspberrypi kernel: [98924.211041] ------------[ cut here ]------------
Sep  2 16:44:53 raspberrypi kernel: [98924.211194] WARNING: CPU: 2 PID: 25783 at net/wireless/sme.c:752 __cfg80211_connect_result+0x3c0/0x410 [cfg80211]
Sep  2 16:44:53 raspberrypi kernel: [98924.211202] Modules linked in: bonding nft_counter ipt_REJECT nf_reject_ipv4 xt_tcpudp xt_state xt_conntrack nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 nft_compat nf_tables nfnetlink evdev r820t rt$
Sep  2 16:44:53 raspberrypi kernel: [98924.211337] CPU: 2 PID: 25783 Comm: kworker/u8:2 Tainted: G        WC        4.19.66-v7+ #1253
Sep  2 16:44:53 raspberrypi kernel: [98924.211341] Hardware name: BCM2835
Sep  2 16:44:53 raspberrypi kernel: [98924.211446] Workqueue: cfg80211 cfg80211_event_work [cfg80211]
Sep  2 16:44:53 raspberrypi kernel: [98924.211468] [<80111f38>] (unwind_backtrace) from [<8010d4b0>] (show_stack+0x20/0x24)
Sep  2 16:44:53 raspberrypi kernel: [98924.211482] [<8010d4b0>] (show_stack) from [<808191e0>] (dump_stack+0xd4/0x118)
Sep  2 16:44:53 raspberrypi kernel: [98924.211494] [<808191e0>] (dump_stack) from [<801209c8>] (__warn+0x104/0x11c)
Sep  2 16:44:53 raspberrypi kernel: [98924.211503] [<801209c8>] (__warn) from [<80120b18>] (warn_slowpath_null+0x50/0x58)
Sep  2 16:44:53 raspberrypi kernel: [98924.211598] [<80120b18>] (warn_slowpath_null) from [<7f28b6f8>] (__cfg80211_connect_result+0x3c0/0x410 [cfg80211])
[... cut ...]
Sep  2 16:44:53 raspberrypi kernel: [98924.212238] 5fc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
Sep  2 16:44:53 raspberrypi kernel: [98924.212242] 5fe0: 00000000 00000000 00000000 00000000 00000013 00000000
Sep  2 16:44:53 raspberrypi kernel: [98924.212275] ---[ end trace e699c425d6179cb6 ]---
Sep  2 16:44:53 raspberrypi kernel: [98924.240971] bond0: link status up for interface wlan0, enabling it in 200 ms
Sep  2 16:44:53 raspberrypi kernel: [98924.461041] bond0: link status definitely up for interface wlan0, 0 Mbps full duplex
`

scyto, 2019-11-11:
@DenyDarko  I am also getting net/wireless/sme.c:752 __cfg80211_connect_result+0x3c0/0x410 [cfg80211] on a pi4 i just installed using wla0.

i see the commands for the debug trace
```
sudo sh -c "echo 1 > /sys/kernel/debug/tracing/events/cfg80211/enable"
```
if i run this will a reboot of the pi wipe the trace or does it persist reboots (the pi4 is only wifi connected given its location).

scyto, 2019-11-11:
DenyDark and mine could be related to this https://github.com/raspberrypi/linux/issues/3318 as this is the same connect_result of 0x3c0/0x410

pelwell, 2019-11-11:
Event tracing enabled through sysfs is not persistent. You can normally set it persistently via cmdline.txt - use the `trace_event=` option (N.B. not `trace_events` as you might think). However, cfg80211 appears to be loaded so late that this mechanism doesn't work.

scyto, 2019-11-12:
@pelwell thanks for the info, i have tried disabling all features on my wifi equipment i thought could contribute (Unscheduled Automatic Power Save Delivery, fast roaming 80211r, multicast enhancements).  My next test will be to disable CCMP encryption.... not sure what is left after that.

When the unit is inside on 5ghz i have no issue, when the unit is outside and only able to make a 2.4ghz connection i have issues.

What interesting is though the pi has no TCP/UDP connectivity I can still see it roaming between APs if i take the AP it is joined to offline.  So my gut says this is related to wifi protocol and features itself.

Any suggestions how to capture the issue in the trace without stringing a 100m CAT6 to outside in the rain :-)

ilinty74, 2020-04-21:
Same issue for me, seems raspberry is not suitable for any interesting project. 
any suggestions ?

Apr 21 21:23:37 berryone kernel: [  192.602922] ------------[ cut here ]------------
Apr 21 21:23:37 berryone kernel: [  192.604422] WARNING: CPU: 0 PID: 122 at net/wireless/sme.c:1088 __cfg80211_disconnected+0x398/0x400 [cfg80211]
Apr 21 21:23:37 berryone kernel: [  192.604444] Modules linked in: psnap llc ax25 8192cu sha256_generic libsha256 cfg80211 rfkill raspberrypi_hwmon hwmon bcm2835_codec(C) bcm2835_v4l2(C) v4l2_mem2mem bcm2835_mmal_vchiq(C) videobuf2_vmalloc videobuf2_dma_contig videobuf2_memops videobuf2_v4l2 snd_bcm2835(C) videobuf2_common snd_pcm snd_timer snd videodev vc_sm_cma(C) mc uio_pdrv_genirq fixed uio i2c_dev ip_tables x_tables ipv6 nf_defrag_ipv6
Apr 21 21:23:37 berryone kernel: [  192.604649] CPU: 0 PID: 122 Comm: kworker/u2:2 Tainted: G         C        5.4.32+ #1308

JamesH65, 2020-04-22:
> any interesting project

That's quite a broad statement.

scyto, 2020-04-22:
i have a thesis, i have yet to test fully

I had similar issues with a old carrier thermostat where it would associate with the AP but never connect.

After  doing captures at the AP it was apparent that the thermostat didn't like certain types of broadcast DHCP offers from a win 2k19 dhcp server.
adding a DHCP relay to my network fixed it (even though it is all one subnet).
changing DHCP server to my UniFi router also fixed it.  
(note i was forced to analyze this because the thermostats static addressing didn't work at all!)

So my thesis is there is an issue in some of the open source dhcp client stacks that are incompatible/have bug with certain types of offer packets.  I still need to validate if the relay fixed my pi + wifi issues or not.

This may not be your issue YMMV.

BuMRK, 2020-05-20:
[   18.130122] WARNING: CPU: 2 PID: 37 at net/wireless/sme.c:1088 __cfg80211_disconnected+0x554/0x5a0 [cfg80211]
[   18.130133] Modules linked in: w1_therm w1_gpio wire cn ip6table_filter ip6_tables xt_MASQUERADE iptable_nat nf_nat nf_conntrack nf_defrag_ipv4 xt_comment xt_tcpudp xt_multiport iptable_filter binfmt_misc joydev spidev 8192cu sha256_generic libsha256 evdev cfg80211 rfkill raspberrypi_hwmon raspberrypi_ts hwmon input_polldev bcm2835_codec(C) bcm2835_isp(C) bcm2835_v4l2(C) v4l2_mem2mem i2c_bcm2835 bcm2835_mmal_vchiq(C) spi_bcm2835 videobuf2_vmalloc videobuf2_dma_contig videobuf2_memops videobuf2_v4l2 snd_bcm2835(C) videobuf2_common snd_pcm snd_timer videodev snd mc vc_sm_cma(C) fixed rpi_backlight backlight uio_pdrv_genirq uio i2c_bcm2708 i2c_dev ip_tables x_tables ipv6 nf_defrag_ipv6
[   18.130342] CPU: 2 PID: 37 Comm: kworker/u8:1 Tainted: G         C        5.4.42-v7+ #1319
[   18.130347] Hardware name: BCM2835
[   18.130664] Workqueue: cfg80211 cfg80211_event_work [cfg80211]
[   18.130673] Backtrace:
[   18.130697] [<8010d458>] (dump_backtrace) from [<8010d750>] (show_stack+0x20/0x24)
[   18.130708]  r6:aa2b6000 r5:00000000 r4:80d93a34 r3:b477c3ae
[   18.130721] [<8010d730>] (show_stack) from [<808ae684>] (dump_stack+0xe0/0x124)
[   18.130734] [<808ae5a4>] (dump_stack) from [<8011fd24>] (__warn+0xec/0x104)
[   18.130745]  r8:00000440 r7:00000009 r6:7f310168 r5:00000000 r4:00000000 r3:b477c3ae
[   18.130756] [<8011fc38>] (__warn) from [<8011fdf4>] (warn_slowpath_fmt+0xb8/0xc0)
[   18.130767]  r9:7f310168 r8:00000440 r7:7f2e58c8 r6:00000009 r5:00000000 r4:80d04f48
[   18.131017] [<8011fd40>] (warn_slowpath_fmt) from [<7f2e58c8>] (__cfg80211_disconnected+0x554/0x5a0 [cfg80211])
[   18.131030]  r9:00000000 r8:00000000 r7:a4c3fc24 r6:a4c3fc00 r5:aa08b800 r4:a8a1c1c0
[   18.131415] [<7f2e5374>] (__cfg80211_disconnected [cfg80211]) from [<7f2b0b34>] (cfg80211_process_wdev_events+0x13c/0x160 [cfg80211])
[   18.131426]  r10:00000100 r9:00000122 r8:a4c3fc8c r7:a4c3fc24 r6:a4c3fc94 r5:a4c3fc00
[   18.131431]  r4:a2cf2680
[   18.131807] [<7f2b09f8>] (cfg80211_process_wdev_events [cfg80211]) from [<7f2b0b98>] (cfg80211_process_rdev_events+0x40/0x98 [cfg80211])
[   18.131819]  r10:00000000 r9:00000100 r8:00000000 r7:a9a3a900 r6:aa080400 r5:a8a1c460
[   18.131823]  r4:a4c3fc00
[   18.132198] [<7f2b0b58>] (cfg80211_process_rdev_events [cfg80211]) from [<7f2aa1b8>] (cfg80211_event_work+0x24/0x2c [cfg80211])
[   18.132205]  r5:aa262f80 r4:a8a1c0e4
[   18.132403] [<7f2aa194>] (cfg80211_event_work [cfg80211]) from [<8013bd28>] (process_one_work+0x17c/0x4b4)
[   18.132409]  r4:a8a1c0e4 r3:7f2aa194
[   18.132421] [<8013bbac>] (process_one_work) from [<8013c590>] (worker_thread+0x54/0x5b4)
[   18.132432]  r10:aa262f80 r9:aa2b6038 r8:80d03d00 r7:aa080418 r6:00000088 r5:aa262f94
[   18.132436]  r4:aa080400
[   18.132448] [<8013c53c>] (worker_thread) from [<80142974>] (kthread+0x13c/0x168)
[   18.132460]  r10:aa21e19c r9:aa103e74 r8:8013c53c r7:aa262f80 r6:00000000 r5:aa2766c0
[   18.132464]  r4:aa21e180
[   18.132475] [<80142838>] (kthread) from [<801010ac>] (ret_from_fork+0x14/0x28)
[   18.132481] Exception stack(0xaa2b7fb0 to 0xaa2b7ff8)
[   18.132490] 7fa0:                                     00000000 00000000 00000000 00000000
[   18.132499] 7fc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[   18.132507] 7fe0: 00000000 00000000 00000000 00000000 00000013 00000000
[   18.132518]  r10:00000000 r9:00000000 r8:00000000 r7:00000000 r6:00000000 r5:80142838
[   18.132524]  r4:aa2766c0 r3:80104648
[   18.132531] ---[ end trace 6661c2c8b5d2b731 ]---
[   18.225331] can: controller area network core (rev 20170425 abi 9)
[   18.225745] NET: Registered protocol family 29
[   18.259644] can: raw protocol (rev 20170425)
[   18.628518] smsc95xx 1-1.1:1.0 eth0: link up, 100Mbps, full-duplex, lpa 0xC1E1

pelwell, 2020-05-20:
Which network interfaces were active at the time this exception occurred?

Have you tried disabling other devices such as the CAN bus interface?

BuMRK, 2020-05-20:
eth0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 192.168.1.168  netmask 255.255.255.0  broadcast 192.168.1.255
        inet6 fd1c:56e9:61ee:0:ba27:ebff:fe51:ab75  prefixlen 64  scopeid 0x0<global>
        inet6 fe80::ba27:ebff:fe51:ab75  prefixlen 64  scopeid 0x20<link>
        inet6 2a02:c7f:a42d:5300:ba27:ebff:fe51:ab75  prefixlen 64  scopeid 0x0<global>
        ether b8:27:eb:51:ab:75  txqueuelen 1000  (Ethernet)
        RX packets 2351  bytes 214429 (209.4 KiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 1168  bytes 174957 (170.8 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

lo: flags=73<UP,LOOPBACK,RUNNING>  mtu 65536
        inet 127.0.0.1  netmask 255.0.0.0
        inet6 ::1  prefixlen 128  scopeid 0x10<host>
        loop  txqueuelen 1000  (Local Loopback)
        RX packets 266  bytes 22440 (21.9 KiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 266  bytes 22440 (21.9 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

tun0: flags=4305<UP,POINTOPOINT,RUNNING,NOARP,MULTICAST>  mtu 1500
        inet 10.8.0.1  netmask 255.255.255.0  destination 10.8.0.1
        inet6 fe80::2efa:47ef:2cc3:811a  prefixlen 64  scopeid 0x20<link>
        unspec 00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00  txqueuelen 100  (UNSPEC)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 9  bytes 432 (432.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

wlan0: flags=4099<UP,BROADCAST,MULTICAST>  mtu 1500
        ether 00:13:ef:50:14:24  txqueuelen 1000  (Ethernet)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 23  overruns 0  frame 0
        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
[   49.523565] tun: Universal TUN/TAP device driver, 1.6
[  363.092546] ------------[ cut here ]------------
[  363.093204] WARNING: CPU: 1 PID: 364 at net/wireless/sme.c:1088 __cfg80211_disconnected+0x554/0x5a0 [cfg80211]
[  363.093221] Modules linked in: tun can_raw can w1_therm w1_gpio wire cn ip6table_filter ip6_tables xt_MASQUERADE iptable_nat nf_nat nf_conntrack nf_defrag_ipv4 xt_comment xt_tcpudp xt_multiport iptable_filter binfmt_misc joydev spidev 8192cu sha256_generic libsha256 evdev cfg80211 rfkill raspberrypi_hwmon raspberrypi_ts hwmon input_polldev bcm2835_codec(C) bcm2835_isp(C) bcm2835_v4l2(C) v4l2_mem2mem i2c_bcm2835 bcm2835_mmal_vchiq(C) spi_bcm2835 videobuf2_vmalloc videobuf2_dma_contig videobuf2_memops videobuf2_v4l2 snd_bcm2835(C) videobuf2_common snd_pcm snd_timer videodev snd mc vc_sm_cma(C) fixed rpi_backlight backlight uio_pdrv_genirq uio i2c_bcm2708 i2c_dev ip_tables x_tables ipv6 nf_defrag_ipv6
[  363.093538] CPU: 1 PID: 364 Comm: kworker/u8:3 Tainted: G        WC        5.4.42-v7+ #1319
[  363.093544] Hardware name: BCM2835
[  363.093888] Workqueue: cfg80211 cfg80211_event_work [cfg80211]
[  363.093899] Backtrace:
[  363.093926] [<8010d458>] (dump_backtrace) from [<8010d750>] (show_stack+0x20/0x24)
[  363.093941]  r6:a2db0000 r5:00000000 r4:80d93a34 r3:b477c3ae
[  363.093959] [<8010d730>] (show_stack) from [<808ae684>] (dump_stack+0xe0/0x124)
[  363.093979] [<808ae5a4>] (dump_stack) from [<8011fd24>] (__warn+0xec/0x104)
[  363.093996]  r8:00000440 r7:00000009 r6:7f310168 r5:00000000 r4:00000000 r3:b477c3ae
[  363.094013] [<8011fc38>] (__warn) from [<8011fdf4>] (warn_slowpath_fmt+0xb8/0xc0)
[  363.094029]  r9:7f310168 r8:00000440 r7:7f2e58c8 r6:00000009 r5:00000000 r4:80d04f48
[  363.094343] [<8011fd40>] (warn_slowpath_fmt) from [<7f2e58c8>] (__cfg80211_disconnected+0x554/0x5a0 [cfg80211])
[  363.094362]  r9:00000000 r8:00000000 r7:a4c3fc24 r6:a4c3fc00 r5:aa08b800 r4:a8a1c1c0
[  363.095024] [<7f2e5374>] (__cfg80211_disconnected [cfg80211]) from [<7f2b0b34>] (cfg80211_process_wdev_events+0x13c/0x160 [cfg80211])
[  363.095043]  r10:00000100 r9:00000122 r8:a4c3fc8c r7:a4c3fc24 r6:a4c3fc94 r5:a4c3fc00
[  363.095050]  r4:a1b39d80
[  363.095689] [<7f2b09f8>] (cfg80211_process_wdev_events [cfg80211]) from [<7f2b0b98>] (cfg80211_process_rdev_events+0x40/0x98 [cfg80211])
[  363.095708]  r10:00000000 r9:00000100 r8:00000000 r7:a9a3a900 r6:aa080400 r5:a8a1c460
[  363.095715]  r4:a4c3fc00
[  363.096307] [<7f2b0b58>] (cfg80211_process_rdev_events [cfg80211]) from [<7f2aa1b8>] (cfg80211_event_work+0x24/0x2c [cfg80211])
[  363.096318]  r5:a46f3780 r4:a8a1c0e4
[  363.096619] [<7f2aa194>] (cfg80211_event_work [cfg80211]) from [<8013bd28>] (process_one_work+0x17c/0x4b4)
[  363.096629]  r4:a8a1c0e4 r3:7f2aa194
[  363.096648] [<8013bbac>] (process_one_work) from [<8013c590>] (worker_thread+0x54/0x5b4)
[  363.096665]  r10:a46f3780 r9:a2db0038 r8:80d03d00 r7:aa080418 r6:00000088 r5:a46f3794
[  363.096671]  r4:aa080400
[  363.096689] [<8013c53c>] (worker_thread) from [<80142974>] (kthread+0x13c/0x168)
[  363.096706]  r10:a2d80fdc r9:a4487e74 r8:8013c53c r7:a46f3780 r6:00000000 r5:a429a6c0
[  363.096713]  r4:a2d80fc0
[  363.096731] [<80142838>] (kthread) from [<801010ac>] (ret_from_fork+0x14/0x28)
[  363.096741] Exception stack(0xa2db1fb0 to 0xa2db1ff8)
[... 78 lines cut ...]
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
Device Qualifier (for other device speed):
[... cut ...]
  bDeviceSubClass         0
  bDeviceProtocol         0
  bMaxPacketSize0        64
  bNumConfigurations      1
Device Status:     0x0000
  (Bus Powered)
T:  Bus=01 Lev=02 Prnt=02 Port=01 Cnt=02 Dev#=  5 Spd=480 MxCh= 0
D:  Ver= 2.00 Cls=00(>ifc ) Sub=00 Prot=00 MxPS=64 #Cfgs=  1
P:  Vendor=0bda ProdID=8176 Rev=02.00
S:  Manufacturer=Realtek
S:  Product=802.11n WLAN Adapter
S:  SerialNumber=00e04c000001
C:  #Ifs= 1 Cfg#= 1 Atr=80 MxPwr=500mA
I:  If#= 0 Alt= 0 #EPs= 4 Cls=ff(vend.) Sub=ff Prot=ff Driver=rtl8192cu

pelwell, 2020-05-21:
You're using an external RealTek WLAN adaptor, and with the downstream 8192cu driver. Aside from writing your own driver, that's about as unsupported as it gets.

This is a different WARN to the original post, and it appears to be as the result of the device disconnecting (possibly forcibly), which would make this a symptom of a bigger problem - why is the adaptor disconnecting? If I had to guess, looking at the list of modules you have loaded, does the Pi have enough power to driver them all? Try with the WLAN adaptor on a powered hub.

xyklex, 2020-07-21:
I think this ticket represent the same issue I have with a RPi4, it is a Yocto generated image with kernel version `4.19.126-v7l` 

Here is the kernel trace
```
[  501.834854] ------------[ cut here ]------------
[  501.839748] WARNING: CPU: 3 PID: 7 at net/wireless/sme.c:756 __cfg80211_connect_result+0x3b0/0x414 [cfg80211]
[  501.849845] Modules linked in: xt_conntrack ipt_MASQUERADE nf_conntrack_netlink nfnetlink xt_addrtype iptable_filter iptable_nat nf_nat_ipv4 nf_nat nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 ip_tables x_tables br_netfilter bridge stp llc overlay ipv6 88x2bu(O) brcmfmac brcmutil bcm2835_codec(C) bcm2835_v4l2(C) v4l2_mem2mem bcm2835_mmal_vchiq(C) vc4 v4l2_common videobuf2_dma_contig videobuf2_vmalloc videobuf2_memops videobuf2_v4l2 videobuf2_common snd_soc_core videodev cfg80211 v3d snd_compress snd_pcm_dmaengine rfkill media snd_pcm gpu_sched raspberrypi_hwmon hwmon snd_timer vc_sm_cma(C) snd rpivid_mem uio_pdrv_genirq uio
[  501.905917] CPU: 3 PID: 7 Comm: kworker/u8:0 Tainted: G        WC O      4.19.126-v7l #1
[  501.914129] Hardware name: BCM2835
[  501.917721] Workqueue: cfg80211 cfg80211_event_work [cfg80211]
[  501.923665] [<c0212fc8>] (unwind_backtrace) from [<c020d4b8>] (show_stack+0x20/0x24)
[  501.931536] [<c020d4b8>] (show_stack) from [<c0a1197c>] (dump_stack+0xd4/0x118)
[  501.938966] [<c0a1197c>] (dump_stack) from [<c0222850>] (__warn.part.0+0xcc/0xe8)
[  501.946569] [<c0222850>] (__warn.part.0) from [<c0222a04>] (warn_slowpath_null+0x54/0x5c)
[  501.955011] [<c0222a04>] (warn_slowpath_null) from [<bf22d8f4>] (__cfg80211_connect_result+0x3b0/0x414 [cfg80211])
[  501.965789] [<bf22d8f4>] (__cfg80211_connect_result [cfg80211]) from [<bf1fd53c>] (cfg80211_process_wdev_events+0x104/0x160 [cfg80211])
[  501.978401] [<bf1fd53c>] (cfg80211_process_wdev_events [cfg80211]) from [<bf1fd5e0>] (cfg80211_process_rdev_events+0x48/0xa0 [cfg80211])
[  501.991092] [<bf1fd5e0>] (cfg80211_process_rdev_events [cfg80211]) from [<bf1f72e0>] (cfg80211_event_work+0x24/0x2c [cfg80211])
[  502.002870] [<bf1f72e0>] (cfg80211_event_work [cfg80211]) from [<c023e058>] (process_one_work+0x200/0x4e0)
[  502.012677] [<c023e058>] (process_one_work) from [<c023f124>] (worker_thread+0x4c/0x568)
[  502.020895] [<c023f124>] (worker_thread) from [<c0244c18>] (kthread+0x140/0x170)
[  502.028406] [<c0244c18>] (kthread) from [<c02010ac>] (ret_from_fork+0x14/0x28)
[  502.035735] Exception stack(0xef917fb0 to 0xef917ff8)
[  502.040864] 7fa0:                                     00000000 00000000 00000000 00000000
[  502.049167] 7fc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[  502.057466] 7fe0: 00000000 00000000 00000000 00000000 00000013 00000000
[  502.064248] ---[ end trace 16e517af7247ed6f ]---
```

I just finished testing the same driver and wifi device (DWA-181) on a RPi3 B+ but with kernel `5.4.51-v7l` and it works without errors, I don't know yet if kernel `5.4.51-v7l` is stable enough for me right now.

Hope this helps for something.

DAMEK86, 2020-09-18:
looks like the same on my pi4 with kernel `5.4.51-v7l`
```
[ 4337.280772] ------------[ cut here ]------------
[ 4337.280940] WARNING: CPU: 0 PID: 10237 at net/wireless/sme.c:756 __cfg80211_connect_result+0x43c/0x4e8 [cfg80211]
[ 4337.280949] Modules linked in: veth xt_nat xt_tcpudp xt_conntrack xt_MASQUERADE nf_conntrack_netlink nft_counter xt_addrtype nft_compat nft_chain_nat nf_nat nf_conntrack nf_defrag_ipv4 nf_tables nfnetlink br_netfilter bridge overlay 8021q garp stp llc sg brcmfmac brcmutil sha256_generic libsha256 cfg80211 vc4 rfkill cec v3d gpu_sched drm_kms_helper raspberrypi_hwmon bcm2835_isp(C) bcm2835_codec(C) bcm2835_v4l2(C) drm v4l2_mem2mem bcm2835_mmal_vchiq(C) videobuf2_dma_contig videobuf2_vmalloc videobuf2_memops videobuf2_v4l2 drm_panel_orientation_quirks videobuf2_common snd_soc_core snd_compress snd_bcm2835(C) snd_pcm_dmaengine vc_sm_cma(C) videodev snd_pcm mc snd_timer snd syscopyarea sysfillrect sysimgblt fb_sys_fops rpivid_mem uio_pdrv_genirq uio ip_tables x_tables ipv6 nf_defrag_ipv6
[ 4337.281165] CPU: 0 PID: 10237 Comm: kworker/u8:2 Tainted: G         C        5.4.51-v7l+ #1333
[ 4337.281171] Hardware name: BCM2711
[ 4337.281310] Workqueue: cfg80211 cfg80211_event_work [cfg80211]
[ 4337.281318] Backtrace: 
[ 4337.281339] [<c020d46c>] (dump_backtrace) from [<c020d768>] (show_stack+0x20/0x24)
[ 4337.281349]  r6:ce61c000 r5:00000000 r4:c129c8f8 r3:e92b754e
[ 4337.281365] [<c020d748>] (show_stack) from [<c0a39a44>] (dump_stack+0xe0/0x124)
[ 4337.281379] [<c0a39964>] (dump_stack) from [<c0221c70>] (__warn+0xec/0x104)
[ 4337.281390]  r8:000002f4 r7:00000009 r6:bf4de168 r5:00000000 r4:00000000 r3:e92b754e
[ 4337.281402] [<c0221b84>] (__warn) from [<c0221d40>] (warn_slowpath_fmt+0xb8/0xc0)
[ 4337.281412]  r9:bf4de168 r8:000002f4 r7:bf4b2d00 r6:00000009 r5:00000000 r4:c1204f88
[ 4337.281549] [<c0221c8c>] (warn_slowpath_fmt) from [<bf4b2d00>] (__cfg80211_connect_result+0x43c/0x4e8 [cfg80211])
[ 4337.281560]  r9:00000122 r8:ce61de54 r7:c1204f88 r6:00000000 r5:d2732e0c r4:ef371004
[ 4337.281818] [<bf4b28c4>] (__cfg80211_connect_result [cfg80211]) from [<bf47eb0c>] (cfg80211_process_wdev_events+0x104/0x160 [cfg80211])
[ 4337.281828]  r8:ef371090 r7:ef371028 r6:ef371098 r5:ef371004 r4:d2732e00
[ 4337.282084] [<bf47ea08>] (cfg80211_process_wdev_events [cfg80211]) from [<bf47eba8>] (cfg80211_process_rdev_events+0x40/0x98 [cfg80211])
[ 4337.282094]  r10:00000000 r9:00000100 r8:00000000 r7:d77fb200 r6:ef846400 r5:d68e8470
[ 4337.282101]  r4:ef371004
[ 4337.282355] [<bf47eb68>] (cfg80211_process_rdev_events [cfg80211]) from [<bf4781b8>] (cfg80211_event_work+0x24/0x2c [cfg80211])
[ 4337.282363]  r5:d24e2c80 r4:d68e80e4
[ 4337.282498] [<bf478194>] (cfg80211_event_work [cfg80211]) from [<c023e244>] (process_one_work+0x17c/0x4b4)
[ 4337.282506]  r4:d68e80e4 r3:bf478194
[ 4337.282519] [<c023e0c8>] (process_one_work) from [<c023eaac>] (worker_thread+0x54/0x5b4)
[ 4337.282529]  r10:d24e2c80 r9:ce61c038 r8:c1203d00 r7:ef846418 r6:00000088 r5:d24e2c94
[ 4337.282535]  r4:ef846400
[ 4337.282548] [<c023ea58>] (worker_thread) from [<c0244e90>] (kthread+0x13c/0x168)
[ 4337.282557]  r10:ce6290dc r9:ce5b3e74 r8:c023ea58 r7:d24e2c80 r6:00000000 r5:cf13ac80
[ 4337.282563]  r4:ce6290c0
[ 4337.282574] [<c0244d54>] (kthread) from [<c02010ac>] (ret_from_fork+0x14/0x28)
[ 4337.282581] Exception stack(0xce61dfb0 to 0xce61dff8)
[ 4337.282589] dfa0:                                     00000000 00000000 00000000 00000000
[ 4337.282598] dfc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[ 4337.282607] dfe0: 00000000 00000000 00000000 00000000 00000013 00000000
[ 4337.282617]  r10:00000000 r9:00000000 r8:00000000 r7:00000000 r6:00000000 r5:c0244d54
[ 4337.282623]  r4:cf13ac80 r3:c0204648
[ 4337.282642] ---[ end trace 458e4b73fffb9813 ]---
[ 4630.055099] ieee80211 phy0: brcmf_run_escan: error (-52)
[ 4630.055119] ieee80211 phy0: brcmf_cfg80211_scan: scan error (-52)
[ 4631.057405] ieee80211 phy0: brcmf_run_escan: error (-52)
[ 4631.057426] ieee80211 phy0: brcmf_cfg80211_scan: scan error (-52)
[ 4632.060151] ieee80211 phy0: brcmf_run_escan: error (-52)
[ 4632.060170] ieee80211 phy0: brcmf_cfg80211_scan: scan error (-52)
[20331.930920] ------------[ cut here ]------------
[20331.931034] WARNING: CPU: 3 PID: 24437 at net/wireless/sme.c:756 __cfg80211_connect_result+0x43c/0x4e8 [cfg80211]
[20331.931039] Modules linked in: veth xt_nat xt_tcpudp xt_conntrack xt_MASQUERADE nf_conntrack_netlink nft_counter xt_addrtype nft_compat nft_chain_nat nf_nat nf_conntrack nf_defrag_ipv4 nf_tables nfnetlink br_netfilter bridge overlay 8021q garp stp llc sg brcmfmac brcmutil sha256_generic libsha256 cfg80211 vc4 rfkill cec v3d gpu_sched drm_kms_helper raspberrypi_hwmon bcm2835_isp(C) bcm2835_codec(C) bcm2835_v4l2(C) drm v4l2_mem2mem bcm2835_mmal_vchiq(C) videobuf2_dma_contig videobuf2_vmalloc videobuf2_memops videobuf2_v4l2 drm_panel_orientation_quirks videobuf2_common snd_soc_core snd_compress snd_bcm2835(C) snd_pcm_dmaengine vc_sm_cma(C) videodev snd_pcm mc snd_timer snd syscopyarea sysfillrect sysimgblt fb_sys_fops rpivid_mem uio_pdrv_genirq uio ip_tables x_tables ipv6 nf_defrag_ipv6
[20331.931156] CPU: 3 PID: 24437 Comm: kworker/u8:2 Tainted: G        WC        5.4.51-v7l+ #1333
[20331.931159] Hardware name: BCM2711
[20331.931232] Workqueue: cfg80211 cfg80211_event_work [cfg80211]
[20331.931236] Backtrace: 
[20331.931248] [<c020d46c>] (dump_backtrace) from [<c020d768>] (show_stack+0x20/0x24)
[... cut ...]
[20331.931898] Exception stack(0xcf249fb0 to 0xcf249ff8)
[20331.931902] 9fa0:                                     00000000 00000000 00000000 00000000
[20331.931907] 9fc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[20331.931912] 9fe0: 00000000 00000000 00000000 00000000 00000013 00000000
[20331.931916]  r10:00000000 r9:00000000 r8:00000000 r7:00000000 r6:00000000 r5:c0244d54
[20331.931920]  r4:cd4e0d00 r3:c0204648
[20331.931924] ---[ end trace 458e4b73fffb9814 ]---
```

BuMRK, 2020-10-28:
new problem pi 3b+  5.4.72-v7+

[   18.737652] ------------[ cut here ]------------
[   18.738181] WARNING: CPU: 3 PID: 169 at net/wireless/sme.c:1088 __cfg80211_disconnected+0x57c/0x5b8 [cfg80211]
[   18.738192] Modules linked in: w1_therm w1_gpio wire cn ip6table_filter ip6_tables xt_multiport xt_tcpudp iptable_filter xt_MASQUERADE xt_comment iptable_nat nf_nat nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 binfmt_misc spidev 8192cu sha256_generic libsha256 cfg80211 rfkill raspberrypi_hwmon bcm2835_codec(C) v4l2_mem2mem bcm2835_v4l2(C) bcm2835_isp(C) i2c_bcm2835 bcm2835_mmal_vchiq(C) videobuf2_vmalloc videobuf2_dma_contig snd_bcm2835(C) videobuf2_memops videobuf2_v4l2 spi_bcm2835 videobuf2_common snd_pcm vc_sm_cma(C) snd_timer snd videodev mc uio_pdrv_genirq uio fixed i2c_bcm2708 i2c_dev ip_tables x_tables ipv6
[   18.738393] CPU: 3 PID: 169 Comm: kworker/u8:2 Tainted: G         C        5.4.72-v7+ #1358
[   18.738397] Hardware name: BCM2835
[   18.738647] Workqueue: cfg80211 cfg80211_event_work [cfg80211]
[   18.738660] Backtrace:
[   18.738686] [<8010dfa4>] (dump_backtrace) from [<8010e324>] (show_stack+0x20/0x24)
[   18.738697]  r7:ffffffff r6:00000000 r5:60000013 r4:80e996f0
[   18.738713] [<8010e304>] (show_stack) from [<80900534>] (dump_stack+0xd8/0x11c)
[   18.738727] [<8090045c>] (dump_stack) from [<80120448>] (__warn+0xe0/0x108)
[   18.738738]  r10:a9b35800 r9:00000009 r8:7f2b4934 r7:00000440 r6:00000009 r5:7f2b4934
[   18.738744]  r4:7f2dd54c r3:00000000
[   18.738756] [<80120368>] (__warn) from [<80120810>] (warn_slowpath_fmt+0x70/0xc0)
[   18.738764]  r7:00000440 r6:7f2dd54c r5:80e04f88 r4:00000000
[   18.738978] [<801207a4>] (warn_slowpath_fmt) from [<7f2b4934>] (__cfg80211_disconnected+0x57c/0x5b8 [cfg80211])
[   18.738990]  r9:00000001 r8:00000000 r7:00000000 r6:a1b38d44 r5:a6616c00 r4:a40199c0
[   18.739350] [<7f2b43b8>] (__cfg80211_disconnected [cfg80211]) from [<7f27fe50>] (cfg80211_process_wdev_events+0x13c/0x160 [cfg80211])
[   18.739362]  r10:00000100 r9:00000122 r8:a6616c8c r7:a6616c24 r6:a6616c94 r5:a6616c00
[   18.739367]  r4:a1b38d00
[   18.739724] [<7f27fd14>] (cfg80211_process_wdev_events [cfg80211]) from [<7f27feb4>] (cfg80211_process_rdev_events+0x40/0x98 [cfg80211])
[   18.739739]  r10:00000000 r9:00000100 r8:00000000 r7:a997f200 r6:aa044400 r5:a4019c60
[   18.739744]  r4:a6616c00
[   18.740110] [<7f27fe74>] (cfg80211_process_rdev_events [cfg80211]) from [<7f2791c0>] (cfg80211_event_work+0x24/0x2c [cfg80211])
[   18.740117]  r5:a7aa2b80 r4:a40198e4
[   18.740306] [<7f27919c>] (cfg80211_event_work [cfg80211]) from [<8013c87c>] (process_one_work+0x250/0x570)
[   18.740313]  r5:a7aa2b80 r4:a40198e4
[   18.740326] [<8013c62c>] (process_one_work) from [<8013cbfc>] (worker_thread+0x60/0x5d0)
[   18.740337]  r10:aa044400 r9:80e03d00 r8:aa044418 r7:00000088 r6:aa044400 r5:a7aa2b94
[   18.740342]  r4:a7aa2b80
[   18.740356] [<8013cb9c>] (worker_thread) from [<80143fbc>] (kthread+0x170/0x174)
[   18.740367]  r10:a8a57e74 r9:8013cb9c r8:a7aa2b80 r7:a6460000 r6:00000000 r5:a806dd80
[   18.740372]  r4:a806ddc0
[   18.740383] [<80143e4c>] (kthread) from [<801010ac>] (ret_from_fork+0x14/0x28)
[   18.740391] Exception stack(0xa6461fb0 to 0xa6461ff8)
[   18.740400] 1fa0:                                     00000000 00000000 00000000 00000000
[   18.740409] 1fc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[   18.740418] 1fe0: 00000000 00000000 00000000 00000000 00000013 00000000
[   18.740429]  r10:00000000 r9:00000000 r8:00000000 r7:00000000 r6:00000000 r5:80143e4c
[   18.740433]  r4:a806dd80
[   18.740532] ---[ end trace 4c7e925fb88fd53a ]---
[   18.937120] can: controller area network core (rev 20170425 abi 9)
[   18.937416] NET: Registered protocol family 29
[   18.971511] can: raw protocol (rev 20170425)
[   19.243479] smsc95xx 1-1.1:1.0 eth0: link up, 100Mbps, full-duplex, lpa 0xC1E1
[  363.096820] ------------[ cut here ]------------
[  363.097493] WARNING: CPU: 1 PID: 169 at net/wireless/sme.c:1088 __cfg80211_disconnected+0x57c/0x5b8 [cfg80211]
[  363.097509] Modules linked in: can_raw can w1_therm w1_gpio wire cn ip6table_filter ip6_tables xt_multiport xt_tcpudp iptable_filter xt_MASQUERADE xt_comment iptable_nat nf_nat nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 binfmt_misc spidev 8192cu sha256_generic libsha256 cfg80211 rfkill raspberrypi_hwmon bcm2835_codec(C) v4l2_mem2mem bcm2835_v4l2(C) bcm2835_isp(C) i2c_bcm2835 bcm2835_mmal_vchiq(C) videobuf2_vmalloc videobuf2_dma_contig snd_bcm2835(C) videobuf2_memops videobuf2_v4l2 spi_bcm2835 videobuf2_common snd_pcm vc_sm_cma(C) snd_timer snd videodev mc uio_pdrv_genirq uio fixed i2c_bcm2708 i2c_dev ip_tables x_tables ipv6
[  363.097807] CPU: 1 PID: 169 Comm: kworker/u8:2 Tainted: G        WC        5.4.72-v7+ #1358
[  363.097814] Hardware name: BCM2835
[  363.098165] Workqueue: cfg80211 cfg80211_event_work [cfg80211]
[  363.098176] Backtrace:
[  363.098210] [<8010dfa4>] (dump_backtrace) from [<8010e324>] (show_stack+0x20/0x24)
[  363.098225]  r7:ffffffff r6:00000000 r5:60000013 r4:80e996f0
[  363.098248] [<8010e304>] (show_stack) from [<80900534>] (dump_stack+0xd8/0x11c)
[  363.098267] [<8090045c>] (dump_stack) from [<80120448>] (__warn+0xe0/0x108)
[  363.098285]  r10:a9b35800 r9:00000009 r8:7f2b4934 r7:00000440 r6:00000009 r5:7f2b4934
[  363.098294]  r4:7f2dd54c r3:00000000
[  363.098311] [<80120368>] (__warn) from [<80120810>] (warn_slowpath_fmt+0x70/0xc0)
[... cut ...]
[  363.100700] [<80143e4c>] (kthread) from [<801010ac>] (ret_from_fork+0x14/0x28)
[  363.100710] Exception stack(0xa6461fb0 to 0xa6461ff8)
[  363.100723] 1fa0:                                     00000000 00000000 00000000 00000000
[  363.100738] 1fc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[  363.100750] 1fe0: 00000000 00000000 00000000 00000000 00000013 00000000
[  363.100766]  r10:00000000 r9:00000000 r8:00000000 r7:00000000 r6:00000000 r5:80143e4c
[  363.100773]  r4:a806dd80
[  363.100938] ---[ end trace 4c7e925fb88fd53b ]---

chacles, 2020-11-03:
Linux raspberrypi 5.4.51-v7l+ #1333 SMP Mon Aug 10 16:51:40 BST 2020 armv7l GNU/Linux

[  320.000988] ------------[ cut here ]------------
[  320.001373] WARNING: CPU: 3 PID: 1058 at /home/pi/RTL8812BU_for_Raspbian/hal/hal_com.c:11513 rtw_lps_state_chk+0x6c/0x70 [88x2bu]
[  320.001386] Modules linked in: 88x2bu(O) fuse cmac rfcomm bnep hci_uart btbcm bluetooth ecdh_generic ecc 8021q garp stp llc hid_logitech_hidpp evdev joydev brcmfmac brcmutil sha256_generic libsha256 hid_logitech_dj cfg80211 v3d rfkill gpu_sched raspberrypi_hwmon vc4 cec drm_kms_helper bcm2835_v4l2(C) bcm2835_codec(C) bcm2835_isp(C) v4l2_mem2mem bcm2835_mmal_vchiq(C) videobuf2_vmalloc videobuf2_dma_contig drm videobuf2_memops videobuf2_v4l2 videobuf2_common drm_panel_orientation_quirks videodev snd_bcm2835(C) mc snd_soc_core snd_compress vc_sm_cma(C) snd_pcm_dmaengine snd_pcm snd_timer snd syscopyarea sysfillrect sysimgblt fb_sys_fops rpivid_mem uio_pdrv_genirq uio i2c_dev ip_tables x_tables ipv6 nf_defrag_ipv6
[  320.001734] CPU: 3 PID: 1058 Comm: RTW_CMD_THREAD Tainted: G        WC O      5.4.51-v7l+ #1333
[  320.001749] Hardware name: BCM2711
[  320.001756] Backtrace: 
[  320.001784] [<c020d46c>] (dump_backtrace) from [<c020d768>] (show_stack+0x20/0x24)
[  320.001797]  r6:d6bac000 r5:00000000 r4:c129c8f8 r3:41b6c424
[  320.001817] [<c020d748>] (show_stack) from [<c0a39a44>] (dump_stack+0xe0/0x124)
[  320.001836] [<c0a39964>] (dump_stack) from [<c0221c70>] (__warn+0xec/0x104)
[  320.001849]  r8:00002cf9 r7:00000009 r6:bf662b34 r5:00000000 r4:00000000 r3:41b6c424
[  320.001864] [<c0221b84>] (__warn) from [<c0221d40>] (warn_slowpath_fmt+0xb8/0xc0)
[  320.001877]  r9:bf662b34 r8:00002cf9 r7:bf59c140 r6:00000009 r5:00000000 r4:c1204f88
[  320.002232] [<c0221c8c>] (warn_slowpath_fmt) from [<bf59c140>] (rtw_lps_state_chk+0x6c/0x70 [88x2bu])
[  320.002245]  r9:d6badf30 r8:00000000 r7:00000001 r6:00000604 r5:f088f000 r4:00000000
[  320.002887] [<bf59c0d4>] (rtw_lps_state_chk [88x2bu]) from [<bf59fc28>] (SetHwReg+0x7c4/0x7ec [88x2bu])
[  320.002903]  r7:f0894000 r6:d6bade5f r5:f0894000 r4:f088f000
[  320.003542] [<bf59f464>] (SetHwReg [88x2bu]) from [<bf5c35a0>] (rtl8822b_sethwreg+0xd54/0xda0 [88x2bu])
[  320.003554]  r7:f0894000 r6:00000069 r5:d6bade5f r4:f088f000
[  320.004187] [<bf5c284c>] (rtl8822b_sethwreg [88x2bu]) from [<bf5c6a74>] (sethwreg+0x1bc/0x1c4 [88x2bu])
[  320.004212]  r9:d6badf30 r8:00000000 r7:00000000 r6:d7085aa4 r5:d6bade5f r4:f088f000
[  320.004851] [<bf5c68b8>] (sethwreg [88x2bu]) from [<bf597018>] (rtw_hal_set_hwreg+0x20/0x24 [88x2bu])
[  320.004861]  r5:d7085000 r4:f088f000
[  320.005494] [<bf596ff8>] (rtw_hal_set_hwreg [88x2bu]) from [<bf54f85c>] (rtw_set_ps_mode+0xe8/0x228 [88x2bu])
[  320.006132] [<bf54f774>] (rtw_set_ps_mode [88x2bu]) from [<bf54fbbc>] (LPS_Leave+0xd8/0xe0 [88x2bu])
[  320.006144]  r8:bf675000 r7:d6bade94 r6:bf65bd60 r5:d7085000 r4:f088f000
[  320.006776] [<bf54fae4>] (LPS_Leave [88x2bu]) from [<bf51a23c>] (lps_ctrl_wk_hdl+0x15c/0x1b4 [88x2bu])
[  320.006791]  r7:f088ff50 r6:d6bac000 r5:d7085000 r4:f088f000
[  320.007429] [<bf51a0e0>] (lps_ctrl_wk_hdl [88x2bu]) from [<bf51c1c0>] (rtw_drvextra_cmd_hdl+0x124/0x3fc [88x2bu])
[  320.007439]  r5:f088f000 r4:d7097200
[  320.008072] [<bf51c09c>] (rtw_drvextra_cmd_hdl [88x2bu]) from [<bf517ec0>] (rtw_cmd_thread+0x188/0x480 [88x2bu])
[  320.008087]  r9:d6badf30 r8:bf675000 r7:f088ff50 r6:d6bac000 r5:f088f000 r4:d72afec0
[  320.008412] [<bf517d38>] (rtw_cmd_thread [88x2bu]) from [<c0244e90>] (kthread+0x13c/0x168)
[  320.008425]  r10:d7052f1c r9:d6b3bb8c r8:bf517d38 r7:f088f000 r6:00000000 r5:d8f85b80
[  320.008433]  r4:d7052f00
[  320.008448] [<c0244d54>] (kthread) from [<c02010ac>] (ret_from_fork+0x14/0x28)
[  320.008457] Exception stack(0xd6badfb0 to 0xd6badff8)
[  320.008472] dfa0:                                     00000000 00000000 00000000 00000000
[  320.008483] dfc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[  320.008494] dfe0: 00000000 00000000 00000000 00000000 00000013 00000000
[  320.008506]  r10:00000000 r9:00000000 r8:00000000 r7:00000000 r6:00000000 r5:c0244d54
[  320.008514]  r4:d8f85b80 r3:c0204648
[  320.008525] ---[ end trace f02041ede1282205 ]---

BuMRK, 2020-12-19:
after kernel upgrade 5.10.1-v7+ everything works fine, no error. It even started working WireGuard

popcornmix, 2020-12-23:
@BuMRK good to know. We should be moving to that by default in the not too distant future.
Can anyone else test that? Moving to 5.10 kernel information [here](https://www.raspberrypi.org/forums/viewtopic.php?f=29&t=288234)

mgrey, 2021-03-11:
Raspberry PI4 @ 5.10.17-v7l+ #1403

Mar 11 20:17:23 gridbox kernel: [   25.699064] ieee80211 phy0: brcmf_update_bss_info: wl dtim_assoc failed (-52)
Mar 11 20:17:23 gridbox kernel: [   25.699323] ------------[ cut here ]------------
Mar 11 20:17:23 gridbox kernel: [   25.699523] WARNING: CPU: 1 PID: 70 at net/wireless/sme.c:757 __cfg80211_connect_result+0x454/0x500 [cfg80211]
Mar 11 20:17:23 gridbox kernel: [   25.699539] Modules linked in: cmac bnep hci_uart btbcm bluetooth ecdh_generic ecc 8021q garp stp llc rtc_ds1307 regmap_i2c brcmfmac brcmutil sha256_generic vc4 v3d cfg80211 gpu_sched raspberrypi_hwmon rfkill bcm2835_codec(C) bcm2835_isp(C) bcm2835_v4l2(C) cec v4l2_mem2mem videobuf2_dma_contig bcm2835_mmal_vchiq(C) videobuf2_vmalloc videobuf2_memops videobuf2_v4l2 videobuf2_common i2c_bcm2835 drm_kms_helper videodev drm mc snd_bcm2835(C) drm_panel_orientation_quirks vc_sm_cma(C) snd_soc_core snd_compress snd_pcm_dmaengine snd_pcm snd_timer rpivid_mem snd syscopyarea sysfillrect sysimgblt fb_sys_fops backlight uio_pdrv_genirq uio nvmem_rmem i2c_dev ip_tables x_tables ipv6
Mar 11 20:17:23 gridbox kernel: [   25.700392] CPU: 1 PID: 70 Comm: kworker/u8:1 Tainted: G         C        5.10.17-v7l+ #1403
Mar 11 20:17:23 gridbox kernel: [   25.700402] Hardware name: BCM2711
Mar 11 20:17:23 gridbox kernel: [   25.700562] Workqueue: cfg80211 cfg80211_event_work [cfg80211]
Mar 11 20:17:23 gridbox kernel: [   25.700579] Backtrace:
Mar 11 20:17:23 gridbox kernel: [   25.700619] [<c0b63050>] (dump_backtrace) from [<c0b633e4>] (show_stack+0x20/0x24)
Mar 11 20:17:23 gridbox kernel: [   25.700635]  r7:ffffffff r6:00000000 r5:60000013 r4:c12e69fc
Mar 11 20:17:23 gridbox kernel: [   25.700654] [<c0b633c4>] (show_stack) from [<c0b67774>] (dump_stack+0xcc/0xf8)
Mar 11 20:17:23 gridbox kernel: [   25.700673] [<c0b676a8>] (dump_stack) from [<c0220bac>] (__warn+0xfc/0x114)
Mar 11 20:17:23 gridbox kernel: [   25.700688]  r10:00000100 r9:00000009 r8:bf3f9828 r7:000002f5 r6:00000009 r5:bf3f9828
Mar 11 20:17:23 gridbox kernel: [   25.700699]  r4:bf4294fc r3:c1205094
Mar 11 20:17:23 gridbox kernel: [   25.700717] [<c0220ab0>] (__warn) from [<c0b63b54>] (warn_slowpath_fmt+0x70/0xd8)
Mar 11 20:17:23 gridbox kernel: [   25.700729]  r7:000002f5 r6:bf4294fc r5:c1205048 r4:00000000
Mar 11 20:17:23 gridbox kernel: [   25.700887] [<c0b63ae8>] (warn_slowpath_fmt) from [<bf3f9828>] (__cfg80211_connect_result+0x454/0x500 [cfg80211])
Mar 11 20:17:23 gridbox kernel: [   25.700901]  r9:00000122 r8:c2291e4c r7:00000000 r6:c1205048 r5:c31d680c r4:c2fec004
Mar 11 20:17:23 gridbox kernel: [   25.701194] [<bf3f93d4>] (__cfg80211_connect_result [cfg80211]) from [<bf3c3ff4>] (cfg80211_process_wdev_events+0x104/0x160 [cfg80211])
Mar 11 20:17:23 gridbox kernel: [   25.701207]  r8:c2fec094 r7:c2fec02c r6:c2fec09c r5:c2fec004 r4:c31d6800
Mar 11 20:17:23 gridbox kernel: [   25.701493] [<bf3c3ef0>] (cfg80211_process_wdev_events [cfg80211]) from [<bf3c4090>] (cfg80211_process_rdev_events+0x40/0x98 [cfg80211])
Mar 11 20:17:23 gridbox kernel: [   25.701507]  r10:00000000 r9:00000000 r8:00000100 r7:c343c000 r6:c180e400 r5:c3464480
Mar 11 20:17:23 gridbox kernel: [   25.701518]  r4:c2fec004
Mar 11 20:17:23 gridbox kernel: [   25.701802] [<bf3c4050>] (cfg80211_process_rdev_events [cfg80211]) from [<bf3bd1c0>] (cfg80211_event_work+0x24/0x2c [cfg80211])
Mar 11 20:17:23 gridbox kernel: [   25.701813]  r5:c224f000 r4:c34640cc
Mar 11 20:17:23 gridbox kernel: [   25.701970] [<bf3bd19c>] (cfg80211_event_work [cfg80211]) from [<c023dc18>] (process_one_work+0x250/0x5a0)
Mar 11 20:17:23 gridbox kernel: [   25.701981]  r5:c224f000 r4:c34640cc
Mar 11 20:17:23 gridbox kernel: [   25.701998] [<c023d9c8>] (process_one_work) from [<c023dfc8>] (worker_thread+0x60/0x5c4)
Mar 11 20:17:23 gridbox kernel: [   25.702013]  r10:c180e400 r9:c1203d00 r8:c180e418 r7:00000088 r6:c180e400 r5:c224f014
Mar 11 20:17:23 gridbox kernel: [   25.702022]  r4:c224f000
Mar 11 20:17:23 gridbox kernel: [   25.702040] [<c023df68>] (worker_thread) from [<c0245a50>] (kthread+0x170/0x174)
Mar 11 20:17:23 gridbox kernel: [   25.702054]  r10:c1921e74 r9:c224f000 r8:c023df68 r7:c2290000 r6:00000000 r5:c220ca40
Mar 11 20:17:23 gridbox kernel: [   25.702064]  r4:c1a46280
Mar 11 20:17:23 gridbox kernel: [   25.702080] [<c02458e0>] (kthread) from [<c02000ec>] (ret_from_fork+0x14/0x28)
Mar 11 20:17:23 gridbox kernel: [   25.702090] Exception stack(0xc2291fb0 to 0xc2291ff8)
Mar 11 20:17:23 gridbox kernel: [   25.702102] 1fa0:                                     00000000 00000000 00000000 00000000
Mar 11 20:17:23 gridbox kernel: [   25.702116] 1fc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
Mar 11 20:17:23 gridbox kernel: [   25.702128] 1fe0: 00000000 00000000 00000000 00000000 00000013 00000000
Mar 11 20:17:23 gridbox kernel: [   25.702142]  r10:00000000 r9:00000000 r8:00000000 r7:00000000 r6:00000000 r5:c02458e0
Mar 11 20:17:23 gridbox kernel: [   25.702152]  r4:c220ca40
Mar 11 20:17:23 gridbox kernel: [   25.702165] ---[ end trace f64785c5f6f94534 ]---