raspberrypi/linux #1989: maxcpus=1 stops Pi2 and Pi3 from booting

pelwell, 2017-05-03:
As inadvertently discovered by @P33M, booting a Pi2 or Pi3 with `maxcpus=1` on the kernel cmdline.txt stops it from booting with abnormally large timestamps in the kernel log and SD timeouts (presumably because the timer is running too fast). This smells like an issue with the secondary CPU startup and/or the local interrupt controller, but I haven't identified the specific cause yet.

pelwell, 2017-05-03:
`maxcpus=4`, `maxcpus=` and an absent setting are the only options that work.

Note that a soft reboot is not enough to show the problem.

popcornmix, 2017-05-03:
I've used ``maxcpus=1`` a number of times, so it certainly has worked in the past.
It is currently working for me (I'm on 4.11 kernel)
```
pi@domnfs:~ $ uname -a
Linux domnfs 4.11.0-v7+ #2483 SMP Wed May 3 15:17:10 BST 2017 armv7l GNU/Linux
pi@domnfs:~ $ cat /proc/cmdline 
8250.nr_uarts=0 bcm2708_fb.fbwidth=1920 bcm2708_fb.fbheight=1080 bcm2708_fb.fbswap=1 vc_mem.mem_base=0x3dc00000 vc_mem.mem_size=0x3f000000  maxcpus=1 console=ttyS0,115200 kgdboc=ttyS0,115200 console=tty1 nfsroot=192.168.4.43:/home/dc4/rootfs ip=dhcp rootwait
pi@domnfs:~ $ cat /proc/cpuinfo 
processor	: 0
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 76.80
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

Hardware	: BCM2835
Revision	: 2a02080
Serial		: 00000000f45df3f2
```

popcornmix, 2017-05-04:
I seem to get a mixture of successful and unsuccessful boots with ``maxcpus=1``.
It seems to depend when the timer jump occurs and by how much whether it is fatal. E.g.
```
[   10.169931] Freeing unused kernel memory: 1024K
[  671.062266] INFO: rcu_sched self-detected stall on CPU
[  671.072361] 	0-...: (2 ticks this GP) idle=1bb/140000000000001/0 softirq=59/59 fqs=0 
[  671.082311] 	 (t=66077 jiffies g=-257 c=-258 q=2)
[  671.092181] rcu_sched kthread starved for 66077 jiffies! g4294967039 c4294967038 f0x0 RCU_GP_WAIT_FQS(3) ->state=0x1
[  671.102573] rcu_sched       S    0     8      2 0x00000000
[  671.112714] [<8074011c>] (__schedule) from [<8074079c>] (schedule+0x50/0xa8)
[  671.123014] [<8074079c>] (schedule) from [<80744060>] (schedule_timeout+0x1e8/0x33c)
[  671.133226] [<80744060>] (schedule_timeout) from [<80184cd4>] (rcu_gp_kthread+0x478/0x950)
[  671.143478] [<80184cd4>] (rcu_gp_kthread) from [<8013cfec>] (kthread+0x13c/0x16c)
[  671.153944] [<8013cfec>] (kthread) from [<80108148>] (ret_from_fork+0x14/0x2c)
```

pelwell, 2017-05-05:
The mystery deepens. Pi2 and Pi3 both use the ARM architectural timers as their clock source, accessed from via the p15 co-processor interface by the `arch_counter_get_cntpct` function. Logging the values returned by this function shows a strange discontinuity around the time everything starts to go wrong:
```
[    4.919827]  mmcblk0: p1 p2
[  894.754884] arm_arch_timer: Timer history (400000029):
[  894.754884] arm_arch_timer:  5b54ec1
[  894.754880] arm_arch_timer:  5b55768
[  894.754878] arm_arch_timer:  5b55830
[  894.754882] arm_arch_timer:  5b560c4
[  894.754882] arm_arch_timer:  5b56179
[  894.754880] arm_arch_timer:  5b56a26
[  894.754881] arm_arch_timer:  5b56ae2
[  894.754882] arm_arch_timer:  400000020
```
And from later on:
```
[  894.754884] arm_arch_timer: Timer history (40000000e):
[  894.754881] arm_arch_timer:  400000019
[  894.754882] arm_arch_timer:  400000042
[  894.754878] arm_arch_timer:  400000055
[  894.754879] arm_arch_timer:  400000003
[  894.754879] arm_arch_timer:  40000000e
[  894.754881] arm_arch_timer:  400000062
[  894.754879] arm_arch_timer:  400000000
[  894.754879] arm_arch_timer:  4000000ab
```
Notice how the timestamps are now shuffling back and forth rather than monotonically increasing. It is as if this register has been switched into another mode, but I don't know what that mode would be and I can't see any evidence of such a switch; the frequency (CNTFRQ) remains the same (19.2MHz), as does the control register (CNTKCTL).

Are there any ARM gurus out there who can shed some light? @lategoodbye? @ED6E0F17?

pelwell, 2017-05-05:
@swarren?

swarren, 2017-05-05:
In the maxcpus!=4 case, some CPUs will permanently be executing the loop in the armstub waiting to be released. Hence, the content of this memory is permanently important. Is the kernel accidentally re-using that page for something else, thus causing the unused secondary CPUs to go off an execute random code? I don't know why that would always cause arch timer issues though.

When all 4 CPUs are in use, all the CPUs jump out of that code very early on, thus making it quite unlikely any such bug would be hit.

The value 400000020 looks special, as if it's a 400000000 flag plus value 20. Perhaps you can add code into writel and friends that checks for that value (or something in that range) and prints a backtrace in case it's caused by a stray register write?

pelwell, 2017-05-05:
Ooh, interesting thought:
```
c0000000: 0000004d 00000000 00104278 000e822c M.......|xB..,...
c0000010: 00000000 0009dfe4 00000000 00000000 ........|........
c0000020: 0000004d 00000000 000f1664 00000000 M.......|d.......
c0000030: 00000000 0009de90 00000000 00000000 ........|........
c0000040: 0000004d 00000000 00104288 00000000 M.......|.B......
c0000050: 00000000 0009c494 00000000 00000000 ........|........
c0000060: 0000004d 00000000 00104294 00000000 M.......|.B......
c0000070: 00000000 0009d4a8 00000000 00000000 ........|........
c0000080: 0000004d 00170000 0010429c 00000000 M.......|.B......
c0000090: 00000000 0009d384 00000000 00000000 ........|........
c00000a0: 0000004d 00170000 001042a4 00000000 M.......|.B......
c00000b0: 00000000 0009d260 00000000 00000000 ....`...|........
c00000c0: 0000004d 00170000 001042b0 00000000 M.......|.B......
c00000d0: 00000000 0009d13c 00000000 00000000 ....<...|........
c00000e0: 0000004d 00170000 001042b8 00000000 M.......|.B......
c00000f0: 00000000 0009d01c 00000000 00000000 ........|........
```
The stub is loaded to 0xc0000000 (0x00000000 uncached, as seen from the VPU), and it has clearly been trashed by something.

Thanks for the suggestion - I'll follow up with an explanation when I find one.

pelwell, 2017-05-05:
I have a fix which is somewhere between a solution and a workaround:
```
	reserved-memory {
		#address-cells = <1>;
		#size-cells = <1>;
		ranges = <>;

		rpi,stubs {
			reg = <0x0 0x1000>;
		};
	};
```
This bit of Device Tree magic will declare the first page as off-limits.

lategoodbye, 2017-05-05:
@pelwell You were faster than me. I think this could be an acceptable way.

lategoodbye, 2017-05-05:
Is the size of the armstub variable?

swarren, 2017-05-05:
I wonder if the kernel code should reserve this itself; the memory can be released if it's known that all possible CPUs have been released from the spin loop. Let met check...

swarren, 2017-05-05:
Looks like this is basically the correct approach, although the kernel docs recommend using /memreserve/ instead:

https://www.kernel.org/doc/Documentation/arm64/booting.txt

```
The boot loader is expected to enter the kernel on each CPU in the
following manner:

- The primary CPU must jump directly to the first instruction of the
  kernel image.  The device tree blob passed by this CPU must contain
  an 'enable-method' property for each cpu node.  The supported
  enable-methods are described below.

  It is expected that the bootloader will generate these device tree
  properties and insert them into the blob prior to kernel entry.

- CPUs with a "spin-table" enable-method must have a 'cpu-release-addr'
  property in their cpu node.  This property identifies a
  naturally-aligned 64-bit zero-initalised memory location.

  **These CPUs should spin outside of the kernel in a reserved area of
  memory (communicated to the kernel by a /memreserve/ region in the
  device tree)** polling their cpu-release-addr location, which must be
  contained in the reserved region.  A wfe instruction may be inserted
  to reduce the overhead of the busy-loop and a sev will be issued by
  the primary CPU.  When a read of the location pointed to by the
  cpu-release-addr returns a non-zero value, the CPU must jump to this
  value.  The value will be written as a single 64-bit little-endian
  value, so CPUs must convert the read value to their native endianness
  before jumping to it.
```

popcornmix, 2017-05-05:
The armstub only uses 0x0-0xff.
ATAGS (when enabled) are at 0x100.

lategoodbye, 2017-05-05:
According to the binding document and this [example](http://lists.infradead.org/pipermail/linux-arm-kernel/2017-January/478570.html), we should add a "no-map" property to the stubs node.

pelwell, 2017-05-08:
Thanks, @swarren @lategoodbye, having had time to read around I think the `/memreserve/` option is preferable. Interestingly, it is already conditionally enabled for 64-bit (bcmrpi3_defconfig) builds of bcm2710-rpi-3-b.dts, so I propose to move this to either bcm283x.dtsi. Since bcm2835-based devices only have a single ARM core and tend to be more limited in RAM, I could make it conditional on CONFIG_SMP, but that seems unnecessary for the sake of 4KB.

pelwell, 2017-05-08:
I have a patch adding /memreserve/ for all Pis waiting to go - I think it's safe to merge.

While we're in this area, I think it's time to add a `wfe` to the ARMv and ARMv8-32 stubs to reduce power consumption and bus traffic during startup (and later, if `maxcpus=<n>` has been used to keep some cores offline). The corresponding kernel-side change (adding an `sev` to the secondary-core-start function) is harmless, but a wfe with an sev will cause the secondary cores to sleep indefinitely.

@popcornmix - I propose merging both kernel side patches now, and I'll send you the armstubs patches - one for armstub7.S, and one for the generated header - for perusal and possible delaying.

popcornmix, 2017-05-08:
@pelwell agree that adding the wfe/sev is desirable, but the wfe alone will break older kernels.
Adding the sev for now sounds good.
The wfe is trickier, but should be added in the future.

pelwell, 2017-05-08:
I've pushed the kernel-side commits.

lategoodbye, 2017-05-08:
So these changes must go upstream or we are in trouble after release of the new armstubs?

pelwell, 2017-05-09:
People using the upstream kernel can always put copies of the old stub
files on their SD card/network boot directory, but that isn't ideal. I'll
upstream these patches as well.

swarren, 2017-05-09:
Having to copy old firmware files sounds like a nasty regression that can cause user-visible problems. I'd like to see this rolled out in a smoother fashion if at all possible. (1) Get the required sev present in all OSs (this includes RPi Foundation Linux, upstream Linux, Linux distro kernels, all non-Linux OSs that exist and use SMP), then wait a decent amount of time, and only then update the stubs in a way that requires an OS update. Alternatively, conditionalize the feature on some setup.txt flag that the RPi disk images can set, but nobody else does yet. The default value of the flag could be flipped in e.g. 6 months after people have time to convert.

pelwell, 2017-05-09:
We're considering adding flags or a "breaking step" counter to our DTB that the firmware can check, which would then allow it to nop-out the `wfe` for old kernels. However, I have a feeling that wouldn't upstream, since they can be very picky about what constitutes acceptable DT content.

lategoodbye, 2017-05-09:
Adding a temporary flag to the DTB won't be accepted upstream. Is adding a flag to config.txt not possible here?

popcornmix, 2017-05-10:
The dtb files change with the kernel, so having the flag there means it is just work transparently.
config.txt could be used, but that would need manually updating when kernel changes.

lategoodbye, 2017-05-10:
I understand, but as i said this dts change won't be accepted in upstream.

pelwell, 2017-05-10:
Then I won't try to submit it.

pelwell, 2017-05-10:
I propose to add a new property, `/chosen/rpi,firmware-abi-version`. At boot time, the firmware reads the property from the .dtb, establishing the age of the kernel, then overwrites it with its ABI version. This allows the firmware and (theoretically at least) the kernel to make adjustments for an out-of-date peer.

The ABI version starts at 1 - absent and 0 are treated the same - and is incremented with every breaking change to the ABI (which should be a rare occurrence). I'm using `chosen` because, as the kernel docs say:

    The chosen node does not represent a real device, but serves as a place
    for passing data between firmware and the operating system, like boot
    arguments.

I understand that the author was thinking of firmware->kernel data flow, but this location should at least hint at firmware interaction - that and the property name.

lategoodbye, 2017-05-10:
The idea is good, but i can't speak for all upstream guys.

@anholt @nullr0ute Any comments?

swarren, 2017-05-10:
There's already a /soc/firmware node upstream. It would make most sense to put any new DT property here, since it's a property of the firmware. This proposal should be reviewed upstream (on the mailing lists) by the DT binding and ARM maintainers. Also, what about OSs that use SMP but don't use DT (RISC OS, any bare metal users, ... How will this non-backwards-compatible change be communicated to them?)

```
        soc {
                firmware: firmware {
                        compatible = "raspberrypi,bcm2835-firmware";
                        mboxes = <&mailbox>;
                };

                power: power {
                        compatible = "raspberrypi,bcm2835-power";
                        firmware = <&firmware>;
                        #power-domain-cells = <1>;
                };
        };

```

pelwell, 2017-05-10:
I thought of /soc/firmware, but didn't choose it because chosen is specifically for values from the firmware. I'll reconsider now.

Ah - an interesting observation: without the kernel-side sev, a Pi 2 with a wfe in the spin loop still boots. Presumably there are enough other sevs scattered through the early code that omitting this one doesn't matter very much.

This makes the whole issue of signalling an incompatible kernel moot for now - but I'd still like to work through how we might add such a mechanism.

zougloub, 2017-05-11:
Just wanted to say I'm so glad there is a fix available for this, this issue only drove me crazy for a few hours. @pelwell I'm thanking you once again today.

JamesH65, 2017-05-19:
@pelwell Closable?

pelwell, 2017-05-19:
Yes - my patch is making its way upstream, and we'll eventually apply the firmware patch (although I've found that there are enough other sevs that we could probably update the firmware now).

closed 2017-05-19 by pelwell

Phlamethrower, 2017-07-22:
> Also, what about OSs that use SMP but don't use DT (RISC OS, any bare metal users, ... How will this non-backwards-compatible change be communicated to them?)

FWIW, RISC OS should be fine. When I first wrote the code to break the other cores out of the spin loop in the boot stub, I made sure to include a DSB + SEV just in case future versions of the boot stub added a WFE.

popcornmix, 2017-07-22:
That was good thinking.

pelwell, 2017-07-24:
Sufficient time (and a few Raspbian updates) have passed since the `sev` went into the kernels, so I've merged the armstubs patch that adds the corresponding `wfe`. The next firmware release is likely to include the updated built-in default stubs.

lategoodbye, 2017-07-25:
@pelwell Since "[PATCH] irq_bcm2836: Send event when onlining sleeping cores" has never been applied upstream, do you plan to rebase the patch on top of 4.13-rc1 ( https://github.com/anholt/linux/tree/bcm2835-soc-cleanup )?

pelwell, 2017-07-25:
I will, but it won't be for a week or two as I am out of the office. In practise we found that there are sufficient events that you don't really need an explicit one.

nullr0ute, 2017-07-26:
As long as upstream isn't going to break with the new firmware I'm fine with that, if there's a chance of breakage I think we need to be cautious

anholt, 2017-07-26:
I tried what I thought was the obvious port of your patch on top of bcm2835-soc-cleanup (right before the iounmap), and it makes my system hang late in boot (after init, before ssh comes up).

I wanted to compare the early boot state, but earlycon=uart8250,mmio32,0x3f215040 has been giving me garbage on the pi3 for a while (possibly related to a firmware update?  config.txt has enable_uart=1 and force_turbo=1)  firmware is ca9116ffd0cea06fdeb876ac7d19790bf2631d8c

pelwell, 2017-07-27:
Anyone who wants to try the stubs before we update the firmware can grab the source code from the tools repo and build it (there's a Makefile) then copy the .bins into the boot partition - the firmware will the use external files in preference to those compiled in.

pelwell, 2017-08-07:
I see no problems with the dsb+sev patch, but I did have a non-bootable system for hours because of a recent firmware change that was breaking simplefb. See [here](https://github.com/raspberrypi/firmware/issues/849#issuecomment-319488391) for an explanation and a temporary firmware to fix it.

I find it odd that there is only one upstream defconfig and that you have to modify the configuration to enable SMP.

lategoodbye, 2017-08-07:
The Raspberry Pi has a separate defconfig upstream and that is unusual.

RPI 1 + Zero: arch/arm/configs/bcm2835_defconfig
RPI 2 + 3 (32 bit): arch/arm/configs/multi_v7_defconfig
RPI 3 (64 bit): arch/arm64/configs/defconfig

pelwell, 2017-08-07:
Thanks - my searches didn't uncover that, but I can find references to it now.

I can confirm that building bcm2835-smp-soc-cleanup with multi_v7_defconfig, the sev patch and the fixed firmware results in a working SMP Pi3.

lategoodbye, 2017-08-07:
Okay, only RPI 2 and RPI 3 (64 bit) left ;-)