raspberrypi/linux #1992: nr_uarts=0 patch now superceeded by mainline change.

paulgortmaker, 2017-05-04:
I believe the rpi patch "serial: 8250: Don't crash when nr_uarts is 0" can be dropped from the patch queue for kernels 4.2 and above.  Mainline has this change:

commit 59cfc45f17d6d1dda2990e6f5a94df24a18330b8
Author: Jan Kiszka <jan.kiszka@siemens.com>
Date:   Tue May 5 08:26:27 2015 +0200

    serial: 8250: Do nothing if nr_uarts=0
    
    When nr_uarts was set to 0 (via config or 8250_core.nr_uarts), we crash
    early on x86 because serial8250_isa_init_ports dereferences base_ops
    which remains NULL. In fact, there is nothing to do for all the callers
    of serial8250_isa_init_ports if there are no uarts.
    
    Based on suggestions by Peter Hurley.
    
    Signed-off-by: Jan Kiszka <jan.kiszka@siemens.com>
    Signed-off-by: Greg Kroah-Hartman <gregkh@linuxfoundation.org>

popcornmix, 2017-05-05:
Yes, that does appear to address the same issue. I've reverted our commit from 4.9, 4.10 and 4.11 trees.
Thanks for pointer.

JamesH65, 2017-05-19:
Closing this issue as questions answered/issue resolved.

closed 2017-05-19 by JamesH65