raspberrypi/linux #1993: mmc1 error while using WiFi on CM3

sidsapien, 2017-05-04:
Hi, I am trying to use RTL8723BS wifi module with CM3 SDIO (pins 22-27) as in: https://hackaday.io/project/9934-raspberry-pi-rtl8723bs-sdio-gpio-150mbs-wifi-bt

I did dtoverlay in /boot/config.txt using: dtoverlay=sdio,poll_once=false
I also tried different sdio_overclock values but always getting the following error (repeating every second):

mmc-bcm2835 3f300000.mmc: no support for card's volts
mmc1: error -22 whilst initializing SDIO card
mmc1: Controller never released inhibit bit(s).
mmc1: Got command interrupt 0x00030000 even though no command operation was in progress.

Surprisingly in the tests I have done until now same circuit and dtoverlay works for WiFi with Raspberry Pi 2 but giving above error with Compute Module 3. Please help.

pelwell, 2017-05-07:
There is no reason why CM3 should behave any differently than Pi2 in this respect, so it is possible that your wiring to the CM3 may be faulty.

sidsapien, 2017-05-19:
@pelwell Writing my finding here for anyone interested:

It turns out that the internal GPIO wiring is probably somewhat different in CM3 than in RPi B+ as in the above link. Having capacitors in parallel to the SDIO data lines is not a problem in RPi B+ but in CM3 this leads to the SDIO chip not being detected or to problems while initializing.

JamesH65, 2017-05-19:
@pelwell @Roger-Thornton Any thoughts? I thought the GPIO wiring was the same/similar.

pelwell, 2017-05-19:
What is the purpose of the inline capacitors on the module?

Roger-Thornton, 2017-05-19:
The GPIO pins on CM3 are just connected to the pins on the edge of the boards.

pelwell, 2017-05-19:
Yes, I was talking about the SDIO module.

pelwell, 2017-05-22:
@sidsapien Do you have a schematic for the SDIO module?

sidsapien, 2017-05-22:
The schematic for SDIO module is here: [](https://easyeda.com/normal/RTL8723BSSchematic-oNHBvpHZh.png)

closed 2017-09-13 by JamesH65