raspberrypi/linux #1995: USB not working on rpi3 with arm64 kernel

ryncsn, 2017-05-07:
Hi, I'm trying to compile a arm64 kernel for rpi3, but USB is not working (on branch rpi-4.9.y).
After some research I found commit 8b68f431a6168c65f67f0c6b645e9c925f90a7c7 add depend on FIQ for dwc_otg, but seems there is no CONFIG_FIQ available with arm64 as fiq not implemented on arm64.
Commit 26ade7747b261cad2db5207a603cffafe9c845d0 just add a work around by using regular irq on arm64, so maybe dwc_otg should only depend on fiq on arm, not on arm64?

johannes-krude, 2017-05-10:
I can confirm this. I bisected this on rpi-4.10.y to 117407f76605efb83743294e1858f59d61853aaf .
I tested: reverting 8b68f431a6168c65f67f0c6b645e9c925f90a7c7 on rpi-4.9.y fixes this as well as reverting 117407f76605efb83743294e1858f59d61853aaf on rpi-4.10.y .

johannes-krude, 2017-05-10:
I created a pull request: #2001

johannes-krude, 2017-05-10:
2f2836b5100a80c4738097e4f2b41d31a7ea8281 solves this for the rpi-4.9.y branch

ryncsn, 2017-05-10:
Great to see it's working well again, closing this issue.

closed 2017-05-10 by ryncsn