raspberrypi/linux #1996: VCHIQ functional test broken with multi_v7_defconfig

lategoodbye, 2017-05-08:
Hi,
in mainline the only way to get a suitable config for a Raspberry Pi 2 is multi_v7_defconfig. Unfortunately i'm not able to get VCHIQ function test running with this defconfig (starting from 4.9 until f526a0155be1ad6666203db8dbb0be214f2838a2). The error pattern is mostly data corruption, but i also saw NULL pointer dereference in memcpy of free_pagelist and vchiq_test hangs.

Here is a output for 4.11 with latest vchiq fix:
```
pi@raspberrypi:~$ vchiq_test -f 10
Functional test - iters:10
======== iteration 1 ========
0002e594,0: 00 <-> ff
0002e595,1: 01 <-> ff
0002e596,2: 02 <-> ff
0002e597,3: 03 <-> ff
0002e598,4: 04 <-> ff
0002e599,5: 05 <-> ff
0002e59a,6: 06 <-> ff
0002e59b,7: 07 <-> ff
0002e59c,8: 08 <-> ff
0002e59d,9: 09 <-> ff
0002e59e,a: 0a <-> ff
0002e59f,b: 0b <-> ff
mem_check failed - buffer 2e594, size 1024
0002ed80,3ec: ec <-> ff
0002ed81,3ed: ed <-> ff
0002ed82,3ee: ee <-> ff
0002ed83,3ef: ef <-> ff
0002ed84,3f0: f0 <-> ff
0002ed85,3f1: f1 <-> ff
0002ed86,3f2: f2 <-> ff
0002ed87,3f3: f3 <-> ff
0002ed88,3f4: f4 <-> ff
0002ed89,3f5: f5 <-> ff
0002ed8a,3f6: f6 <-> ff
0002ed8b,3f7: f7 <-> ff
0002ed8c,3f8: f8 <-> ff
0002ed8d,3f9: f9 <-> ff
0002ed8e,3fa: fa <-> ff
0002ed8f,3fb: fb <-> ff
0002ed90,3fc: fc <-> ff
0002ed91,3fd: fd <-> ff
0002ed92,3fe: fe <-> ff
mem_check failed - buffer 2e994, size 1024
Testing bulk transfer for alignment.
vchiq_test: 0: Data corrupted at 0-0 (datalen 1, align 0, server_align 0) -> ff
vchiq_test: 891: func_data_test(service, size, align & 31, srvr_align & 31) != VCHIQ_SUCCESS
```

Currently i wasn't able to narrow down to the relevant config setting(s) which causes the issue.

I already tried

`pagelistinfo->dma_dir =  DMA_BIDIRECTIONAL;`

but this hasn't influence on the issue.

@pelwell Any ideas?

pelwell, 2017-05-09:
Which DTB are you using? Or, more precisely, how are you activating vchiq, since it isn't referenced in the upstream DTBs?

lategoodbye, 2017-05-09:
[This](https://github.com/lategoodbye/rpi-zero/commit/b7ddd1d721ba60da325f331f53a59572f22b4132) plus [that](https://github.com/lategoodbye/rpi-zero/commit/5ad7765a39bf81af79573184d8d63b9949dff32e). Unfortunately we can't upstream these changes as long as the driver is in staging and dt-binding would be a longer discussion ...

pelwell, 2017-05-09:
I needed to add CONFIG_BCM2835_VCHIQ=y in order to build VCHIQ. I then realised that the resulting image uses HIGHMEM, which breaks VCHIQ. After adding CONFIG_VMSPLIT_2G=y the functional test is passing for me.

lategoodbye, 2017-05-09:
Sorry, i don't have deep knowledge about vchiq.
Could you please explain why HIGHMEM=n and CONFIG_VMSPLIT_2G=y are necessary?
Do you see a chance to make it work with these options?

pelwell, 2017-05-09:
HIGHMEM is a problem because you can't use get_user_pages on pages in
HIGHMEM. From what I recall, the code is ugly and the performance suffers.

It shouldn't be necessary to use both config settings - with just the
VMSPLIT_2G it won't use HIGHMEM. It will also run with just HIGHMEM=n, but
with reduced memory.

lategoodbye, 2017-05-09:
Isn't there way to enforce pages to be not from HIGHMEM area?
Otherwise we need to fix module dependency or give a warning when a page is from HIGHMEM area.

pelwell, 2017-05-09:
For a general user page? That sounds unlikely, but I'm happy to be proved wrong.

lategoodbye, 2017-05-09:
I'm wondering there are so many users of get_user_pages*, i can't believe that all of them break on HIGHMEM.

http://elixir.free-electrons.com/linux/latest/ident/get_user_pages

lategoodbye, 2017-08-26:
Great news, i think i've found the solution for the problem. In free_pagelist() the first and last page are accessed with page_address(). This okay as long as this a low mem page or the page is mapped. Otherwise this function returns NULL and cause trouble. So we better use kmap:

https://gist.github.com/lategoodbye/7922e7d08166f5daf43cf66aa738a5dc

popcornmix, 2017-08-27:
Interesting. @pelwell should we do this?

pelwell, 2017-08-28:
I'd like to know the performance impact of the kmap, particularly on BCM2835 devices that will never need it. Reworking the patch to use kmap as a fallback would be an option.

lategoodbye, 2017-08-28:
The performance impact should be little, because kmap would call page_address() in low mem case ( http://elixir.free-electrons.com/linux/latest/source/arch/arm/mm/highmem.c#L37 ). So it doesn't make sense to call kmap as fallback.

Is it right, that free_pagelist() is never called from interrupt context?
Is there a preferred performance measure scenario?

lategoodbye, 2017-10-09:
Notice: My patch has been applied upstream for 4.14 and will be queued for the upcoming 4.9 and 4.13.

closed 2017-10-29 by lategoodbye