raspberrypi/linux #1997: RPI Zero doesn't get enumerated in case g_serial is builtin

lategoodbye, 2017-05-08:
Hi,
i already posted this issue on [linux-usb](http://marc.info/?l=linux-usb&m=149227909111970&w=2), but didn't get any feedback. I tested g_serial with a Raspberry Pi Zero (OTG mode, Linux 4.11-rc1) as gadget and a Linux Laptop as host (Linux 4.10.9). In case g_serial is build as a kernel module the RPi Zero gets enumerated on the Linux host and in case of g_serial builtin not. In both cases i didn't provide any gadget configuration via configfs or kernel commandline.

**Gadget builtin (bad case)**

dmesg (Linux Host)
```
[19755.687770] usb 2-1.1.3: new high-speed USB device number 20 using ehci-pci
[19756.055743] usb 2-1.1.3: new high-speed USB device number 21 using ehci-pci
[19761.271755] usb 2-1.1.3: device descriptor read/64, error -110
[19776.887724] usb 2-1.1.3: device descriptor read/64, error -110
[19777.075724] usb 2-1.1.3: new high-speed USB device number 22 using ehci-pci
[19782.263715] usb 2-1.1.3: device descriptor read/64, error -110
[19797.879679] usb 2-1.1.3: device descriptor read/64, error -110
[19798.067654] usb 2-1.1.3: new high-speed USB device number 23 using ehci-pci
[19808.759654] usb 2-1.1.3: device not accepting address 23, error -110
[19808.843564] usb 2-1.1.3: new high-speed USB device number 24 using ehci-pci
[19819.511555] usb 2-1.1.3: device not accepting address 24, error -110
[19819.511699] usb 2-1.1-port3: unable to enumerate USB device
```

.config (RPi Zero)
```
CONFIG_USB_DWC2=y
CONFIG_USB_DWC2_DUAL_ROLE=y
CONFIG_USB_PHY=y
CONFIG_NOP_USB_XCEIV=y
CONFIG_USB_GADGET=y
CONFIG_USB_LIBCOMPOSITE=y
CONFIG_USB_F_ACM=y
CONFIG_USB_U_SERIAL=y
CONFIG_USB_F_SERIAL=y
CONFIG_USB_F_OBEX=y
CONFIG_USB_G_SERIAL=y
```

dmesg (RPi Zero)

```
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 4.11.0-rc1+ (user@Raspberry) (gcc version 4.7.1 20120402 \
                (prerelease) (crosstool-NG 1.15.2) ) #4 Sat Apr 15 12:31:27 UTC 2017
...
[    0.691449] udc-core: couldn't find an available UDC - added [g_serial] to list of \
                pending drivers
...
[    2.265701] dwc2 20980000.usb: EPs: 8, dedicated fifos, 4080 entries in SPRAM
[    2.280590] g_serial gadget: Gadget Serial v2.4
[    2.291998] g_serial gadget: g_serial ready
[    2.302774] dwc2 20980000.usb: bound driver g_serial
[    2.314624] dwc2 20980000.usb: DWC OTG Controller
[    2.325990] dwc2 20980000.usb: new USB bus registered, assigned bus number 1
[    2.339981] dwc2 20980000.usb: irq 33, io mem 0x20980000
[    2.353571] hub 1-0:1.0: USB hub found
[    2.364030] hub 1-0:1.0: 1 port detected
```

ED6E0F17, 2017-05-09:
The obvious difference between your set-up and mine is 5110feed84810fec42bd6d5f1b4b220841fcfb82 - I have not tested with CONFIG_NOP_USB_XCEIV 

I have not had any problems with composite serial enumerating on the downstream kernel since https://github.com/raspberrypi/linux/pull/1825 , but I believe that issue may have been caused by not having a phy.

lategoodbye, 2017-05-09:
It works for me, if i build g_serial as a kernel module with the same setup.

So did you test it with g_serial as builtin and dwc2 as dual role?

ED6E0F17, 2017-05-13:
I was able to add CONFIG_NOP_USB_XCEIV and the device-tree change to my builds with no issues.

I seem to be using "CONFIG_USB_CDC_COMPOSITE=y" with "CONFIG_USB_U_SERIAL=y" but NOT "CONFIG_USB_G_SERIAL" or "CONFIG_USB_F_SERIAL", so I will try to get closer to your config.

ED6E0F17, 2017-05-13:
It is still working for me with G_SERIAL and F_SERIAL in the config.
````
< CONFIG_USB_U_ETHER=y
< CONFIG_USB_F_ECM=y
< # CONFIG_USB_G_SERIAL is not set
< CONFIG_USB_CDC_COMPOSITE=y

> CONFIG_U_SERIAL_CONSOLE=y
> CONFIG_USB_F_SERIAL=y
> CONFIG_USB_F_OBEX=y
> CONFIG_USB_G_SERIAL=y
> # CONFIG_USB_CDC_COMPOSITE is not set
````
I get the expected dmesg results, and lose g_ether:
````
[265007.592665] usb 1-2: New USB device found, idVendor=0525, idProduct=a4a7
[265007.592676] usb 1-2: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[265007.592683] usb 1-2: Product: Gadget Serial v2.4
[265007.592688] usb 1-2: Manufacturer: Linux 4.11.0-armv6+ with 20980000.usb
[265007.599067] cdc_acm 1-2:2.0: ttyACM0: USB ACM device

[    2.305187] hub 1-0:1.0: USB hub found
[    2.329449] dwc2 20980000.usb: bound driver g_serial
[    2.536831] dwc2 20980000.usb: new device is high-speed
[    2.696995] dwc2 20980000.usb: new device is high-speed
[    2.823849] g_serial gadget: high-speed config #2: CDC ACM config
[    2.826397] gs_console_connect: port num [0] is not support console
````
That is with DUAL_ROLE, "otg" not forced "peripheral" mode, and booting with the USB data cable as the only Power Supply.

I do not get any messages for "udc".

lategoodbye, 2017-05-13:
Maybe this issue is not reproducible with rpi-4.11 tree? I tested only the mainline tree so far.

Steps to reproduce the config:
make bcm2835_defconfig
make menuconfig
* enable OTG support
* enable NOP_USB_XCEIV
* enable USB gadget
* select serial gadget (with CDC ACM and CDC OBEX support)
* select DWC2 dual role

The dmesg of your Raspberry Pi Zero shows a different order. In my case the gadget is registered before the hub is found.

lategoodbye, 2017-05-13:
If i apply this hacky [patch](https://gist.github.com/lategoodbye/5e842f22c1f5a94fde24b466569a7d10) (no proper solution!) the RPi Zero gets enumerated without any error messages. So it looks like a timing issue.

ED6E0F17, 2017-05-14:
I thought it was odd that your g_serial device was being loaded (and failing) so much earlier than mine.

I appear to have hit this issue on 4.12-rc1, where it failed to enumerate with mode "otg", but did enumerate with mode "peripheral", and now gives the udc message:
`````
[    1.444363] udc-core: couldn't find an available UDC - added [g_serial] to list of pending drivers

lategoodbye, 2017-05-14:
It seems like a race between the gadget enumeration and host init. I looked at some other otg driver (dwc3, chipidea, isp1760, mtu3, musb) and most of them init the host first. Unfortunately i didn't get a reply from the Synopsys guys at linux-usb.

Regarding to your observations in 4.12-rc1, what do mean with mode "otg"? The dr_mode in devicetree or Kconfig setting of the dwc2 driver? These have a different influence on the runtime behavior.

Btw i prepare the patch series to enable OTG support for Raspberry Pi for Linux 4.13. So this should make it a little bit easier in the future.

ED6E0F17, 2017-05-15:
````
with mode "otg"? 
````
With the downstream kernel you can change the "dr_mode" in device-tree with an overlay setting in the boot partition config.txt - I have not done enough testing to be sure that this change was significant, it was just the easiest thing to change.

ED6E0F17, 2017-05-15:
g_serial in now consistently working after rebuilding 4.12  - and there is a timing difference between the two builds:
````
[    1.566134] usbcore: registered new interface driver smsc95xx
[    2.082970] usbcore: registered new interface driver cdc_acm
[    2.090916] usbcore: registered new interface driver usb-storage

and

[    1.440205] usbcore: registered new interface driver smsc95xx
[    1.443838] usbcore: registered new interface driver cdc_acm
[    1.444131] usbcore: registered new interface driver usb-storage
[    1.444363] udc-core: couldn't find an available UDC - added [g_serial] to list of pending drivers
````
The build that does not have a delay before loading "cdc_acm" will enumerate g_serial when dr_mode is 
 set to "peripheral"

JamesH65, 2017-09-13:
@ED6E0F17  Can this be closed?

JamesH65, 2018-06-27:
Closing this issue as questions answered/issue resolved.

closed 2018-06-27 by JamesH65

ED6E0F17, 2018-06-28:
Hi James, I am sorry that I missed some Github alerts last year. This looks like an upstream race condition so there was no need to keep the issue open here - and this should not affect Raspbian kernels because usb gadgets are built as modules.

I cannot see that this has been fixed upstream.

maksim-ka, 2020-04-29:
Sry not sure if I better should open another issue, but I think this still holds up to the latest kernel in the buster image. enumeration is only possible on and full reboot for me. I mean the pi is fully bootet and i reboot my host machine. pluggin the pizero in on an booted host always results in an `unable to enumerate` error

pelwell, 2020-04-29:
What is your host machine, and why do you think the failure to enumerate is because of a problem with the Zero and not the host?

maksim-ka, 2020-04-29:
I tested it on my with own machine which runs an Manjaro with Linux 5.6.6
some endpoints got enumerated others didn't. Everything enumerated only
when I did reboot (guess there's some service I could have restarted, but
don't know which). That's what dmesg claimed. than I tried a friends
windows machine which recognized only the masstorage. Again the  stretch
image work right away on manjaro so far - that's what I'm working with
right now