raspberrypi/linux #1998: USB issue during audio playback and webcam usage

PeterPablo, 2017-05-09:
Dear team,

due to recent advances regarding USB transfers I wanted to give the new `rpi-update` kernel/firmware `Linux rpi2 4.9.25-v7+ #995 SMP Thu May 4 15:52:25 BST 2017 armv7l GNU/Linux` a try on my Raspberry Pi 3 (do not get distracted by the name... I upgraded hardware from a model 2).

Here is the lsusb output (HDD, HifiBerry DAC+ Pro, Logitech Webcam):
```
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=dwc_otg/1p, 480M
    |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/5p, 480M
        |__ Port 1: Dev 3, If 0, Class=Vendor Specific Class, Driver=smsc95xx, 480M
        |__ Port 2: Dev 6, If 0, Class=Mass Storage, Driver=usb-storage, 480M
        |__ Port 3: Dev 4, If 2, Class=Human Interface Device, Driver=usbhid, 12M
        |__ Port 3: Dev 4, If 0, Class=Human Interface Device, Driver=usbhid, 12M
        |__ Port 3: Dev 4, If 1, Class=Human Interface Device, Driver=usbhid, 12M
        |__ Port 5: Dev 5, If 0, Class=Vendor Specific Class, Driver=sonixj, 12M
        |__ Port 5: Dev 5, If 1, Class=Audio, Driver=snd-usb-audio, 12M
        |__ Port 5: Dev 5, If 2, Class=Audio, Driver=snd-usb-audio, 12M
```

I take a webcam picture every minute (via cron) and this sometimes leads to:
```
May  8 09:54:03 rpi2 kernel: [83801.672246] Transfer to device 5 endpoint 0x1 frame 231 failed - FIQ timed out. Data may have been lost.
May  8 09:54:03 rpi2 kernel: [83801.695007] gspca_main: ISOC data error: [0] len=0, status=-63
```

In the evening I started playing music over my HifiBerry DAC+ Pro which continuously lead to a lot of messages like:
````
May  8 23:07:34 rpi2 kernel: [131412.568253] bcm2835-dma 3f007000.dma: DMA transfer could not be terminated
May  8 23:07:40 rpi2 kernel: [131418.574853] bcm2835-i2s 3f203000.i2s: I2S SYNC error!
May  8 23:07:40 rpi2 kernel: [131418.588710] bcm2835-dma 3f007000.dma: DMA transfer could not be terminated
May  8 23:07:40 rpi2 kernel: [131418.589744] bcm2835-i2s 3f203000.i2s: I2S SYNC error!
May  8 23:07:42 rpi2 kernel: [131420.609891] bcm2835-dma 3f007000.dma: DMA transfer could not be terminated
May  8 23:07:44 rpi2 kernel: [131422.806334] bcm2835-dma 3f007000.dma: DMA transfer could not be terminated
May  8 23:07:46 rpi2 kernel: [131425.035625] bcm2835-dma 3f007000.dma: DMA transfer could not be terminated
May  8 23:07:48 rpi2 kernel: [131427.241291] bcm2835-dma 3f007000.dma: DMA transfer could not be terminated
May  8 23:07:50 rpi2 kernel: [131429.453291] bcm2835-dma 3f007000.dma: DMA transfer could not be terminated
```

Regarding my energy supply:
```
vcgencmd measure_temp; vcgencmd get_throttled; vcgencmd measure_temp
temp=49.4'C
throttled=0x0
temp=49.4'C
```

I'll glady supply more detailed information, if required.

PeterPablo, 2017-05-09:
I just noticed that yet another updated kernel/firmware is available, so I updated, started music and `dmesg` looks clean after the first few minutes of  playback/webcam usage:
```
pi@rpi2 ~ $ uname -a; uptime; dmesg
Linux rpi2 4.9.26-v7+ #996 SMP Fri May 5 17:58:35 BST 2017 armv7l GNU/Linux
 08:07:02 up 3 min,  2 users,  load average: 0,75, 0,62, 0,27
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 4.9.26-v7+ (dc4@dc4-XPS13-9333) (gcc version 4.9.3 (crosstool-NG crosstool-ng-1.22.0-88-g8460611) ) #996 SMP Fri May 5 17:58:35 BST 2017
[    0.000000] CPU: ARMv7 Processor [410fd034] revision 4 (ARMv7), cr=10c5383d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] OF: fdt:Machine model: Raspberry Pi 3 Model B Rev 1.2
[    0.000000] cma: Reserved 8 MiB at 0x36800000
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] On node 0 totalpages: 225280
[    0.000000] free_area_init_node: node 0, pgdat 80c6edc0, node_mem_map b603a000
[    0.000000]   Normal zone: 1980 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 225280 pages, LIFO batch:31
[    0.000000] percpu: Embedded 14 pages/cpu @b5ff4000 s25600 r8192 d23552 u57344
[    0.000000] pcpu-alloc: s25600 r8192 d23552 u57344 alloc=14*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 223300
[    0.000000] Kernel command line: 8250.nr_uarts=0 bcm2708_fb.fbwidth=1824 bcm2708_fb.fbheight=1016 bcm2708_fb.fbswap=1 vc_mem.mem_base=0x3dc00000 vc_mem.mem_size=0x3f000000  dwc_otg.lpm_enable=0 console=ttyS0,115200 console=tty1 root=/dev/mmcblk0p2 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait
[    0.000000] PID hash table entries: 4096 (order: 2, 16384 bytes)
[    0.000000] Dentry cache hash table entries: 131072 (order: 7, 524288 bytes)
[    0.000000] Inode-cache hash table entries: 65536 (order: 6, 262144 bytes)
[    0.000000] Memory: 871344K/901120K available (7168K kernel code, 483K rwdata, 2012K rodata, 1024K init, 778K bss, 21584K reserved, 8192K cma-reserved)
[    0.000000] Virtual kernel memory layout:
    vector  : 0xffff0000 - 0xffff1000   (   4 kB)
    fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
    vmalloc : 0xb7800000 - 0xff800000   (1152 MB)
    lowmem  : 0x80000000 - 0xb7000000   ( 880 MB)
    modules : 0x7f000000 - 0x80000000   (  16 MB)
      .text : 0x80008000 - 0x80800000   (8160 kB)
      .init : 0x80b00000 - 0x80c00000   (1024 kB)
      .data : 0x80c00000 - 0x80c78f94   ( 484 kB)
       .bss : 0x80c7a000 - 0x80d3ca64   ( 779 kB)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] Hierarchical RCU implementation.
[    0.000000] 	Build-time adjustment of leaf fanout to 32.
[    0.000000] NR_IRQS:16 nr_irqs:16 16
[    0.000000] arm_arch_timer: Architected cp15 timer(s) running at 19.20MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x46d987e47, max_idle_ns: 440795202767 ns
[    0.000008] sched_clock: 56 bits at 19MHz, resolution 52ns, wraps every 4398046511078ns
[    0.000025] Switching to timer-based delay loop, resolution 52ns
[    0.000336] Console: colour dummy device 80x30
[    0.001231] console [tty1] enabled
[    0.001281] Calibrating delay loop (skipped), value calculated using timer frequency.. 38.40 BogoMIPS (lpj=192000)
[    0.001350] pid_max: default: 32768 minimum: 301
[    0.001711] Mount-cache hash table entries: 2048 (order: 1, 8192 bytes)
[    0.001754] Mountpoint-cache hash table entries: 2048 (order: 1, 8192 bytes)
[    0.002773] Disabling cpuset control group subsystem
[    0.002965] CPU: Testing write buffer coherency: ok
[    0.003036] ftrace: allocating 22387 entries in 66 pages
[    0.054657] CPU0: update cpu_capacity 1024
[    0.054722] CPU0: thread -1, cpu 0, socket 0, mpidr 80000000
[    0.054818] Setting up static identity map for 0x100000 - 0x100034
[    0.057050] CPU1: update cpu_capacity 1024
[    0.057057] CPU1: thread -1, cpu 1, socket 0, mpidr 80000001
[    0.057779] CPU2: update cpu_capacity 1024
[    0.057786] CPU2: thread -1, cpu 2, socket 0, mpidr 80000002
[    0.058454] CPU3: update cpu_capacity 1024
[    0.058461] CPU3: thread -1, cpu 3, socket 0, mpidr 80000003
[    0.058544] Brought up 4 CPUs
[    0.058718] SMP: Total of 4 processors activated (153.60 BogoMIPS).
[    0.058748] CPU: All CPU(s) started in HYP mode.
[    0.058774] CPU: Virtualization extensions available.
[    0.059584] devtmpfs: initialized
[    0.070766] VFP support v0.3: implementor 41 architecture 3 part 40 variant 3 rev 4
[    0.071082] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[... 200 lines cut ...]
[    5.637917]  sda: sda1 sda2
[    5.639586] sd 0:0:0:0: Attached scsi generic sg0 type 0
[    5.642261] sd 0:0:0:0: [sda] Attached SCSI disk
[    6.333598] random: crng init done
[    8.892286] EXT4-fs (sda2): mounted filesystem with ordered data mode. Opts: user_xattr,acl
[    9.044675] systemd-journald[148]: Received request to flush runtime journal from PID 1
[   19.242631] logitech-hidpp-device 0003:046D:4024.0004: HID++ 2.0 device connected.
[   19.292918] uart-pl011 3f201000.serial: no DMA platform data
[... cut ...]
[   24.213332] Bluetooth: HCI UART driver ver 2.3
[   24.213341] Bluetooth: HCI UART protocol H4 registered
[   24.213344] Bluetooth: HCI UART protocol Three-wire (H5) registered
[   24.213457] Bluetooth: HCI UART protocol Broadcom registered
[   24.808980] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[   24.808987] Bluetooth: BNEP filters: protocol multicast
[   24.809001] Bluetooth: BNEP socket layer initialized
[   27.824890] Bluetooth: HIDP (Human Interface Emulation) ver 1.2
[   27.824910] Bluetooth: HIDP socket layer initialized
```

pelwell, 2017-05-09:
Thanks for the timely feedback. Feel free to close this issue when you are happy the problem is solved.

PeterPablo, 2017-05-09:
I'll close it right away and might reopen, if the issue reappears. Thank you!
(Possibly it helps somebody else to identify his issue using the search.)

PeterPablo, 2017-05-09:
Oh boy, I was too quick. I just had another look at the log and the tail of dmesg reads:
```
[   27.824910] Bluetooth: HIDP socket layer initialized
[ 1102.686758] Transfer to device 5 endpoint 0x1 frame 1544 failed - FIQ timed out. Data may have been lost.
[ 1102.709519] gspca_main: ISOC data error: [0] len=0, status=-63
```

PeterPablo, 2017-05-09:
Here is the detailed output regarding the usb device. It is not a Logitech but rather a Microsoft device, I just noticed:
```
pi@rpi2 ~ $ lsusb -v -s 5

Bus 001 Device 005: ID 045e:00f7 Microsoft Corp. LifeCam VX-1000
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               1.10
  bDeviceClass            0 (Defined at Interface level)
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x045e Microsoft Corp.
  idProduct          0x00f7 LifeCam VX-1000
  bcdDevice            1.01
  iManufacturer           0 
  iProduct                1 
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength          382
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           3
      bInterfaceClass       255 Vendor Specific Class
      bInterfaceSubClass    255 Vendor Specific Subclass
      bInterfaceProtocol    255 Vendor Specific Protocol
      iInterface              0 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            1
          Transfer Type            Isochronous
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0000  1x 0 bytes
        bInterval               1
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0001  1x 1 bytes
[... 406 lines cut ...]
        bBitResolution         16
        bSamFreqType            2 Discrete
        tSamFreq[ 0]         8000
        tSamFreq[ 1]        16000
      Endpoint Descriptor:
        bLength                 9
        bDescriptorType         5
        bEndpointAddress     0x84  EP 4 IN
        bmAttributes            1
          Transfer Type            Isochronous
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
        bRefresh                0
        bSynchAddress           0
        AudioControl Endpoint Descriptor:
          bLength                 7
          bDescriptorType        37
          bDescriptorSubtype      1 (EP_GENERAL)
          bmAttributes         0x01
            Sampling Frequency
          bLockDelayUnits         0 Undefined
          wLockDelay              0 Undefined
```

trulio, 2017-05-10:
Hi,

i'm getting usb audio issues since 4.9 kernel upgrade.
i submitted it here.

[USB audio stuttering since update to 4.9 kernel (USB packet management regression?)](https://github.com/raspberrypi/firmware/issues/753)

your problem could also be mine.

Thanks to the foundation for your efforts and great job.

Regards,

Trulio

P33M, 2017-05-15:
Leaving aside the I2S errors, does the USB webcam still work after one of these messages appears in the log?

PeterPablo, 2017-05-15:
Yes, the Webcam keeps working after one of the messages is logged.

P33M, 2017-05-15:
Then the messages are of no consequence - they are merely warnings (due to the documented limitations of the OTG core when performing split-isochronous transfers).

Do you still see I2S DMA errors on the updated kernel?

PeterPablo, 2017-05-15:
Unfortunately I am unable to ssh into my Pi currently for some unknown
reason. I'll report back tomorrow evening.

PeterPablo, 2017-05-16:
In order to be up-to-date with regards to the recent USB fixes I updated both my raspbian via `apt` and ran `rpi-update` and rebooted. This is the output of dmesg after quite some time of music playback and webcam activity:
```
~ $ dmesg; uname -a; uptime; vcgencmd get_throttled
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 4.9.28-v7+ (dc4@dc4-XPS13-9333) (gcc version 4.9.3 (crosstool-NG crosstool-ng-1.22.0-88-g8460611) ) #999 SMP Tue May 16 14:03:19 BST 2017
[    0.000000] CPU: ARMv7 Processor [410fd034] revision 4 (ARMv7), cr=10c5383d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] OF: fdt:Machine model: Raspberry Pi 3 Model B Rev 1.2
[    0.000000] cma: Reserved 8 MiB at 0x36800000
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] On node 0 totalpages: 225280
[    0.000000] free_area_init_node: node 0, pgdat 80c6edc0, node_mem_map b603a000
[    0.000000]   Normal zone: 1980 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 225280 pages, LIFO batch:31
[    0.000000] percpu: Embedded 14 pages/cpu @b5ff4000 s25600 r8192 d23552 u57344
[    0.000000] pcpu-alloc: s25600 r8192 d23552 u57344 alloc=14*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 223300
[    0.000000] Kernel command line: 8250.nr_uarts=0 bcm2708_fb.fbwidth=1824 bcm2708_fb.fbheight=1016 bcm2708_fb.fbswap=1 vc_mem.mem_base=0x3dc00000 vc_mem.mem_size=0x3f000000  dwc_otg.lpm_enable=0 console=ttyS0,115200 console=tty1 root=/dev/mmcblk0p2 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait
[    0.000000] PID hash table entries: 4096 (order: 2, 16384 bytes)
[    0.000000] Dentry cache hash table entries: 131072 (order: 7, 524288 bytes)
[    0.000000] Inode-cache hash table entries: 65536 (order: 6, 262144 bytes)
[    0.000000] Memory: 871340K/901120K available (7168K kernel code, 483K rwdata, 2012K rodata, 1024K init, 778K bss, 21588K reserved, 8192K cma-reserved)
[    0.000000] Virtual kernel memory layout:
    vector  : 0xffff0000 - 0xffff1000   (   4 kB)
    fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
    vmalloc : 0xb7800000 - 0xff800000   (1152 MB)
    lowmem  : 0x80000000 - 0xb7000000   ( 880 MB)
    modules : 0x7f000000 - 0x80000000   (  16 MB)
      .text : 0x80008000 - 0x80800000   (8160 kB)
      .init : 0x80b00000 - 0x80c00000   (1024 kB)
      .data : 0x80c00000 - 0x80c78f94   ( 484 kB)
       .bss : 0x80c7a000 - 0x80d3ca64   ( 779 kB)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] Hierarchical RCU implementation.
[    0.000000] 	Build-time adjustment of leaf fanout to 32.
[    0.000000] NR_IRQS:16 nr_irqs:16 16
[    0.000000] arm_arch_timer: Architected cp15 timer(s) running at 19.20MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x46d987e47, max_idle_ns: 440795202767 ns
[    0.000008] sched_clock: 56 bits at 19MHz, resolution 52ns, wraps every 4398046511078ns
[    0.000025] Switching to timer-based delay loop, resolution 52ns
[    0.000330] Console: colour dummy device 80x30
[    0.001222] console [tty1] enabled
[    0.001270] Calibrating delay loop (skipped), value calculated using timer frequency.. 38.40 BogoMIPS (lpj=192000)
[    0.001341] pid_max: default: 32768 minimum: 301
[    0.001708] Mount-cache hash table entries: 2048 (order: 1, 8192 bytes)
[    0.001751] Mountpoint-cache hash table entries: 2048 (order: 1, 8192 bytes)
[    0.002771] Disabling cpuset control group subsystem
[    0.002961] CPU: Testing write buffer coherency: ok
[    0.003029] ftrace: allocating 22386 entries in 66 pages
[    0.056050] CPU0: update cpu_capacity 1024
[    0.056117] CPU0: thread -1, cpu 0, socket 0, mpidr 80000000
[    0.056219] Setting up static identity map for 0x100000 - 0x100034
[    0.058462] CPU1: update cpu_capacity 1024
[    0.058469] CPU1: thread -1, cpu 1, socket 0, mpidr 80000001
[    0.059192] CPU2: update cpu_capacity 1024
[    0.059199] CPU2: thread -1, cpu 2, socket 0, mpidr 80000002
[    0.059864] CPU3: update cpu_capacity 1024
[    0.059871] CPU3: thread -1, cpu 3, socket 0, mpidr 80000003
[    0.059954] Brought up 4 CPUs
[    0.060128] SMP: Total of 4 processors activated (153.60 BogoMIPS).
[    0.060158] CPU: All CPU(s) started in HYP mode.
[    0.060185] CPU: Virtualization extensions available.
[    0.060997] devtmpfs: initialized
[    0.072174] VFP support v0.3: implementor 41 architecture 3 part 40 variant 3 rev 4
[    0.072491] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.072553] futex hash table entries: 1024 (order: 4, 65536 bytes)
[    0.073122] pinctrl core: initialized pinctrl subsystem
[... 231 lines cut ...]
[ 2088.454627] gspca_main: ISOC data error: [7] len=0, status=-71
Linux rpi2 4.9.28-v7+ #999 SMP Tue May 16 14:03:19 BST 2017 armv7l GNU/Linux
 21:42:33 up  1:14,  2 users,  load average: 0,13, 0,41, 0,35
throttled=0x0
```
Sound playback is fine and webcam continues to take pictures every minute.

Unfortunately though the internal Wifi of my Raspberry Pi 3B stopped working after updating to the bleeding-edge kernel/firmware and I had to use an old wifi stick and disable the internal wifi. My `config.txt` now reads as follows:
```
~ $ grep -vxE '[[:blank:]]*([#;].*)?' /boot/config.txt
overscan_top=-16
overscan_bottom=-16
overscan_scale=1
hdmi_group=1
hdmi_mode=16
hdmi_force_hotplug=1 
decode_MPG2=0xMyKey
decode_WVC1=0xMyKey
dtparam=audio=no
dtoverlay=hifiberry-dacplus
gpu_mem=128
dtoverlay=pi3-disable-wifi
```

Do you regard the issue as resolved?

P33M, 2017-05-17:
As the USB devices are still working, yes - the messages are advisory.

closed 2017-05-17 by P33M

PeterPablo, 2017-05-22:
Dear community,
I wanted to give a brief update -- my Raspberry Pi 3B internal wifi did not work as my 2.4 GHz wifi was automatically set to channel 13 in the meanwhile and could not be discovered using `sudo iwlist wlan0 scan | grep ESSID`. Manually setting it to run on channel 7 in my router made it work.

Dmesg is clean now apart from the previously mentioned spuriod (harmless) messages during playback:
```
...
[143055.026312] gspca_main: ISOC data error: [17] len=0, status=-71
[143955.548135] Transfer to device 6 endpoint 0x1 frame 246 failed - FIQ reported NYET. Data may have been lost.
[143955.564194] gspca_main: ISOC data error: [7] len=0, status=-71
[146655.237269] Transfer to device 6 endpoint 0x1 frame 671 failed - FIQ timed out. Data may have been lost.
[146775.837272] Transfer to device 6 endpoint 0x1 frame 439 failed - FIQ timed out. Data may have been lost.
[148155.666147] Transfer to device 6 endpoint 0x1 frame 1964 failed - FIQ reported NYET. Data may have been lost.
[148155.676427] gspca_main: ISOC data error: [13] len=0, status=-71
[154155.895166] Transfer to device 6 endpoint 0x1 frame 1553 failed - FIQ reported NYET. Data may have been lost.
```