raspberrypi/linux #2000: different /etc/machine-id on each restart

andrewmichaelsmith, 2017-05-09:
With a fresh image of Raspbian Jessie Lite 2017-04-10 I find the machine-id changes on each boot.

I've tried this with a Pi 2 and a Pi Zero W and notice the same behaviour:

- Boot for first time observe machine-id X
- sudo reboot
- Observe machine id Y

The machine-id changes on each reboot. I expected it stay the same. This becomes a problem especially when you use journald, which depends on the machine-id staying the same.

Here's what the system logs tell me:
```
pi@raspberrypi:~ $ sudo journalctl | grep machine
Apr 10 09:58:36 raspberrypi systemd[1]: Initializing machine ID from random generator.
Apr 10 09:58:36 raspberrypi systemd[1]: Installed transient /etc/machine-id file.
```

And machine-id is actually a tmpfs mount:

```
pi@raspberrypi:~ $ mount | grep machine
tmpfs on /etc/machine-id type tmpfs (ro,mode=755)
```
Anything else I can check?

popcornmix, 2017-05-09:
This should really be reported at https://github.com/RPi-Distro/repo as it doesn't sound like a linux kernel issue. 
I've just checked and my machine-id is persistent.
```
pi@domnfs:~ $ sudo journalctl | grep machine
pi@domnfs:~ $ mount | grep machine
pi@domnfs:~ $ ls -l /etc/machine-id 
-r--r--r-- 1 root root 33 Feb  1  2016 /etc/machine-id
pi@domnfs:~ $ cat machine-id 
a36570840c1b4696abcad1da4fe24a2d
```
You might want to read:
https://unix.stackexchange.com/questions/260367/why-does-etc-machine-id-changes-at-reboot
which gives a solution to this issue.

andrewmichaelsmith, 2017-05-09:
thanks @popcornmix I can move it.

Are you using lite?

I have figured out a way to solve it manually, but it feels like manual setup I have to do for every install is a bit of a bug.

andrewmichaelsmith, 2017-05-09:
Closing, still an issue but raising here: https://github.com/RPi-Distro/repo/issues/66

closed 2017-05-09 by andrewmichaelsmith

popcornmix, 2017-05-09:
It was a full jessie image I posted results from.
I've just ssh-ed into a pi at home which is running jessie lite and that also looks to be persistent.
```
pi@dvbdonkey:~ $ sudo journalctl | grep machine
pi@dvbdonkey:~ $ mount | grep machine-id
pi@dvbdonkey:~ $ ls -l /etc/machine-id 
-r--r--r-- 1 root root 33 Mar  3 15:23 /etc/machine-id
pi@dvbdonkey:~ $ cat /etc/machine-id 
8c5604e85c924195aef34a3e4079998d
```
But anyway the other repo is the better place for this report.

andrewmichaelsmith, 2017-05-09:
OK thanks. I can consistently reproduce with the base lite image.