raspberrypi/linux #2002: Failed to load w1-gpio overlay with non-default gpiopin value

softerra, 2017-05-10:
In short, the issue looks like:
```
pi@raspberrypi:~ $ sudo dtoverlay w1-gpio gpiopin=13
* Failed to apply overlay '0_w1-gpio' (kernel)

pi@raspberrypi:~ $ dmesg
...
[  145.010455] OF: resolver: __of_adjust_tree_phandle_references: Could not find target child 'onewire' @/__local_fixups__/fragment@0/__overlay__
[  145.010472] create_overlay: Failed to resolve tree
```
I used the RPi image of 2017-04-10 and w1-gpio loaded well. 
Then, since I did apt-get upgrade I've got the issue.

It seems that something was broken after kernel update 2c47f0d73ddb7ca4639a119ffce804e14ecf131a
Is it a kernel issue? Or it's a problem of dtoverlay utility?

pelwell, 2017-05-10:
Interesting - it's not immediately obvious why it is failing, but I think some part of the kernel overlay application mechanism has changed. I'll report back when I know more.

pelwell, 2017-05-11:
It turns out that this is a problem in the dtoverlay utility, caused by failure to take account of a difference between how the firmware merges overlays and how the kernel does it.

Overlays have to be "fixed up" - ensuring that label numbers don't clash with the base DTB - when loaded. They also need to have their parameters applied before they are merged with the base DTB. The firmware sequence is:
1. Load
2. Fix up
3. Apply parameters
4. Merge

The standard kernel overlay mechanism uses a different sequence:
1. Receive (something else does the loading)
2. Fix up
3. Merge

Note that the kernel doesn't understand parameters, so that stage is performed by the dtoverlay utility, leading to a combined sequence of:
1. (A) Load
2. (A) Apply parameters
3. (A) Send to kernel
4. (K) Receive
5. (K) Fix up
6. (K) Merge

where (A) means the dtoverlay application and (K) means the kernel.

The problem occurs because, in the most recent images, a parameter can change a node name, and node names form part of the information using during the fixing up process. Here's an extract from the w1-gpio overlay, where you can see the onewire@0 in both sets of fix-ups:
```
    __local_fixups__ {
        fragment@0 {
            __overlay__ {
                onewire@0 {
                    pinctrl-0 = <0x00000000>;
                };
            };
        };
        __overrides__ {
            gpiopin = <0x00000000 0x0000000c 0x00000016 0x00000026>;
            pullup = <0x00000000>;
        };
    };
    __fixups__ {
        gpio = "/fragment@0/__overlay__/onewire@0:gpios:0", "/fragment@1:target:0";
    };
```
What is needed is an extension to the "Apply parameters" step that also patches the fix-ups as necessary. It's not that difficult, but it may be a while before I get around to it.

pelwell, 2017-05-15:
You will find an updated libdtovl.so [here](https://drive.google.com/file/d/0B_P-i4u-SLBXODBlbklBZlB5UUU/view?usp=sharing). Download it and copy it into /opt/vc/lib (you will need root permission, and you might want to save the original version in case you want to revert), then try the failing command again.

Provided you don't find any problems they fixed library will be in the next firmware release.

pelwell, 2017-05-16:
Yesterday's rpi-update release includes the new libdtovl.so, so you can run `sudo rpi-update` instead of manually patching your installation.

closed 2017-05-18 by P33M