raspberrypi/linux #2003: mmc0: unexpected command 5 error

zougloub, 2017-05-11:
I'm plugging in a Pi 3 [a BRCM43340 Wifi SDIO module](http://www.inventeksys.com/products-page/wifi-eval-kits/ism43340-l77-evb/) which is working fine on my ThinkPad's SD/SDIO slot.

There are MMC issues, `mmc0: unexpected command 5 error` from `drivers/mmc/host/bcm2835-sdhost.c` (ping @pelwell) while the card is inserted.

pelwell, 2017-05-11:
The SDHOST interface - the one that drives the SD card - is unlikely to work with an SDIO card because it turns off the SD clock between bus transactions.

zougloub, 2017-05-11:
OK, good to know!

The other one (I have an [MMC-HAT](http://mmchat.uk/) on the way), does it do SDIO?

zougloub, 2017-05-11:
The interwebs says that yes, so I'm closing the issue. Thanks @pelwell for the prompt response.

pelwell, 2017-05-11:
I'll give you a brief version of the whole story. The BCM2835 family of processors has two MMC/SD interfaces - `mmc` (also known as the Arasan block)  is suitable for SDIO, `sdhost` isn't. There are 3 GPIO groups where `mmc` can be used (22-27, 34-39 and 48-53), and two for `sdhost` (34-39, 48-53); the SD slot (or EMMC) is on 48-53. We used to use `mmc` for everything (SD, uSD and EMMC) but then we switched to `sdhost` because a) it can be made quicker, and b) it frees up the SDIO-capable `mmc`. Then along comes the Pi3 and uses `mmc` to drive the WiFi chip via SDIO.

The assignment of `mmc`/`sdhost` interfaces to pins on the BCM2835 under Linux is controlled by Device Tree. Device Tree overlays can be used to change the group of pins used by each interface, and even force `mmc` to be used on the SD/uSD slot. This means that my first answer didn't give you the full picture. You could use a DT overlay to make `mmc` drive the uSD slot, which would make it SDIO capable, _but_ then you would lose WiFi. Similarly, the MMC-HAT is capable of SDIO - it's just wiring pins 22-27 to the new uSD slot, which means it must be using `mmc`) but you won't also be able to use WiFi on the Pi3.

zougloub, 2017-05-11:
I see. If I can't get the card to work on i.MX 6, I'll come back to the Pi and try to make a dtbo for my use case. Thanks again,

pelwell, 2017-05-11:
The standard `sdio` overlay (which I've just simplified - https://github.com/raspberrypi/linux/blob/rpi-4.9.y/arch/arm/boot/dts/overlays/sdio-overlay.dts) ought to do what you want, or at least it should be a good starting point.

zougloub, 2017-05-11:
(couldn't get my iMX6 board to underclock whereas the Pi3 is OK doing that with config.txt, so I'm back)

While I don't have the MMC-HAT, I tried the following, with the intent of routing `mmc` to the SD slot: ```
diff --git a/arch/arm/boot/dts/overlays/sdhost-overlay.dts b/arch/arm/boot/dts/overlays/sdhost-overlay.dts
index de3d1b0a5e40..305272b377b5 100644
--- a/arch/arm/boot/dts/overlays/sdhost-overlay.dts
+++ b/arch/arm/boot/dts/overlays/sdhost-overlay.dts
@@ -7 +7 @@
-       compatible = "brcm,bcm2708";
+       compatible = "brcm,bcm2837", "brcm,bcm2836";
diff --git a/arch/arm/boot/dts/overlays/sdio-overlay.dts b/arch/arm/boot/dts/overlays/sdio-overlay.dts
index 215d5e3e8a8c..28ded260878e 100644
--- a/arch/arm/boot/dts/overlays/sdio-overlay.dts
+++ b/arch/arm/boot/dts/overlays/sdio-overlay.dts
@@ -4 +4 @@
-/* Enable SDIO from MMC interface via GPIOs 22-27. Includes sdhost overlay. */
+/* Enable SDIO from MMC interface via GPIOs 48-53. Includes sdhost overlay. */
@@ -7 +7 @@
-       compatible = "brcm,bcm2708";
+       compatible = "brcm,bcm2837", "brcm,bcm2836";
@@ -25 +25 @@
-                               brcm,pins = <22 23 24 25 26 27>;
+                               brcm,pins = <48 49 50 51 52 53>;
```
but it looks like it's not doing anything (checked that `/proc/device-tree/soc/gpio@*/sdio_pins/brcm,pins` is indeed outputting the right values) to make the error go away.

Did I do anything wrong?

zougloub, 2017-05-11:
(reopening the issue, as there should be a way to get SDIO in the SD-card slot)

pelwell, 2017-05-11:
Try `dtoverlay=mmc`.

zougloub, 2017-05-11:
OK, I'll try to make it work and attach my dts.

zougloub, 2017-05-11:
Cool, `mmc-overlay.dtbs` is working as it is.

Note that `brcm.pull` is `brcm,pull` in `sdio-overlay.dtbs`, I'm not sure how `brcm,bcm2708` is still making it work, and IMHO adding comments to these overlays could be useful for those who aren't brute-forcing their way with web search engines.

closed 2017-05-11 by zougloub

pelwell, 2017-05-11:
Thanks for spotting the typo - I've fixed that now. Using `brcm,bcm2708` as the compatible string makes sense because that is the name of the family of SoCs, but in reality nothing is checking the string value.

Are you aware of the `dtoverlay` command in Raspbian? `dtoverlay -a` lists all known overlays, and `dtoverlay -h <overlay>` gives usage information for a named overlay. It reads /boot/overlays/README, so you may prefer to search it manually.