raspberrypi/linux #2004: kernel-1.20170427-1 : wireless networking stalls, hangs, is unusable

edsantiago, 2017-05-11:
Wireless networking on raspberrypi-kernel-1.20170427-1 is unusable. Network comes up, then hangs, sometimes coming back on its own, sometimes not until my 10-minute watchdog script resets the network.

There is nothing useful in the systemd journal, even with `rtl8192cu` loaded with `debug=2`. (At higher levels, it's too much volume for me to filter). I regret not having noticed or logged the output of `ifconfig` to check for packet loss, but I don't think this is #1954 because of the timing. I realize this is not much to go on.

Reverting (`dpkg -i  /var/cache/apt/archives/raspberrypi-kernel_1.20170405-1_armhf.deb`) and rebooting brings both my systems back to normal: no hangs in at least twelve hours. (With 20170427, the network would not last two hours without a hang. Often it would hang in less than ten minutes).

Two units, an original Pi and a Pi 2. Both with Edimax  EW-7811Un USB dongles; driver module is `8192cu` in the older kernel, renamed to `rtl8192cu` in newer, but I didn't have `modprobe.d` settings anyway so it's probably not a module parameter issue (at least not on my configuration).

I don't have easy access to wired Ethernet on either system so I was unable to test if this was a networking issue or wireless or 8192-specific.

I have reverted both systems, and am reluctant to "upgrade" back to the broken kernel, but will do so if there's useful information I can poke around for.

popcornmix, 2017-05-11:
Read https://github.com/raspberrypi/linux/issues/1866
I suspect you are missing the blacklist (default setting in raspbian) which should give you the older driver.

edsantiago, 2017-05-11:
@popcornmix thank you, I'll give that a try. Obviously I didn't search well enough.

edsantiago, 2017-05-11:
Followup: confirmed. Blacklisting `rtl8192cu` resolves the problem.

For benefit of anyone finding this thread in the future: the actual solution on raspbian is to install the **`raspberrypi-sys-mods`** package, which is what provides that blacklist (and more).

closed 2017-05-11 by edsantiago