raspberrypi/linux #2006: Need build-in CONFIG_SQUASHFS_LZ4=y and other.

monetny, 2017-05-12:
Because default kernel use other systems, good idea add several parametrs:
```
$ make bcm2709_defconfig
$ echo "CONFIG_ZSWAP=y
CONFIG_ZPOOL=y
CONFIG_Z3FOLD=y
CONFIG_OVERLAY_FS=y
CONFIG_SQUASHFS=y
CONFIG_SQUASHFS_LZ4=y
CONFIG_CRYPTO_LZ4HC=m" >> .config
$ make oldconfig
```
or
```
$ diff .config.new .config.orig
560,561c560,561
< CONFIG_ZSWAP=y
< CONFIG_ZPOOL=y
---
> # CONFIG_ZSWAP is not set
> # CONFIG_ZPOOL is not set
563d562
< CONFIG_Z3FOLD=y
5220c5219
< CONFIG_OVERLAY_FS=y
---
> CONFIG_OVERLAY_FS=m
5300c5299
< CONFIG_SQUASHFS=y
---
> CONFIG_SQUASHFS=m
5308c5307
< CONFIG_SQUASHFS_LZ4=y
---
> # CONFIG_SQUASHFS_LZ4 is not set
5769c5768
< CONFIG_CRYPTO_LZO=y
---
> CONFIG_CRYPTO_LZO=m
5772c5771
< CONFIG_CRYPTO_LZ4HC=m
---
> # CONFIG_CRYPTO_LZ4HC is not set
5841d5839
< CONFIG_LZ4HC_COMPRESS=m
```

1. build-in overlayfs need for boot RO rootfs w/o initramfs
2. build-in squashfs need for boot w/o initramfs
3. lz4hc is very fast for squashfs (squashfs_lz4 include lz4hc)
4. zswap is very good idea for raspberry ;)

JamesH65, 2017-05-18:
Do you have any figures for impact on kernel size and performance? Given a limited resources device like the Pi we like to ensure that kernel changes for smaller use cases do not have a detrimental impact.

Ferroin, 2017-05-18:
1 may impose a small overhead on file operations (one of it's dependencies adds some extra code in many places in the VFS layer), but it isn't much impact even on systems with very high-speed storage (read as high-end NVMe SSD's), so it likely won't be noticeable on a Pi, 2 and 3 should have no impact on performance, 4 will have near zero performance impact when not turned on (and it defaults to disabled), although it will result in pathologically bad memory usage if enabled when using zram for swap.

As far as space usage, I can't' give much advice, except that if SquashFS is built-in, you should make sure all the supported compression algorithms are enabled, otherwise you will get complaints from peopel wanting to use other algorithms.

monetny, 2017-05-18:
lz4 is very-very small, and very-very fast
it designed for small and slow systems
lz4hc is good for one time compress many time decompress

minimum change is 
CONFIG_ZSWAP=m
CONFIG_ZPOOL=m
CONFIG_Z3FOLD=m
CONFIG_SQUASHFS_LZ4=y
CONFIG_CRYPTO_LZ4HC=m

monetny, 2017-05-18:
my raspberry pi with [SanDiskUltra](url=http://www.pidramble.com/wiki/benchmarks/microsd-cards)
[f2fs+zswap+swapfs->loopdev->sparsefile+tmpfs](http://4pda.ru/forum/index.php?showtopic=308317&view=findpost&p=59145629)
is faster then big comp with hdd now

monetny, 2017-05-18:
for example, then I compile kernel for upgrade
(/tmp is tmpfs,1Gb + zram 1Gb or zswap 2Gb)
`cd /tmp
wget -q -P ~ https://github.com/raspberrypi/linux/archive/rpi-4.9.y.zip
unzip ~/rpi-4.9.y.zip
mksquashfs linux-rpi-4.9.y ~/rpi-4.9.y.squash -comp lz4  -Xhc -noI
rm -R linux-rpi-4.9.y
mkdir -p linux/low linux/up linux/work
sudo mount ~/rpi-4.9.y.squash linux/low
sudo mount -t overlay overlay -olowerdir=linux/low,upperdir=linux/up,workdir=linux/work linux
cd linux
make bcm2709_defconfig
echo "CONFIG_ZSWAP=y
CONFIG_ZPOOL=y
CONFIG_Z3FOLD=y
CONFIG_OVERLAY_FS=y
CONFIG_SQUASHFS=y
CONFIG_SQUASHFS_LZ4=y
CONFIG_CRYPTO_LZ4HC=m" >> .config
make oldconfig
make -j4 zImage modules dtbs
make modules_install
...`
it's very fast, because did not write to SD then compile,
read and decompress faster than only read

monetny, 2017-05-19:
time and size mksquashfs raspbian-full:
1. lzo 17:11 1564Mb
2. lz4 2:54 1912Mb
3. lz4hc 6:05 1668Mb
lzo is very long time for me

JamesH65, 2017-05-19:
@popcornmix @pelwell You guys have any thoughts on this?

IComplainInComments, 2017-05-21:
I actually do the same thing, and I can also confirm faster build times along with better boot performance on the kernels as well, on average it shaves off 4 seconds from the default kernels.

Along with that, system performance is also a lot better. Using raspian stretch repos, LxQT, LightDM, watchdog, UFW, SSH, VC4 KMS, and VNC, I use 190MB on boot.

popcornmix, 2017-05-22:
We are unlikely to move from modules to builtins as it affects the memory used and kernel load time for every pi user (when the vast majority don't use the features).
It is also a slippery slope for every user requesting their favourite modules are built in.

We'd need very compelling evidence for what the benefits are for the majority of pi users.
The output of ``free`` after booting with stock kernel and with modifies kernel would be useful information to provide.

monetny, 2017-05-22:
You are right. Not everything modules is need built in.
I'll think about evidence. what is to need and not to need.

monetny, 2017-05-23:
I am shure need add `CONFIG_SQUASHFS_LZ4=y
CONFIG_CRYPTO_LZ4HC=m`
because it is modules
1) SQUASHFS_LZ4 faster then SQUASHFS_LZO
2) Then add CONFIG_CRYPTO_LZ4HC=m
automaticaly add CONFIG_LZ4_COMPRESS=m
and then for ZRAM and zswap can select this compressor

monetny, 2017-08-18:
Raspbian Stretch was released
squashfs-tools support lz4 now
I think need add CONFIG_SQUASHFS_LZ4=y for kernel support lz4 too
PS good idea for add CONFIG_CRYPTO_LZ4HC=m too (for zram)

jayslovak, 2017-09-15:
Hi all,

I have just hit this issue while preparing my custom OS image, I was hoping to switch to LZ4 (LZ4HC makes more sense) to speed up my embedded rpis. I already use LZ4 for ZRAM compression, so this would complement it nicely.

```
CONFIG_SQUASHFS_LZ4=y
CONFIG_CRYPTO_LZ4HC=m
```

jayslovak, 2017-10-05:
Hi @popcornmix 

As requested here is a comparison between vanilla git kernel, and after the above enabled options:

```
Linux raspberrypi-vanilla 4.9.52+ #1 Thu Oct 5 19:23:11 CEST 2017 armv6l GNU/Linux

              total        used        free      shared  buff/cache   available
Mem:         444492       21204      333952        5856       89336      369816
Swap:        102396           0      102396
```

```
Linux raspberrypi-lz4 4.9.52+ #1 Thu Oct 5 19:44:51 CEST 2017 armv6l GNU/Linux

              total        used        free      shared  buff/cache   available
Mem:         444492       21168      336796        3076       86528      372648
Swap:        102396           0      102396
```

So I would say there is no memory impact.

Please enable:
```
CONFIG_SQUASHFS_LZ4=y
CONFIG_CRYPTO_LZ4HC=m
```

Thanks!

jayslovak, 2017-11-08:
Hi @pelwell,

Is there anything I can do to help get this changed?

Thanks!

monetny, 2017-11-09:
@jayslovak, Thx
English is not easy for me.

JamesH65, 2017-12-04:
@popcornmix @pelwell Any further thoughts?

howardqiao, 2017-12-11:
Volumio2 is a good and popular system that uses SQUASHFS.

jayslovak, 2018-02-09:
Hi @popcornmix @pelwell 

Could we get some feedback from you guys?

Thanks

JamesH65, 2018-06-27:
@pelwell @popcornmix Any thoughts? Or have we already done this?

JamesH65, 2019-07-30:
These are not currently set on the 4.19 kernel, so presumably we are not inclined to set them.

This issue will be closed within 30 days unless further interactions are posted. If you wish this issue to remain open, please add a comment. A closed issue may be reopened if requested.

closed 2019-09-18 by JamesH65