raspberrypi/linux #2008: 4.9 | /proc/cpuinfo always shows Hardware = BCM2835

Fourdee, 2017-05-13:
As title, in ```4.9.27```, ``` /proc/cpuinfo``` always shows ```BCM2835``` for ```Hardware``` info. Confirmed with:
- RPi 3
- RPi Zero
- RPi Zero W

In ```4.4.50```
- RPi 2/3 = BCM2709 (incorrect?https://www.raspberrypi.org/documentation/hardware/raspberrypi/bcm2836/README.md https://www.raspberrypi.org/documentation/hardware/raspberrypi/bcm2837/README.md)
- RPi 1/zero = BCM2708 (incorrect? https://www.raspberrypi.org/documentation/hardware/raspberrypi/bcm2835/README.md)

This is an issue for us, as we use the that information to determine which RPi is actually running, and, how we optimize for those devices (minus cpu core detection ```$(nproc --all)``` for software installs such as php-fpm thread count).
https://github.com/Fourdee/DietPi/issues/936#issuecomment-301055299

Any chance we could see a possible fix for correct ````Hardware``` information, reported in ```/proc/cpuinfo```?

Revision codes do seem to be correct in ```4.9.24```: https://github.com/Fourdee/DietPi/blob/master/dietpi/misc/rpi_boardinfo

clivem, 2017-05-13:
#1725

pelwell, 2017-05-13:
This is an unintended but unavoidable consequence of using less downstream, Pi-specific code in newer kernels. I suggest you use /proc/device-tree/model (human readable) or extract the information you need from the Revision field of /proc/cpuinfo. There is a bit in the revision number indicating that the it's in the new format: if it's set then the CPU type is in another group of bits, otherwise it must be a 2708.

Fourdee, 2017-05-14:
@pelwell 

Thanks for the info 👍 

I've made the switch to using revision codes for board detection:
https://github.com/Fourdee/DietPi/blob/testing/dietpi/dietpi-obtain_hw_model#L125-L364

I'll mark as closed, [dupe] of https://github.com/raspberrypi/firmware/issues/705

closed 2017-05-14 by Fourdee

karu2003, 2017-07-22:
popcornmix
Thank you!!!!