raspberrypi/linux #2013: Ecryptfs does not recognize files and directories after last update

ufa, 2017-05-15:
Hi,
I have a USB thumb encrypted with ecryptfs and after the update I cannot access my files anymore. 
I can mount it normally, but when I access it, it throws me:
ufa@picam:~ $ cd /bau
ufa@picam:/bau $ ls
ls: cannot access programas: No such file or directory
ls: cannot access imagens: No such file or directory
ls: cannot access jogos: No such file or directory
ls: cannot access tmp: No such file or directory
ls: cannot access backup: No such file or directory
d?????????  ? ?    ?       ?            ? .Trash-1001
d?????????  ? ?    ?       ?            ? tmp
d?????????  ? ?    ?       ?            ? programas
d?????????  ? ?    ?       ?            ? jogos

etc

Is there a way to fix this or to downgrade the kernel?

ufa, 2017-05-15:
$uname -a
Linux picam 4.9.24+ #993 Wed Apr 26 17:56:54 BST 2017 armv6l GNU/Linux

ufa, 2017-05-15:
Just an update.
Plugging the USB thumb in another not updated pi, I could mount and see the files and folders, so the issue is with the new kernel (almost certain with the 4.9).

As it appears to me as a critical bug (one cannot rollback the kernel version), it should have high priority on the bug fix list.

Thanks for all the hard work making Raspbian/Pixel a great OS.

popcornmix, 2017-05-16:
```
sudo rpi-update 52241088c1da59a359110d39c1875cda56496764
```
will get you the last kernel before we moved to 4.9. Does that work?

popcornmix, 2017-05-16:
Capturing dmesg output on 4.9 after mounting the drive and trying to access it would be useful.

ufa, 2017-05-16:
I will try to downgrade the kernel and attatch the DMESG output too, but I did saw the messages and there wasn't nothing wrong there. 
Just to clarify, it does mount the partition without visible errors, but the file names and directories are messed up,so I cannot see or navigate in it.

popcornmix, 2017-05-16:
I'd be interested what ``lsmod`` shows after mounting on both 4.4 and 4.9 kernels.
It may be worth trying ``sudo modprobe ecryptfs`` on 4.9 before mounting.

I did find a report that 4.7 works and 4.8 does not for one user:
https://forum.manjaro.org/t/kernel-4-8-breaks-encryption/11971

popcornmix, 2017-05-16:
Another similar report:
https://bugs.launchpad.net/ubuntu/+source/linux/+bug/1682410

popcornmix, 2017-05-16:
Possibly this:

> The ability to look up plaintext and encrypted filenames, under the same eCryptfs mount point, has been disabled with the following commit in 4.7

https://unix.stackexchange.com/questions/314255/ecryptfs-mounting-failure-from-kernel-4-7-onwards/314266

ufa, 2017-05-16:
Yeah, that could well be the case here. I have plain and encrypted file names.
So, I will try to rename the ones unencrypted to see if it gets fixed.

ufa, 2017-05-16:
Ok, I'm moving the files to another partition, and then move them to the all filename encrypted partition again. It will take a while, but I will post results as soon as it finishes

ufa, 2017-05-18:
Just a follow up, I confirm that the problem is related with kernel changes on 4.7 and above. After making all files filename-encrypted, it worked again. 
So you cannot mix files with plain and encrypted filenames in the same partition. 
Maybe a warning should be added in the update from kernels below 4.7.

closed 2017-05-18 by ufa