raspberrypi/linux #2014: Enable edt-ft5x06 touch screen driver module in default config

blacktigersoftware, 2017-05-16:
The Adafruit PiTFT 2.8 Capacitive touchscreen uses this driver for touch responses.  I can confirm it works with 4.9..28+ on the Pi Zero W.  This driver replaces the ft6236 (which also works :).  And probably a bunch more.

cayman54, 2017-05-16:
Hi,
I was using the driver ft6236, but with the last upgrade it has disapeared and the pitft 2.8 capacitive do not work anymore.
How can I use this driver instead ?
Thanks in advance

popcornmix, 2017-05-16:
Can you confirm that config option ``CONFIG_TOUCHSCREEN_EDT_FT5X06`` is was is wanted?

cayman54, 2017-05-16:
I don't know, I just have my touchscreen not working anymore...

blacktigersoftware, 2017-05-16:
@popcornmix I'm mostly sure.  Don't have my config file handy at the moment.  I thought about a pull request, if that is what you prefer, I can have it ready tomorrow morning.

@cayman54 Yes, the ft6236 disappeared around version 4.5.  Took a bunch of time sleuthing where it went.  After you get the module, if it doesn't work, you may have to look at the dts overlay files.  For me, it was the 'pitft28-capacitive-overlay.dts'.  I don't know why the driver doesn't have it's own overlay.

pelwell, 2017-05-16:
> I don't know why the driver doesn't have it's own overlay.

Although Raspberry Pi didn't invent the concept or syntax of overlays, all the overlays in the raspberrypi/linux tree are put there by us or contributed by our users.

cayman54, 2017-05-16:
The screen works properly, but the touch doesn't.
Where can I specify to use this driver ?
For the moment I have the dtoverlay options in /boot/config.txt and the /etc/udev/rules.d/95-ft6236.rules file that doesn't have any use anymore?

blacktigersoftware, 2017-05-16:
@cayman54 --- 
you basically have a few options:
1 - download kernel source, enable driver, build and deploy it
2 - wait until the kernel changes ( I'll be testing Pi Zero, Pi Zero W/ Pi 2+ and Pi 3 this evening), and raspberrypi.org provides a new image
3 - go back to an older kernel (probably 4.5 or earlier, where everything works).

This is why I opened the issue.  If you PM me with details about what display you are using, I might be able to help you out.  Additionally, you can go to Adafruit forum where I helped someone else:
https://forums.adafruit.com/viewtopic.php?f=50&t=116774

closed 2017-05-16 by blacktigersoftware

popcornmix, 2017-05-17:
For anyone reading this the option was added here:
https://github.com/raspberrypi/linux/commit/cf149324f685d9f6716cc6d374bbb7d9ce3453e3

popcornmix, 2017-05-21:
Should be in latest rpi-update kernel.

blacktigersoftware, 2017-05-21:
rpi-update confirmed working