raspberrypi/linux #2015: kernel memory corruption __kmalloc, kfree

fourtrax, 2017-05-16:
Hi, I trying to find out the reason for a kernel panic on my CM3.
The following kernel panic occurs randomly after a few hours of runtime, most of the times in kmalloc:

> [14886.929616] Unable to handle kernel paging request at virtual address 6f616943
> [24793.191864 14859.478431] [14886.936863] pgd = a740c000
> [24793.193363 0.001499] [14886.939573] [6f616943] *pgd=2adba835, *pte=00000000, *ppte=00000000
> [24793.196859 0.003496] [14886.945863] Internal error: Oops: 17 [#1] SMP ARM
> [24793.199464 0.002605] [14886.950569] Modules linked in: evdev joydev edt_ft5x06(O) arc4 ecb md4 md5 hmac nls_utf8 cifs s1d13781fb(O) spi_config(O) bnep cfg80211 snd_soc_wm8731 spidev snd_soc_bcm2835_i2s snd_soc_rpi_proto snd_soc_core snd_compress snd_pcm_dmaengine snd_pcm i2c_bcm2835 snd_timer snd spi_bcm2835 bcm2835_gpiomem btusb btrtl btintel btbcm bluetooth rfkill uio_pdrv_genirq uio fixed bcm2835_v4l2 videobuf2_vmalloc videobuf2_memops videobuf2_v4l2 videobuf2_core v4l2_common videodev media i2c_dev ipv6
> [24793.244143 0.044679] [14886.993678] CPU: 3 PID: 1056 Comm: queue1:src Tainted: GF          O    4.9.20-v7+ #985
> [24793.246447 0.002304] [14887.001679] Hardware name: BCM2835
> [24793.247362 0.000915] [14887.005079] task: aacb49c0 task.stack: a7556000
> [24793.250032 0.002670] [14887.009617] PC is at __kmalloc+0x9c/0x274
> [24793.264085 0.014053] [14887.013626] LR is at __kmalloc+0x30/0x274
> [24793.265160 0.001075] [14887.017636] pc : [<8025dbac>]    lr : [<8025db40>]    psr: 20000093
> [24793.266802 0.001642] [14887.017636] sp : a75578b8  ip : a75578b8  fp : a75578fc
> [24793.269737 0.002935] [14887.029112] r10: ad801e40  r9 : 8058ffbc  r8 : 02088020
> [24793.284257 0.014520] [14887.034335] r7 : 0000002c  r6 : acc23418  r5 : ad801e40  r4 : 6f616943
> [24793.285919 0.001662] [14887.040860] r3 : 00000000  r2 : 80b81420  r1 : 2e45a000  r0 : a75578b8
> [24793.289286 0.003367] [14887.047387] Flags: nzCv  IRQs off  FIQs on  Mode SVC_32  ISA ARM  Segment user
> [24793.304254 0.014968] [14887.054608] Control: 10c5383d  Table: 2740c06a  DAC: 00000055
> [24793.305650 0.001396] [14887.060352] Process queue1:src (pid: 1056, stack limit = 0xa7556210)
> [24793.309260 0.003610] [14887.066704] Stack: (0xa75578b8 to 0xa7558000)
> [24793.310320 0.001060] [14887.071059] 78a0:                                                       ad88dd00 8025db40
> [24793.325299 0.014979] [14887.079241] 78c0: 00000003 002f8d29 a755791c a75578d8 8025db40 ad138000 9cc99040 acc23418
> [24793.329348 0.004049] [14887.087421] 78e0: 9cc99040 ad13c400 acc233f8 9cc5d680 a755790c a7557900 8058ffbc 8025db1c
> [24793.344493 0.015145] [14887.095601] 7900: a7557924 a7557910 8058a5b4 8058ffa4 ad138000 00000001 a7557954 a7557928
> [24793.346367 0.001874] [14887.103781] 7920: 80583a14 8058a594 a755797c 80556128 a7557954 00000000 00000000 9cc99040
> [24793.363720 0.017353] [14887.111961] 7940: ad138000 ad13c400 a75579ac a7557958 80585780 805839c8 000005ae 00000000
> [24793.366893 0.003173] [14887.120140] 7960: 00000000 00000001 00000000 80114234 00000000 acc23418 00000000 60000093
> [24793.369763 0.002870] [14887.128321] 7980: 80c02d00 9cc5d680 ad13c400 9cc5d688 00000000 02080020 00000003 40000013
> [24793.385084 0.015321] [14887.136504] 79a0: a7557a64 a75579b0 80558080 805855c8 aacb4de4 80b7e940 80c0442c 00000000
> [24793.387910 0.002826] [14887.144684] 79c0: a7557a44 a75579d0 80151348 8014f7ac ad803000 80c03180 00164256 00000003
> [24793.403928 0.016018] [14887.152866] 79e0: 2224f3b0 00000d8a a7557a04 a75579f8 8014d294 8015a88c a7557a74 a7557a08
> [24793.406766 0.002838] [14887.161047] 7a00: 8015a88c 80717bb4 a7557a44 80b7e940 acf13880 ad943b00 aefd8940 80713434
> [24793.410003 0.003237] [14887.169228] 7a20: a7557a74 805045e0 ad27688c a7556028 8013e11c 60000093 00000005 9cc5d680
> [24793.425370 0.015367] [14887.177408] 7a40: 02080020 00000002 00000000 ad204c00 00000003 40000013 a7557aac a7557a68
> [24793.428138 0.002768] [14887.185589] 7a60: 80559be4 80557fc4 a7557a94 a7557a78 805045f0 807180c4 ad276800 00000001
> [24793.444196 0.016058] [14887.193769] 7a80: ad135800 ace5dd80 9cc5d680 ad135800 ad135dec ad135d00 00000000 40000013
> [24793.446904 0.002708] [14887.201949] 7aa0: a7557ae4 a7557ab0 8054c4c4 805598f0 a7557b04 00000000 aacb49c0 ace5dd80
> [24793.450320 0.003416] [14887.210129] 7ac0: ad1a1800 ad135800 ace5dd80 000005a2 00000000 ad1a1800 a7557b3c a7557ae8
> [24793.465556 0.015236] [14887.218309] 7ae0: 80629ca4 8054c328 a7557b2c 00000000 ad1a1800 a7557b44 80c75ae4 80c02d00
> [24793.469073 0.003517] [14887.226489] 7b00: 80c049b8 80c0312c a7556028 00000000 80629ba8 acd2ce00 ad1a1800 acd2ce90
> [24793.484496 0.015423] [14887.234669] 7b20: ace5dd80 ad135800 00000003 ad1a1800 a7557b6c a7557b40 80651178 80629be4
> [24793.487113 0.002617] [14887.242849] 7b40: 80461490 00000010 1b42ebbd 00000001 1b42e61b 00000001 00000000 acd2ce00
> [24793.490585 0.003472] [14887.251029] 7b60: a7557bc4 a7557b70 8062a460 80651094 acd2ce90 00000001 000005a2 00000000
> [24793.505918 0.015333] [14887.259209] 7b80: 00000000 acd2ce90 a7557ba4 fffffff4 00000000 8046125c a7557be4 a75452a0
> [24793.509236 0.003318] [14887.267389] 7ba0: a7545200 00000002 0000000e 80c64d80 000005dc ace5dd80 a7557bd4 a7557bc8
> [24793.524640 0.015404] [14887.275569] 7bc0: 8062a6ac 80629f2c a7557c0c a7557bd8 8066b934 8062a69c a7557c04 1bbca8c0
> [24793.527094 0.002454] [14887.283749] 7be0: 8066d25c ace5dd80 80c64d80 aaf958c0 00000000 80c64d80 000005dc 1bbca8c0
> [24793.543565 0.016471] [14887.291929] 7c00: a7557c44 a7557c10 8066d568 8066b7b4 8066e018 000d34a8 00000000 ace5dd80
> [24793.545965 0.002400] [14887.300109] 7c20: 80c64d80 000d34a8 00000000 aaf958c0 ad135800 1bbca8c0 a7557c94 a7557c48
> [24793.549470 0.003505] [14887.308289] 7c40: 8066e0dc 8066d43c 8066d70c 8066b740 00000000 80c64d80 8066c6bc 00000000
> [24793.565071 0.015601] [14887.316469] 7c60: aaf958c0 a7557d50 00000580 8066d1bc ace5dd80 aaf958c0 80c64d80 ad1d6834
> [24793.567415 0.002344] [14887.324650] 7c80: 00000000 a7557d50 a7557cb4 a7557c98 8066d7f4 8066e024 000017ff ace5dd80
> [24793.583739 0.016324] [14887.332832] 7ca0: 80c64d80 00000580 a7557ccc a7557cb8 8066eab8 8066d7b8 ace5dd80 aaf958c0
> [24793.586053 0.002314] [14887.341014] 7cc0: a7557cf4 a7557cd0 80698360 8066eaa0 aaf958c0 ace5dd80 a7557f4c 00000578
> [24793.589783 0.003730] [14887.349195] 7ce0: a7557d50 00000000 a7557dec a7557cf8 80699d50 80698148 00000580 00000008
> [24793.605035 0.015252] [14887.357375] 7d00: a7557d34 a7557d30 00004040 0000001c 80c64d80 00000000 00008e13 8066d1bc
> [24793.606467 0.001432] [14887.365556] 7d20: 00000000 00000000 aaf95b10 00000000 00000000 a7557e1c 1f330000 1bbca8c0
> [24793.623753 0.017286] [14887.373736] 7d40: 00000000 00000000 ffff0000 00000d8a 00000002 00000001 00000000 00110000
> [24793.624979 0.001226] [14887.381916] 7d60: 00000000 a7557e48 00000000 00000000 19bca8c0 1bbca8c0 62858e13 a7557d88
> [24793.629553 0.004574] [14887.390097] 7d80: 80271450 804614f0 a7557f54 00000000 a7557e24 00000001 00000003 00004000
> [24793.644567 0.015014] [14887.398277] 7da0: a7557de4 a7557db0 804695e8 802713e4 a7557e48 a7557dbc 00000010 a7557e48
> [24793.649061 0.004494] [14887.406457] 7dc0: a7557de4 aaf958c0 00000000 ac1a6fc0 00000000 00000000 00004000 a7557e28
> [24793.663947 0.014886] [14887.414637] 7de0: a7557e0c a7557df0 806a3f90 80699ae4 a7557e24 a7557e24 a7557f4c 00000000
> [24793.665180 0.001233] [14887.422817] 7e00: a7557e1c a7557e10 806098a8 806a3f00 a7557f34 a7557e20 80609f88 80609890
> [24793.669587 0.004407] [14887.430997] 7e20: acf13880 00000000 00000000 00000000 00000000 ffffffff 60000013 ffffffff
> [24793.684703 0.015116] [14887.439177] 7e40: a7557eac 80713434 6f9676a0 0000000c 6f963218 00000002 6f84b07f 0000056a
> [24793.689180 0.004477] [14887.447358] 7e60: 024080c0 ad801cc0 a7557e88 ad60d518 ad60d5f0 9b0aa490 a7557eb4 a7557e88
> [24793.704066 0.014886] [14887.455537] 7e80: 8028cdb4 80286af8 8e130002 1bbca8c0 00000000 00000000 00000000 9b0aa490
[... 157 lines cut ...]
> [  119.944722] Padding adb520f8: 5a 5a 5a 5a 5a 5a 5a 5a                          ZZZZZZZZ
> [  119.944728] CPU: 1 PID: 764 Comm: slabinfo Tainted: G    B           4.9.28-v7+ #998
> [  119.944731] Hardware name: BCM2835
> [  119.944742] [<8010fb3c>] (unwind_backtrace) from [<8010c058>] (show_stack+0x20/0x24)
> [  119.944751] [<8010c058>] (show_stack) from [<804552c0>] (dump_stack+0xd4/0x118)
> [  119.944759] [<804552c0>] (dump_stack) from [<8025b338>] (print_trailer+0x184/0x268)
[... cut ...]
> [  119.944841] [<80271334>] (SyS_write) from [<801080c0>] (ret_fast_syscall+0x0/0x1c)
> [  119.944845] FIX kmalloc-512: Restoring 0xadb51e08-0xadb51e0c=0xcc

I am not an expert on kernel debugging but this seems to be related to the kernel panics.

I have tried a fresh installation of 2017-04-10-raspbian-jessie-lite without any modifications and slabinfo shows the same memory corruption, so it doesn't seem to be related to anything I am running on the CM3.

Any input on how to debug this is highly appreciated.
thanks in advance

pelwell, 2017-05-16:
Memory corruptions like this are usually the result of an undervoltage situation.

1. How are you powering your CM3?
2. Is it in the CMIO board? Which version?
3. Try running `vcgencmd get_throttled` and verifying that the result is 0.

N.B. The original CMIO board (V1.x) isn't capable of adequately powering a CM3 under heavy load - you need a V3 board.

fourtrax, 2017-05-16:
thanks for you help. You are probably right that it is a hardware problem. What surprises me a little though is that the overwritten bytes reported by slabinfo are always identical.

1. I tested powering it via a 5V/2A wall and a 5A bench supply set to 5V, that made no difference
2. It is a CMIO v1.2 board . The kernel panics happened under load but the reported corruption by slabinfo is there right after start up without any load.
3. vcgencmd get_throttled returns 0 for some time and then returns throttled=0x20002 
   The 0x20002 seems to indicate a thermal problem.

I will test with a V3 board and better cooling.

popcornmix, 2017-05-16:
I don't think CM3 will report under-voltage through ``get_throttled``.
You'll probably need to use a voltmeter/scope when the processor is under stress to confirm if there is a voltage drop.

JamesH65, 2017-09-13:
@fourtrax Can this be closed?

closed 2018-06-27 by JamesH65