raspberrypi/linux #2016: HiFiBerry Amp+ 48kHz not working in 4.9

hifiberry, 2017-05-16:
I've been informed that 48kHz on the HiFiBerry Amp+ isn't working with the latest kernel anymore. So I started debugging this.
The issue seems to be related to the 4.9 kernel. The latest 4.4 kernel is still working.

The easiest way to reproduce the issue is using sox:
```
sudo apt-get install sox
play -r 44100 -n synth sine 2000
play -r 48000 -n synth sine 2000
```

Working with 4.4:
```
sudo rpi-update 52241088c1da59a359110d39c1875cda56496764
```

Not working with 4.9 (this is the first 4.9 release from rpi-update)
```
sudo rpi-update 13711648ecf6f2e5aed380baaed00d47f56bcc3f
```

I had a short look on the I2S signal lines and there is no visible difference between 4.4 and 4.9. As 44.1kHz is working fine, I also expect that the I2C communication is working fine (did not yet check this with a logic analyser)

AFAIK, the driver itself hasn't changed between 4.4 and 4.9 
Any idea what has changed in 4.9 ?

popcornmix, 2017-05-16:
Is this just on the Amp or also Dac/Digi?
I've not seen any other complaints.
@HiassofT have you seen any issues?

HiassofT, 2017-05-16:
I'm currently not aware of any issues.

There were a few changes between 4.4 and 4.9 that could be
worth looking into: I2C switched to the upstream driver and
I2S switched to using the clock framework.

The latter change now leads to a "jumping" I2S clock when RPi
is clock master and the rate is from the 48kHz family.

eg on 48kHz with 64fs the clock is derived from 19.2MHz with
a fractional divisor of 6.25 - IOW every 4th cycle will be
about 364.6ns long, the other 3 312.5ns.

This could be problematic, eg when the codec runs a PLL off that
signal.

On kernel 4.4 the 19.2MHz OSC clock was only used in integer
divisor mode. All other clocks frequencies that couldn't be exactly
generated by that were derived from the 500MHz PLLD, with fractional
divisors allowed - which leads to a much cleaner clock signal
(still some jitter, but no "jumping"  as with the 19.2MHz clock).

@hifiberry could you do a test on kernel 4.9 with the hacky patch below  - it's a quick hack to run all clocks off PLLD and disable OSC, I2S clock for 48kHz should be the same as with kernel 4.4
```
--- a/drivers/clk/bcm/clk-bcm2835.c
+++ b/drivers/clk/bcm/clk-bcm2835.c
@@ -1564,7 +1564,7 @@ static const char *const bcm2835_clock_per_parents[] = {

 static const char *const bcm2835_pcm_per_parents[] = {
        "-",
-       "xosc",
+       "-",
        "-",
        "-",
        "-",
```

BTW: of course, the issue could be caused by something else, these were just the first things that came to my mind.

hifiberry, 2017-05-16:
Thanks for this. I will test this. This clock change could be the cause.
 
What was the specific reason to move to the other clock source in 4.9? Having a bit clock with a lot more jitter seems not to be a good thing.

hifiberry, 2017-05-16:
Ok, I can confirm that the clock change fixes the problem. A clock that has a jitter of almost 15% really is a problem for the PLL in the TAS5713. While PLLs of other chips might be a bit more robust, I don't think it's a good idea to have that much jitter on the signal if there are better clocks available. 

@HiassofT How can we go back to the old behaviour? Can the quick hack above be used or does it need major changes in the clock code?

pelwell, 2017-05-16:
I don't think the above change is any more hacky than the patch that added those lines - it's just a refinement. If there are no adverse effects of removing OSC as a potential source then I'm happy to make that change official, but it would be nicer to allow integer divisors of OSC. @anholt ?

HiassofT, 2017-05-16:
As all downstream drivers here would only use OSC in integer mode at 8kHz/16bit/stereo (both with 4.4 and 4.9 kernels) I'd say restricting PCM clock to only the PLLD parent is indeed a proper fix for the issue.

Using OSC in fractional mode is causing a real bug and the benefit of keeping OSC as a possible parent only for 8kHz/16bit is next to zero.

So, let's apply this here.

If someone else wants to go down the rabbit hole of implementing a generic PCM clock selection scheme that can be upstreamed and follows the "only use fractional divisors on the highest frequency clock source" rule he/she is welcome, but that someone won't be me :)

anholt, 2017-05-16:
The generic parent selection scheme sounds nice, but I would be happy to see a patch sent upstream that limited PCM to PLLD_PER only until someone wanted to do better.

clivem, 2017-05-16:
> eg on 48kHz with 64fs the clock is derived from 19.2MHz with a fractional divisor of 6.25 - IOW every 4th cycle will be about 364.6ns long, the other 3 312.5ns.

> I don't think it's a good idea to have that much jitter on the signal if there are better clocks available.

> If there are no adverse effects of removing OSC as a potential source then I'm happy to make that change official, but it would be nicer to allow integer divisors of OSC. 

Sometimes I just want to bang my head against the wall out of complete frustration! By the minute, it becomes more and more obvious that every single minute I spent last summer was a complete waste of time!!!!!!!!!!!!!!!

clivem, 2017-05-17:
@msperl Martin, did anything ever get beyond the "talking about it stage", regarding the ability to be able to have audio driver request specific things via the clock framework, eg. MASH1/2/3/4 for fractional dividers, only integer dividers otherwise fallback to PLLD parent?

msperl, 2017-05-17:
it did not go any further - besides that "generic" case in simple-card that could have easily get extended to include methods exported from i2s to handle those kind of things.
On the otherhand the agreement was that if someone is audiophil enough he would buy the cards/hats/... which come with their own clocks and drive the i2s-clock themselves.

Obviously the "policy" could also get added to the bcm2835-i2s driver itself to get configured via configfs as this is not supposed to describe HW, but policy.

clivem, 2017-05-17:
>  besides that "generic" case in simple-card that could have easily get extended to include methods exported from i2s to handle those kind of things.

That's kind of dead in the water, in as much as it was already vetoed upstream.

> Obviously the "policy" could also get added to the bcm2835-i2s driver itself to get configured via configfs as this is not supposed to describe HW, but policy.

I had a "simple" patchset that could be used via overlay, (for MASH selection and prefer int div), much as the enable MMAP overlay could be used with bcm2835-i2s driver up to 4.4.y..... Abandoned post 4.4.y with the switch to clockmgr..... But now that the clocks are not directly accessible from the platform driver, trying to resurrect that, wouldn't fly anyway......

pelwell, 2017-05-17:
I think the real problem is that the clock driver doesn't model jitter. If bcm2835_clock_choose_div_and_prate returned a range of frequencies then the best rate selection code could optionally minimise the difference from the ideal frequency to the worst-case (rather than the average), the choice of scheme being fixed for each clock.

hifiberry, 2017-05-22:
@pelwell Can you integrate the patch from @HiassofT ?

clivem, 2017-05-22:
Oh yeah.... That's flippin' genius! Disable OSC as a clock source all PCM, even when it could be chosen with an integer divider...... (That's a regression over rpi-4.4.y, for any hardware for which OSC with an integer divider would have been chosen.) 

The best part about this, is that I had a Skype conversation with Daniel a year ago, (last week of May/first week June, thereabouts in 2016), where he was made aware of the coming changes, that they had been fully tested with his original 'B' DAC, Digi, DAC+, DAC+ Pro, (as part of the wider testing I did with 20+ boards), but I couldn't test with anything else, like his AMP board, as I didn't have the hardware, and that he should test any hardware of his I didn't have. A year later, "I've been informed that 48kHz on the HiFiBerry Amp+ isn't working with the latest kernel anymore." So much for having done any testing a year ago!

pelwell, 2017-05-22:
I have the code open now with a view to implementing a minimal-deviation search, initially just for "pcm".

HiassofT, 2017-05-22:
@pelwell it'd probably be easiest to change the clock selection for the PCM clock to allow fractional mode only on the highest frequency parent and use integer mode on all others - only one additional loop over the parents and a slight change to the "find best parent" loop is needed:

1. iterate over all all allowed parents, find the one with highest frequency
2. when calculating rates from parents use integer mode on all except the one found in step 1

As this follows the recommendation from the BCM2835 datasheet (p. 105) it should be easy to argue with upstream why it's implemented that way

> The fractional divider operates by periodically dropping source clock pulses, therefore the
output frequency will periodically switch between
> source _ frequency / DIVI   and  source _ frequency / (DIVI + 1)
> Jitter is therefore reduced by increasing the source clock frequency. In applications where
jitter is a concern, the fastest available clock source should be used.

pelwell, 2017-05-22:
I had just decided to add a low_jitter flag to the clock data, set only for PCM initially which scores each parent by:
1. calculating the average rate using the available fraction bits,
2. finding the largest deviation from the average of the nearest integer divisors, and
3. subtracting that deviation from the average and using the result.

In this way the existing logic of choosing the clock closest to the ideal but without exceeding it can still be used, and the fastest clock doesn't have to be treated specially.

pelwell, 2017-05-22:
Can you try [this patch](https://gist.github.com/pelwell/43559fb95cbee6fde5449fb4fd217053)? It chooses PLLD_PER for me when playing a 48KHz clip, but it should still select XOSC it if there is a suitable integer divisor or if the jitter is very small.

HiassofT, 2017-05-22:
@pelwell thanks a lot, your patch is looking good!

I hacked up a quick test to check the common sample- and bclk-rates and the output looks as expected.

If someone wants to play with it, the tree is here:
https://github.com/HiassofT/rpi-linux/tree/rpi-4.9.y-fractional-clk-test

When bcm2835-i2s is loaded it checks the rates and prints the result in kernel log.

pelwell, 2017-05-23:
For the curious, the output from @HiassofT's test (thanks!) is:
```
# Without the clk-bcm2835.c patch
set clk  11025 *  32 =    352800 got    352799 parent: osc (19200000) FRACT
set clk  11025 *  48 =    529200 got    529199 parent: plld_per (500000006) FRACT
set clk  11025 *  50 =    551250 got    551249 parent: plld_per (500000006) FRACT
set clk  11025 *  64 =    705600 got    705599 parent: plld_per (500000006) FRACT
set clk  11025 * 100 =   1102500 got   1102499 parent: plld_per (500000006) FRACT
set clk  22050 *  32 =    705600 got    705599 parent: plld_per (500000006) FRACT
set clk  22050 *  48 =   1058400 got   1058399 parent: plld_per (500000006) FRACT
set clk  22050 *  50 =   1102500 got   1102499 parent: plld_per (500000006) FRACT
set clk  22050 *  64 =   1411200 got   1411199 parent: plld_per (500000006) FRACT
set clk  22050 * 100 =   2205000 got   2204998 parent: plld_per (500000006) FRACT
set clk  44100 *  32 =   1411200 got   1411199 parent: plld_per (500000006) FRACT
set clk  44100 *  48 =   2116800 got   2116798 parent: plld_per (500000006) FRACT
set clk  44100 *  50 =   2205000 got   2204998 parent: plld_per (500000006) FRACT
set clk  44100 *  64 =   2822400 got   2822398 parent: plld_per (500000006) FRACT
set clk  44100 * 100 =   4410000 got   4409991 parent: plld_per (500000006) FRACT
set clk  88200 *  32 =   2822400 got   2822398 parent: plld_per (500000006) FRACT
set clk  88200 *  48 =   4233600 got   4233591 parent: osc (19200000) FRACT
set clk  88200 *  50 =   4410000 got   4409991 parent: plld_per (500000006) FRACT
set clk  88200 *  64 =   5644800 got   5644796 parent: plld_per (500000006) FRACT
set clk  88200 * 100 =   8820000 got   8819982 parent: plld_per (500000006) FRACT
set clk 176400 *  32 =   5644800 got   5644796 parent: plld_per (500000006) FRACT
set clk 176400 *  48 =   8467200 got   8467183 parent: osc (19200000) FRACT
set clk 176400 *  50 =   8820000 got   8819982 parent: plld_per (500000006) FRACT
set clk 176400 *  64 =  11289600 got  11289593 parent: plld_per (500000006) FRACT
set clk 176400 * 100 =  17640000 got  17639965 parent: plld_per (500000006) FRACT
set clk 352800 *  32 =  11289600 got  11289593 parent: plld_per (500000006) FRACT
set clk 352800 *  48 =  16934400 got  16934297 parent: plld_per (500000006) FRACT
set clk 352800 *  50 =  17640000 got  17639965 parent: plld_per (500000006) FRACT
set clk 352800 *  64 =  22579200 got  22579187 parent: plld_per (500000006) FRACT
set clk 352800 * 100 =  35280000 got  35279931 parent: plld_per (500000006) FRACT
set clk   8000 *  32 =    256000 got    256000 parent: osc (19200000) INT
set clk   8000 *  48 =    384000 got    384000 parent: osc (19200000) INT
set clk   8000 *  50 =    400000 got    400000 parent: osc (19200000) INT
set clk   8000 *  64 =    512000 got    512000 parent: osc (19200000) FRACT
set clk   8000 * 100 =    800000 got    800000 parent: osc (19200000) INT
set clk  12000 *  32 =    384000 got    384000 parent: osc (19200000) INT
set clk  12000 *  48 =    576000 got    575999 parent: plld_per (500000006) FRACT
set clk  12000 *  50 =    600000 got    600000 parent: osc (19200000) INT
set clk  12000 *  64 =    768000 got    768000 parent: osc (19200000) INT
set clk  12000 * 100 =   1200000 got   1200000 parent: osc (19200000) INT
set clk  16000 *  32 =    512000 got    512000 parent: osc (19200000) FRACT
set clk  16000 *  48 =    768000 got    768000 parent: osc (19200000) INT
set clk  16000 *  50 =    800000 got    800000 parent: osc (19200000) INT
set clk  16000 *  64 =   1024000 got   1024000 parent: osc (19200000) FRACT
set clk  16000 * 100 =   1600000 got   1600000 parent: osc (19200000) INT
set clk  24000 *  32 =    768000 got    768000 parent: osc (19200000) INT
set clk  24000 *  48 =   1152000 got   1151999 parent: plld_per (500000006) FRACT
set clk  24000 *  50 =   1200000 got   1200000 parent: osc (19200000) INT
set clk  24000 *  64 =   1536000 got   1536000 parent: osc (19200000) FRACT
set clk  24000 * 100 =   2400000 got   2400000 parent: osc (19200000) INT
set clk  48000 *  32 =   1536000 got   1536000 parent: osc (19200000) FRACT
set clk  48000 *  48 =   2304000 got   2303999 parent: plld_per (500000006) FRACT
set clk  48000 *  50 =   2400000 got   2400000 parent: osc (19200000) INT
set clk  48000 *  64 =   3072000 got   3072000 parent: osc (19200000) FRACT
set clk  48000 * 100 =   4800000 got   4800000 parent: osc (19200000) INT
set clk  96000 *  32 =   3072000 got   3072000 parent: osc (19200000) FRACT
set clk  96000 *  48 =   4608000 got   4607994 parent: plld_per (500000006) FRACT
set clk  96000 *  50 =   4800000 got   4800000 parent: osc (19200000) INT
set clk  96000 *  64 =   6144000 got   6144000 parent: osc (19200000) FRACT
set clk  96000 * 100 =   9600000 got   9600000 parent: osc (19200000) INT
set clk 192000 *  32 =   6144000 got   6144000 parent: osc (19200000) FRACT
set clk 192000 *  48 =   9216000 got   9215967 parent: plld_per (500000006) FRACT
set clk 192000 *  50 =   9600000 got   9600000 parent: osc (19200000) INT
set clk 192000 *  64 =  12288000 got  12287975 parent: plld_per (500000006) FRACT
set clk 192000 * 100 =  19200000 got  19199940 parent: plld_per (500000006) FRACT
set clk 384000 *  32 =  12288000 got  12287975 parent: plld_per (500000006) FRACT
set clk 384000 *  48 =  18432000 got  18431852 parent: plld_per (500000006) FRACT
[... 54 lines cut ...]
set clk  24000 *  50 =   1200000 got   1200000 parent: osc (19200000) INT
[... cut ...]
set clk 384000 *  32 =  12288000 got  12287975 parent: plld_per (500000006) FRACT
set clk 384000 *  48 =  18432000 got  18431852 parent: plld_per (500000006) FRACT
set clk 384000 *  50 =  19200000 got  19199940 parent: plld_per (500000006) FRACT
set clk 384000 *  64 =  24576000 got  24575803 parent: plld_per (500000006) FRACT
set clk 384000 * 100 =  38400000 got  38399520 parent: plld_per (500000006) FRACT
```
Although the patch doesn't explicitly preclude the use of fractional divisors with OSC, in practice that is what it does.

pelwell, 2017-05-23:
Any objections before I merge the patch? @clivem?

clivem, 2017-05-23:
@pelwell Sorry Phil, I'm not in a position to do any testing at the moment, (let alone what I was doing last year)...... If it results in the OSC with an integer divider being preferred, when possible, and PLLD for everything else..... That's cool. And results in the same behavior as was had back in the rpi-4.4.y branch, that was the default behavior for many years, so no complaints from me, just a thank you for the doing the work!

pelwell, 2017-05-23:
Thanks, Clive. I've merged to rpi-4.9.y, and I'll start the upstreaming process.

HiassofT, 2017-05-23:
@pelwell thanks a lot!

BTW: could you push the fix to the 4.11 and 4.12 trees as well?

pelwell, 2017-05-23:
I was going to give it a day or two after the next firmware release, but the non-stable trees are easy enough to patch - done. 4.11 and 4.12 also needed the patch to fix #1949.

popcornmix, 2017-05-26:
Should be in latest rpi-update kernel. Please test and close if happy.

hifiberry, 2017-06-02:
Thanks for the great work! This is working and the issue can be closed.

closed 2017-06-02 by pelwell