raspberrypi/linux #2020: USB Audio Issue with Raspberry Pi Zero W

byarmis, 2017-05-17:
When I play music (either `mpg123` on an mp3 file or pink noise via `speaker-test`), there is popping and crackling in the audio output.  If I let `speaker-test` run for a bit (20 seconds or so), the popping and crackling will go away.  The crackling does not go away after playing a song for an even longer amount of time (still crackling after several minutes).

I have tentatively narrowed it down to the USB frequency.  If I run `watch -n0 "cat /proc/asound/card1/stream0"`, I can see that the "Momentary freq" is reported to be 50 Hz higher or lower than it should be (it varies whether it's higher or lower) when there's crackling:

```
AudioQuest AudioQuest DragonFly Black v1.5 at usb-20980000.usb-1, full speed : U
SB Audio

Playback:
  Status: Running
    Interface = 1
    Altset = 1
    Packet Size = 360
    Momentary freq = 48050 Hz (0x30.0ccc)
    Feedback Format = 10.14
  Interface 1
    Altset 1
    Format: S24_3LE
    Channels: 2
    Endpoint: 1 OUT (ASYNC)
    Rates: 44100, 48000, 88200, 96000
```
When the crackling goes away (when running a `speaker-test`), the momentary frequency drops down (or increases) to where it should be:

```
AudioQuest AudioQuest DragonFly Black v1.5 at usb-20980000.usb-1, full speed : U
SB Audio

Playback:
  Status: Running
    Interface = 1
    Altset = 1
    Packet Size = 360
    Momentary freq = 48000 Hz (0x30.0000)
    Feedback Format = 10.14
  Interface 1
    Altset 1
    Format: S24_3LE
    Channels: 2
    Endpoint: 1 OUT (ASYNC)
    Rates: 44100, 48000, 88200, 96000
```
In [Issue #888](https://github.com/raspberrypi/linux/issues/888), I found a suggestion to try adding either `dwc_otg.fiq_fsm_enable=0` or `dwc_otg.fiq_fsm_mask=0x3` to `/boot/cmdline.txt`; neither fixed the issue.
 
## Hardware

* Raspberry Pi Zero W
* USB OTG Adapter
* [Audioquest Dragonfly Black USB DAC](http://www.audioquest.com/dragonfly-series/)

## Outputs
### uname -a

```Linux raspberrypi 4.9.28+ #999 Tue May 16 13:59:16 BST 2017 armv6l GNU/Linux```

### lsusb -v

```
Bus 001 Device 002: ID 21b4:0083  
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 (Defined at Interface level)
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x21b4 
  idProduct          0x0083 
[... 405 lines cut ...]
[    6.911234] usbcore: registered new interface driver brcmfmac
[    7.084123] brcmfmac: Firmware version = wl0: May 27 2016 00:13:38 version 7.45.41.26 (r640327) FWID 01-df77e4a7
[    7.910484] systemd-journald[116]: Received request to flush runtime journal from PID 1
[   11.927566] hid-generic 0003:21B4:0083.0001: hiddev0,hidraw0: USB HID v1.11 Device [AudioQuest AudioQuest DragonFly Black v1.5] on usb-20980000.usb-1/input2
[   12.074092] usbcore: registered new interface driver snd-usb-audio
[   12.661915] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[   12.661951] brcmfmac: power management disabled
[   13.540709] uart-pl011 20201000.serial: no DMA platform data
[   14.193051] Adding 102396k swap on /var/swap.  Priority:-1 extents:5 across:200700k SSFS
[   15.325576] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[   17.999845] Bluetooth: Core ver 2.22
[   18.000040] NET: Registered protocol family 31
[   18.000050] Bluetooth: HCI device and connection manager initialized
[   18.000080] Bluetooth: HCI socket layer initialized
[   18.000101] Bluetooth: L2CAP socket layer initialized
[   18.000162] Bluetooth: SCO socket layer initialized
[   18.009574] Bluetooth: HCI UART driver ver 2.3
[   18.009593] Bluetooth: HCI UART protocol H4 registered
[   18.009599] Bluetooth: HCI UART protocol Three-wire (H5) registered
[   18.016108] Bluetooth: HCI UART protocol Broadcom registered
[   18.378919] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[   18.378931] Bluetooth: BNEP filters: protocol multicast
[   18.378963] Bluetooth: BNEP socket layer initialized
[   56.540057] random: crng init done
```

P33M, 2017-05-18:
The dwc2 driver is being loaded and used in host mode. Do you have an overlay set in config.txt?

byarmis, 2017-05-18:
Yes, I'm sorry, I forgot that I added `dtoverlay=dwc2` to the end of my config.txt.  If I comment that out and reboot, I get no sound coming out at all and `/proc/asound/card1/stream0` doesn't change from how it looks if I'm not running `speaker-test`:

```
AudioQuest AudioQuest DragonFly Black v1.5 at usb-20980000.usb-1, full speed : U
SB Audio

Playback:
  Status: Stop
  Interface 1
    Altset 1
    Format: S24_3LE
    Channels: 2
    Endpoint: 1 OUT (ASYNC)
    Rates: 44100, 48000, 88200, 96000
```

P33M, 2017-05-18:
Ok, no sound coming out with dwc_otg is definitely a bug. The root port should work with a single device in full-speed mode.

I have an earlier version of this device, but I think from the descriptors that it is substantially different to yours. Where is this device commercially available (preferably an EU seller)?

byarmis, 2017-05-18:
Sorry, I should probably include a little more info (realized that I should do that after submitting the comment): so the reason that I added that to config.txt was that I dug it up from [this](https://volumio.org/forum/with-new-rpi-zero-t6050-50.html) forum post.

With that line removed, my `dmesg` output gets a bit more interesting (and I was struggling with the below errors for quite a while):

```
pi@raspberrypi:~ $ dmesg 
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 4.9.28+ (dc4@dc4-XPS13-9333) (gcc version 4.9.3 (crosstool-NG crosstool-ng-1.22.0-88-g8460611) ) #999 Tue May 16 13:59:16 BST 2017
[    0.000000] CPU: ARMv6-compatible processor [410fb767] revision 7 (ARMv7), cr=00c5387d
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT nonaliasing instruction cache
[    0.000000] OF: fdt:Machine model: Raspberry Pi Zero W Rev 1.1
[    0.000000] cma: Reserved 8 MiB at 0x1b400000
[    0.000000] Memory policy: Data cache writeback
[    0.000000] On node 0 totalpages: 114688
[    0.000000] free_area_init_node: node 0, pgdat c0914834, node_mem_map db010000
[    0.000000]   Normal zone: 1008 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 114688 pages, LIFO batch:31
[    0.000000] pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
[    0.000000] pcpu-alloc: [0] 0 
[    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 113680
[    0.000000] Kernel command line: 8250.nr_uarts=0 bcm2708_fb.fbwidth=656 bcm2708_fb.fbheight=416 bcm2708_fb.fbswap=1 smsc95xx.macaddr=B8:27:EB:8E:B1:B6 vc_mem.mem_base=0x1ec00000 vc_mem.mem_size=0x20000000  dwc_otg.lpm_enable=0 console=ttyS0,115200 console=tty1 root=PARTUUID=1ca0687a-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait
[    0.000000] PID hash table entries: 2048 (order: 1, 8192 bytes)
[    0.000000] Dentry cache hash table entries: 65536 (order: 6, 262144 bytes)
[    0.000000] Inode-cache hash table entries: 32768 (order: 5, 131072 bytes)
[    0.000000] Memory: 435940K/458752K available (5943K kernel code, 489K rwdata, 1948K rodata, 404K init, 734K bss, 14620K reserved, 8192K cma-reserved)
[    0.000000] Virtual kernel memory layout:
    vector  : 0xffff0000 - 0xffff1000   (   4 kB)
    fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
    vmalloc : 0xdc800000 - 0xff800000   ( 560 MB)
    lowmem  : 0xc0000000 - 0xdc000000   ( 448 MB)
    modules : 0xbf000000 - 0xc0000000   (  16 MB)
      .text : 0xc0008000 - 0xc05d5efc   (5944 kB)
      .init : 0xc083f000 - 0xc08a4000   ( 404 kB)
      .data : 0xc08a4000 - 0xc091e6e8   ( 490 kB)
       .bss : 0xc091e6e8 - 0xc09d5ee8   ( 734 kB)
[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
[    0.000000] NR_IRQS:16 nr_irqs:16 16
[    0.000028] sched_clock: 32 bits at 1000kHz, resolution 1000ns, wraps every 2147483647500ns
[    0.000071] clocksource: timer: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 1911260446275 ns
[    0.000172] bcm2835: system timer (irq = 27)
[    0.000641] Console: colour dummy device 80x30
[    0.001403] console [tty1] enabled
[    0.001459] Calibrating delay loop... 697.95 BogoMIPS (lpj=3489792)
[    0.060344] pid_max: default: 32768 minimum: 301
[    0.060804] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.060873] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.061972] Disabling cpuset control group subsystem
[    0.062243] CPU: Testing write buffer coherency: ok
[    0.062352] ftrace: allocating 21708 entries in 64 pages
[    0.177592] Setting up static identity map for 0x8200 - 0x8238
[    0.179636] devtmpfs: initialized
[    0.188429] VFP support v0.3: implementor 41 architecture 1 part 20 variant b rev 5
[    0.188914] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.189014] futex hash table entries: 256 (order: -1, 3072 bytes)
[    0.190324] pinctrl core: initialized pinctrl subsystem
[    0.191827] NET: Registered protocol family 16
[    0.194318] DMA: preallocated 1024 KiB pool for atomic coherent allocations
[    0.203943] hw-breakpoint: found 6 breakpoint and 1 watchpoint registers.
[    0.204036] hw-breakpoint: maximum watchpoint size is 4 bytes.
[    0.204177] Serial: AMBA PL011 UART driver
[    0.206979] bcm2835-mbox 2000b880.mailbox: mailbox enabled
[    0.207885] uart-pl011 20201000.serial: could not find pctldev for node /soc/gpio@7e200000/uart0_pins, deferring probe
[    0.254726] bcm2835-dma 20007000.dma: DMA legacy API manager at dc80d000, dmachans=0x1
[    0.257453] SCSI subsystem initialized
[    0.257943] usbcore: registered new interface driver usbfs
[    0.258117] usbcore: registered new interface driver hub
[    0.258317] usbcore: registered new device driver usb
[    0.267660] raspberrypi-firmware soc:firmware: Attached to firmware from 2017-05-15 16:57
[    0.269601] clocksource: Switched to clocksource timer
[... 162 lines cut ...]
[   16.941219] Bluetooth: BNEP socket layer initialized
[   63.557620] INFO:: schedule_periodic: Insufficient periodic bandwidth for periodic transfer.

[   63.557644] ERROR::dwc_otg_hcd_urb_enqueue:526: DWC OTG HCD URB Enqueue failed adding QTD. Error status -1

[   63.557666] usb 1-1: cannot submit urb 0, error -1: unknown error
[   63.558321] INFO:: schedule_periodic: Insufficient periodic bandwidth for periodic transfer.

[... cut ...]
[   63.559042] ERROR::dwc_otg_hcd_urb_enqueue:526: DWC OTG HCD URB Enqueue failed adding QTD. Error status -1

[   63.559061] usb 1-1: cannot submit urb 0, error -1: unknown error
[   63.568345] INFO:: schedule_periodic: Insufficient periodic bandwidth for periodic transfer.

...
```
and the errors keep repeating for quite a bit.  It looks like they start only after I started running the `speaker-test`.

---

If you're asking about the DAC, I just bought mine off of Amazon.  I'm in the US, but it looks like it's also on EU Amazon sites.

P33M, 2017-05-18:
I get the same result with the v1.0 Dragonfly:

```
pi@raspberrypi:~$ speaker-test -c 2 -D default:CARD=DragonFly

speaker-test 1.0.28

Playback device is default:CARD=DragonFly
Stream parameters are 48000Hz, S16_LE, 2 channels
Using 16 octaves of pink noise
ALSA lib pcm_direct.c:999:(snd1_pcm_direct_initialize_slave) unable to install hw params
ALSA lib pcm_dmix.c:1034:(snd_pcm_dmix_open) unable to initialize slave
Playback open error: -32,Broken pipe
pi@raspberrypi:~$ dmesg | tail
[  226.679438] INFO:: schedule_periodic: Insufficient periodic bandwidth for periodic transfer.

[  226.679467] ERROR::dwc_otg_hcd_urb_enqueue:526: DWC OTG HCD URB Enqueue failed adding QTD. Error status -1

[  226.679494] usb 1-1: cannot submit urb 0, error -1: unknown error
[  249.263687] INFO:: schedule_periodic: Insufficient periodic bandwidth for periodic transfer.

[  249.263717] ERROR::dwc_otg_hcd_urb_enqueue:526: DWC OTG HCD URB Enqueue failed adding QTD. Error status -1

[  249.263741] usb 1-1: cannot submit urb 0, error -1: unknown error

```

P33M, 2017-05-18:
In the dwc2 case, is the wlan0 interface being used at all? Is the only system activity the mpg321 test?

byarmis, 2017-05-18:
I'm SSH'd in which is how I've been running all these tests

P33M, 2017-05-18:
Our microframe scheduler is a bit myopic when it comes to the bus being in full-speed mode: https://github.com/raspberrypi/linux/blob/rpi-4.9.y/drivers/usb/host/dwc_otg/dwc_otg_hcd_queue.c#L539

There's no determination other than if the endpoint in question is FS or HS.

As a test, what happens if you boot without the dwc2 overlay (i.e. use dwc_otg) but set `dwc_otg.microframe_schedule=0` in /boot/cmdline.txt?

byarmis, 2017-05-18:
Good news, audio plays and there are no `dmesg` errors!  Bad news, there's still the momentary frequency weirdness going on-- it starts at 48050 Hz and then after a moment or so, drops down to 48000 Hz.  I'm not at home at the moment so I can't confirm that the audio is crackling at 48050 Hz and not at 48000 Hz, but I can check in 5 hours or so.

P33M, 2017-05-18:
Variance in the reported playback frequency rate is not inherently a cause for concern. The way most USB-audio devices work is that they report how fast they are consuming data via a feedback endpoint. This allows the host controller to send variable amounts of data in response to the device's requirements and keep a constant output samplerate - so instantaneous variations of ~0.1% are indicative of either momentary data rate correction or dropped packets.

byarmis, 2017-05-18:
Ah, that makes sense.  I'll report back this evening how the audio sounds.  Thanks for all the help with this, by the way.

byarmis, 2017-05-18:
Alright, just got home and there's still popping / crackling in the audio.  Would it help if I made a recording and posted it to Youtube or something like that?

P33M, 2017-05-18:
If I'm not able to confirm the same on my v1.0 device tomorrow, then it may help to do so. All that the continuing corruption tells me is that there is more than one bug affecting this use case.

byarmis, 2017-05-19:
[Here's](https://www.dropbox.com/s/vap8k3hm5fobuuj/crackles.wav?dl=0) a link to download a .wav file I recorded.  Volume was set at 20%, 3.5mm cable going from the DAC to the front-panel mic input on my desktop

P33M, 2017-05-20:
Do the two failing cases (dwc2 vs dwc_otg.microframe_schedule=0) sound the same? I.e. is the sound corrupted in the same fashion?

byarmis, 2017-05-20:
To my ear (and eye) they appear the same.  [Here](https://www.dropbox.com/s/b7xdr4qcg5nintk/crackles_with_dwc.wav?dl=0) is a link to a recording with `dwc_otg.microframe_schedule=0` removed from `/boot/cmdline.txt` and `dtoverlay=dwc2` added back to `/boot/config.txt`

P33M, 2017-05-22:
I get far less corruption if I test with a Pi 2 with `dwc_otg.speed=1 dwc_otg.microframe_schedule=0`, but I still see obvious gaps in audio data on the scope. As a test, do you have a hub you can plug between the Pi Zero and the device?

byarmis, 2017-05-22:
I'll try both options this evening.  I do have a hub, but not a powered one-- do you think that's alright?

P33M, 2017-05-22:
The DAC is unlikely to require significant power - it should be OK on a bus-powered hub.

P33M, 2017-05-22:
I have a potential source of the issue...

With a Pi0 and just the Dragonfly attached, I see a burst of missing data mostly when speaker-test starts. There's the occasional dropout when speaker-test is running but it's not severe.

If I add `dwc_otg.fiq_enable=0 dwc_otg.fiq_fsm_enable=0` to /boot/cmdline.txt, the playback starts without glitches. I suspect that hcd->frame_number isn't being updated often as the FIQ will mask start-of-frame interrupts. 

Frame number is used extensively to reference isochronous transfers: http://elixir.free-electrons.com/linux/latest/source/sound/usb/pcm.c#L1543

The frame number is critical to asynchronous endpoint operation - the feedback endpoint momentary sample rate is tied to the notional "bus" sample rate. If snd_usb_audio is getting stale numbers, bad things are likely to happen.

Can you test and report back if the two settings I've suggested make any difference?

byarmis, 2017-05-23:
All tests were done with [this](https://www.amazon.com/gp/product/B002MRF7OI/ref=oh_aui_search_detailpage?ie=UTF8&th=1) hub and nothing else plugged in aside from the Dragonfly DAC.  Between each test, I shutdown the Pi (Zero W), unplugged the power cable, and plugged it back in.

| `cmdline.txt` | USB Hub? | Results | Link to file|
| ---------------- | -------------- | -----------| ------------- |
| `dwc_otg.lpm_enable=0 console=serial0,115200 console=tty1 root=PARTUUID=1ca0687a-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait dwc_otg.fiq_enable=0 dwc_otg.fiq_fsm_enable=0 dwc_otg.microframe_schedule=0` | Yes | Static | [Link](https://www.dropbox.com/s/qb4q1cwii3dag66/hub_and_fiq.wav?dl=0) |
| `dwc_otg.lpm_enable=0 console=serial0,115200 console=tty1 root=PARTUUID=1ca0687a-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait dwc_otg.fiq_enable=0 dwc_otg.fiq_fsm_enable=0 dwc_otg.microframe_schedule=0` | No | Static | [Link](https://www.dropbox.com/s/b49gp31kbklr4rc/no_hub_fiq.wav?dl=0) |
| `dwc_otg.lpm_enable=0 console=serial0,115200 console=tty1 root=PARTUUID=1ca0687a-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait dwc_otg.microframe_schedule=0` | Yes | **No static** | [Link |
| `dwc_otg.lpm_enable=0 console=serial0,115200 console=tty1 root=PARTUUID=1ca0687a-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait dwc_otg.microframe_schedule=0` | No | **No Static** | Link |

So... these results are a bit unexpected :-/ .  Suffice it to say, I have no idea what's going on.  I'll do more testing when I return home later this evening.

byarmis, 2017-05-23:
Although, the sound issues sounds different-- the first three have somewhat regular periods of nothing for about 0.01 sec

[image]

But nothing visible on the macro level
[image]

The last one (the control, as it were), has somewhat frequent periods of clipping
[image]

And are certainly visible on the macro level
[image]

byarmis, 2017-05-23:
So, I also repeated the tests with a clip of Vivaldi (and can hear no difference with or without hub)

| `dwc_otg.fiq_enable=0`/ `dwc_otg.fiq_fsm_enable=0` | Hub | Link |
| ----------------------------------------------------------------------- | ----- | ------ |
| Yes | Yes | [Link](https://www.dropbox.com/s/sn8l121v6hm89ic/vivaldi_fiq_hub.wav?dl=0) |
| Yes | No | [Link](https://www.dropbox.com/s/xw5chz76uhb0iuc/vivaldi_fiq_no_hub.wav?dl=0) |
| No | Yes | [Link](https://www.dropbox.com/s/rhai4oqit3614gh/vivaldi_no_fiq_hub.wav?dl=0) |
| No | No | [Link](https://www.dropbox.com/s/1xne7kib8eagyws/vivaldi_no_fiq_no_hub.wav?dl=0) |

P33M, 2017-05-23:
Thanks for the testing. This is enough information to convince me I need to acquire a v1.5 device and compare the two.

P33M, 2017-05-23:
Deviation noticed: with normal commandline and microframe_schedule=0, in the last few milliseconds before the audio stream is terminated, two packets are seen on the bus in a single frame. This is a bug with fiq_fsm's isochronous handling.

P33M, 2017-05-23:
Returning stale hcd->frame_num is a red herring, the driver always reads the frame number directly from the count register on each invocation of get_frame_number.

I have found a definite discontinuity in the data - I had the luck to get a corrupt output within ~10ms of the identifiable start of stream so that saves me from searching through piles of transfers.

The smoking gun is that data becomes misaligned when viewed via USB analyzer. In the previous packet we had this:
```
00 33 00 
00 00 00 
00 8B 08 
00 00 00 
00 BF 10 
00 00 00 
00 AB 18 
00 00 00 
00 29 20 
00 00 00 
00 19 27 
00 00 00 
00 61 2D 
00 00 00 
00 E1 32 
00 00 00 
00 81 37 
00 00 00 
```
Where each line is a separate channel, MSByte on the right to LSByte on the left.

In the packet immediately following this one, we get:
```
00 00 00 
6C 08 00 
00 00 00 
9F 10 00 
00 00 00 
8B 18 00 
00 00 00 
0D 20 00 
00 00 00 
00 27 00 
00 00 00 
49 2D 00 
00 00 00 
CD 32 00 
00 00 00 
73 37 00 
00 00 00 
23 3B 00 
00 00 00 
```
I.e. the data has now spilled into the other channel and shifted by a byte. The only difference is that the last packet sent before the garbled one was of length 294 and the majority of the others is length 288.

P33M, 2017-05-24:
Ah. It's ISO buffer offsets vs core alignment requirements.

The usbmon trace says:

```
USB URB
    [Source: host]
    [Destination: 1.2.1]
    URB id: 0x00000000d8f01100
    URB type: URB_SUBMIT ('S')
    URB transfer type: URB_ISOCHRONOUS (0x00)
    Endpoint: 0x01, Direction: OUT
    Device: 2
    URB bus id: 1
    Device setup request: not relevant ('-')
    Data: present (0)
    URB sec: 1495630005
    URB usec: 72320
    URB status: Operation now in progress (-EINPROGRESS) (-115)
    URB length [bytes]: 1446
    Data length [bytes]: 1526
    [Response in: 66]
    [bInterfaceClass: Unknown (0xffff)]
    ISO error count: 0
    Number of ISO descriptors: 5
    Interval: 1
    Start frame: 0
    Copy of Transfer Flags: 0x00000004
    Number of ISO descriptors: 5
    USB isodesc 0 [Cross-device link (-EXDEV)] (288 bytes)
        Status: Cross-device link (-EXDEV) (-18)
        Offset [bytes]: 0
        Length [bytes]: 288
        ISO Data: 0021c30000000018c20000000043c100000000a3c0000000...
        Padding: 0x00000000
    USB isodesc 1 [Cross-device link (-EXDEV)] (288 bytes)
        Status: Cross-device link (-EXDEV) (-18)
        Offset [bytes]: 288
        Length [bytes]: 288
        ISO Data: 00503100000000943300000000ac35000000009737000000...
        Padding: 0x00000000
    USB isodesc 2 [Cross-device link (-EXDEV)] (288 bytes)
        Status: Cross-device link (-EXDEV) (-18)
        Offset [bytes]: 576
        Length [bytes]: 288
        ISO Data: 002ae100000000fddd00000000edda00000000fcd7000000...
        Padding: 0x00000000
    USB isodesc 3 [Cross-device link (-EXDEV)] (294 bytes)
        Status: Cross-device link (-EXDEV) (-18)
        Offset [bytes]: 864
        Length [bytes]: 294
        ISO Data: 00050800000000a90b00000000440f00000000d112000000...
        Padding: 0x00000000
    USB isodesc 4 [Cross-device link (-EXDEV)] (288 bytes)
        Status: Cross-device link (-EXDEV) (-18)
        Offset [bytes]: 1158
        Length [bytes]: 288
        ISO Data: 00520c00000000af08000000000505000000005701000000...
        Padding: 0x00000000

```
This is with speaker-test -c2, so one channel is silent and the other has 16 bits shifted into a 24 bit sample.

The first submitted data on iso_desc[4] was 
```
00 52 0c 
00 00 00 
00 af 08 
00 00 00
``` 
but the analyzer captured 
```
00 00 00 
52 0C 00 
00 00 00 
AF 08 00 
```

The buffer offset is the issue. 1158 isn't divisible by 4 so the DWC core DMA engine fetches data from the wrong offset. This issue doesn't appear when the root port is in HS mode because we use bounce buffers for all split-transaction iso transfer data.

This bug is in both dwc_otg and dwc2, which explains why changing the driver makes no difference.

lategoodbye, 2017-05-24:
@P33M Any suggestions to fix this issue on dwc2?

P33M, 2017-05-24:
@lategoodbye there's a significant difference in the way dwc handles misaligned start addresses.

In dwc2, only the start address of the URB's buffer is considered when allocating a bounce buffer for a transfer - see dwc2_hc_init_xfer() and dwc2_alloc_dma_aligned_buffer().

In dwc_otg, a more complicated flow results in assign_and_init_hc() - a dword-aligned buffer is considered for each phase of the isochronous transaction.

For dwc2, dwc2_alloc_dma_aligned_buffer() needs to scan iso_descs[n] and allocate if any urb->transfer_dma + offset is not dword-aligned. dwc2_hc_init_xfer() needs to expand the bounce buffer case for isochronous transfers.

lategoodbye, 2017-05-24:
Sounds complicated. Recently there was a discussion how unaligned transfer buffer on dwc2 should be handled.

http://lists.infradead.org/pipermail/linux-rpi-kernel/2017-March/005795.html
http://lists.infradead.org/pipermail/linux-rpi-kernel/2017-March/006078.html

Unfortunately i can't see any update on this issue.

P33M, 2017-05-24:
Regarding the microframe_schedule=1 case, our bandwidth search is bone-headed when it comes to the bus being in full-speed mode. It tries to fit a full-speed transfer's worth of bus usecs into a high-speed slot, which is never going to work e.g. in find_multi_uframe(): `INFO:: ret=-1 qh->usecs=466 hcd->frame_usecs=[100 100 100 100 100 100 30 0]`.

popcornmix, 2017-05-26:
Latest rpi-update contains a fix that may help. Can you test?

P33M, 2017-05-26:
Note that this should be tested without any additional command-line options in cmdline.txt or `dtoverlay=dwc2` in config.txt

byarmis, 2017-05-26:
I owe you a beer :+1: !

Both mp3 files as well as `speaker-test` work without any pops or crackles from what I can tell

P33M, 2017-05-27:
Closing as works with our preferred host-mode driver.

closed 2017-05-27 by P33M

trulio, 2017-05-31:
Hi, 

if we consider the root cause of this issue as the same as the one of the thread 
[https://github.com/raspberrypi/firmware/issues/753](url)
upgrade to 4.9.30 kernel didn't solved the issue.

Regards,

P33M, 2017-05-31:
The issue with adaptive DACs is separate to this one. I will update the other thread when there is more progress.