raspberrypi/linux #2021: device descriptor read error using dwc_otg

YuGiOhJCJ, 2017-05-18:
Hello,

I bought an USB2.0 HUB that is working fine on my laptop and desktop computers but it fails on my Raspberry Pi 3 Model B.

Here is with my laptop computer:
```
$ uname -a
Linux laptop 3.18.20-laptop #1 SMP Tue Aug 18 16:41:13 CEST 2015 i686 AMD A4-1200 APU with Radeon(TM) HD Graphics AuthenticAMD GNU/Linux
$ # I connect the USB hub to the laptop
$ sudo dmesg
[...]
[ 3822.771053] usb 1-1: USB disconnect, device number 5
[ 6723.673880] usb 3-1: new high-speed USB device number 2 using ehci-pci
[ 6723.789141] usb 3-1: New USB device found, idVendor=214b, idProduct=7000
[ 6723.789163] usb 3-1: New USB device strings: Mfr=0, Product=1, SerialNumber=0
[ 6723.789175] usb 3-1: Product: USB2.0 HUB
[ 6723.791039] hub 3-1:1.0: USB hub found
[ 6723.791623] hub 3-1:1.0: 4 ports detected
$ # I connect the touchpad to the USB hub
$ sudo dmesg
[...]
[ 6928.085730] usb 3-1.4: new low-speed USB device number 3 using ehci-pci
[ 6928.166181] usb 3-1.4: New USB device found, idVendor=1ddd, idProduct=2819
[ 6928.166203] usb 3-1.4: New USB device strings: Mfr=0, Product=0, SerialNumber=0
[ 6938.286449] input: HID 1ddd:2819 as /devices/pci0000:00/0000:00:12.2/usb3/3-1/3-1.4/3-1.4:1.0/0003:1DDD:2819.0002/input/input19
[ 6938.287059] hid-generic 0003:1DDD:2819.0002: input,hidraw0: USB HID v1.10 Keyboard [HID 1ddd:2819] on usb-0000:00:12.2-1.4/input0
[ 6938.290193] input: HID 1ddd:2819 as /devices/pci0000:00/0000:00:12.2/usb3/3-1/3-1.4/3-1.4:1.1/0003:1DDD:2819.0003/input/input20
[ 6938.290882] hid-generic 0003:1DDD:2819.0003: input,hiddev0,hidraw1: USB HID v1.10 Mouse [HID 1ddd:2819] on usb-0000:00:12.2-1.4/input1
```

Here is with my desktop computer:
```
$ uname -a
Linux desktop 4.9.24-desktop #1 SMP Mon Apr 24 10:54:11 CEST 2017 x86_64 Intel(R) Celeron(R) CPU G3920 @ 2.90GHz GenuineIntel GNU/Linux
$ # I connect the USB hub to the desktop
$ sudo dmesg
[...]
[ 7945.533264] usb 1-9: new high-speed USB device number 7 using xhci_hcd
[ 7945.701356] usb 1-9: New USB device found, idVendor=214b, idProduct=7000
[ 7945.701361] usb 1-9: New USB device strings: Mfr=0, Product=1, SerialNumber=0
[ 7945.701363] usb 1-9: Product: USB2.0 HUB
[ 7945.701959] hub 1-9:1.0: USB hub found
[ 7945.702025] hub 1-9:1.0: 4 ports detected
$ # I connect the touchpad to the USB hub
$ sudo dmesg
[...]
[ 7982.251297] usb 1-9.4: new low-speed USB device number 8 using xhci_hcd
[ 7982.331526] usb 1-9.4: New USB device found, idVendor=1ddd, idProduct=2819
[ 7982.331531] usb 1-9.4: New USB device strings: Mfr=0, Product=0, SerialNumber=0
[ 7992.648017] input: HID 1ddd:2819 as /devices/pci0000:00/0000:00:14.0/usb1/1-9/1-9.4/1-9.4:1.0/0003:1DDD:2819.0004/input/input20
[ 7992.700709] hid-generic 0003:1DDD:2819.0004: input,hidraw3: USB HID v1.10 Keyboard [HID 1ddd:2819] on usb-0000:00:14.0-9.4/input0
[ 7992.709588] input: HID 1ddd:2819 as /devices/pci0000:00/0000:00:14.0/usb1/1-9/1-9.4/1-9.4:1.1/0003:1DDD:2819.0005/input/input21
[ 7992.761675] hid-generic 0003:1DDD:2819.0005: input,hiddev0,hidraw4: USB HID v1.10 Mouse [HID 1ddd:2819] on usb-0000:00:14.0-9.4/input1
```

Here is with my Raspberry Pi 3 Model B:
```
$ uname -a
Linux raspberrypi 4.9.24-raspberrypi #1 SMP Tue Apr 25 16:41:02 BST 2017 armv7l BCM2835 GNU/Linux
$ # I connect the USB hub to the Raspberrypi 3 Model B
$ sudo dmesg
[...]
[175337.788021] usb 1-1.5: new high-speed USB device number 52 using dwc_otg
[175337.918640] usb 1-1.5: New USB device found, idVendor=214b, idProduct=7000
[175337.918658] usb 1-1.5: New USB device strings: Mfr=0, Product=1, SerialNumber=0
[175337.918666] usb 1-1.5: Product: USB2.0 HUB
[175337.919623] hub 1-1.5:1.0: USB hub found
[175337.919790] hub 1-1.5:1.0: 4 ports detected
$ # I connect the touchpad to the USB hub
$ sudo dmesg
[...]
[175377.667953] usb 1-1.5.4: new low-speed USB device number 53 using dwc_otg
[175377.767927] usb 1-1.5.4: device descriptor read/64, error -32
[175377.987934] usb 1-1.5.4: device descriptor read/64, error -32
[175378.217863] usb 1-1.5.4: new low-speed USB device number 54 using dwc_otg
[175378.317946] usb 1-1.5.4: device descriptor read/64, error -32
[175378.537948] usb 1-1.5.4: device descriptor read/64, error -32
[175378.757948] usb 1-1.5.4: new low-speed USB device number 55 using dwc_otg
[175379.208044] usb 1-1.5.4: device not accepting address 55, error -32
[175379.307964] usb 1-1.5.4: new low-speed USB device number 56 using dwc_otg
[175379.747898] usb 1-1.5.4: device not accepting address 56, error -32
[175379.748085] usb 1-1.5-port4: unable to enumerate USB device
```

You can remark that on the laptop computer the ehci-pci USB controller is used, on the desktop computer the xhci_hcd USB controller is used and on the Raspberry Pi 3 Model B the dwc_otg USB controller is used.
Is it a bug in the Raspberry Pi Linux kernel?

Thank you.
Best regards.

Raspberry Pi Linux kernel used: 6cfb40f8ec37e92ced573755f9111c5ac23c20b3

P33M, 2017-05-18:
Does the touchpad work when plugged directly into the Pi's USB ports?

YuGiOhJCJ, 2017-05-18:
Yes, the touchpad works when plugged directly into the Pi's USB ports:
```
$ # I connect the touchpad to the Raspberrypi 3 Model B
$ sudo dmesg
[...]
[228278.328115] usb 1-1.4: new low-speed USB device number 58 using dwc_otg
[228278.464441] usb 1-1.4: New USB device found, idVendor=1ddd, idProduct=2819
[228278.464460] usb 1-1.4: New USB device strings: Mfr=0, Product=0, SerialNumber=0
[228288.573881] input: HID 1ddd:2819 as /devices/platform/soc/3f980000.usb/usb1/1-1/1-1.4/1-1.4:1.0/0003:1DDD:2819.001F/input/input30
[228288.639213] hid-generic 0003:1DDD:2819.001F: input,hidraw0: USB HID v1.10 Keyboard [HID 1ddd:2819] on usb-3f980000.usb-1.4/input0
[228288.692442] input: HID 1ddd:2819 as /devices/platform/soc/3f980000.usb/usb1/1-1/1-1.4/1-1.4:1.1/0003:1DDD:2819.0020/input/input31
[228288.769076] hid-generic 0003:1DDD:2819.0020: input,hiddev0,hidraw1: USB HID v1.10 Mouse [HID 1ddd:2819] on usb-3f980000.usb-1.4/input1
```

P33M, 2017-05-26:
I'd say the answer to this issue is "use a hub from a reputable brand". There are any number of commodity devices out there that are known to work with the Pi, closing as a workaround is available.

closed 2017-05-26 by P33M