raspberrypi/linux #2022: Building kernel fails because of issues in wireless drivers code

Radoslaw-K, 2017-05-18:
Building kernel with default configuration as shown below:

```
KERNEL=kernel7
make ARCH=arm CROSS_COMPILE=arm-linux-gnueabihf- bcm2709_defconfig
make ARCH=arm CROSS_COMPILE=arm-linux-gnueabihf- zImage modules dtbs
```

Fails with the following errors:

```
drivers/net/wireless/realtek/rtl8192cu/os_dep/linux/ioctl_cfg80211.c: In function ‘rtw_cfg80211_indicate_connect’:
drivers/net/wireless/realtek/rtl8192cu/os_dep/linux/ioctl_cfg80211.c:649:12: error: initialization from incompatible pointer type [-Werror=incompatible-pointer-types]
     .bss = cur_network->network.MacAddress,
            ^
drivers/net/wireless/realtek/rtl8192cu/os_dep/linux/ioctl_cfg80211.c:649:12: note: (near initialization for ‘roam_info.bss’)
  CC [M]  drivers/staging/fbtft/fb_ssd1306.o
drivers/net/wireless/realtek/rtl8192cu/os_dep/linux/ioctl_cfg80211.c: At top level:
drivers/net/wireless/realtek/rtl8192cu/os_dep/linux/ioctl_cfg80211.c:5234:22: error: initialization from incompatible pointer type [-Werror=incompatible-pointer-types]
  .add_virtual_intf = cfg80211_rtw_add_virtual_intf,
                      ^
drivers/net/wireless/realtek/rtl8192cu/os_dep/linux/ioctl_cfg80211.c:5234:22: note: (near initialization for ‘rtw_cfg80211_ops.add_virtual_intf’)
```

popcornmix, 2017-05-18:
Are you on 4.12 branch? Can you try now - there was an update earlier today.

pelwell, 2017-05-18:
1. What is the head commit of your kernel tree (`git rev-parse HEAD`)?
2. Which toolchain are you using (`arm-linux-gnueabi-gcc -v`)?

closed 2017-05-18 by popcornmix