raspberrypi/linux #2024: Invalid CRC in USB packets on Pi Zero

yoctopuce, 2017-05-19:
We have found a bug in the USB stack of the Raspberry PI image.

Using an USB analyser we have found that sometime the Raspberry Pi Zero send USB packets with Invalid CRC. The strange thing is that this issue affects only Raspberry Pi Zero with Raspbian image after March 2016. The same image work perfectly on a Raspberry Pi 2 or 3.

We have publish on GitHub a complete description of the problem and a program that exhibits this issue:
https://github.com/yoctopuce-examples/raspberry_pi_zero_bug

JamesH65, 2017-05-19:
@P33M One for you I think?

@yoctopuce Can you confirm March **2016** as the date it appears to start going wrong.

P33M, 2017-05-19:
There's two ways to interpret the statement. Has the Pi Zero ever been reliable on any version of firmware? If so, what was the last good firmware revision/distro?

lategoodbye, 2017-05-20:
What is the behavior on a Raspberry Pi 1?
Is it possible to reproduce the issue without a Yoctopuce device?

P33M, 2017-05-22:
To narrow down if a specific firmware version broke functionality, can you test firmware revisions retrieved via rpi-update:

https://github.com/Hexxeh/rpi-firmware/commits/master

Running `sudo rpi-update <hash>` will get the firmware revision corresponding to the hash in the repo above.  A rudimentary pseudo-bisect should get to the first broken revision relatively quickly.

yoctopuce, 2017-05-23:
Hi all,

Sorry to have took so long to answers, but we are quite busy these days.

@JamesH65 Official image of 18 March 2016 is the last one that *IS WORKGING". Wee log bellow:
         
```
pi@raspberrypi:~/raspberry_pi_zero_bug $ sudo ./bug_pizero
this is a simple program that exhibit a bug in the USB stack
of the Raspberry PI Zero and any Yoctopuce device
Use libUSB v1.0.19.10903
List all Yoctopuce devices present on this host:
 - USB dev: RELAYHI1-85D28 (24E0:D:0)
1 Yoctopuce devices are present
\ Test device RELAYHI1-85D28 (24E0:D)
+ Send USB pkt no 0
- rd_callback for RELAYHI1-85D28 : response 0 is valid (len=64)
+ Send USB pkt no 1
- rd_callback for RELAYHI1-85D28 : response 1 is valid (len=64)
+ Send USB pkt no 2
- rd_callback for RELAYHI1-85D28 : response 2 is valid (len=64)
+ Send USB pkt no 3
- rd_callback for RELAYHI1-85D28 : response 3 is valid (len=64)
+ Send USB pkt no 4
- rd_callback for RELAYHI1-85D28 : response 4 is valid (len=64)
+ Send USB pkt no 5
- rd_callback for RELAYHI1-85D28 : response 5 is valid (len=64)
+ Send USB pkt no 6
- rd_callback for RELAYHI1-85D28 : response 6 is valid (len=64)
+ Send USB pkt no 7
- rd_callback for RELAYHI1-85D28 : response 7 is valid (len=64)
+ Send USB pkt no 8
- rd_callback for RELAYHI1-85D28 : response 8 is valid (len=64)
+ Send USB pkt no 9
- rd_callback for RELAYHI1-85D28 : response 9 is valid (len=64)
+ Send USB pkt no 10
- rd_callback for RELAYHI1-85D28 : response 10 is valid (len=64)
+ Send USB pkt no 11
- rd_callback for RELAYHI1-85D28 : response 11 is valid (len=64)
+ Send USB pkt no 12
- rd_callback for RELAYHI1-85D28 : response 12 is valid (len=64)
+ Send USB pkt no 13
- rd_callback for RELAYHI1-85D28 : response 13 is valid (len=64)
+ Send USB pkt no 14
- rd_callback for RELAYHI1-85D28 : response 14 is valid (len=64)
= result: 15 pkt sent vs 15 pkt received
Cancel and free all pending transactions
pi@raspberrypi:~/raspberry_pi_zero_bug $ uname -a
Linux raspberrypi 4.1.19+ #858 Tue Mar 15 15:52:03 GMT 2016 armv6l GNU/Linux
```

@lategoodbye: I will double check but the Raspberry Pi 1 does not have the same issue. 
I've not been able to reproduce this issue without Yoctopuce devices. But if you need to do
some test we can send you an device or we can setup an remote access to our test bench.

I'm still running the remaining tests and should have all the result this evening.

lategoodbye, 2017-05-23:
@yoctopuce Thanks for for the offer, but i think it's to soon to send me test devices. Btw i'm not member of the Pi Foundation.

According to the output the good release was on March 15th not 18th. Could you please narrow down to the very last good firmware version as P33M suggested and also name the first bad firmware version?
Can you estimate in how many cases the CRC is invalid in case of a Raspberry Pi Zero?
Are you able to provide some invalid CRC values and their corresponding valid ones?
Did you try to use the dwc2 USB driver instead of dwc_otg?

yoctopuce, 2017-05-26:
Hi,

I've done some regression testing and I've found that the CRC issue has been injected the 11 April 2016 with
the following changelist: Hexxeh/rpi-firmware@c05b9e6fba9cc2ce9e47abf41b8d413b2a69d952

Before this the USB stack is working correctly.

P33M, 2017-05-26:
It looks like this is the first 4.4 kernel we shipped. This will have a lot of kernel code churn which is going to be difficult to sort through

@yoctopuce can you try testing BRANCH=next firmware revisions? These are preview versions that we build prior to a kernel branch change or new firmware feature.

https://github.com/Hexxeh/rpi-firmware/commits/next?after=f4cb84915874f32d88e2bfe793620d7afe25c755+69

You'll probably want to start around the time that master was broken - f4e7f4731f25ae289d2f983157024bc83e88cf02

and work backwards until it starts working again. If it's the move to 4.4 that caused the breakage then I would expect the last good commit to be cbb6fb920da75da94c1a86f8694f2b3137113011

 `sudo BRANCH=next rpi-update <hash>`

phishstang65, 2017-06-03:
Any progress on this? I update to the latest kernel with rpi-update on my Zero W and it still does not recognize these devices.

lategoodbye, 2017-06-06:
@phishstang65 Any help that narrow down the issue regarding to the affecting commit is welcome.

yoctopuce, 2017-06-07:
HI,

I've done some regression testing and the problem has been introduce in the changelist : Hexxeh/rpi-firmware@2ef7dc7175adcd761c95b619fcba435ad9fa3b70 (kernel: BCM270X_DT: Remove explicit claiming of UART pins).

I've look to the changes made in this revision and I've found the culprit : the hack for Split Interrupt transactions has been enabled by default (raspberrypi/linux@7956536a3d78ba0ef8ec990651b315664ed70f90).

If I disable this hack in the /boot/cmdline.txt file everything work as expected.
```
dwc_otg.fiq_fsm_mask=7 dwc_otg.lpm_enable=0 console=serial0,115200 console=tty1 root=/dev/mmcblk0p2 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait
```

P33M, 2017-06-07:
Interesting. In the failing case, what is the output of `sudo lsusb -v`? That code should have no effect if the Zero is plugged directly into a FS device.

yoctopuce, 2017-06-07:
Here is the output of  `sudo lsusb -v` on the Pi Zero with the firmware Hexxeh/rpi-firmware@2ef7dc7175adcd761c95b619fcba435ad9fa3b70 with no modification 
to the commandline.txt.

```
Bus 001 Device 004: ID 05ac:1402 Apple, Inc. Ethernet Adapter [A1277]
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass          255 Vendor Specific Class
  bDeviceSubClass       255 Vendor Specific Subclass
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x05ac Apple, Inc.
  idProduct          0x1402 Ethernet Adapter [A1277]
  bcdDevice            0.01
  iManufacturer           1 Apple Inc.      
  iProduct                2 Apple USB Ethernet Adapter
  iSerial                 3 12E1F6
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           39
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          4 0
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              250mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           3
      bInterfaceClass       255 Vendor Specific Class
      bInterfaceSubClass    255 Vendor Specific Subclass
      bInterfaceProtocol      0 
      iInterface              7 0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval              11
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x03  EP 3 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
[... 201 lines cut ...]
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0004  1x 4 bytes
        bInterval              12
Hub Descriptor:
  bLength               9
  bDescriptorType      41
  nNbrPorts             1
  wHubCharacteristic 0x0008
    Ganged power switching
    Per-port overcurrent protection
    TT think time 8 FS bits
  bPwrOn2PwrGood        1 * 2 milli seconds
  bHubContrCurrent      0 milli Ampere
  DeviceRemovable    0x00
  PortPwrCtrlMask    0xff
 Hub Port Status:
   Port 1: 0000.0503 highspeed power enable connect
Device Status:     0x0001
  Self Powered

  Self Powered
```

I've also done some test with a Raspberry PI Zero W. If the Yoctopuce device is plugged directly on the PI everything is working but if the Yoctopuce device is plugged on a USB hub I've got IO error. So the problem seems to be the handling of USB Split Transactions.

P33M, 2017-06-07:
Does your USB analyzer work at high-speed? Can you capture a trace of a (failed) transfer when snooping the bus between the Pi and the hub?

Note that if you have the Ethernet device active, your trace is likely to be full of network packet spam. Running the test via the UART would be recommended.

yoctopuce, 2017-06-08:
Hi,

I've manage to capture the High Speed trafic between the Raspberry Pi Zero and the USB Hub (http://www.uugear.com/product/zero4u/). USB traces are available here: https://www.yoctopuce.com/tmp/bug_split_transaction.zip and  can be viewed with the Visual USB Software from Ellisys (http://www.ellisys.com/products/usbex200/download.php).

The capture file `usb_hub_trafic_mask_f.ufo` is the one that failed. The cmdline.txt is untouched and
the hack for Split Interrupt transactions is enabled.

The capture file `usb_hub_trafic_mask_7.ufo` is the one that work. I've added `dwc_otg.fiq_fsm_mask=7` to the cmdline.txt and the hack for Split Interrupt transactions is disabled.

Both capture have been done with the same hardware and the same USB devices with the latest offical image:
``
Linux raspberrypi 4.4.50+ #970 Mon Feb 20 19:12:50 GMT 2017 armv6l GNU/Linux
``

P33M, 2017-06-12:
OK. That's a sufficiently confusing result. The first SSPLIT OUT is accepted (ACK) and the host subsequently tries to queue another SSPLIT IN. This is rejected (NAK) meaning the TT has no space available. We then try to complete the first OUT transfer with a CSPLIT and get... no response. 

That's a common hub chipset, so I would expect that if there were a general issue with the hack + this hub chipset then everyone would be screaming at once. There has to be some sort of interaction between the behaviour of the device + hub (or how much data we're sending).

I think to investigate this further, I'll need one of these devices and the hub in question.

P33M, 2017-06-12:
@yoctopuce do you have an e-mail address where we can continue the discussion?

yoctopuce, 2017-06-12:
You can contact support@yoctopuce.com

P33M, 2017-06-19:
I can confirm that the bug only happens with a specific type of hub. From our selection of (cheap) hub devices, the Terminus Tech single-TT hub (FE1.1s) appears to be the only chip that exhibits this behaviour.

There are two main failure modes. The first is as reported previously, where an IN start-split is sent, followed by an OUT start-split. The second is NAK'd (no TT buffer space) but subsequent transfers to the TT all result in transfer errors. On the downstream port, the bus ends up suspended (no traffic for 3ms) and the device is inaccessible until it's reconnected.

The second is where the hub reports a STALL response for the IN transfer, but this does not result in a suspend state on the downstream port.

There are two main failure modes. The first is as reported previously, where an IN start-split is sent, followed by an OUT start-split. The second is NAK'd (no TT buffer space) but subsequent transfers to the TT all result in transfer errors. On the downstream port, the bus ends up suspended (no traffic for 3ms) and the device is inaccessible until it's reconnected.

The second is where the hub reports a STALL response for the IN transfer, but this does not result in a suspend state on the downstream port.

P33M, 2017-06-19:
[image]

This is the first failure mode - note the timestamps matching between the two windows.

P33M, 2017-06-19:
[image]

In the second failure mode that results in a STALL handshake, the only thing that is different is the ordering of complete-splits. The second SSPLIT is issued before the first CSPLIT, which results in a stall. The second CSPLIT results in an ACK.

I believe the hub only has a single TT buffer for non-periodic transfers. Or alternatively, it cannot differentiate between inbound/outbound transfers for non-periodic endpoints with the same address and endpoint number.

@yoctopuce an easy test would be if one of the endpoints could be configured with a different number. Is it possible to get a testing firmware that I can load that changes one of the endpoints to be e.g. EP 2 IN?

yoctopuce, 2017-06-19:
@P33M yes. I will send you that firmware by email.

P33M, 2017-06-19:
With the custom firmware, this is confirmed to be an issue with how the hub is responding to multiple transfers to the same device/endpoint. Both the in direction and the out direction are considered to be the same pipe as far as the hub is concerned.

The response when the endpoints are different follows a more sane flow - the CSPLIT for the OUT transfer receives a NYET response instead of everything ending up broken.

[image]

P33M, 2017-06-20:
Thinking a bit wider, this hub limitation could be tripped over if any combination of non-periodic endpoint types (or Interrupt munged into Control types) share endpoint number on a device.

I have a nominal fix for the issue, will make a PR shortly.

popcornmix, 2017-06-22:
Latest rpi-update kernel contains @P33M's latest fix.

yoctopuce, 2017-06-23:
I confirm that the latest firmware fix this issue. Will this fix be includes in the next Raspbian update ?

popcornmix, 2017-06-23:
Yes, the next time apt-get firmware is bumped it will include this.
The next raspbian sdcard image after this will include this.

(Note there was an updated raspbian image two days ago, so it may be a little while until the next bump)

closed 2017-06-28 by P33M