raspberrypi/linux #2025: NETDEV_WATCHDOG_eth0_smsc95xx_transmit_queue_0_timed_out

andersthomson, 2017-05-19:
While compiling latest 4.9.y, I got this one on the console. 

Didn't get the first half, but it seems a mmc error (bad card?) got something stuck and ev the networking stuff choked on it. Uninformed guess suggests that their sharing on USB is part of this faith sharing.

[NETDEV_WATCHDOG_eth0_smsc95xx_transmit_queue_0_timed_out.txt](https://github.com/raspberrypi/linux/files/1013840/NETDEV_WATCHDOG_eth0_smsc95xx_transmit_queue_0_timed_out.txt)

JamesH65, 2017-05-22:
Again, without the setup of the device and what you are doing, its going to be difficult to fathom this out.

andersthomson, 2017-05-23:
What do you need in terms of setup? The box has two powered usb hubs, two tv cards, two HDs, two serial readers. It's running 4 containers (with tvheadend, nginx, openhab, deamons, and a spare work one). No idea what triggers it. It can happen in the middle of a compile (off the HD), with no TV traffic, so net stuff should be low.

JamesH65, 2017-05-23:
I suspect this is related to #2027. Something has stopped the ethernet driver from emptying its queue in a timely manner. 

Note that the SD interface is not USB, so it going wrong as well sort of implies a somewhat wider failure than just the USB.

If this was a one off I suspect it it going to be very difficult to reproduce and hence figure out.

andersthomson, 2017-05-28:
It probably is. As it happens, I keep /usr/src/linux on a USB attached HD. I've never seen this one happening again.

JamesH65, 2017-07-14:
I'm inclined to close this since its was a one off. If it does turn out to happen again, please reopen and report.

closed 2017-07-14 by JamesH65