raspberrypi/linux #2026: kernel 4.9 fault

danyconnect, 2017-05-19:
```
May 17 01:52:15 raspberrypi kernel: [ 3922.764656] Unable to handle kernel paging request at virtual address 591b7f9e
May 17 01:52:15 raspberrypi kernel: [ 3922.768109] pgd = 9de48000
May 17 01:52:15 raspberrypi kernel: [ 3922.769814] [591b7f9e] *pgd=00000000
May 17 01:52:15 raspberrypi kernel: [ 3922.771480] Internal error: Oops: 5 [#1] SMP ARM
May 17 01:52:15 raspberrypi kernel: [ 3922.773139] Modules linked in: cfg80211 rfkill ftdi_sio usbserial spidev i2c_bcm2835 spi_bcm2835 bcm2835_gpiomem uio_pdrv_genirq uio fixed rtc_ds1307 hwmon i2c_dev i2c_bcm2708 snd_bcm2835 snd_pcm snd_timer snd fuse ipv6
May 17 01:52:15 raspberrypi kernel: [ 3922.778811] CPU: 1 PID: 25260 Comm: smbd Not tainted 4.9.28-v7+ #999
May 17 01:52:15 raspberrypi kernel: [ 3922.780734] Hardware name: BCM2835
May 17 01:52:15 raspberrypi kernel: [ 3922.782651] task: 9dd4ac40 task.stack: a067e000
May 17 01:52:15 raspberrypi kernel: [ 3922.784603] PC is at locks_remove_posix+0x34/0x150
May 17 01:52:15 raspberrypi kernel: [ 3922.786528] LR is at filp_close+0x68/0x8c
May 17 01:52:15 raspberrypi kernel: [ 3922.788456] pc : [<802c5b80>]    lr : [<8026d760>]    psr: 20000013
May 17 01:52:15 raspberrypi kernel: [ 3922.788456] sp : a067fea8  ip : a067ff58  fp : a067ff54
May 17 01:52:15 raspberrypi kernel: [ 3922.792432] r10: 00000000  r9 : a067e000  r8 : 80108244
May 17 01:52:15 raspberrypi kernel: [ 3922.794431] r7 : 00000006  r6 : bc8a0800  r5 : bb1cda18  r4 : 00000000
May 17 01:52:15 raspberrypi kernel: [ 3922.796442] r3 : 591b7f92  r2 : b9f4c640  r1 : bc8a0800  r0 : bc27cd80
May 17 01:52:15 raspberrypi kernel: [ 3922.798441] Flags: nzCv  IRQs on  FIQs on  Mode SVC_32  ISA ARM  Segment user
May 17 01:52:15 raspberrypi kernel: [ 3922.800504] Control: 10c5387d  Table: 1de4806a  DAC: 00000055
May 17 01:52:15 raspberrypi kernel: [ 3922.802582] Process smbd (pid: 25260, stack limit = 0xa067e210)
May 17 01:52:15 raspberrypi kernel: [ 3922.804688] Stack: (0xa067fea8 to 0xa0680000)
May 17 01:52:15 raspberrypi kernel: [ 3922.806781] fea0:                   00000000 00000000 80c06cc0 00000000 a067fedc a067fec8
May 17 01:52:15 raspberrypi kernel: [ 3922.811093] fec0: 8013e218 00000000 00000800 00000008 a067ff44 80c09474 0000080f 807192b8
May 17 01:52:15 raspberrypi kernel: [ 3922.815441] fee0: 55a19a2c a067ffb0 00000001 764f3100 a067ffac a067ff00 801011e8 807192c4
May 17 01:52:15 raspberrypi kernel: [ 3922.820029] ff00: a067ff54 801495a4 3cd249d3 00000391 80cb30f0 00000000 80b79300 80454600
May 17 01:52:15 raspberrypi kernel: [ 3922.824832] ff20: 80000013 80454600 80000013 00000000 bc27cd80 bc8a0800 00000006 80108244
May 17 01:52:15 raspberrypi kernel: [ 3922.829650] ff40: a067e000 00000000 a067ff74 a067ff58 8026d760 802c5b58 00000001 00000008
May 17 01:52:15 raspberrypi kernel: [ 3922.834562] ff60: bc8a0800 bc27cd80 a067ff94 a067ff78 80290000 8026d704 9dd4ac40 55a0b768
May 17 01:52:15 raspberrypi kernel: [ 3922.839697] ff80: 768f5104 55a0b768 a067ffa4 a067ff98 8026d7b4 8028ff44 00000000 a067ffa8
May 17 01:52:15 raspberrypi kernel: [ 3922.845080] ffa0: 801080c0 8026d790 55a0b768 768f5104 00000008 764b1504 00000071 00000001
May 17 01:52:15 raspberrypi kernel: [ 3922.850663] ffc0: 55a0b768 768f5104 55a0b768 00000006 55a0b7e8 7ef35930 00000000 76ee8e1c
May 17 01:52:15 raspberrypi kernel: [ 3922.856496] ffe0: 00000000 7ef357dc 764b7b7c 76f39d3c 60000010 00000008 00000000 00000000
May 17 01:52:15 raspberrypi kernel: [ 3922.861576] [<802c5b80>] (locks_remove_posix) from [<8026d760>] (filp_close+0x68/0x8c)
May 17 01:52:15 raspberrypi kernel: [ 3922.863437] [<8026d760>] (filp_close) from [<80290000>] (__close_fd+0xc8/0xec)
May 17 01:52:15 raspberrypi kernel: [ 3922.865301] [<80290000>] (__close_fd) from [<8026d7b4>] (SyS_close+0x30/0x58)
May 17 01:52:15 raspberrypi kernel: [ 3922.867137] [<8026d7b4>] (SyS_close) from [<801080c0>] (ret_fast_syscall+0x0/0x1c)
May 17 01:52:15 raspberrypi kernel: [ 3922.868971] Code: e59530e8 f57ff05b e3530000 0a000025 (e5b3200c) 
May 17 01:52:15 raspberrypi kernel: [ 3922.875677] ---[ end trace c275cccca39f4f71 ]---
May 17 01:52:15 raspberrypi kernel: [ 3922.881250] Unable to handle kernel paging request at virtual address 591b7f9e
May 17 01:52:15 raspberrypi kernel: [ 3922.884251] pgd = 80004000
May 17 01:52:15 raspberrypi kernel: [ 3922.887208] [591b7f9e] *pgd=00000000
May 17 01:52:15 raspberrypi kernel: [ 3922.890160] Internal error: Oops: 5 [#2] SMP ARM
May 17 01:52:15 raspberrypi kernel: [ 3922.893117] Modules linked in: cfg80211 rfkill ftdi_sio usbserial spidev i2c_bcm2835 spi_bcm2835 bcm2835_gpiomem uio_pdrv_genirq uio fixed rtc_ds1307 hwmon i2c_dev i2c_bcm2708 snd_bcm2835 snd_pcm snd_timer snd fuse ipv6
May 17 01:52:15 raspberrypi kernel: [ 3922.902324] CPU: 0 PID: 25260 Comm: smbd Tainted: G      D         4.9.28-v7+ #999
May 17 01:52:15 raspberrypi kernel: [ 3922.905287] Hardware name: BCM2835
May 17 01:52:15 raspberrypi kernel: [ 3922.908162] task: 9dd4ac40 task.stack: a067e000
May 17 01:52:15 raspberrypi kernel: [ 3922.910988] PC is at locks_remove_posix+0x34/0x150
May 17 01:52:15 raspberrypi kernel: [ 3922.913746] LR is at filp_close+0x68/0x8c
May 17 01:52:15 raspberrypi kernel: [ 3922.916422] pc : [<802c5b80>]    lr : [<8026d760>]    psr: 20000013
May 17 01:52:15 raspberrypi kernel: [ 3922.916422] sp : a067fb38  ip : a067fbe8  fp : a067fbe4
May 17 01:52:15 raspberrypi kernel: [ 3922.921610] r10: 802c5b82  r9 : b5facc00  r8 : 00000000
May 17 01:52:15 raspberrypi kernel: [ 3922.924114] r7 : bc8a0800  r6 : bc8a0800  r5 : bb1cda18  r4 : 00000000
May 17 01:52:15 raspberrypi kernel: [ 3922.926580] r3 : 591b7f92  r2 : bc27c6c0  r1 : bc8a0800  r0 : bc27c6c0
May 17 01:52:15 raspberrypi kernel: [ 3922.928979] Flags: nzCv  IRQs on  FIQs on  Mode SVC_32  ISA ARM  Segment user
May 17 01:52:15 raspberrypi kernel: [ 3922.931351] Control: 10c5387d  Table: 1de4806a  DAC: 00000055
May 17 01:52:15 raspberrypi kernel: [ 3922.933673] Process smbd (pid: 25260, stack limit = 0xa067e210)
May 17 01:52:15 raspberrypi kernel: [ 3922.935964] Stack: (0xa067fb38 to 0xa0680000)
May 17 01:52:15 raspberrypi kernel: [ 3922.938184] fb20:                                                       a067fb5c 802448f8
May 17 01:52:15 raspberrypi kernel: [ 3922.942441] fb40: 20000013 ffffffff a067fb94 bc889240 bbf19108 bc8a0800 00000041 80d03602
May 17 01:52:15 raspberrypi kernel: [ 3922.946698] fb60: 000062ac 60000013 0000020f 00000000 00000000 00048456 bbc9c840 a067fb88
May 17 01:52:15 raspberrypi kernel: [ 3922.951183] fb80: 00000000 00000000 ffffffff 7fffffff a067fbc4 a067fba0 8047952c 00000000
May 17 01:52:15 raspberrypi kernel: [ 3922.955932] fba0: 00000000 00000000 00000001 bbc3d880 00000001 bbc3d8b8 a067fc34 00000000
May 17 01:52:15 raspberrypi kernel: [ 3922.960828] fbc0: bc27c6c0 bc8a0800 bc8a0800 00000000 b5facc00 802c5b82 a067fc04 a067fbe8
May 17 01:52:15 raspberrypi kernel: [ 3922.966013] fbe0: 8026d760 802c5b58 bbf5e838 0003ffff 00000038 00000000 a067fc2c a067fc08
May 17 01:52:15 raspberrypi kernel: [ 3922.971415] fc00: 8028fb2c 8026d704 9dd4ac40 9dd4b140 bc8a0800 bbc3d880 00000001 bbc3d8b8
May 17 01:52:15 raspberrypi kernel: [ 3922.976978] fc20: a067fc4c a067fc30 8028fc3c 8028fa84 00000000 9dd4ac40 9dd4b140 80c08998
May 17 01:52:15 raspberrypi kernel: [ 3922.982740] fc40: a067fc8c a067fc50 801212b4 8028fbf0 80b7b85c 80c7a2c4 60000113 80c08998
May 17 01:52:15 raspberrypi kernel: [ 3922.988750] fc60: 0000000b 80c7a2c4 60000113 80c08998 0000000b 802c5b84 00000001 802c5b82
May 17 01:52:15 raspberrypi kernel: [ 3922.995026] fc80: a067fd14 a067fc90 8010c2b0 80120f70 a067e210 0000000b bbc3d880 8094de5c
May 17 01:52:15 raspberrypi kernel: [ 3923.001543] fca0: 00000008 7f000000 00000000 80c08998 65171f00 33353935 20386530 66373566
May 17 01:52:15 raspberrypi kernel: [ 3923.008287] fcc0: 62353066 35336520 30303033 61302030 30303030 28203532 33623565 63303032
[... 23 lines cut ...]
May 17 01:52:15 raspberrypi kernel: [ 3923.181899] ffc0: 55a0b768 768f5104 55a0b768 00000006 55a0b7e8 7ef35930 00000000 76ee8e1c
May 17 01:52:15 raspberrypi kernel: [ 3923.189152] ffe0: 00000000 7ef357dc 764b7b7c 76f39d3c 60000010 00000008 00000000 00000000
May 17 01:52:15 raspberrypi kernel: [ 3923.196427] [<802c5b80>] (locks_remove_posix) from [<8026d760>] (filp_close+0x68/0x8c)
May 17 01:52:15 raspberrypi kernel: [ 3923.200084] [<8026d760>] (filp_close) from [<8028fb2c>] (put_files_struct+0xb4/0x10c)
[... cut ...]
May 17 01:52:15 raspberrypi kernel: [ 3923.269434] [<80290000>] (__close_fd) from [<8026d7b4>] (SyS_close+0x30/0x58)
May 17 01:52:15 raspberrypi kernel: [ 3923.272276] [<8026d7b4>] (SyS_close) from [<801080c0>] (ret_fast_syscall+0x0/0x1c)
May 17 01:52:15 raspberrypi kernel: [ 3923.275042] Code: e59530e8 f57ff05b e3530000 0a000025 (e5b3200c) 
May 17 01:52:15 raspberrypi kernel: [ 3923.277933] ---[ end trace c275cccca39f4f72 ]---
May 17 01:52:15 raspberrypi kernel: [ 3923.280772] Fixing recursive fault but reboot is needed!``
```

popcornmix, 2017-05-19:
Please provide information about what you were doing when fault occurred. Any peripherals plugged in?
Are you overclocking? Any other non-default config.txt settings?
What does ``vcgencmd get_throttled`` report after running under load for a while?
Report output of:
```
vcgencmd version
uname -a
```

danyconnect, 2017-05-19:
i have two reader ftdi in usb.

**root@raspberrypi:~# vcgencmd get_throttled
throttled=0x0
root@raspberrypi:~# uname -a
Linux raspberrypi 4.9.28-v7+ #999 SMP Tue May 16 14:03:19 BST 2017 armv7l GNU/Linux
root@raspberrypi:~# vcgencmd version
May 15 2017 16:58:04 
Copyright (c) 2012 Broadcom
version 9469ea3706e34c4de62f38a5008f69a429b4b43e (clean) (release)**

i use ajenti core , unifi and oscam.
i reverted back to previous kernel 4.4.50 and all is ok
root@raspberrypi:~# vcgencmd versionFeb 20 2017 13:55:19 
Copyright (c) 2012 Broadcom
version 272c4de2358f2469414f38a1dea7179b5f09656c (clean) (release)
root@raspberrypi:~# uname -a
Linux raspberrypi 4.4.50-v7+ #970 SMP Mon Feb 20 19:18:29 GMT 2017 armv7l GNU/Linux
[config.txt](https://github.com/raspberrypi/linux/files/1014618/config.txt)

popcornmix, 2017-05-19:
On the 4.9 kernel how long does it take to get the failure?
Can you confirm if failure only occurs when ftdi usb devices are connected?

danyconnect, 2017-05-19:
i can not test without usb because the server is remotely. but i can confirm that some services(like ajenti,unifi) are frozen after a few hours from startup.
restart services frozen does not work.only work after restart server. this is on rpi2b.

mitsuhito, 2017-06-20:
my pi3 also have same issue.
kernel 4.9.30-v7+

danyconnect, 2017-12-02:
fixed in new kernel 4.9.57 
uptime 30 days no problem

closed 2017-12-02 by danyconnect