raspberrypi/linux #2027: smsc95xx 1-1.1:1.0 eth0: kevent 0 may have been dropped

andersthomson, 2017-05-19:
On a recent install of rpi-4.9.y, 4.9.28, I get this after a period of constant tv streaming.

[smsc95xx_eth0_kevent_0_may_have_been_dropped.txt](https://github.com/raspberrypi/linux/files/1014415/smsc95xx_eth0_kevent_0_may_have_been_dropped.txt)

Didn't get it all, so I attach frech dmesg as well to get a view on the early parts.

[4.9.28_dmesg.txt](https://github.com/raspberrypi/linux/files/1014422/4.9.28_dmesg.txt)

JamesH65, 2017-05-19:
Interesting. The kevent dropped message occurs when the system is trying to deal with a particular scenario using kevents (using the system queue), then the scenario occurs again before the previous one has been handled. However, to get that many in the log implies that the entire processing of those messages has died completely. Usually just one every now and again. Is this repeatable? Did the TV streaming die?

andersthomson, 2017-05-20:
Yep it died. Probaby as en effect, not cause, of this. No network connectivity after this event. I've seen it on vanilla 4.11 and, iirc, rpi-4.9.y 4.9.27 as well.

JamesH65, 2017-05-22:
yes, if you get that kevent dropped message, the network stack is probably FUBARd.

The log seems to imply bridging is set up - can you describe how the device is set up please?

popcornmix, 2017-05-22:
Does increasing ``vm.min_free_kbytes`` in ``etc/sysctl.d/98-rpi.conf`` help?
e.g. double it to 32678.

andersthomson, 2017-05-23:
It's running 4 systemd-nspawn containers via its bridging support. I'll try the min_free_kbytes when I get back to the device end of the week. This one is fairly easily triggered. I've seen it about 6 times already this week on both .26 and .27 kernels. .17 does not have this issue, so my plan was to move upwards from that to see what triggered it.

JamesH65, 2017-05-23:
What are you running in the containers? I wonder if the CPU load of running all the containers starves the main system kevent queue of time, so the kevent required to process the ethernet packet problem isn't run in time, or whether the system queue is simply swamped with events that need processing.

andersthomson, 2017-05-28:
One containter with tvheadend, one with openhab, one with nginx, one for misc cli stuff. In terms of load, tvheadend does the most net traffic. nginx is virtually dormant, so is openhab.

andersthomson, 2017-06-20:
Hi!
Any update on this? (What does 'Assigned for implementation' mean?)

I just updated to rpi-4.9.33 and have had two instances like this in like 24h or so. 
I previously ran a patched (the cgroup thing) 4.9.30 which ran fine for a couple of weeks.

I'd be happy to test out any patches you have...

JamesH65, 2017-06-20:
There was a fix for an issue that caused this when bridging a month back,
so it's possible you already have it. There may be further error paths that
cause it though. Did you provide a means of replication?

andersthomson, 2017-06-21:
If it's in 4.9.33 I should have it. Unfortunately, I have no means to reproduce it. No idea what triggers it (last one seems to have started at 18:00 sharp, fwiw). Any chance I can sprinkle a few printfs in selected places to get more context out?

JamesH65, 2017-07-25:
Is anyone still seeing this issue? I've not seen it for a while. Wondering whether recent changes to drivers/kernel have fixed it.

andersthomson, 2017-07-26:
I'm on 4.9.33-v7+ and I'm seeing it regularly. Like once every 4 days or so. I see we're t 4.9.39 now so I should probably do an upgrade. I'll report back after that.

JamesH65, 2017-07-26:
We do have a 4.13 in github, bleeding edge, but does seem to work so if
anyone fancies trying that....or 4.12 if you want something that has had
more testing.

andersthomson, 2017-07-26:
I'll queue that after the test of 4.9.39. Last time I tested higher than 4.9 it got stuck in the userspace boot phase. My memory is failing but it was either a usb driver thing (the other driver being the default and not working ok for TV transfers??), or lvm2 stuff not being in-kernel. Any update on the usb driver side? I recall a discussion on that...

JamesH65, 2017-07-26:
Not familiar with any USB issues, or what you mean by TV transfers. I've been using 4.12 a lot, and have tried 4.13 briefly (since it has a number of driver changes for the Wifi), seems to work OK for me.

andersthomson, 2017-07-27:
Isochronous transfers. See the discussion in #1649.

JamesH65, 2017-07-27:
I guessing lots of changes in all those areas, so might be worth giving 4.13 (or 4.12) a go.

andersthomson, 2017-07-27:
Great! I'm on 4.9.33 as of this morning. I'll let that cook for a while first (and will compile a 4.13 kernel)

andersthomson, 2017-07-28:
I can report that 4.9.33 has the same issue:
[81236.624055] smsc95xx 1-1.1:1.0 eth0: kevent 0 may have been dropped
[81236.661888] smsc95xx 1-1.1:1.0 eth0: kevent 0 may have been dropped
[81237.264041] smsc95xx 1-1.1:1.0 eth0: kevent 0 may have been dropped
[81238.544051] smsc95xx 1-1.1:1.0 eth0: kevent 0 may have been dropped

intermixed I now also see rcu errors. Not sure if they are related:

Eg.
[68777.769173] INFO: rcu_sched self-detected stall on CPU
[68777.806275]  3-...: (1957831 ticks this GP) idle=a4b/140000000000001/0 softirq=1539842/1539842 fqs=978154
[68777.878128]   (t=2025363 jiffies g=922236 c=922235 q=704310)
[68777.915933] Task dump for CPU 3:
[68777.949589] kswapd0         R  running task        0    40      2 0x00000002
[68777.986446] [<8010fb3c>] (unwind_backtrace) from [<8010c058>] (show_stack+0x20/0x24)
[68778.054711] [<8010c058>] (show_stack) from [<8014a8b8>] (sched_show_task+0xb8/0x10c)
[68778.124776] [<8014a8b8>] (sched_show_task) from [<8014cf84>] (dump_cpu_task+0x48/0x4c)
[68778.195656] [<8014cf84>] (dump_cpu_task) from [<80209e74>] (rcu_dump_cpu_stacks+0x9c/0xd0)
[68778.266975] [<80209e74>] (rcu_dump_cpu_stacks) from [<80180e3c>] (rcu_check_callbacks+0x800/0x9c0)
[68778.339345] [<80180e3c>] (rcu_check_callbacks) from [<80186640>] (update_process_times+0x40/0x6c)
[68778.411844] [<80186640>] (update_process_times) from [<80198834>] (tick_sched_handle+0x58/0x64)
[68778.484460] [<80198834>] (tick_sched_handle) from [<80198898>] (tick_sched_timer+0x58/0x9c)
[68778.557282] [<80198898>] (tick_sched_timer) from [<80187140>] (__hrtimer_run_queues+0x16c/0x338)
[68778.632636] [<80187140>] (__hrtimer_run_queues) from [<80187b9c>] (hrtimer_interrupt+0xd4/0x230)
[68778.710212] [<80187b9c>] (hrtimer_interrupt) from [<805db7a8>] (arch_timer_handler_phys+0x40/0x48)
[68778.788632] [<805db7a8>] (arch_timer_handler_phys) from [<801783f8>] (handle_percpu_devid_irq+0x88/0x23c)
[68778.869164] [<801783f8>] (handle_percpu_devid_irq) from [<80173050>] (generic_handle_irq+0x34/0x44)
[68778.950758] [<80173050>] (generic_handle_irq) from [<80173690>] (__handle_domain_irq+0x6c/0xc4)
[68779.032225] [<80173690>] (__handle_domain_irq) from [<8010150c>] (bcm2836_arm_irqchip_handle_irq+0xa8/0xac)
[68779.114847] [<8010150c>] (bcm2836_arm_irqchip_handle_irq) from [<8071acfc>] (__irq_svc+0x5c/0x7c)
[68779.196568] Exception stack(0xb9b3fcc8 to 0xb9b3fd10)
[68779.237623] fcc0:                   39412000 00004198 000021c3 00002a4b 8eed6220 b9b3fd58
[68779.316971] fce0: 80b7ce1c 8c45a660 8c45a6b0 00080400 8c45a6c8 b9b3fd4c b9b3fd18 b9b3fd18
[68779.395745] fd00: 80287acc 80287844 a00f0013 ffffffff
[68779.435626] [<8071acfc>] (__irq_svc) from [<80287844>] (shrink_dentry_list+0x11c/0x3e0)
[68779.512870] [<80287844>] (shrink_dentry_list) from [<80289e70>] (prune_dcache_sb+0x58/0x64)
[68779.590460] [<80289e70>] (prune_dcache_sb) from [<802741a0>] (super_cache_scan+0x11c/0x198)
[68779.667968] [<802741a0>] (super_cache_scan) from [<80221b8c>] (shrink_slab+0x2b4/0x504)
[68779.745056] [<80221b8c>] (shrink_slab) from [<802253c4>] (shrink_node+0x128/0x364)
[68779.821713] [<802253c4>] (shrink_node) from [<802264dc>] (kswapd+0x360/0x804)
[68779.863631] [<802264dc>] (kswapd) from [<8013c8e8>] (kthread+0x10c/0x124)
[68779.904671] [<8013c8e8>] (kthread) from [<80108148>] (ret_from_fork+0x14/0x2c)
[68792.448500] Unable to handle kernel paging request at virtual address 0cb70008
[68792.461609] pgd = 9a754000
[68792.469849] [0cb70008] *pgd=00000000
[68792.479199] Internal error: Oops: 80000005 [#1] SMP ARM
[68792.489996] Modules linked in: dm_mod rc_pinnacle_pctv_hd em28xx_rc rc_core si2157 si2168 i2c_mux tda18271 cxd2820r em28xx_dvb bridge stp llc veth dvb_core em28xx tveeprom v4l2_common videodev media evdev bcm2835_gpiomem fixed uio_pdrv_genirq uio ftdi_sio usbserial sch_fq_codel nfsd ip_tables x_tables ipv6
[68792.539619] CPU: 1 PID: 16586 Comm: make Not tainted 4.9.39-v7+ #36
[68792.550837] Hardware name: BCM2835
[68792.559137] task: 9822d000 task.stack: aee76000
[68792.568667] PC is at 0xcb70008
[68792.576455] LR is at vfs_getattr_nosec+0x3c/0x50
[68792.585887] pc : [<0cb70008>]    lr : [<80275740>]    psr: 200f0013
[68792.585887] sp : aee77ee0  ip : 8f3e34f0  fp : aee77ef4
[68792.607274] r10: 00000000  r9 : 00000000  r8 : aee77f40
[68792.617312] r7 : 7edc1e70  r6 : ffffff9c  r5 : 00000001  r4 : 0cb7000a
[68792.628525] r3 : 8c45a660  r2 : aee77f40  r1 : 8c45a660  r0 : b5324710
[68792.639709] Flags: nzCv  IRQs on  FIQs on  Mode SVC_32  ISA ARM  Segment user
[68792.651602] Control: 10c5387d  Table: 1a75406a  DAC: 00000055
[68792.662165] Process make (pid: 16586, stack limit = 0xaee76210)
[68792.672783] Stack: (0xaee77ee0 to 0xaee78000)
[68792.681784] 7ee0: aee77e60 00000000 aee77f2c aee77ef8 80275838 80275710 00000000 23b43423
[68792.699699] 7f00: b5324710 8c45a660 7edc1ea8 7edc2f40 00000018 000000c3 80108244 aee76000
[68792.717642] 7f20: aee77f3c aee77f30 802758a0 802757d4 aee77fa4 aee77f40 80276038 80275884
[68792.735921] 7f40: 00046693 00000000 0fe0000a 000081a4 00000001 00000000 00000000 00000000
[68792.755167] 7f60: 00002dc6 00000000 597a21a1 231b9d97 597a21a1 00000013 7edc2f40 00000018
[68792.775223] 7f80: 000000c3 800f0010 00000000 aee77f98 801081a0 00000013 00000000 aee77fa8
[68792.795712] 7fa0: 801080c0 80276020 00000013 7edc2f40 7edc1e70 7edc1ea8 7edc1ea8 00035704
[68792.816848] 7fc0: 00000013 7edc2f40 00000018 000000c3 7edc3138 00000000 000771cb 7edc1f2c
[... cut ...]
[68752.968763] [<80289e70>] (prune_dcache_sb) from [<802741a0>] (super_cache_scan+0x11c/0x198)
[68753.049355] [<802741a0>] (super_cache_scan) from [<80221b8c>] (shrink_slab+0x2b4/0x504)
[68753.131283] [<80221b8c>] (shrink_slab) from [<802253c4>] (shrink_node+0x128/0x364)
[68753.213023] [<802253c4>] (shrink_node) from [<802264dc>] (kswapd+0x360/0x804)
[68753.256701] [<802264dc>] (kswapd) from [<8013c8e8>] (kthread+0x10c/0x124)
[68753.300688] [<8013c8e8>] (kthread) from [<80108148>] (ret_from_fork+0x14/0x2c)

JamesH65, 2017-07-28:
OK, thanks for checking. I can only think that this might be an overloaded system queue being unable to process a kevent from the network stack before it happens again. Perhaps down to the system being very busy as it seems to be in your case. I'm not able to take any detailed look at it for the near future I'm afraid.

andersthomson, 2017-07-28:
The system hung in the middle of the night, while compiling the 4.12 kernel using make -j3 (on an usb hdd). I've also seen it hang in the middle of the night when there should be zero load.

It's not overloaded in the sense of the scheduler's run queue. No TV streaming, so not much usb traffic either.

The system has executed the same task for about 3 years. This error started to happen circa 4.9.15 or so.

JamesH65, 2017-07-28:
What I don't understand is why you are seeing it frequently, and others are
not.

andersthomson, 2017-07-28:
If that's the case, let's see what might differ:

>cat /boot/config.txt
#gpu_mem=192
#dtoverlay=dwc2
#device_tree=bcm2836-rpi-2-b.dtb 
#device_tree=dtbs/4.9.25/bcm2836-rpi-2-b.dtb
#kernel=vmlinuz-4.9.25
#device_tree=dtbs/4.10.14/bcm2836-rpi-2-b.dtb
#kernel=vmlinuz-4.10.14
#device_tree=dtbs/4.11.1/bcm2836-rpi-2-b.dtb
#kernel=vmlinuz-4.11.1
#kernel=vmlinuz-4.9.27-v7+
#kernel=vmlinuz-4.9.21-v7-00048-g1e7deb9
#device_tree=dtbs/4.9.21-v7+/bcm2836-rpi-2-b.dtb
#kernel=vmlinuz-4.9.21-v7+
#kernel=vmlinuz-4.9.19-v7+
#kernel=vmlinuz-4.9.30-v7+
#kernel=vmlinuz-4.9.33-v7+
#device_tree=dtbs/4.9.39-v7+/bcm2836-rpi-2-b.dtb
kernel=vmlinuz-4.9.39-v7+
#kernel=kernel7-new

As you can see I'm not using the dtbs stuff (whenever I try to enable it as per the net's instructions, it doesn't boot at all). I never got the hung of how the device tree stuff is expected to be used.

> ls -ltr /boot/
total 211092
-rwxr-xr-x  1 root root 4224800 Jan 13  2017 kernel7_orig.img
-rwxr-xr-x  1 root root 4012608 Jan 13  2017 kernel
-rwxr-xr-x  1 root root 7142324 Jan 13  2017 kernel7
-rwxr-xr-x  1 root root 3929604 Mar 13 13:11 start_x.elf
-rwxr-xr-x  1 root root 2846692 Mar 13 13:11 start.elf
-rwxr-xr-x  1 root root 4983140 Mar 13 13:11 start_db.elf
-rwxr-xr-x  1 root root  654980 Mar 13 13:11 start_cd.elf
-rwxr-xr-x  1 root root    1494 Mar 13 13:11 LICENCE.broadcom
-rwxr-xr-x  1 root root 4375736 Mar 13 13:11 kernel.img
-rwxr-xr-x  1 root root 4576688 Mar 13 13:11 kernel7.img
-rwxr-xr-x  1 root root    9775 Mar 13 13:11 fixup_x.dat
-rwxr-xr-x  1 root root    9777 Mar 13 13:11 fixup_db.dat
-rwxr-xr-x  1 root root    6646 Mar 13 13:11 fixup.dat
-rwxr-xr-x  1 root root    2558 Mar 13 13:11 fixup_cd.dat
-rwxr-xr-x  1 root root   18693 Mar 13 13:11 COPYING.linux
-rwxr-xr-x  1 root root   50268 Mar 13 13:11 bootcode.bin
-rwxr-xr-x  1 root root 4578440 Mar 28 14:24 kernel7-new
-rwxr-xr-x  1 root root   15660 May  1 10:22 bcm2836-rpi-2-b.dtb
-rwxr-xr-x  1 root root   14773 May  1 10:22 bcm2835-rpi-zero.dtb
-rwxr-xr-x  1 root root   15070 May  1 10:22 bcm2835-rpi-b-rev2.dtb
-rwxr-xr-x  1 root root   15194 May  1 10:22 bcm2835-rpi-b-plus.dtb
-rwxr-xr-x  1 root root   14937 May  1 10:22 bcm2835-rpi-b.dtb
-rwxr-xr-x  1 root root   14901 May  1 10:22 bcm2835-rpi-a-plus.dtb
-rwxr-xr-x  1 root root   14785 May  1 10:22 bcm2835-rpi-a.dtb
-rwxr-xr-x  1 root root   21285 May  1 10:22 bcm2710-rpi-cm3.dtb
-rwxr-xr-x  1 root root   22493 May  1 10:22 bcm2710-rpi-3-b.dtb
-rwxr-xr-x  1 root root   21402 May  1 10:22 bcm2709-rpi-2-b.dtb
-rwxr-xr-x  1 root root   19831 May  1 10:22 bcm2708-rpi-cm.dtb
-rwxr-xr-x  1 root root   20335 May  1 10:22 bcm2708-rpi-b-plus.dtb
-rwxr-xr-x  1 root root   20076 May  1 10:22 bcm2708-rpi-b.dtb
-rwxr-xr-x  1 root root   20565 May  1 10:22 bcm2708-rpi-0-w.dtb
drwxr-xr-x  2 root root   12288 May  1 10:22 overlays
-rwxr-xr-x  1 root root 4426744 May  1 11:16 vmlinuz-4.11.0-rc8+
-rwxr-xr-x  1 root root 1941138 May  1 11:16 System.map-4.11.0-rc8+
-rwxr-xr-x  1 root root  147224 May  1 11:16 config-4.11.0-rc8+
-rwxr-xr-x  1 root root 4513104 May  1 16:31 kernel7-test
-rwxr-xr-x  1 root root 7178384 May  3 15:06 vmlinuz-4.9.25.old
-rwxr-xr-x  1 root root 3598849 May  3 15:06 System.map-4.9.25.old
-rwxr-xr-x  1 root root  170247 May  3 15:06 config-4.9.25.old
-rwxr-xr-x  1 root root 7178584 May  3 20:34 vmlinuz-4.9.25
-rwxr-xr-x  1 root root 3598849 May  3 20:34 System.map-4.9.25
-rwxr-xr-x  1 root root  170247 May  3 20:34 config-4.9.25
-rwxr-xr-x  1 root root 7000136 May  5 14:09 vmlinuz-4.10.13.old
[... 35 lines cut ...]
-rwxr-xr-x  1 root root 4578232 May 28 22:00 vmlinuz-4.9.19-v7+
-rwxr-xr-x  1 root root 2223278 May 28 22:00 System.map-4.9.19-v7+
-rwxr-xr-x  1 root root  145477 May 28 22:00 config-4.9.19-v7+
-rwxr-xr-x  1 root root 4579680 May 29 20:22 vmlinuz-4.9.29-v7+
-rwxr-xr-x  1 root root 2223276 May 29 20:22 System.map-4.9.29-v7+
-rwxr-xr-x  1 root root  145581 May 29 20:22 config-4.9.29-v7+
-rwxr-xr-x  1 root root 4579832 May 31 20:24 vmlinuz-4.9.30-v7+
-rwxr-xr-x  1 root root 2223368 May 31 20:24 System.map-4.9.30-v7+
-rwxr-xr-x  1 root root  145581 May 31 20:24 config-4.9.30-v7+
-rwxr-xr-x  1 root root     192 Jun 15 20:42 cmdline.txt
-rwxr-xr-x  1 root root 4580808 Jun 19 21:52 vmlinuz-4.9.33-v7+
-rwxr-xr-x  1 root root 2223622 Jun 19 21:52 System.map-4.9.33-v7+
-rwxr-xr-x  1 root root  145630 Jun 19 21:52 config-4.9.33-v7+
-rwxr-xr-x  1 root root 4583304 Jul 26 22:27 vmlinuz-4.9.39-v7+
-rwxr-xr-x  1 root root 2224273 Jul 26 22:27 System.map-4.9.39-v7+
-rwxr-xr-x  1 root root  145662 Jul 26 22:27 config-4.9.39-v7+
drwxr-xr-x 16 root root    4096 Jul 26 22:28 dtbs
-rwxr-xr-x  1 root root     589 Jul 27 09:20 config.txt

Quite a few kernels etc, but perhaps you can see if the initial boot files are too old or something?

cat /boot/cmdline.txt
root=/dev/mmcblk0p2 smsc95xx.turbo_mode=N rootdelay=2 console=ttyAMA0,115200 console=tty0 init=/usr/lib/systemd/systemd ro systemd.unit=multi-user.target systemd.journald.forward_to_console=1

Anything I should adjust in this area to become more aligned with what others might have?

JamesH65, 2017-07-28:
Not the foggiest. In Raspbian, device tree should "just work". It's been
there by default for a couple of years.

andersthomson, 2017-07-30:
Hi,
I can report that 4.12.y have these kevent errors too.

andersthomson, 2017-08-30:
I can now also report that this happens on the foundation provided 4.9.45-v7+ kernel

excerpt of /boot
-rwxr-xr-x  1 root root   50248 Aug 25 11:12 bootcode.bin
-rwxr-xr-x  1 root root    6694 Aug 25 11:13 fixup.dat
-rwxr-xr-x  1 root root    2594 Aug 25 11:13 fixup_cd.dat
-rwxr-xr-x  1 root root 2868644 Aug 25 11:13 start.elf
-rwxr-xr-x  1 root root  666596 Aug 25 11:14 start_cd.elf
-rwxr-xr-x  1 root root   16523 Aug 25 12:32 bcm2709-rpi-2-b.dtb
-rwxr-xr-x  1 root root 4581760 Aug 28 14:23 kernel7.img

JamesH65, 2017-08-31:
Ok, thanks for the update. I wasn't expecting any movement on this, I've
not seen anything on the linux mailing lists and there have been no driver
changes.

radfish, 2017-10-08:
I've experienced this TX queue stall consistently (reproducible within several hours of heavy network load) on Odroid U3 (armv7h with smsc95xx chip) on kernel on 4.9.52, but not on 4.4.89, and not on 4.14-rc3. The issue did reproduce on this fork of 4.13 https://github.com/tobiasjakobi/linux-odroid-public . So, please do try it on 4.14-rc3 (and 4.4.89) mainline, maybe the issue will go away for you too.

There is a chance that something in the .config was different across my tests (I tried to keep it mostly the same, based on exynos_defconfig), or maybe the network load was different (or weather was diffferent -- it got cooler over time and maybe smsc95 overheated, who knows), but it's been running under similar load for 2.5 days without this nasty networking stall.

aphirst, 2017-10-08:
I've just experienced this on my XU4 running 4.9.51-1-ARCH. Was viewing video off its NFS share when apparently the entire network connectivity of the device stopped. Should have logged in via UART first, but I just assumed it was a hard crash since I couldn't login via SSH, and did a power cycle.

https://pastebin.com/ZVAgLFJH When looking through my logs I noticed I was being SSH dictionary attacked, but after about line 100 you can see the same sort of error (which is when the NFS shares disappeared).

```Oct 08 18:35:24 alarm kernel: fb_ili9340 spi1.0: I/O Error: rx-0 tx-1 res:rx-p tx-f len-4096
Oct 08 18:35:24 alarm kernel: fb_ili9340 spi1.0: SPI transfer failed: -5
Oct 08 18:35:24 alarm kernel: spi_master spi1: failed to transfer one message from queue
Oct 08 18:35:24 alarm kernel: fb_ili9340 spi1.0: fbtft_update_display: write_vmem failed to update display buffer
Oct 08 18:35:46 alarm kernel: INFO: rcu_preempt detected stalls on CPUs/tasks:
Oct 08 18:35:46 alarm kernel:         4-...: (1 GPs behind) idle=75b/140000000000000/0 softirq=4893732/4893733 fqs=1986
Oct 08 18:35:46 alarm kernel:         (detected by 1, t=4202 jiffies, g=2562895, c=2562894, q=602)
Oct 08 18:35:46 alarm kernel: Task dump for CPU 4:
Oct 08 18:35:46 alarm kernel: kworker/4:0     R  running task        0 30198      2 0x00000002
Oct 08 18:35:46 alarm kernel: Workqueue: events fb_deferred_io_work
Oct 08 18:35:46 alarm kernel: [<c0968994>] (__schedule) from [<00000001>] (0x1)
Oct 08 18:36:38 alarm systemd[1]: systemd-timesyncd.service: Watchdog timeout (limit 3min)!
Oct 08 18:36:38 alarm systemd[1]: systemd-timesyncd.service: Killing process 343 (systemd-timesyn) with signal SIGABRT.
Oct 08 18:36:49 alarm kernel: INFO: rcu_preempt detected stalls on CPUs/tasks:
Oct 08 18:36:49 alarm kernel:         4-...: (1 GPs behind) idle=75b/140000000000000/0 softirq=4893732/4893733 fqs=8125
Oct 08 18:36:49 alarm kernel:         (detected by 0, t=16809 jiffies, g=2562895, c=2562894, q=974)
Oct 08 18:36:49 alarm kernel: Task dump for CPU 4:
Oct 08 18:36:49 alarm kernel: kworker/4:0     R  running task        0 30198      2 0x00000002
Oct 08 18:36:49 alarm kernel: Workqueue: events fb_deferred_io_work
Oct 08 18:36:49 alarm kernel: [<c0968994>] (__schedule) from [<00000001>] (0x1)
Oct 08 18:37:38 alarm systemd[1]: systemd-networkd.service: Watchdog timeout (limit 3min)!
Oct 08 18:37:38 alarm systemd[1]: systemd-networkd.service: Killing process 283 (systemd-network) with signal SIGABRT.
Oct 08 18:37:52 alarm kernel: INFO: rcu_preempt detected stalls on CPUs/tasks:
Oct 08 18:37:52 alarm kernel:         4-...: (1 GPs behind) idle=75b/140000000000000/0 softirq=4893732/4893733 fqs=14073
Oct 08 18:37:52 alarm kernel:         (detected by 1, t=29412 jiffies, g=2562895, c=2562894, q=1116)
Oct 08 18:37:52 alarm kernel: Task dump for CPU 4:
Oct 08 18:37:52 alarm kernel: kworker/4:0     R  running task        0 30198      2 0x00000002
Oct 08 18:37:52 alarm kernel: Workqueue: events fb_deferred_io_work
Oct 08 18:37:52 alarm kernel: [<c0968994>] (__schedule) from [<00000001>] (0x1)
-- Reboot --```

If there's any more info I can provide, or somewhere else I should post this, feel free to ask/say.

radfish, 2017-10-17:
I had to move back to 4.13 because 4.14 resulted in yet another error (), and the network stalls are back. But, new information is: 'ip set link down eth0' and 'ip set link up eth0' (and adding routs with 'ip route') brings the network back up. I executed this from the serial console. But, this suggests a workaround: add a systemd  service that will check for network connectivity and attempt to reset link state as above.

6by9, 2017-10-18:
Just an observation in @andersthomson 's config, cmdline has ```smsc95xx.turbo_mode=N```, whilst the default is Y.
Description for that parameter is "Enable multiple frames per Rx transaction". Any particular reason why that has been disabled? It may be totally unrelated, but it's a difference.

andersthomson, 2017-10-22:
It's been like that the last 4 years or so, since I got the machine. Either I got it as part of the default install, or from the net somewhere. Is the default these days smsc95xx.turbo_mode=y on all Pis? Or am I better off removing this altogether?

andersthomson, 2017-10-22:
As a further note. Ever since I switched to PiF built kernels (ca 4.9.30) I've not had these issues. Maybe I should build my own again an see what might be triggering it....

karneaud, 2018-01-19:
I'm experiencing this on my pi3 b model running ubuntu-server 16.0.4 trying to pull docker images

JamesH65, 2018-01-19:
Interesting - this one hasn't reared its head for some time. What kernel
version are you using?

ZaMaZaN4iK, 2018-05-04:
@JamesH65 I have this issue on kernel 4.9.35-v7+

Is there any way to fix it?

JamesH65, 2018-05-05:
Not that I am aware of. It has become much much rarer over the last year. Worth trying the latest kernel release.

bcutter, 2018-06-06:
Possibly related: https://github.com/raspberrypi/linux/issues/2447

I also saw and see this on my Pi 3 B+. Very frustrating, especially cause after more than one year this issue is still not solved... :-(

JamesH65, 2018-06-06:
Completely different drivers so not the same issue. That error messages appears when a kernel work thread is being asked to service something whilst it is already servicing the same thing - it therefor drops the request. The different kevent number means a different workload. 

We haven't had any reports of the kevent0 issue for over a year IIRC.

bcutter, 2019-03-02:
Here we go again: Pi had to upload a quite big file (1,55 GB) via Ethernet to a client on the internet, which was interrupted and from that time the Pi was completely unresponsive on all network services. In the evening checked the console - system up and running. No chance to reset the ethernet (e. g. by ifdown/ifup), needed to reboot the system.

```
Feb 26 14:29:14 raspberrypi kernel: [198193.678073] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Feb 26 14:30:01 raspberrypi kernel: [198240.428064] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Feb 26 14:30:01 raspberrypi kernel: [198240.428089] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Feb 26 14:30:01 raspberrypi kernel: [198240.428182] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Feb 26 14:30:01 raspberrypi kernel: [198240.428226] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Feb 26 14:30:01 raspberrypi kernel: [198240.628603] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Feb 26 14:30:01 raspberrypi kernel: [198240.628678] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Feb 26 14:30:01 raspberrypi kernel: [198240.628717] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Feb 26 14:30:01 raspberrypi kernel: [198240.628765] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Feb 26 14:30:01 raspberrypi kernel: [198240.628804] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Feb 26 14:30:01 raspberrypi kernel: [198240.628841] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Feb 26 14:30:01 raspberrypi kernel: [198240.628888] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Feb 26 14:30:01 raspberrypi kernel: [198240.628927] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Feb 26 14:30:01 raspberrypi kernel: [198240.628965] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Feb 26 14:30:01 raspberrypi kernel: [198240.629013] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Feb 26 14:30:01 raspberrypi kernel: [198240.629052] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Feb 26 14:30:01 raspberrypi kernel: [198240.829514] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Feb 26 14:30:01 raspberrypi kernel: [198240.829623] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Feb 26 14:30:01 raspberrypi kernel: [198240.829669] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Feb 26 14:30:01 raspberrypi kernel: [198240.829722] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Feb 26 14:30:01 raspberrypi kernel: [198240.829771] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Feb 26 14:34:24 raspberrypi kernel: [198503.992292] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Feb 26 14:39:29 raspberrypi kernel: [198808.894392] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Feb 26 14:39:29 raspberrypi kernel: [198808.894574] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Feb 26 14:39:34 raspberrypi kernel: [198813.899593] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Feb 26 14:39:34 raspberrypi kernel: [198813.899807] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Feb 26 14:39:39 raspberrypi kernel: [198818.904917] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Feb 26 14:39:39 raspberrypi kernel: [198818.905092] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Feb 26 14:41:14 raspberrypi kernel: [198913.742733] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Feb 26 14:44:56 raspberrypi kernel: [199135.864108] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Feb 26 14:44:56 raspberrypi kernel: [199135.864213] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Feb 26 14:44:56 raspberrypi kernel: [199135.864301] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Feb 26 14:44:56 raspberrypi kernel: [199135.864422] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Feb 26 14:44:56 raspberrypi kernel: [199135.864466] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Feb 26 14:44:56 raspberrypi kernel: [199136.064917] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Feb 26 14:44:56 raspberrypi kernel: [199136.065041] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Feb 26 14:44:56 raspberrypi kernel: [199136.065088] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Feb 26 14:44:56 raspberrypi kernel: [199136.065205] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Feb 26 14:44:56 raspberrypi kernel: [199136.065299] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Feb 26 14:44:56 raspberrypi kernel: [199136.065506] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Feb 26 14:44:56 raspberrypi kernel: [199136.065611] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Feb 26 14:44:56 raspberrypi kernel: [199136.065747] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Feb 26 14:44:56 raspberrypi kernel: [199136.065838] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Feb 26 14:44:56 raspberrypi kernel: [199136.065937] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Feb 26 14:44:57 raspberrypi kernel: [199136.266493] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Feb 26 14:44:57 raspberrypi kernel: [199136.266596] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Feb 26 14:44:57 raspberrypi kernel: [199136.266690] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
[... cut ...]
Feb 26 20:37:01 raspberrypi kernel: [220261.056188] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Feb 26 20:37:01 raspberrypi kernel: [220261.056233] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Feb 26 20:37:01 raspberrypi kernel: [220261.056280] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Feb 26 20:37:01 raspberrypi kernel: [220261.056332] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Feb 26 20:37:30 raspberrypi kernel: [    0.000000] Booting Linux on physical CPU 0x0
```

6by9, 2019-03-02:
@bcutter NB the issue title of "**smsc95xx** 1-1.1:1.0 eth0: kevent 0 may have been dropped" vs your logging lines containing "lan78xx". Different chip, different drivers, different fixes required. JamesH65 had already pointed that out to you [before](https://github.com/raspberrypi/linux/issues/2027#issuecomment-395207481)

You also don't say what kernel version you're running. rpi-update will now be giving you a 4.19 kernel, whilst Raspbian and apt is still on 4.14. Without that we have insufficient information to do anything.

The warning is due to something kicking a worker thread faster than the worker thread can respond. The majority of people aren't seeing issues, so there seems to be something about your network that causes it. Please provide details of your network.

bcutter, 2019-04-09:
As it finally happened again a few minutes ago (kern.log flooded, network completely dead, only system reboot works around this issue): do you want me to open a new issue with the correct chip or to give more detailed information in this issue tracker?

E. g. there‘s no pattern, it happens randomly but usually after a longer System uptime, this time more than 11 days.

JamesH65, 2019-04-09:
Please start a new issue with all the details, including specifying the Pi/chip combo in the subject. Thanks.

bcutter, 2019-04-09:
Done. https://github.com/raspberrypi/linux/issues/2928

andersthomson, 2019-04-10:
Hi,

Fwiw, i've not gotten these the last year or so. When i got these (iirc) i was running self-compiled kernels and 4 systemd nspawn containers on my rpi2. "System pressure' might be a factor.

Anders