raspberrypi/linux #2029: cgourp WARNING

andersthomson, 2017-05-21:
Got this one on a fresh rpi-4.9.y kernel. Did not happen on 3.4.17.
[cgroup_warning_3779.txt](https://github.com/raspberrypi/linux/files/1017226/cgroup_warning_3779.txt)

JamesH65, 2017-05-22:
Can you give a bit more detail on setup of the device? You appear to be using bridging?

popcornmix, 2017-05-22:
Also this is not a stock raspbian kernel:
```
[    0.000000] Linux version 4.9.27-v7+ (root@rpi2.lan) (gcc version 4.9.4 (Gentoo 4.9.4 p1.0, pie-0.6.4) ) #23 SMP Mon May 15 09:20:08 CEST 2017
```
If you are able to test on raspbian if you get the same issue it may help to narrow it down.

andersthomson, 2017-05-23:
HI,

it's a gentoo with two powered usb bridges with two tv tuners, two HDs and two serial card readers. I run 4 systemd-nspawn containers via a bridge.

There was a reason for compiling my own kernel in the past. A CONFIG I needed, but I forgot which. I'll see if I can get the binary kernel working, but I won;t be near the device until end of the week.

closed 2017-09-13 by JamesH65