raspberrypi/linux #2030: Cross-compile aarch64 using 4.12y No act/pwr LED support

gbkelley, 2017-05-22:
Cross compiling to aarch64 using 4.12y source. Still no led0 or led1 directories in /sys/class/leds/ so unable to control either act or pwr led. Standard upgrade of Jessie Lite on rpi3 now shows both leds for control but compiling from source doesn't.

popcornmix, 2017-05-22:
It is working with 32-bit kernel:
```
pi@domnfs:~ $ uname -a
Linux domnfs 4.12.0-rc2-v7+ #2518 SMP Mon May 22 13:19:35 BST 2017 armv7l GNU/Linux
pi@domnfs:~ $ ls /sys/class/leds/
input1::capslock  input1::numlock  input1::scrolllock  led0  led1
```

gbkelley, 2017-05-22:
Yes, as mentioned in OP 32-bit upgrade of Jessie Lite distro now shows both leds. Compiling from source (cross-compile to aarch64) somehow is not setting things correctly for same result. Tried 4.11 and 4.12 branches. Compiles fine, boots fine. Ideas? Thanks.

Linux RasPi3B 4.11.0+pi64-v8+ #1 SMP PREEMPT Wed Apr 26 20:23:29 CEST 2017 aarch64 GNU/Linux

pelwell, 2017-05-22:
That's because both LEDs on a Pi 3 are attached to a GPIO expander addressable by I2C, but the arm64 Pi3 DTB is still trying to use GPIO 47. The GPIO expander drivers aren't upstreamed, which is why the upstream DTB can't use them.

gbkelley, 2017-05-22:
Ah, makes sense. Any plans to upstream as part of a future 64-bit distro for Pi3?

pelwell, 2017-05-22:
Not at the moment. Two kernels is more than we'd like, so we won't be adding a third unless somebody can demonstrate clear benefits to using 64-bit on Pi3. So far, nobody has - some things are quicker, some things suffer.

gbkelley, 2017-05-22:
OK, thanks for the input folks. Using Pi3 as a CLOAK coin POS staking node and no real speed difference between 32 and 64 although boots quicker and apps load quicker.

closed 2017-05-22 by gbkelley