raspberrypi/linux #2033: NULL pointer dereferences with full-speed root port dequeues

P33M, 2017-05-23:
With an active isochronous OUT transfer on a Pi Zero with the port in full-speed, you can get crashes like this by doing Ctrl-C while speaker-test is running:

```
[  599.275530] Unable to handle kernel NULL pointer dereference at virtual address 0000002f
[  599.288144] pgd = c0004000
[  599.292977] [0000002f] *pgd=00000000
[  599.298706] Internal error: Oops: 817 [#1] ARM
[  599.305241] Modules linked in: bnep hci_uart btbcm bluetooth snd_usb_audio snd_hwdep snd_usbmidi_lib snd_rawmidi snd_seq_device snd_pcm snd_timer snd brcmfmac brcmutil cfg80211 rfkill bcm2835_gpiomem uio_pdrv_genirq uio fixed i2c_dev fuse ipv6
[  599.333806] CPU: 0 PID: 0 Comm: swapper Not tainted 4.9.28+ #999
[  599.342087] Hardware name: BCM2835
[  599.347727] task: c08aa2d8 task.stack: c08a4000
[  599.354558] PC is at usb_api_blocking_completion+0x20/0x2c
[  599.362367] LR is at __usb_hcd_giveback_urb+0x80/0x160
[  599.369796] pc : [<c041c380>]    lr : [<c0418d68>]    psr: 60000193
[  599.369796] sp : c08a5de8  ip : c08a5df8  fp : c08a5df4
[  599.385742] r10: 00000001  r9 : 00000100  r8 : 00000000
[  599.393115] r7 : 00000000  r6 : 00000000  r5 : 60000113  r4 : da4b2280
[  599.401829] r3 : 00000023  r2 : 00000000  r1 : 00000560  r0 : 00000023
[  599.410485] Flags: nZCv  IRQs off  FIQs on  Mode SVC_32  ISA ARM  Segment user
[  599.419912] Control: 00c5387d  Table: 18e9c008  DAC: 00000055
[  599.427775] Process swapper (pid: 0, stack limit = 0xc08a4188)
[  599.435728] Stack: (0xc08a5de8 to 0xc08a6000)
[  599.442085] 5de0:                   c08a5e14 c08a5df8 c0418d68 c041c36c d8cf1160 daf8b800
[  599.454455] 5e00: da4b2280 00000000 c08a5e24 c08a5e18 c0418e90 c0418cf4 c08a5e4c c08a5e28
[  599.467000] 5e20: c0443504 c0418e54 c0418148 20000113 c08b5420 daf62280 00000000 c08adefc
[  599.479530] 5e40: c08a5e5c c08a5e50 c045212c c04434a4 c08a5e7c c08a5e60 c00266f8 c0452118
[  599.492142] 5e60: c091ee80 c08f2d2c 00000001 00000000 c08a5ec4 c08a5e80 c00095cc c0026680
[  599.504955] 5e80: dac20cc0 dac20cd0 00000001 00200100 000074df 0000000a 0004ee8d 00000000
[  599.517943] 5ea0: c08f2d2c 00000000 00000001 dac1e000 c08a4000 dbfff940 c08a5ed4 c08a5ec8
[  599.531052] 5ec0: c00262e4 c00094b4 c08a5efc c08a5ed8 c005e16c c0026200 c08a5f18 c08a5f18
[  599.544489] 5ee0: c08a7cc4 ffffffff c08a5f4c 00000001 c08a5f14 c08a5f00 c0009420 c005e118
[  599.558067] 5f00: c00107c8 60000013 c08a5f74 c08a5f18 c05d3c1c c0009404 00000000 00000000
[  599.571775] 5f20: 00000000 c08a7760 c08a4000 00000000 c08a68ac c08bbcc2 00000001 c08bbcc2
[  599.585573] 5f40: dbfff940 c08a5f74 c08a5f68 c08a5f68 c00107c4 c00107c8 60000013 ffffffff
[  599.599346] 5f60: c00107c4 00000000 c08a5f84 c08a5f78 c05d3acc c00107a4 c08a5f9c c08a5f88
[  599.613121] 5f80: c00506cc c05d3aa4 c05d7594 c08a6820 c08a5fac c08a5fa0 c05ceee8 c005064c
[  599.626942] 5fa0: c08a5ff4 c08a5fb0 c083fc84 c05cee88 ffffffff ffffffff 00000000 c083f6d4
[  599.640772] 5fc0: 00000000 c0881a28 00000000 c091e894 c08a683c c0881a24 c08abfd0 00004008
[  599.654599] 5fe0: 410fb767 0087f77c 00000000 c08a5ff8 00008074 c083f954 00000000 00000000
[  599.668440] [<c041c380>] (usb_api_blocking_completion) from [<c0418d68>] (__usb_hcd_giveback_urb+0x80/0x160)
[  599.683946] [<c0418d68>] (__usb_hcd_giveback_urb) from [<c0418e90>] (usb_hcd_giveback_urb+0x48/0x10c)
[  599.698827] [<c0418e90>] (usb_hcd_giveback_urb) from [<c0443504>] (completion_tasklet_func+0x6c/0x98)
[  599.713739] [<c0443504>] (completion_tasklet_func) from [<c045212c>] (tasklet_callback+0x20/0x24)
[  599.728321] [<c045212c>] (tasklet_callback) from [<c00266f8>] (tasklet_hi_action+0x84/0xd8)
[  599.742374] [<c00266f8>] (tasklet_hi_action) from [<c00095cc>] (__do_softirq+0x124/0x31c)
[  599.756250] [<c00095cc>] (__do_softirq) from [<c00262e4>] (irq_exit+0xf0/0x140)
[  599.766558] [<c00262e4>] (irq_exit) from [<c005e16c>] (__handle_domain_irq+0x60/0xb8)
[  599.779974] [<c005e16c>] (__handle_domain_irq) from [<c0009420>] (bcm2835_handle_irq+0x28/0x48)
[  599.794338] [<c0009420>] (bcm2835_handle_irq) from [<c05d3c1c>] (__irq_svc+0x5c/0x7c)
[  599.807781] Exception stack(0xc08a5f18 to 0xc08a5f60)
[  599.815651] 5f00:                                                       00000000 00000000
[  599.829164] 5f20: 00000000 c08a7760 c08a4000 00000000 c08a68ac c08bbcc2 00000001 c08bbcc2
[  599.842645] 5f40: dbfff940 c08a5f74 c08a5f68 c08a5f68 c00107c4 c00107c8 60000013 ffffffff
[  599.856141] [<c05d3c1c>] (__irq_svc) from [<c00107c8>] (arch_cpu_idle+0x30/0x40)
[  599.868842] [<c00107c8>] (arch_cpu_idle) from [<c05d3acc>] (default_idle_call+0x34/0x48)
[  599.882192] [<c05d3acc>] (default_idle_call) from [<c00506cc>] (cpu_startup_entry+0x8c/0xe8)
[  599.895945] [<c00506cc>] (cpu_startup_entry) from [<c05ceee8>] (rest_init+0x6c/0x84)
[  599.909027] [<c05ceee8>] (rest_init) from [<c083fc84>] (start_kernel+0x33c/0x3b0)
[  599.921850] Code: e8bd4000 e5903074 e5902038 e1a00003 (e583200c)
[  599.930667] ---[ end trace 0e1fcde3c8dca79d ]---
[  599.937929] Kernel panic - not syncing: Fatal exception in interrupt
[  599.946936] ---[ end Kernel panic - not syncing: Fatal exception in interrupt

```

IComplainInComments, 2017-05-24:
Just so its easier for everyone; Can you provide which kernel youre using? Along with it being a customize kernel, or if it was just the default ARMv6 Raspberry Pi Kernel?

P33M, 2017-06-28:
Not seen a recurrence of this after improving locking around disconnect callbacks.

closed 2017-06-28 by P33M