raspberrypi/linux #2034: Preventing watchdogd from running

vanfanel, 2017-05-24:
Hi,

I'd like to disable watchdogd for experimentation pourposes.
So, I tried "systemd mask watchdogd", "systemd mask watchdog", etc... with no luck.

My guess then is that "watchdogd" is a kernel daemon. Any boot parameters to disable it?
According to:
https://wiki.archlinux.org/index.php/Power_management#Disabling_NMI_watchdog
...the nmi_watchdog=0 parameter should disable it, but it does not.

EDIT: It seems the watchdog module is responsible, according to this:
https://www.raspberrypi.org/forums/viewtopic.php?t=154055&p=1008798
The problem is that in current kernel config bcm2835_wdt module is not loaded, but built-in... why? Any way to disable it then?

Any ideas?

Thanks!

pelwell, 2017-05-24:
In config.txt:
```
dtparam=watchdog=off
```

vanfanel, 2017-05-24:
@pelwell : thanks, that did it!
BTW, there a dtparam to disable the ethernet without disabling USB?
Just like in the typical:
echo 1-1.1:1.0 > /sys/bus/usb/drivers/smsc95xx/unbind

pelwell, 2017-05-24:
Not that I am aware of. I tried the obvious things like putting `status = "disabled";` in the usbether node, but it didn't make any difference.

vanfanel, 2017-05-24:
Is there any way to get the CRTL+ALT+SUP combination work without the watchdogd running? (sudo reboot does not work either)

pelwell, 2017-05-24:
Rebooting uses the watchdog, so if the watchdog is completely disabled then you can't reboot. Have you tried disabling the watchdog at runtime?
```
sudo sh -c "echo V > /dev/watchdog"
```
That should disarm the watchdog until something that doesn't contain a 'V' is written to wake it up again.

pelwell, 2017-05-24:
A stock Raspbian has the watchdog enabled but not armed. It isn't clear yet what watchdogd is doing but it isn't periodically resetting the watchdog, as you can see if you write something not containing V to it and then wait 10-15 seconds.

Do you know what is arming your watchdog? Are you actually seeing it triggering or are you trying to prevent a theoretical problem?

vanfanel, 2017-05-24:
I don't know if my watchdog is armed or not, really. I am just trying to understand how it works and how to disable it because I am experimenting on what can be disabled in the kernel for faster boot times, etc.

pelwell, 2017-05-24:
I presume you have listed all processes and found one called `watchdogd`? This is actually a thread associated with a kernel workqueue - the string `watchdogd` only appears in one place in the kernel source, and changing that occurrence also changes the name of the process. Since kernel threads are very lightweight and disabling it would require a custom kernel build, I suggest you just accept it.

Please close this issue if you are happy your query has been answered.

vanfanel, 2017-05-24:
@pelwell : Yes, that was the case. All my questions are answered... in the best way possible. So I deeply thank you :+1:

closed 2017-05-24 by vanfanel