raspberrypi/linux #2036: Backport joystick polling rate commandline option

vanfanel, 2017-05-24:
I have found that, at last, the polling rate will be configurable for USB joysticks in 4.12 kernel:
https://patchwork.kernel.org/patch/9591909/

It seems to have been merged upstream, so, could that be backported to Pi's current kernel? It will be some time until we are at 4.12 as default, and the feature is interesting enough.

Thanks

popcornmix, 2017-05-24:
@pelwell I've tested and 4.12 patch applies without much grief. Any objections?

pelwell, 2017-05-24:
Go ahead - it looks safe.

popcornmix, 2017-05-24:
Added: https://github.com/raspberrypi/linux/commit/3db4135e6fbae417e0b297afa680c6caeaf52c19
Will be in next rpi-update kernel.

vanfanel, 2017-05-26:
Thanks for the backport! When can we expect the next rpi-update with this?

popcornmix, 2017-05-26:
Should be in latest rpi-update kernel. Please test and close if happy.

vanfanel, 2017-05-27:
Works great, 500Hz according to evhz! 
Thanks, closing!

closed 2017-05-27 by vanfanel