raspberrypi/linux #2037: rpi-4.12.y: cc1plus: page allocation failure

D-Albers, 2017-05-25:
Hi,

the rpi3 stucks on regular basis with the following error. The root fs is mounted through nfs and the cpu has a high load on all the cores.

arch: arm64
os: debian stretch  

[config-4.12.0-rc2-gaf4bd9b.txt](https://github.com/raspberrypi/linux/files/1029449/config-4.12.0-rc2-gaf4bd9b.txt)
```
[ 6824.578063] smsc95xx 1-1.1:1.0 eth0: kevent 2 may have been dropped
[ 6824.659496] cc1plus: page allocation failure: order:0, mode:0x1082020(GFP_ATOMIC|__GFP_MEMALLOC), nodemask=(null)
[ 6824.766185] cc1plus cpuset=/ mems_allowed=0
[ 6824.836504] CPU: 3 PID: 8240 Comm: cc1plus Tainted: G        W       4.12.0-rc2-v8+ #1
[ 6824.929782] Hardware name: Raspberry Pi 3 Model B Rev 1.2 (DT)
[ 6825.011010] Call trace:
[ 6825.072041] [<ffffff9bc0a8a108>] dump_backtrace+0x0/0x258
[ 6825.150871] [<ffffff9bc0a8a444>] show_stack+0x24/0x30
[ 6825.226930] [<ffffff9bc0e11e14>] dump_stack+0xac/0xe8
[ 6825.301982] [<ffffff9bc0baa670>] warn_alloc+0xe0/0x160
[ 6825.377061] [<ffffff9bc0bab4c8>] __alloc_pages_nodemask+0xd48/0xd98
[ 6825.458731] [<ffffff9bc0bab75c>] page_frag_alloc+0x14c/0x168
[ 6825.536521] [<ffffff9bc0ff4678>] __netdev_alloc_skb+0xd8/0x160
[ 6825.615291] [<ffffff9bc0f19170>] rx_submit+0x48/0x240
[ 6825.689469] [<ffffff9bc0f19824>] rx_complete+0x1fc/0x228
[ 6825.765239] [<ffffff9bc0f23d18>] __usb_hcd_giveback_urb+0x90/0x140
[ 6825.846467] [<ffffff9bc0f23ef0>] usb_hcd_giveback_urb+0x50/0x118
[ 6825.926620] [<ffffff9bc0f51188>] completion_tasklet_func+0x70/0xb8
[ 6826.007710] [<ffffff9bc0f60a70>] tasklet_callback+0x20/0x30
[ 6826.085094] [<ffffff9bc0aa48ec>] tasklet_hi_action+0xac/0x158
[ 6826.163703] [<ffffff9bc0a8198c>] __do_softirq+0x164/0x3a0
[ 6826.240128] [<ffffff9bc0aa4474>] irq_exit+0xc4/0xd8
[ 6826.313235] [<ffffff9bc0af1ef0>] __handle_domain_irq+0x90/0x100
[ 6826.392663] [<ffffff9bc0a81598>] bcm2836_arm_irqchip_handle_irq+0x68/0xc8
[ 6826.477282] Exception stack(0xffffffc644d9fec0 to 0xffffffc644d9fff0)
[ 6826.560181] fec0: 0000007f961a8e00 0000000000000000 0000007f95ce3478 000000003a55d120
[ 6826.652149] fee0: 00000000000002f1 000000003a9b9c60 0000000000000000 00000007bef7ff7f
[ 6826.744272] ff00: 0000000000000001 0000000000000000 0000007f95f57140 00000000013ed000
[ 6826.836362] ff20: 0000000000000000 0000000000000001 0000000000000001 0000007f9672b2d0
[ 6826.928368] ff40: 00000000013d0408 0000007f985e82c0 0000000000000001 00000000000000bc
[ 6827.019976] ff60: 000000003a814d40 00000000000000bd 0000007f95c1da28 00000000000000bd
[ 6827.111298] ff80: 000000003a7f4730 0000000000000000 000000003a55d120 0000000000000001
[ 6827.202139] ffa0: 000000003a7119f0 0000007ffde54d80 00000000007b21d0 0000007ffde54d80
[ 6827.292912] ffc0: 00000000007b4c78 0000000080000000 000000003aaa1000 ffffffffffffffff
[ 6827.383789] ffe0: 0000000000000000 0000000000000000
[ 6827.456928] [<ffffff9bc0a8355c>] el0_irq_naked+0x50/0x64
[ 6827.532866] Mem-Info:
[ 6827.590467] active_anon:163256 inactive_anon:53553 isolated_anon:34
[ 6827.590467]  active_file:946 inactive_file:428 isolated_file:0
[ 6827.590467]  unevictable:3 dirty:0 writeback:21 unstable:35
[ 6827.590467]  slab_reclaimable:9425 slab_unreclaimable:3121
[ 6827.590467]  mapped:1401 shmem:94 pagetables:727 bounce:0
[ 6827.590467]  free:983 free_pcp:64 free_cma:0
[ 6828.054795] Node 0 active_anon:653024kB inactive_anon:214212kB active_file:3784kB inactive_file:1712kB unevictable:12kB isolated(anon):136kB isolated(file):0kB mapped:5604kB dirty:0kB writeback:84kB shmem:376kB writeback_tmp:0kB unstable:140kB all_unreclaimable? no
[ 6828.290190] DMA free:3932kB min:3848kB low:4808kB high:5768kB active_anon:653024kB inactive_anon:214112kB active_file:3684kB inactive_file:1712kB unevictable:12kB writepending:84kB present:966656kB managed:937600kB mlocked:12kB slab_reclaimable:37700kB slab_unreclaimable:12484kB kernel_stack:2144kB pagetables:2908kB bounce:0kB free_pcp:256kB local_pcp:52kB free_cma:0kB
[ 6828.588776] lowmem_reserve[]: 0 0 0
[ 6828.661886] DMA: 8*4kB (H) 1*8kB (H) 6*16kB (H) 5*32kB (H) 5*64kB (H) 2*128kB (H) 2*256kB (H) 3*512kB (H) 1*1024kB (H) 0*2048kB 0*4096kB = 3944kB
[ 6828.794252] 1792 total pagecache pages
[ 6828.869376] 336 pages in swap cache
[ 6828.940858] Swap cache stats: add 204741, delete 204407, find 25737/49613
[ 6829.030875] Free swap  = 1026300kB
[ 6829.098855] Total swap = 1048572kB
[ 6829.165260] 241664 pages RAM
[ 6829.227298] 0 pages HighMem/MovableOnly
[ 6829.294022] 7264 pages reserved
[ 6829.356589] 2048 pages cma reserved
```

pelwell, 2017-05-25:
You appear to be running with swap enabled over nfs, which I've found to be problematic. You're also running a 64-bit kernel (which we don't support) and a bleeding edge kernel.

How many jobs are you running in parallel in your build? Try reducing that number.

D-Albers, 2017-05-25:
> You appear to be running with swap enabled over nfs, which I've found to be problematic.

Can be changed to microSD.

>  You're also running a 64-bit kernel (which we don't support) and a bleeding edge kernel.

Yes, I know it. If this type of reports are not wanted, just say it and I will close it. I see this more as an information for you, that there has been a problem.

> How many jobs are you running in parallel in your build? Try reducing that number.

Only 4, one for each core.

I will move the swap to the µSD now and see what happens.

D-Albers, 2017-05-25:
Moving the swap to a µSD doesn't fix the problem. The same error appears again.

popcornmix, 2017-05-26:
> Yes, I know it. If this type of reports are not wanted, just say it and I will close it. I see this more as an information for you, that there has been a problem.

As we don't typically run 64-bit kernels here we're unlikely to be able to help much (unless the backtrace shows something we are familiar with). But there are a few 64-bit kernel users who comment on these issues so I don't mind leaving this here for now.

Can you test if an upstream kernel has the same problem? If it does then this may interest some upstream folk who keep an eye out here.

D-Albers, 2017-05-26:
The problem didn't appear with the updated version of the last foundation raspbian image.

The last upstream kernel has been build and I will execute the test later.
What may be the best way to get in contact with some upstream folk?

popcornmix, 2017-05-26:
[linux-rpi-kernel](https://lists.infradead.org/mailman/listinfo/linux-rpi-kernel) is probably best for upstream pi kernel issues.

JamesH65, 2017-09-13:
Closing this issue as questions answered/issue resolved. Reopen if necessary.

closed 2017-09-13 by JamesH65