raspberrypi/linux #2039: RPI Zero W can`t found wlan0

LeonardLee, 2017-05-26:
Hi, all

I build 4.9.x kernel and burn into my rpi zero w, but I can not found wlan0 with any way.
I guess it is a driver issue. How to fix it?  dmesg is here:

pi@raspberrypi:/etc/modprobe.d$ dmesg
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 4.9.29+ (pi@raspberrypi) (gcc version 4.9.2 (Raspbian 4.9.2-10) ) #1 Fri May 26 05:53:15 UTC 2017
[    0.000000] CPU: ARMv6-compatible processor [410fb767] revision 7 (ARMv7), cr=00c5387d
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT nonaliasing instruction cache
[    0.000000] OF: fdt:Machine model: Raspberry Pi Zero W Rev 1.1
[    0.000000] cma: Reserved 8 MiB at 0x1b400000
[    0.000000] Memory policy: Data cache writeback
[    0.000000] On node 0 totalpages: 114688
[    0.000000] free_area_init_node: node 0, pgdat c0914854, node_mem_map db010000
[    0.000000]   Normal zone: 1008 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 114688 pages, LIFO batch:31
[    0.000000] pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
[    0.000000] pcpu-alloc: [0] 0 
[    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 113680
[    0.000000] Kernel command line: 8250.nr_uarts=1 bcm2708_fb.fbwidth=1824 bcm2708_fb.fbheight=984 bcm2708_fb.fbswap=1 smsc95xx.macaddr=B8:27:EB:6F:2C:D9 vc_mem.mem_base=0x1ec00000 vc_mem.mem_size=0x20000000  dwc_otg.lpm_enable=0 console=ttyS0,115200 console=tty1 root=PARTUUID=94fa391f-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait quiet splash plymouth.ignore-serial-consoles
[    0.000000] PID hash table entries: 2048 (order: 1, 8192 bytes)
[    0.000000] Dentry cache hash table entries: 65536 (order: 6, 262144 bytes)
[    0.000000] Inode-cache hash table entries: 32768 (order: 5, 131072 bytes)
[    0.000000] Memory: 435940K/458752K available (5945K kernel code, 489K rwdata, 1948K rodata, 400K init, 734K bss, 14620K reserved, 8192K cma-reserved)
[    0.000000] Virtual kernel memory layout:
    vector  : 0xffff0000 - 0xffff1000   (   4 kB)
    fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
    vmalloc : 0xdc800000 - 0xff800000   ( 560 MB)
    lowmem  : 0xc0000000 - 0xdc000000   ( 448 MB)
    modules : 0xbf000000 - 0xc0000000   (  16 MB)
      .text : 0xc0008000 - 0xc05d665c   (5946 kB)
      .init : 0xc0840000 - 0xc08a4000   ( 400 kB)
      .data : 0xc08a4000 - 0xc091e708   ( 490 kB)
       .bss : 0xc091e708 - 0xc09d5f08   ( 734 kB)
[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
[    0.000000] NR_IRQS:16 nr_irqs:16 16
[    0.000029] sched_clock: 32 bits at 1000kHz, resolution 1000ns, wraps every 2147483647500ns
[    0.000060] clocksource: timer: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 1911260446275 ns
[    0.000153] bcm2835: system timer (irq = 27)
[    0.000615] Console: colour dummy device 80x30
[    0.000641] console [tty1] enabled
[    0.000676] Calibrating delay loop... 697.95 BogoMIPS (lpj=3489792)
[    0.060301] pid_max: default: 32768 minimum: 301
[    0.060723] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.060741] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.061803] Disabling cpuset control group subsystem
[    0.062014] CPU: Testing write buffer coherency: ok
[    0.062074] ftrace: allocating 21708 entries in 64 pages
[    0.178635] Setting up static identity map for 0x8200 - 0x8238
[    0.180618] devtmpfs: initialized
[    0.190201] VFP support v0.3: implementor 41 architecture 1 part 20 variant b rev 5
[    0.190620] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.190649] futex hash table entries: 256 (order: -1, 3072 bytes)
[    0.191921] pinctrl core: initialized pinctrl subsystem
[    0.193363] NET: Registered protocol family 16
[    0.195923] DMA: preallocated 1024 KiB pool for atomic coherent allocations
[    0.206017] hw-breakpoint: found 6 breakpoint and 1 watchpoint registers.
[    0.206034] hw-breakpoint: maximum watchpoint size is 4 bytes.
[    0.206140] Serial: AMBA PL011 UART driver
[    0.208984] bcm2835-mbox 2000b880.mailbox: mailbox enabled
[    0.209705] uart-pl011 20201000.serial: could not find pctldev for node /soc/gpio@7e200000/uart0_pins, deferring probe
[    0.210142] irq: no irq domain found for /soc/aux@0x7e215000 !
[    0.257071] bcm2835-dma 20007000.dma: DMA legacy API manager at dc80d000, dmachans=0x1
[    0.259856] SCSI subsystem initialized
[    0.260117] usbcore: registered new interface driver usbfs
[    0.260237] usbcore: registered new interface driver hub
[    0.260434] usbcore: registered new device driver usb
[    0.268698] raspberrypi-firmware soc:firmware: Attached to firmware from 2017-04-05 11:49
[... 137 lines cut ...]
[    1.778156] systemd[1]: Set hostname to <raspberrypi>.
[    1.778772] systemd[1]: Initializing machine ID from random generator.
[    1.779397] systemd[1]: Installed transient /etc/machine-id file.
[    2.554563] systemd[1]: Cannot add dependency job for unit regenerate_ssh_host_keys.service, ignoring: Unit regenerate_ssh_host_keys.service failed to load: No such file or directory.
[    2.561580] systemd[1]: Expecting device dev-ttyS0.device...
[    2.562522] systemd[1]: Starting Forward Password Requests to Wall Directory Watch.
[    2.563133] systemd[1]: Started Forward Password Requests to Wall Directory Watch.
[    3.883039] systemd-udevd[112]: starting version 215
[    6.638291] EXT4-fs (mmcblk0p2): re-mounted. Opts: (null)
[    8.601283] systemd-journald[113]: Received request to flush runtime journal from PID 1
[   14.294309] uart-pl011 20201000.serial: no DMA platform data
[   15.101295] Adding 102396k swap on /var/swap.  Priority:-1 extents:5 across:225280k SSFS
[   17.630768] random: crng init done
[  103.881382] usb 1-1: USB disconnect, device number 2
[  107.331222] Indeed it is in host mode hprt0 = 00041901
[  107.531106] usb 1-1: new low-speed USB device number 3 using dwc_otg
[  107.532184] Indeed it is in host mode hprt0 = 00041901
[  107.767726] usb 1-1: New USB device found, idVendor=046d, idProduct=c31c
[  107.767751] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[  107.767759] usb 1-1: Product: USB Keyboard
[  107.767769] usb 1-1: Manufacturer: Logitech
[  107.808666] input: Logitech USB Keyboard as /devices/platform/soc/20980000.usb/usb1/1-1/1-1:1.0/0003:046D:C31C.0003/input/input2
[  107.872703] hid-generic 0003:046D:C31C.0003: input,hidraw0: USB HID v1.10 Keyboard [Logitech USB Keyboard] on usb-20980000.usb-1/input0
[  107.901762] input: Logitech USB Keyboard as /devices/platform/soc/20980000.usb/usb1/1-1/1-1:1.1/0003:046D:C31C.0004/input/input3
[  107.971597] hid-generic 0003:046D:C31C.0004: input,hidraw1: USB HID v1.10 Device [Logitech USB Keyboard] on usb-20980000.usb-1/input1

LeonardLee, 2017-05-26:
Hi,

By the way, maybe I lost a lot of driver or module. many function can not work, such of sound control, keyboard etc. Has it tested on RPI zero W? How to build a right image for it?

mkreisl, 2017-05-26:
@LeonardLee 
Did you have modules installed?

LeonardLee, 2017-05-27:
@mkreisl
I will try it again for double check.
But I found another issue.
If using old .config as config file for building 4.9.x version kernel and module, it will report undefine when compile, the error is here, and I just comment these code in kernel:

  **LD      arch/arm/boot/compressed/vmlinux
  OBJCOPY arch/arm/boot/zImage
  Kernel: arch/arm/boot/zImage is ready
ERROR: "rpi_firmware_get" [drivers/video/backlight/rpi_backlight.ko] undefined!
ERROR: "rpi_firmware_property" [drivers/video/backlight/rpi_backlight.ko] undefined!
scripts/Makefile.modpost:91: recipe for target '__modpost' failed
make[1]: *** [__modpost] Error 1
Makefile:1196: recipe for target 'modules' failed
make: *** [modules] Error 2**

mkreisl, 2017-05-27:
@LeonardLee 
If you're trying .config from 4.4 kernel for your build, this does not work.

popcornmix, 2017-05-27:
You need to use the right 4.9 config. E.g.
```
make ARCH=arm bcmri_defconfig
```
https://www.raspberrypi.org/documentation/linux/kernel/building.md

If you have custom .config settings, then start with the default working options and then once it is confirmed working add in your changes.

LeonardLee, 2017-05-29:
@mkreisl

As you said, I have test 4.9.x and anything like official tutorial.
I don`t have screen now. so I can`t to see boot information and will test 4.4 kernel.

closed 2017-09-13 by JamesH65