raspberrypi/linux #2042: Kernel >= 4.8.y  / failed to add service - already in use? QT Applications

LogiSch17, 2017-05-30:
Hi guys I am working on an RPI CM with a buildroot busybox.
When i update the Kernel to a Version higher then 4.7.y I can not run any QT Apps.
Only error i get is : 
*  failed to add service - already in use?

This is how my config.txt looks like:
kernel=zImage
disable_splash=1
framebuffer_width=800
framebuffer_height=480
framebuffer_depth=24
hdmi_force_hotplug=1
hdmi_group=2
hdmi_mode=1
hdmi_mode=87
hdmi_cvt=800 480 60 6 0 0 0
max_usb_current=1
config_hdmi_boost=7

device_tree=bcm2708-rpi-cm.dtb
device_tree_address=0x100
kernel_address=0x8000
disable_commandline_tags=1

dtparam=spi=on

dtparam=i2c0=on
dtparam=watchdog=on
dtparam=i2c_arm=on
dtparam=i2c0_baudrate=400000
dtoverlay=mcp2515-can0,oscillator=16000000,interrupt=4

cmdline.txt:
dwc_otg.lpm_enable=0 oops=panic panic=-1 console=tty3 root=/dev/mmcblk0p2 rootfstype=ext4 elevator=deadline rootwait logo.nologo loglevel=1

pelwell, 2017-05-30:
Try deleting these lines:
```
device_tree=bcm2708-rpi-cm.dtb
device_tree_address=0x100
kernel_address=0x8000
disable_commandline_tags=1
```
You don't need them, and they could be preventing Device Tree from being handled correctly.

LogiSch17, 2017-05-31:
Hi,
thanks for your response.

I tried to delete this lines, but when I am deleting : 
`device_tree=bcm2708-rpi-cm.dtb`
Kernel does not load anymore and i don´t get any kernel logs.

So do i have to add the device tree some where esle?

I am using the bcmrpi defconfig

And at buildroot i have the options:

```
Device tree source  ( Use a device tree present in the kernel.) 
[*] Build a Device Tree Blob (DTB)
(bcm2708-rpi-b bcm2708-rpi-b-plus bcm2708-rpi-cm) Device Tree Source file names
```

LogiSch17, 2017-07-10:
Hi guys,
I am still have these problem that i can´t start qt apps on kernel higher equal 4.8.y.
Do you think it is because the device tree isn´t loaded correctly? Cause all my modules i am using out of device tree and device tree overlay are loaded fine.

Do you have any hint how i can find out what is causing these problem?

I would be very thankfull for any advice.

JensC, 2017-07-24:
I have the same problem and I am stuck with it. Any idea how to debug that error? Thanks

pelwell, 2017-07-24:
@JensC Please provide the same information as @LogiSch17, just to be sure.

@LogiSch17 None of those config.txt settings mentioned above should be necessary. The `device_tree` value provided should be the default value on a CM, but if you say it is necessary then leave it (for now) and just remove the others.

Both - the error message comes from the Khronos (OpenGL) client library which is failing to connect to the graphics stack on the VPU. What do you get when you run the following:
```
dmesg | grep -i vchiq
vcgencmd version
```

JensC, 2017-07-24:
Ok, here my results:

pi@powerpi:~ $ dmesg | grep -i vchiq
[    0.735647] vchiq: vchiq_init_state: slot_zero = 0x9ed80000, is_master = 0

pi@powerpi:~ $ vcgencmd version
Jul  3 2017 14:16:43 
Copyright (c) 2012 Broadcom
version 4139c62f14cafdb7d918a3eaa0dbd68cf434e0d8 (tainted) (release)

in my config.txt everything is disabled except:
dtparam=spi=on
tparam=audio=on
dtoverlay=vc4-kms-v3d
gpu_mem=256

I use a natively compiled qt 5.9.2. If you need the config.summary let me know.

Thanks for the quick response and your help in that matter.
Jens

pelwell, 2017-07-24:
vchiq (the mains comms channel between the ARM and the VPU) looks to be up and working, which rules out one possible cause.

Some more runes to type:
```
sudo apt-get install pastebinit
sudo sh -c "echo info > /sys/kernel/debug/vchiq/log/msg"
sudo dmesg -C

<run qt app here>

dmesg | pastebinit
```
The output should be a URL to paste.debian.net.

JensC, 2017-07-24:
Ok, checkout: http://paste.debian.net/977931/

And thanks again.

pelwell, 2017-07-24:
Interesting. How about these:
```
<run qt app>
vcgencmd vcos log set vchiq_core trace
/opt/vc/bin/vchiq_test -s KHAN -p
sudo vcdbg log msg |& pastebinit
```

JensC, 2017-07-24:
Here we go: http://paste.debian.net/977947/

Thanks

pelwell, 2017-07-24:
We have a winner:
```
001912.822: brfs: File read: /mfs/sd/overlays/vc4-kms-v3d.dtbo
001960.188: Loaded overlay 'vc4-kms-v3d'
```
The `vc4-kms-v3d` overlay enables the experimental ARM-side GL driver and disables the VPU-side driver (you can't have both running at the same time). The OpenGL DLL that has been loaded is the one that talks to the VPU driver.

1) Are you using vc4-kms-v3d overlay intentionally?
2) Have you used the ARM-side GL driver with Qt on kernels prior to 4.8?

I suggest removing the dtoverlay line, rebooting and trying again.

JensC, 2017-07-24:
Wow. I really thought I tried it without vc4-kms-v3d  as well, but obviously I didn't! It works now. Now as far as I understand the QT5 app runs natively on the LinuxFB, right? Is there a possibility to run the app under X11 as well (maybe with additional cli params)? As you see the whole QT5 and display driver topic is pretty confusing to me. Thanks so much for your precious time. I really appreciate it! 
I had QT5 5.6 running on a older kernel as well, but that is some time ago and I can't recall the details.

JensC, 2017-07-24:
ok it was simple. I used for example: ./tabdialog --platform xcb --display :0
Thanks again!

JamesH65, 2017-09-13:
Closing this issue as questions answered/issue resolved.

closed 2017-09-13 by JamesH65