raspberrypi/linux #2043: VCHIQ-related oopses in 4.4.50-v7+ (GStreamer 1.12 + camera)

gohai, 2017-05-30:
I haven't seen those before, but recently I made two changes to my development setup:
* updated to Raspbian April 2017 from January 2017
* updated GStreamer build from 1.10 to 1.12

Since then opening the camera via the bcm2835_v4l2 module often times fails, and I receive the following VCHIQ-related oops:

    Internal error: Oops: 2807 [#1] SMP ARM
    Process VCHIQ-0 (pid: 70, stack limit = 0xa9006210)
    Stack: (0xa9007e78 to 0xa9008000)
    7e60:                                                       a9007e88 8006956c
    7e80: 00000124 aa889c00 00004003 aa889bf8 a7d63f00 00000002 aa889bf8 00000000
    7ea0: a7d80200 aa889bf8 a9007ed4 a9007eb8 803d6264 7f0f70e4 00000130 803d6190
    7ec0: 45057003 808abf74 a9007f5c a9007ed8 803ce8e4 803d619c 00000000 80879458
    7ee0: 808658c4 a9052564 a9007f44 a9007ef8 805b8368 80049b94 fff9fee9 a7c5425c
    7f00: 805f0768 aa880194 aa880020 00000124 00026d28 00000003 00000124 aa880194
    7f20: fff9fee9 ffffffff 00000000 00000000 00000000 00000000 a9003540 80962aa4
    7f40: 803cd9dc 00000000 00000000 00000000 a9007fac a9007f60 80042954 803cd9e8
    7f60: 2a775000 00000000 55554545 80962aa4 00000000 00000000 a9007f78 a9007f78
    7f80: 00000000 00000000 a9007f88 a9007f88 a9003540 80042868 00000000 00000000
    7fa0: 00000000 a9007fb0 8000fbe8 80042874 00000000 00000000 00000000 00000000
    7fc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
    7fe0: 00000000 00000000 00000000 00000000 00000013 00000000 55555555 55555551

Sometimes this error is first accompanied by the following output from GStreamer

    GLVideo: v4l2src0: Could not read from resource.
    Debugging information: gstv4l2bufferpool.c(1064): gst_v4l2_buffer_pool_poll (): 
    /GstPipeline:pipeline0/GstV4l2Src:v4l2src0:
    poll error 1: Invalid argument (22)
    GLVideo: v4l2src0: Failed to allocate a buffer
    Debugging information: gstv4l2src.c(866): gst_v4l2src_create (): 
    /GstPipeline:pipeline0/GstV4l2Src:v4l2src0
    GLVideo: v4l2src0: Internal data stream error.
    Debugging information: gstbasesrc.c(2939): gst_base_src_loop (): 
    /GstPipeline:pipeline0/GstV4l2Src:v4l2src0:
    streaming stopped, reason error (-5)

So my question: are there some known issues with this functionality in the 4.4.50-v7+ kernel? I can't rule out that the GStreamer update might also have had something to do, but the kernel oops sounds a bit extreme... BTW: I am running with 320 MB GPU memory (before I never had similar issues with 256 MB even).

lategoodbye, 2017-05-30:
Are you able to reproduce this issue with 256 MB GPU memory?

gohai, 2017-05-30:
@lategoodbye Appreciate your work on bcm2835. Thanks for that!

I first saw this with 256 MB GPU, yes. Because of the wording of the GStreamer error I bumped it up even further, but it didn't make a difference. This is all on a 800x480 px LCD, so not huge frame buffer sizes...

It's a Raspberry Pi 3. Vanilla Raspbian April 2017.

6by9, 2017-05-31:
Can you give us the full OOPS, and also the GStreamer pipeline you're running?
The main supported kernel is now 4.9, so an apt-get update, apt-get upgrade should update to 4.9. It'd be useful to know if the issue was still present on 4.9

gohai, 2017-06-01:
I just reproduced this on 4.9.24-v7+.

I've learned that the initial error I am receiving from GStreamer is actually associated with an error and a stacktrace as well:

    [  156.560318] bcm2835-v4l2: Failed enabling camera, ret -2
    [  156.560324] bcm2835-v4l2: Failed to enable camera
    [  156.560328] ------------[ cut here ]------------
    [  156.560352] WARNING: CPU: 1 PID: 1316 at drivers/media/v4l2-core/videobuf2-core.c:1342 vb2_start_streaming+0xe4/0x160 [videobuf2_core]
    [  156.560355] Modules linked in: bnep bluetooth brcmfmac brcmutil cfg80211 rfkill snd_bcm2835 snd_pcm snd_timer bcm2835_gpiomem snd evdev joydev rpi_ft5406 rpi_backlight uio_pdrv_genirq fixed uio bcm2835_v4l2 videobuf2_vmalloc videobuf2_memops videobuf2_v4l2 videobuf2_core v4l2_common videodev media i2c_dev fuse ipv6
    [  156.560421] CPU: 1 PID: 1316 Comm: v4l2src0:src Not tainted 4.9.24-v7+ #993
    [  156.560424] Hardware name: BCM2835
    [  156.560442] [<8010fb3c>] (unwind_backtrace) from [<8010c058>] (show_stack+0x20/0x24)
    [  156.560451] [<8010c058>] (show_stack) from [<80455200>] (dump_stack+0xd4/0x118)
    [  156.560460] [<80455200>] (dump_stack) from [<8011d2c4>] (__warn+0xf8/0x110)
    [  156.560467] [<8011d2c4>] (__warn) from [<8011d3ac>] (warn_slowpath_null+0x30/0x38)
    [  156.560485] [<8011d3ac>] (warn_slowpath_null) from [<7f0fe644>] (vb2_start_streaming+0xe4/0x160 [videobuf2_core])
    [  156.560515] [<7f0fe644>] (vb2_start_streaming [videobuf2_core]) from [<7f0ffb74>] (vb2_core_streamon+0x130/0x178 [videobuf2_core])
    [  156.560543] [<7f0ffb74>] (vb2_core_streamon [videobuf2_core]) from [<7f10d260>] (vb2_streamon+0x40/0x60 [videobuf2_v4l2])
    [  156.560568] [<7f10d260>] (vb2_streamon [videobuf2_v4l2]) from [<7f10d2d0>] (vb2_ioctl_streamon+0x50/0x54 [videobuf2_v4l2])
    [  156.560641] [<7f10d2d0>] (vb2_ioctl_streamon [videobuf2_v4l2]) from [<7f0b6e50>] (v4l_streamon+0x28/0x2c [videodev])
    [  156.560749] [<7f0b6e50>] (v4l_streamon [videodev]) from [<7f0bb0c0>] (__video_do_ioctl+0x2a0/0x314 [videodev])
    [  156.560854] [<7f0bb0c0>] (__video_do_ioctl [videodev]) from [<7f0baa60>] (video_usercopy+0x274/0x610 [videodev])
    [  156.560959] [<7f0baa60>] (video_usercopy [videodev]) from [<7f0bae1c>] (video_ioctl2+0x20/0x24 [videodev])
    [  156.561064] [<7f0bae1c>] (video_ioctl2 [videodev]) from [<7f0b5720>] (v4l2_ioctl+0xd4/0xe8 [videodev])
    [  156.561123] [<7f0b5720>] (v4l2_ioctl [videodev]) from [<802831cc>] (do_vfs_ioctl+0xac/0x820)
    [  156.561132] [<802831cc>] (do_vfs_ioctl) from [<80283984>] (SyS_ioctl+0x44/0x6c)
    [  156.561142] [<80283984>] (SyS_ioctl) from [<801080c0>] (ret_fast_syscall+0x0/0x1c)
    [  156.561147] ---[ end trace 804b73e8edab3aeb ]---

And the OOPS seems to happen after this initial issue (with the kernel already tainted). Here is it in its entirety:

    [  265.565442] Unable to handle kernel NULL pointer dereference at virtual address 00000000
    [  265.565455] pgd = 9d62c000
    [  265.565460] [00000000] *pgd=26b15835, *pte=00000000, *ppte=00000000
    [  265.565476] Internal error: Oops: 17 [#1] SMP ARM
    [  265.565482] Modules linked in: bnep bluetooth brcmfmac brcmutil cfg80211 rfkill snd_bcm2835 snd_pcm snd_timer bcm2835_gpiomem snd evdev joydev rpi_ft5406 rpi_backlight uio_pdrv_genirq fixed uio bcm2835_v4l2 videobuf2_vmalloc videobuf2_memops videobuf2_v4l2 videobuf2_core v4l2_common videodev media i2c_dev fuse ipv6
    [  265.565548] CPU: 2 PID: 1486 Comm: v4l2src0:src Tainted: G        W       4.9.24-v7+ #993
    [  265.565553] Hardware name: BCM2835
    [  265.565558] task: a68b9d80 task.stack: 9d706000
    [  265.565580] PC is at vchiq_mmal_port_enable+0x70/0x124 [bcm2835_v4l2]
    [  265.565592] LR is at mutex_lock_interruptible+0x1c/0x68
    [  265.565598] pc : [<7f123a30>]    lr : [<80714f0c>]    psr: 00000013
sp : 9d707cd0  ip : 9d707cb8  fp : 9d707cfc
    [  265.565605] r10: a93d6cc0  r9 : 00000000  r8 : 7f11f8e0
    [  265.565611] r7 : a8cb4410  r6 : a8cb4004  r5 : a8cb4000  r4 : a9b0f000
    [  265.565616] r3 : 00000001  r2 : 00000000  r1 : a8cb4498  r0 : 00000000
    [  265.565623] Flags: nzcv  IRQs on  FIQs on  Mode SVC_32  ISA ARM  Segment user
    [  265.565628] Control: 10c5383d  Table: 1d62c06a  DAC: 00000055
    [  265.565633] Process v4l2src0:src (pid: 1486, stack limit = 0x9d706210)
    [  265.565639] Stack: (0x9d707cd0 to 0x9d708000)
    [  265.565646] 7cc0:                                     a8cb4410 a9b0f000 a9b0f620 00000000
    [  265.565654] 7ce0: 00000000 7f0d16a0 00000000 a93d6cc0 9d707d2c 9d707d00 7f11fed4 7f1239cc
    [  265.565662] 7d00: 9d707d0c 9d707d10 7f0fd228 00000008 a9b0f4a0 a9b0f5c4 a9b0f4e8 00000012
    [  265.565670] 7d20: 9d707d4c 9d707d30 7f0fe5cc 7f11fde0 a9b0f4e8 00000001 a9b0f058 00000012
    [  265.565678] 7d40: 9d707d64 9d707d50 7f0ffb74 7f0fe56c a93d6cc0 00000001 9d707d74 9d707d68
    [  265.565686] 7d60: 7f10d260 7f0ffa50 9d707d8c 9d707d78 7f10d2d0 7f10d22c 00000001 40045612
    [  265.565694] 7d80: 9d707d9c 9d707d90 7f0b6e50 7f10d28c 9d707e04 9d707da0 7f0bb0c0 7f0b6e34
    [  265.565702] 7da0: 00000001 00a18000 7f124c6c 00000048 00000001 7f0d1864 9d707e18 9d472940
    [  265.565710] 7dc0: 8023d07c 8016ae88 00000207 9d707e3c 00000000 9d707e18 00000000 40045612
    [  265.565718] 7de0: 9d707e18 00000000 6413a0cc a93d6cc0 7f0bae20 00000004 9d707ec4 9d707e08
    [  265.565726] 7e00: 7f0baa60 7f0bae2c 80c15eca 00000001 00000004 00000001 00000001 8019a0ac
[... cut ...]
    [  265.566611] Code: e1a01007 e5b12088 e1520001 0a000003 (e5922000) 
    [  265.566620] ---[ end trace 804b73e8edab3aed ]---

My video pipeline is fairly conventional, except that it is using `glupload` to retrieve the frame as a GL texture, which is then passed to, and made use of in my GLES2 application. I'd be happy to provide you with a reproducer (it's all free software) - but from the looks of it (especially the `Failed enabling camera`), I am not sure if the issue happens not already before any of this would come into play?

6by9, 2017-06-01:
Yes there appears to be a kernel bug here as we shouldn't OOPS like that, but failing to enable the camera is going to be the root cause of your issue.
Can I suggest you check with ```vcgencmd get_camera``` that the camera module is detected correctly, and then with raspivid or raspistill that you can capture images from it?

Something a little bit funny is going on as the V4L2 driver should have checked during probe that the camera has been detected, so failing on enable probably points to being out of memory (there are two heaps on the GPU, and gpu_mem only sets one of them. The other you normally have to work fairly hard to deplete).
Could another application already be using the camera? It only supports a single user at a time, so something else running in the background and using it through MMAL would block V4L2.
Final thought is there were a couple of odd hardware failures caught a while back - https://www.raspberrypi.org/forums/viewtopic.php?f=43&t=147744. Again, debugging with raspistill will be easier than V4L2.

gohai, 2017-06-01:
I tried again on top of the Raspbian release from January (4.4.34-v7+), and no matter how hard I try I can trigger the issue. So to me this strongly points to something that either changed in kernel-land between 4.4.34 and 4.4.50. Or something else in the base distribution that changed, that exposes a timing-related bug. Any ideas?

`vcgencmd get_camera` outputs `supported=1 detected=1`.

I can't see anything else attempting to use the camera, but my program does initialize GLES2 (using the "legacy" driver) right before opening the camera. Could one block the other?

6by9, 2017-06-01:
Can you try the 4.4.34 kernel against the 4.4.50 firmware? Taking the /boot/[*.elf|*.dat] files from the new image and putting them into /boot on the old is probably quickest (backup the old ones first).

4.4.34 was 21/11/16 whilst 4.4.50 was 18/2/17 - 3 months is a long time to try and bisect, particularly as we can't reproduce the issue.
There is only one kernel change to the V4L2 driver in that period, and that is just a conformance test fix where we weren't setting a flag - very unlikely to have any significant effect.

The other thing to try is adding ```start_debug=1``` to /boot/config.txt. That enables assert logging on the GPU, and ```sudo vcdbg log assert``` can then be used to read them out. It'd be interesting to know if anything nasty is being logged there.

gohai, 2017-06-01:
I was able to trigger it with the 4.4.34 kernel and the 4.4.50 firmware, as suggested.

Here are the lines from the assert log

    237552.241: assert( 0 ) failed; ../../../../../vcfw/drivers/device/camera/imx219.c::imx219_open line 431 rev 3ca4cf4
    vcdbg_ctx_get_dump_stack: dump_stack failed
    ----------------
    237552.276: assert( open_camera_driver: camera open failed ) failed; 
    ../../../../../middleware/camplus/cdi/cdi_camera.c::open_camera_driver line 704 rev 3ca4cf4
    vcdbg_ctx_get_dump_stack: dump_stack failed
    ----------------
    237552.313: assert( cdi_camera_open: Failed to open camera driver/perihperal ) failed; 
    ../../../../../middleware/camplus/cdi/cdi_camera.c::cdi_camera_open line 541 rev 3ca4cf4
    vcdbg_ctx_get_dump_stack: dump_stack failed
    ----------------
    237552.345: assert( retcode == 0 ) failed; 
    ../../../../../middleware/openmaxil/components/camera.c::load_and_open_cdi line 11913 rev 3ca4cf4
    vcdbg_ctx_get_dump_stack: dump_stack failed
    ----------------
    237552.374: assert( id->cdi!=NULL && id->cdi_handle!=NULL ) failed; 
    ../../../../../middleware/openmaxil/components/camera.c::open_camplus line 11278 rev 3ca4cf4
    vcdbg_ctx_get_dump_stack: dump_stack failed
    ----------------
    242195.563: assert( elem ) failed; 
    ../../../../../middleware/dispmanx/dispmanx.c::dispmanx_element_change_source line 5593 rev 3ca4cf4
    vcdbg_ctx_get_dump_stack: dump_stack failed

Can you work off this information?

Here's my reproducer:

* make sure the `bcm2835_v4l2` driver is loaded, and `gpu_mem` is adequate
* install Processing with `curl https://processing.org/download/install-arm.sh | sudo sh`
* `mkdir -p ~/sketchbook ~/sketchbook/examples ~/sketchbook/libraries ~/sketchbook/modes ~/sketchbook/tools`
* `cd ~/sketchbook/libraries`
* `wget https://github.com/gohai/processing-glvideo/releases/download/latest/processing-glvideo.zip`
* `unzip processing-glvideo.zip`
* ``processing-java --sketch=`pwd`/glvideo/examples/SimpleCapture --run``

gohai, 2017-06-01:
@6by9 Googling for the first assert I found a message by you indicating that the asserts relates to a security chip check failing... I got the camera as part of a Raspberry Pi Zero Camera Pack from Adafruit, so I'd be surprised if it wasn't a genuine part. This issue also doesn't _always_ bite.

6by9, 2017-06-01:
```237552.241: assert( 0 ) failed; ../../../../../vcfw/drivers/device/camera/imx219.c::imx219_open line 431 rev 3ca4cf4````
is the crypto chip check failing. That code hasn't been touched since May/June 2016, so there's no specific reason for this to work on 4.4.34 and not on 4.4.50.
If it passes at all then the chip itself is programmed (a few escaped the factory totally unprogrammed). If you bought it as a Pi Zero camera pack, then you've found a cable from elsewhere. How long is that cable? Unreliability/interference on the I2C has the potential to cause this issue.

gohai, 2017-06-02:
Cable is approximately 12 cm long. Not sure if it is the cable from the v1 camera I had bought some time before, or indeed from some "aftermarket" source.

I can't rule out that my attribution to newer firmware versions was flawed - I did have to physically handle the Pi to seat and remove the microSD after all. But I just re-tested the 4.4.34 kernel & firmware 20 times, without seeing the error (in fact I never ever recall receiving this with this version), whereas I did get it with the 4.4.50 firmware (albeit with a lower frequency than running a vanilla Raspbian release a from April).

I am fine if we chalk this up as something that ultimately was caused by I2C instability, but isn't there perhaps the chance that something in later versions just got faster or slower, and that the crypto chip check is failing because something else is now happens to race on the I2C bus (again, I am opening the V3D just right before, and I don't know how much of this initialization [though JOGL] is happening in an extra thread; in my setup I am also using the 7" Pi display if this makes a difference)?

6by9, 2017-06-02:
I'm not saying that it definitely isn't a firmware change, but I would have expected lots of screams if it were a generic flaw in the firmware.

I have had the Pi 7" display lock up the I2C bus fairly impressively and require a reboot to release it. Whilst I have a Pi display connected, I have the touchscreen disabled so it never touches the I2C after initialisation. The crypto chip is time sensitive, so if the display misbehaved at an inopportune moment then it would have the potential to result in a failure. Any chance you could add ```disable_touchscreen=1``` to boot/config.txt and retest?

gohai, 2017-06-02:
@6by9 My LCD has the touchscreen assembly removed (it broke once while travelling).

6by9, 2017-06-02:
My recollection is that unless you've disabled the touchscreen driver in the firmware, it will still be polling it over I2C. I don't believe unloading the ft5406 Linux kernel driver will stop that.

popcornmix, 2017-06-02:
@gohai can you identify the exact update which caused this problem. See:
https://github.com/Hexxeh/rpi-firmware/commits/master

If you click on each commit the end of the url contains a git hash. Run
```sudo rpi-update <hash>```
to revert back to that version. Report the first version with this error.

gohai, 2017-06-07:
I've tested again with a vanilla Raspbian release von April: issue hit the 7th time I restarted the sketch. Then I tried `disable_touchscreen=1`, and I did not see the error for the 20 times or so I tried.

Really unsure if it's just a matter of coincidence whether this bites or not. I still believe that something made this more likely to occur in more recent versions compared to the January release of Raspbian, where I never ever saw it. But since this is so hard to pin down, I don't think I have time to do a bisect with the firmware, sorry.

6by9, 2017-06-07:
I'll do some basic checks to see how frequent the I2C polling is on the display.
I wouldn't expect it to hog the I2C bus for long enough for the crypto chip to time out, but not seeing failures with ```disable_touchscreen=1``` would imply that it is linked.

The other thing that changed relatively recently was the display backlight control, but again I wouldn't expect to see large delays due to that.
@popcornmix The change to bitbash SMPS/GPIO expander control is also within this timeframe. Any chance that is hogging the VPU?

JamesH65, 2017-09-13:
@6by9 @popcornmix Any news on this one?

6by9, 2017-09-20:
Touchscreen polling takes <2ms and then sleeps for 17ms (60fps), so that should be well below the threshold for camera initialisation timeouts.
I still need to profile the SMPS bitbashed I2C.

I still need to look at why the camera failure causes an OOPS from the V4L2 driver.

6by9, 2020-12-09:
Closing due to inactivity and it seemed to be a collision between camera and display touchscreen that hasn't been seen recently.

closed 2020-12-09 by 6by9