raspberrypi/linux #2044: warning in cgroup_get

andersthomson, 2017-05-30:
Got this log on the console while doing compiles on the system (on the sd card).

[warning_cgroup_get_657.txt](https://github.com/raspberrypi/linux/files/1039214/warning_cgroup_get_657.txt)

popcornmix, 2017-05-30:
Looks a bit like this upstream bug:
https://patchwork.kernel.org/patch/9705267/

popcornmix, 2017-05-30:
Can you reproduce this error on demand?
(i.e. in case we want to test a patch)

andersthomson, 2017-05-30:
Not on demand. I've seen many cgroups related errors when moving above 4.9.17 though, so i'd be happy to take a patch for a spin and see what might change.

popcornmix, 2017-05-31:
Can you run rpi-update and then replace /boot/kernel7.img with:
https://drive.google.com/uc?id=0B-6zmEDJwxZEWWtQMmY0bFllSVE&export=download

It contains the linked patch. Run with that a while and report back if you think it is better.

andersthomson, 2017-05-31:
I'd be happy to, but I have no rpi-update here (it's a gentoo box). Can you upload the patch somewhere and indicate which commit  I should apply it to?

popcornmix, 2017-05-31:
This is the patch: http://paste.ubuntu.com/24727401/
Apply to head of rpi-4.9.y tree.

andersthomson, 2017-05-31:
Got 4.9.30 running with the patch. Let's see what happens.

Btw, any reason LOCALVERSION_AUTO is not enabled to get the git hash into
the dmesg? That would help track versions....

popcornmix, 2017-06-01:
We do store the uname string and git hash of tree in [firmware repo](https://github.com/raspberrypi/firmware/tree/master/extra) so we can track any of our builds.

I guess LOCALVERSION_AUTO is another option. @pelwell any thoughts?

pelwell, 2017-06-01:
It would make the module directory names longer and require more frequent tidying up, and guarantee that you have to re-install the modules for every kernel change. I'm not completely against the idea, but it's not very appealing.

popcornmix, 2017-06-01:
Ah yes, I'd missed that it would multiply the number of /lib/modules directories.
It's a no from me then.

andersthomson, 2017-06-01:
Fair enough. Though I, and expect others as well, have multiple
/lib/modules/* directories already and ways to manage that. As far as I
know, there is no guarantee that two different kernel git commit can share
modules. Are you taking steps to ensure that within each vX.Y.Z you never
have to recompile the modules?

14 hours uptime on the patched kernel....

andersthomson, 2017-06-05:
May days without seeing this when using that patch. Seems we have a winner!

popcornmix, 2017-06-05:
Added here: https://github.com/raspberrypi/linux/commit/e54215a9bc4d7b75dc4a991f4bbfc4f42102fbc7
Will be in next rpi-update build.

JamesH65, 2017-09-13:
Please update to the latest kernel which may contain a fix for this issue.

closed 2017-09-13 by JamesH65