raspberrypi/linux #2046: Please enable CONFIG_SENSORS_JC42

rumpelsepp, 2017-05-31:
With this module [this](https://learn.adafruit.com/adafruit-mcp9808-precision-i2c-temperature-sensor-guide/overview) temperature sensor and several others should be accessible. [Here](https://github.com/torvalds/linux/blob/master/Documentation/devicetree/bindings/hwmon/jc42.txt) is the device tree documentation. I will provide a device tree overlay configuration for the `i2c-sensor-overlay` once the option is enabled.

pelwell, 2017-05-31:
10KB for so many sensors isn't bad. The CONFIG_SENSOR_JC42 is now in the RPi defconfigs and will be picked up by the next firmware build.

rumpelsepp, 2017-05-31:
Nice, thanks!

popcornmix, 2017-07-13:
Okay to close this?

JamesH65, 2017-09-13:
Please update to the latest kernel which may contain a fix for this issue.

closed 2017-09-13 by JamesH65