raspberrypi/linux #2051: Problems with dtoverlay=vc4-kms-v3d and dtparam=audio=on

maxnet, 2017-06-02:
It seems the new sound stuff being done in the vc4-kms driver does not play nice with snd_bcm2835

- Plain Raspbian April 2017
- Updated to latest kernel and firmware with rpi-update
- Enabled "experimental GL driver" (vc4-kms-v3d) in raspi-config

On reboot lxpanel (the start menu at the top of screen) does not appear, just a grey bar:

[image]

- "ps" does show lxpanel is running
- attaching a debugger shows it is blocking somewhere inside the libasound library:

```
pi@raspberrypi:~ $ gdb lxpanel `pidof lxpanel`
[..]
(gdb) bt
#0  0x7638ef2c in ioctl () at ../sysdeps/unix/syscall-template.S:81
#1  0x7390bc8c in ?? () from /usr/lib/arm-linux-gnueabihf/libasound.so.2
Backtrace stopped: previous frame identical to this frame (corrupt stack?)
```

lxpanel seems to be running the start menu and all tray icon plug-ins such as "volume control" inside a single thread, so any blocks like this indeed mean nothing is going to work...

- journalctl shows pulseaudio has problems with sound as well:

```
[  243.686009] INFO: task pulseaudio:881 blocked for more than 120 seconds.
[  243.686023]       Not tainted 4.9.30-v7+ #1001
[  243.686030] "echo 0 > /proc/sys/kernel/hung_task_timeout_secs" disables this message.
[  243.686036] pulseaudio      D    0   881      1 0x00000001
[  243.686083] [<80713d80>] (__schedule) from [<807142a4>] (schedule+0x50/0xa8)
[  243.686101] [<807142a4>] (schedule) from [<80717514>] (schedule_timeout+0x220/0x364)
[  243.686119] [<80717514>] (schedule_timeout) from [<80714e18>] (wait_for_common+0xbc/0x17c)
[  243.686138] [<80714e18>] (wait_for_common) from [<80714ef8>] (wait_for_completion+0x20/0x24)
[  243.686190] [<80714ef8>] (wait_for_completion) from [<7f15f70c>] (bcm2835_audio_close+0x90/0x180 [snd_bcm2835])
[  243.686250] [<7f15f70c>] (bcm2835_audio_close [snd_bcm2835]) from [<7f15e404>] (snd_bcm2835_playback_close+0xd0/0xf4 [snd_bcm2835])
[  243.686364] [<7f15e404>] (snd_bcm2835_playback_close [snd_bcm2835]) from [<7f1349a8>] (snd_pcm_release_substream.part.12+0x58/0xbc [snd_pcm])
[  243.686497] [<7f1349a8>] (snd_pcm_release_substream.part.12 [snd_pcm]) from [<7f134ad8>] (snd_pcm_release+0x9c/0xa8 [snd_pcm])
[  243.686575] [<7f134ad8>] (snd_pcm_release [snd_pcm]) from [<80271d8c>] (__fput+0x9c/0x1e8)
[  243.686591] [<80271d8c>] (__fput) from [<80271f48>] (____fput+0x18/0x1c)
[  243.686613] [<80271f48>] (____fput) from [<8013ab04>] (task_work_run+0xcc/0xfc)
[  243.686633] [<8013ab04>] (task_work_run) from [<8010b838>] (do_work_pending+0xcc/0xd0)
[  243.686650] [<8010b838>] (do_work_pending) from [<801080e8>] (slow_work_pending+0xc/0x20)
```

- Issue goes away if removing "dtparam=audio=on" from config.txt

popcornmix, 2017-06-05:
I've just tested this. Ran apt-get dist-upgrade and the rpi-update:
```
pi@domnfs:~ $ uname -a
Linux domnfs 4.9.30-v7+ #1001 SMP Fri May 26 16:09:18 BST 2017 armv7l GNU/Linux
pi@domnfs:~ $ aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: ALSA [bcm2835 ALSA], device 0: bcm2835 ALSA [bcm2835 ALSA]
  Subdevices: 8/8
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
  Subdevice #2: subdevice #2
  Subdevice #3: subdevice #3
  Subdevice #4: subdevice #4
  Subdevice #5: subdevice #5
  Subdevice #6: subdevice #6
  Subdevice #7: subdevice #7
card 0: ALSA [bcm2835 ALSA], device 1: bcm2835 ALSA [bcm2835 IEC958/HDMI]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 1: vc4hdmi [vc4-hdmi], device 0: MAI PCM vc4-hdmi-hifi-0 []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
```
and startx launched okay, and audio worked (e.g. music from neverball through analogue).

@spl237 does this work okay for you?

maxnet, 2017-06-05:
If I configure the system to boot to console and execute "aplay -l" there, I get the same output as you.

However if booting to X (which is the default), and executing it through ssh after X/pulseaudio has started, "aplay -l" hangs as well:

```
pi@raspberrypi:~ $ aplay -l
^C^C^C^C^C
```

And does not feel like terminating:

```
pi@raspberrypi:~ $ ps aux |grep aplay
pi        1087  0.0  0.0   5132   756 pts/0    D+   16:43   0:00 aplay -l
pi        1112  0.0  0.2   4276  2008 pts/1    S+   16:45   0:00 grep --color=auto aplay
pi@raspberrypi:~ $ kill -9 1087
pi@raspberrypi:~ $ ps aux |grep aplay
pi        1087  0.0  0.0   5132   756 pts/0    D+   16:43   0:00 aplay -l
pi        1114  0.0  0.1   4276  1844 pts/1    S+   16:45   0:00 grep --color=auto aplay
```

journalctl output if that's any use: [journaltje.txt](https://github.com/raspberrypi/linux/files/1052607/journaltje.txt)

JamesH65, 2017-09-13:
Going to close due to lack of activity. in a few days unless more comment received.

maxnet, 2017-09-13:
No longer reproducible since Stretch.

closed 2017-09-13 by maxnet