raspberrypi/linux #2052: mini-uart bluetooth on pi3 not working in 4.9

Ing3mar, 2017-06-04:
The mini-uart bluetooth connection on the pi3 has become unstable in kernel 4.9. The bluetooth connection works initially but after as few seconds it breaks down and the connection to the bluetooth device becomes unresponsive. Restaring the hciuart does not work and only a reboot can bring the hciuart connection back. Lowering the baudrate in the hciattach to something like 115200 does not fix it either.   

Running a couple of rpi-update's I have found that the first version that breaks the mini-uart bluetooth is the first bump to kernel 4.9.11 (1371164) from 4.4.50.

bigiot-buildingiot, 2017-06-15:
I have the the same troubles with kernel version 4.9.30-v7+ ... also using bluez version 5.45.
How can safely downgrade my kernel?

pelwell, 2017-06-15:
Why are you using BlueZ 5.45? Raspbian Jessie ships with 5.23, and it includes patches for BCM43438 - does your version include those patches?

You can revert to the 4.4.50 kernel using:
```
sudo rpi-update 52241088c1da59a359110d39c1875cda56496764
```
Note that once you have reverted your kernel you should avoid updating the `raspberrypi-kernel` or `raspberrypi-bootloader` packages via `apt-get`.

bigiot-buildingiot, 2017-06-15:
I tried BlueZ 5.45 to see if it was a problem with 5.23. If I would have seen this bug report right away, I would just have downgraded my kernel. But there is so much information out there on how to "fix" bluetooth issues on RPi 3 that it took me 3 days to find this entry here.

After downgrading my kernel, everything worked fine again, also with BlueZ 5.45.

Follow-question: how shall I then upgrade the kernel in the future - if the problem gets fixed?

pelwell, 2017-06-15:
> Follow-question: how shall I then upgrade the kernel in the future - if the problem gets fixed?

To select the current Raspbian kernel:
```
sudo apt-get update
sudo apt-get upgrade rpi-kernel rpi-bootloader
```
or for the latest testing kernel and firmware:
```
sudo rpi-update
```

pelwell, 2017-06-15:
Edit: corrected Raspbian commands.

pelwell, 2017-06-16:
This is likely to be a more elaborate duplicate of #1855.

bigiot-buildingiot, 2017-06-17:
But didn't you note in your first message in this thread, not to use apt-get for updating the raspberrypi-kernel or raspberrypi-bootloader? Is it okay despite the downgrade?

pelwell, 2017-06-17:
Only avoid the upgrade commands if you want to use a specific version of the kernel.

pelwell, 2017-06-22:
The latest rpi-update firmware (just released) includes a patch that should fix this problem.

JamesH65, 2017-09-13:
Please update to the latest kernel which may contain a fix for this issue.

closed 2017-09-13 by JamesH65