raspberrypi/linux #2054: no ip_vti module

CaineLew, 2017-06-05:
I have tried to create ip tunnel like this:
sudo ip tunnel add ipsec0 local 192.168.1.100 remote X.X.X.X mode vti key 42

But i got a error.
add tunnel "ip_vti0" failed: No such device.

And I could not found the kernel module "ip_vti.ko".

CaineLew, 2017-06-05:
I downloaded the raspbian .img file from official web site,and all thing update to date.

pelwell, 2017-06-05:
That module is enabled with `CONFIG_NET_IPVTI`, but I find it strange that only a handful of legacy (m68k, s390) platforms define it. It's a 16KB module with no obvious side effects, but can you explain what this module allows you to do that can't be done another way?

CaineLew, 2017-06-05:
I have configred a hotspot use interface "wlan0" with ip 192.168.2.1,all traffic via interface "eth0" with ip 192.168.1.10, and installed StrongSwan for VPN client. The IPsec processing is based on policies and add route table 220 by default.I have to added route to table 220 but the traffic not be encrypted.So,I want to configure route based VPNs by VTI devices and manage route manually.

CaineLew, 2017-06-05:
I think the traffic from wlan0 not be encrypted cause of the default ipsec route table.

CaineLew, 2017-06-07:
I fixed it by another way.

pelwell, 2017-06-07:
Care to share, in case somebody else finds this issue?

CaineLew, 2017-06-08:
I found StrongSwan encrypted traffic which source ip address is the virtual ip.I just add a SNAT rule like this:
iptables --table nat --append POSTROUTING --source 192.168.2.0/24 -o eth0 -j SNAT --to-source 10.1.0.100
Alter the source ip from 192.168.2.0/24 to 10.1.0.100.

JamesH65, 2017-09-13:
Closing this issue as questions answered/issue resolved.

closed 2017-09-13 by JamesH65