raspberrypi/linux #2055: Feature request: Overlay for both uarts on GPIO (RPI 3B)

melvyn-sopacua, 2017-06-05:
Since the spec for the BCM 2837 is not available, this may not be possible, as perhaps the chip only allows UARTS to go to 2 sets of 2 pinouts.

If so, then reclassify as a doc update for [`arch/arm/boot/dts/overlays/README`](https://github.com/raspberrypi/linux/blob/rpi-4.9.y/arch/arm/boot/dts/overlays/README) to state that this isn't possible.

If possible, we'd like an overlay that disables bluetooth and routes one UART to as is done in the already available overlays, and makes the second available on a GPIO of your choice. I'm sure that if I see how it's done, it's easy to change the specific GPIO pins.

pelwell, 2017-06-05:
No, this isn't possible - GPIOs 14 and 15 are the only UART-capable GPIOs available on the 40-pin header. In regards to GPIO pin usage, 2837 is identical to 2835.

I don't think this needs to be documented in the overlay README since nothing in the README suggests that this might be possible - only one overlay (uart1) allows the GPIO pins to be changed, and the list of supported pins clearly says 14&15, 32&33 and 40&41.

closed 2017-06-05 by pelwell

melvyn-sopacua, 2017-06-05:
This suggests it's possible:

> /* Switch Pi3 Bluetooth function to use the mini-UART (ttyS0) and restore
   UART0/ttyAMA0 over GPIOs 14 & 15. Note that this may reduce the maximum
usable baudrate.

Except of course that GPIO 32&33 and 40&41 aren't on the header. Then again, I can't work out from any of the schematics [here](https://www.raspberrypi.org/documentation/hardware/raspberrypi/bcm2835/BCM2835-ARM-Peripherals.pdf) what GPIO 40&41 @ alt5 would yield. From what I gather now is that GPIO 32&33 are routed to Bluetooth in model 3B and that would mean uart0 can switch between 14&15 and 32&33, leaving a possible uart0@14&15 + uart1@40&41. Yes, that wouldn't be at the header, but what would that leave me with?

pelwell, 2017-06-05:
On a Compute Module, GPIOs 0 to 45 are available on the module pinout.